realsenseai/realsense-ros #3111: Can't not display compressed RGB/Depth in RViz

pengzhenghao, 2024-05-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking our ROS RealSense Wrapper documentation [README](https://github.com/IntelRealSense/realsense-ros/blob/ros2-development/README.md).
    * Have you looked in our [Discussions](https://github.com/IntelRealSense/realsense-ros/discussions)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/realsense-ros/issues) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D435i                         | 
| Firmware Version                   | (Open RealSense Viewer --> Click info)                          | 
| Operating System & Version |   Linux (Ubuntu 22) 
| Kernel Version (Linux Only)    |  (e.g. 5.4)                                                                       | 
| Platform                                 | PC + NVIDIA Jetson                            |
| Librealsense SDK Version       |  { 2.<?>.<?> }                                                    | 
| Language                               |  {C/C#/labview/opencv/pcl/python/unity }         | 
| Segment			         |  {Robot/Smartphone/VR/AR/others }                             | 
| ROS Distro			         |  {Iron/Humble/Rolling/etc.. }                                           | 
| RealSense ROS Wrapper Version   |  {4.51.1, 4.54.1, etc..}                                                  | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Hi, I want to display the compressed RGB stream in RViz and it says:

[image]

`Expecting Camera Info on topic [/camera/depth/image_rect_raw/camera_info]. No CameraInfo received. Topic may not exist.`

But I found that the camera info is already there.

[image]

I can display the video stream with raw, but not compressed. If using raw video stream, the fluency seems to be identical to the image stream. Is there any recommandation way to visualize? Thanks!

pengzhenghao, 2024-05-28:
I just found that when I monitor the FPS with `ros2 topic hz xxxx`, turn on RViz and render images in real time will greatly reduce the FPS. Sorry for asking some dump questions. Seems like RViz can greatly reduce the FPS? In my understanding RViz is just a consumer why it can reduce the FPS (I am with a high-end PC so let's assume the time spent on rendering is trivial)...

MartyG-RealSense, 2024-05-28:
Hi @pengzhenghao  It is not a dumb question at all, so don't worry.  :)

Does it make a positive difference to FPS if you set RViz to use compressed data in the 'Create visualization' settings, as described at https://github.com/IntelRealSense/realsense-ros/issues/1903#issuecomment-898522352

pengzhenghao, 2024-05-28:
## On the problem of dropped FPS

#### Setup

I have installed D435i in the nvidia `Jetson`, which is connected to a router. I have a `PC` which is also connected to the router and can access D435i topics via network. Launch:

```
ros2 launch realsense2_camera rs_launch.py \
depth_module.profile:=848x480x15 rgb_camera.profile:=848x480x15
```

#### Profile the FPS without RVIZ

Here is the profile result, without rviz. I will This result is mainly for measuring the network bandwidth:

* `ros2 topic hz /camera/color/image_raw` 
  * Jetson: 14.995 HZ
  * PC: 9.586 HZ
* `ros2 topic hz /camera/depth/image_rect_raw`
  * Jetson: 14.983 HZ
  * PC: 14.370 HZ

The above results shows the RGB images might fails to be transported to the PC (maybe because it use 3 channels to represent colors vs 1 channel for depth). 

So let's use compression:

* `ros2 topic hz /camera/color/image_raw` 
  * PC: 9.587 HZ
* `ros2 topic hz /camera/color/image_raw/compressed` 
  * PC: 14.995 HZ

So using compressed RGB images will not make the network becomes bottleneck.

#### What happen when using RViz?

After launching RViz and monitoring `/camera/depth/image_rect_raw` and `/camera/color/image_raw` (without compression!), the FPS from:

* `ros2 topic hz /camera/color/image_raw/compressed` 
  * PC: 14.995 HZ

To:

* `ros2 topic hz /camera/color/image_raw/compressed` 
  * PC: **3.967 HZ**

Shown here:

[image]

The reasong I am monitoring `/camera/color/image_raw` instead of `/camera/color/image_raw/compressed` is because the compressed version can not be displayed as indicated in this issue.

## On the problem of using RViz visualize compressed images

As you suggested `if you set RViz to use compressed data in the 'Create visualization' settings`, I try to do it.

First, I found that there is no compressed option available for  `/camera/color/image_raw/Image`, it is only available for  `/camera/color/image_raw/Camera`:

[image]

Second, if I create the pannel for `/camera/color/image_raw/Camera (compressed)` as:
[image]

it shows nothing:
[image]

The error message indicates the camera_info is not properly read by RViz.

MartyG-RealSense, 2024-05-29:
As you are using a router and a network connection, are you using wireless wi-fi communication instead of a wired connection to the router?

If you are on wi-fi, https://github.com/IntelRealSense/librealsense/issues/10730#issuecomment-1197918767 has advice.  It also refers to how nearby wireless devices such as a wireless mouse can negatively impact performance.

MartyG-RealSense, 2024-06-14:
Case closed due to no further comments received.

closed 2024-06-14 by MartyG-RealSense

epanov, 2025-04-10:
This is broken due to camera_info being published under `/camera/color/camera_info` instead of `/camera/color/image_raw/camera_info`. Case is not closed.

MartyG-RealSense, 2025-04-10:
Hi @epanov  The **Published Topics** section of the RealSense ROS2 wrapper documentation shows that `/camera/camera/color/camera_info` is the correct topic for camera_info in the current wrapper.  

https://github.com/IntelRealSense/realsense-ros?tab=readme-ov-file#published-topics

The other main ROS2 wrapper documentation site also shows the color camera_info to be under **/color/camera_info**

https://dev.intelrealsense.com/docs/ros2-wrapper#d400-stereo-depth-cameras

**********

The topics are now named `/camera/camera/` instead of just `/camera `since updates to the camera_name and namespace system described in the documentation here:

https://github.com/IntelRealSense/realsense-ros?tab=readme-ov-file#camera-name-and-camera-namespace 

If you are using an older ROS2 wrapper version dated before the changes though then the topics will still be named /camera/

********

Are you able to obtain the color camera_info if you set RViz to use `/camera/camera/color/camera_info` instead of /image/raw/camera_info (or use /camera/color/camera_info if you are using an older wrapper version).

epanov, 2025-04-11:
I should have documented more clearly what I meant:

Required Info
 	
| Camera Model                                | D435i                         | 
|-----------------------------------------------|------------------------------|
| Firmware Version                           |                                     |
| Operating System & Version          | Linux (Ubuntu 22)       |
| Kernel Version (Linux Only) 	         |                                   |
| Platform 	                                 | x86 AMD64               |
| Librealsense SDK Version 	         |                                  |
| ROS Distro 	                                 | Humble                   |
| RealSense ROS Wrapper Version |                                |

1) The documentation of the wrapper repo is accurate to the current state of it.
2) I was short handing `/camera/camera` to `/camera` my apologies.
3) The `camera_info` topic is available and readable, the issue is that is in the wrong place for ROS2 infra to use it.

Here is the topic list:
```
$ ros2 topic list
/camera/camera/color/camera_info
/camera/camera/color/image_raw
/camera/camera/color/image_raw/compressed
/camera/camera/color/image_raw/compressedDepth
/camera/camera/color/image_raw/theora
/camera/camera/color/metadata
/camera/camera/depth/camera_info
/camera/camera/depth/image_rect_raw
/camera/camera/depth/image_rect_raw/compressed
/camera/camera/depth/image_rect_raw/compressedDepth
/camera/camera/depth/image_rect_raw/theora
/camera/camera/depth/metadata
/camera/camera/extrinsics/depth_to_color
/parameter_events
/rosout
/tf_static
```
Note that the compressed topics are under the `image_raw` topic, and the `camera_info` topic is outside. Which ever is right or wrong is up to you. Either the `camera_info` topic needs to be insides the `image_raw` topic or the compressed image topics need to be outside. You are correct in saying that the intended topic structure is being published, but I am pointing out that that is not expected structure by `rviz2`.
 
You can see that this in the rviz documentation: 
https://docs.ros.org/en/humble/Tutorials/Intermediate/RViz/RViz-User-Guide/RViz-User-Guide.html#built-in-display-types
And you can see it here in a live example:

[image]

Note that I have the point cloud and rgb images opened to show that the camera is streaming and the tfs of the camera are correctly being published by the driver. The highlighted display field shows the exact error I was commenting on earlier.

Looking into this more I now see this is just causing an issue in the 3d visual and should not effect the 2d display. Let me know if there are other debug steps we can take.

epanov, 2025-04-19:
Thanks for the link. I will try it out as soon as I can.

Sorry about the delayed response.

MartyG-RealSense, 2025-04-19:
It's no trouble at all.  I look forward to your next report.  Good luck!