realsenseai/realsense-ros #3113: Announcement: RealSense ROS wrapper 4.55.1 now available

MartyG-RealSense, 2024-05-28:
Hi everyone,

Version **4.55.1** of the RealSense ROS wrapper is now available.

https://github.com/IntelRealSense/realsense-ros/releases/tag/4.55.1

tonynajjar, 2024-05-29:
It doesn't specify the "Supported RealSense SDK" and "Supported Devices", could you specify? Thanks

tonynajjar, 2024-05-29:
Thanks for the answer, this practically means that librealsense and realsense-ros always have to be upgraded together? It isn't possible e.g. to use librealsense 2.54.2 with realsenes-ros 4.55.1?

MartyG-RealSense, 2024-05-29:
As a general rule, if new SDK version only has a small number of changes from the previous version then you can risk using the new version with a wrapper meant for the previous one.  For example, 2.54.2 was mostly bug-fixes rather than new features and so it still works with wrapper 4.54.1.  However, 2.55.1 has extensive changes and so a new wrapper designed for it is the best choice.

jesusramondovale, 2024-05-31:
Is it now availablevia apt-get install? Keeps giving me the 4.54 on Ubuntu 22.04 LTS jammy amd64

MartyG-RealSense, 2024-05-31:
@jesusramondovale  The wrapper repository lists the current package version as 2.55.1.

https://index.ros.org/p/realsense2_camera/

The latest version of the wrapper should always be able to be built from source code with **colcon build** if problems with the package version are experienced.

tonynajjar, 2024-06-03:
Hello,

One more piece of feedback, it would be nice if in the release notes, the breaking changes would be documented + what action the users should take. As an example, in this [PR](https://github.com/IntelRealSense/realsense-ros/commit/0ca251087a5a5879e3bda86de6871eb8b5f5404c), the parameter `rgb_camera.profile` was renamed to `rgb_camera.color_profile` which breaks the user configuration set for the previous version **silently**.

I found this one but I don't know what else I'm missing which might cause silent bugs in the future.

MartyG-RealSense, 2024-06-04:
Hi @tonynajjar  The ROS wrapper release notes are usually meticulous in their detail.  Whilst it is possible for omissions to occur on rare occasions, in general their documentation of changes is highly reliable.  I appreciate you raising the point.

tonynajjar, 2024-06-04:
I agree they are detailed, the PR is mentioned was not ommited, it is in fact mentioned [here](https://github.com/IntelRealSense/realsense-ros/releases#:~:text=Support%20for%20selecting%20profile%20for%20each%20stream_type%20by%20%40Arun%2DPrasad%2DV%20in%20%233052), my point is not that the Release Notes are problematic, just that it would be helpful to have a Migration Guide i.e. understand which changes in the release notes require a change from the user's configuration. See [Nav2](https://docs.nav2.org/migration/index.html) as an example.

MartyG-RealSense, 2024-06-04:
I will pass your feedback about breaking issues to the RealSense ROS development team.

closed 2024-08-30 by MartyG-RealSense