realsenseai/realsense-ros #3115: Cannot Publish Pointclouds on ROS Noetic under Ubuntu 20.04 on an RPi 4

Austen-G, 2024-06-03:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D435i                       | 
| Firmware Version                   |  5.13.0.50             | 
| Operating System & Version |   Ubuntu 20 | 
| Kernel Version (Linux Only)    |  5.4.0-1109-raspi                                                                       | 
| Platform                                 | Raspberry Pi 4                          |
| Librealsense SDK Version       |  2.50.0                                                     | 
| Language                               |  N/A         | 
| Segment			         |  Robot                          | 
| ROS Distro			         |  Noetic                                           | 
| RealSense ROS Wrapper Version   |  2.3.2                                           | 

### Issue Description
I'm working to get an RPi publishing point-cloud data from a D435i for use with rtab_map on an external system. Before going through the tutorial for using SLAM with D435i, I want to make sure everything is working properly since I've seen a number of issues open on similar topics. 

When running roslaunch realsense2_camera rs_camera.launch filters:=pointcloud initial_reset:=true, using the ros-noetic installation of the ROS1 wrapper I get the following output with nothing on the /camera/depth/color/points topic

I have also verified that the camera works fine when connected to a separate machine with realsense-viewer (The RPi is headless so I can't really do that there) 

I saw some other posts about ensuring different conflicting versions aren't both installed (I get a lot of RUSB and "failed to set power state" when using a librealsense built from source, or when using the version auto-installed when the ros-noetic-realsense2-camera package, but found that installing another version from apt fixed that issue. Here are the installed packages:

librealsense2-dev/focal,now 2.55.1-0~realsense.3336 arm64 [installed]
librealsense2-gl/focal,now 2.55.1-0~realsense.3336 arm64 [installed,automatic]
librealsense2-udev-rules/focal,now 2.55.1-0~realsense.3336 arm64 [installed,automatic]
librealsense2-utils/focal,now 2.55.1-0~realsense.3336 arm64 [installed]
librealsense2/focal,now 2.55.1-0~realsense.3336 arm64 [installed]
ros-noetic-librealsense2/focal,now 2.50.0-1focal.20211115.135230 arm64 [installed,automatic]
ros-noetic-realsense2-camera/focal,now 2.3.2-1focal.20240111.191354 arm64 [installed]

Any ideas? I've sorted through the existing threads and tried a lot of things, but I'm just hoping I've overlooked something. 

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file:
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type:
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
[... 142 lines cut ...]
 03/06 10:42:43,459 WARNING [281472124563824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 10:42:43,510 WARNING [281472124563824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 10:42:43,561 WARNING [281472124563824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 10:42:43,611 WARNING [281472124563824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 10:42:43,662 WARNING [281472124563824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 10:42:43,685 ERROR [281471981953392] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 130
 03/06 10:42:43,712 WARNING [281472124563824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 10:42:43,763 WARNING [281472124563824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 10:42:43,813 WARNING [281472124563824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 10:42:43,864 WARNING [281472124563824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 10:42:43,915 WARNING [281472124563824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 10:42:43,966 WARNING [281472124563824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 10:42:44,319 WARNING [281472124563824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 10:42:44,775 WARNING [281472124563824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ WARN] [1717425764.818181075]: Hardware Notification:Depth stream start failure,1.71743e+12,Error,Hardware Error
 03/06 10:42:44,825 WARNING [281472124563824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 10:42:44,876 WARNING [281472124563824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 10:42:44,926 WARNING [281472124563824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 10:42:45,077 WARNING [281472124563824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 10:42:45,128 WARNING [281472124563824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 10:42:45,279 WARNING [281472124563824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 10:42:45,330 WARNING [281472124563824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 10:42:45,481 WARNING [281472124563824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 10:42:45,532 WARNING [281472124563824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 10:42:46,877 ERROR [281472007131504] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132

MartyG-RealSense, 2024-06-03:
Hi @Austen-G  Raspberry Pi boards can have difficulty with RealSense cameras in regard to features that go beyond providing individual depth and color streams, such as pointclouds and depth-color alignment.

If you are using ROS1 and cannot obtain a pointcloud with rs_camera.launch then an alternate method that works for some RealSense users is to perform an **RGBD** launch.  The steps for doing this are below.

1.  Install support for an RGBD launch on Noetic with this command:

`sudo apt-get install ros-noetic-rgbd-launch`

2.  Launch with the **rs_rgbd.launch** launch file.

`roslaunch realsense2_camera rs_rgbd.launch`

With the rs_rgbd launch the **filters:=pointcloud command is not required.  The pointcloud will be published to `/depth/image/proc`**

Austen-G, 2024-06-03:
I don't see the /depth/image/proc topic, but I do see a /camera/depth_registered/points topic which has data. 

I can't quite visualize this  in rviz yet since I'm still working through getting that to work in WSL1, but since the topic has data I think the issue is resolved, thanks for the quick reply!

Austen-G, 2024-06-03:
Just as a note though, I am getting quite the array of warning/error messages, are these anything to worry about?

 03/06 16:53:12,232 WARNING [281472786579824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 16:53:12,283 WARNING [281472786579824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 16:53:12,334 WARNING [281472786579824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 16:53:12,385 WARNING [281472786579824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 16:53:12,436 WARNING [281472786579824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 16:53:12,486 WARNING [281472786579824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 16:53:12,537 WARNING [281472786579824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 16:53:12,588 WARNING [281472786579824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 16:53:12,639 WARNING [281472786579824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 16:53:12,690 WARNING [281472786579824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 16:53:12,740 WARNING [281472786579824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 16:53:12,791 WARNING [281472786579824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 16:53:12,842 WARNING [281472786579824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 16:53:12,892 WARNING [281472786579824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 16:53:12,943 WARNING [281472786579824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 16:53:12,993 WARNING [281472786579824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 16:53:13,044 WARNING [281472786579824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 16:53:13,094 WARNING [281472786579824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 16:53:13,145 WARNING [281472786579824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 16:53:13,195 WARNING [281472786579824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 16:53:13,246 WARNING [281472786579824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 16:53:13,296 WARNING [281472786579824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 16:53:13,347 WARNING [281472786579824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 16:53:13,397 ERROR [281471604478320] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 130
 03/06 16:53:13,398 WARNING [281472786579824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 16:53:13,448 WARNING [281472786579824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 16:53:13,499 WARNING [281472786579824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 16:53:13,549 WARNING [281472786579824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 16:53:13,601 WARNING [281472786579824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 16:53:13,651 WARNING [281472786579824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ WARN] [1717447994.106889065]: Hardware Notification:Depth stream start failure,1.71745e+12,Error,Hardware Error
 03/06 16:53:14,207 WARNING [281472786579824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[... cut ...]
 03/06 16:55:24,029 WARNING [281472786579824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 16:55:24,080 WARNING [281472786579824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/06 16:55:41,168 WARNING [281472786579824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

MartyG-RealSense, 2024-06-04:
**control_transfer returned** warnings that generate continuously in the log indicate a communication problem between the camera and computer, such as with the USB cable or USB port.  Sometimes topics can be successfully viewed in RViz despite these looping messages though.