realsenseai/realsense-ros #3119: Prefixing camera frames with namespaces in the tf_static topic

TakShimoda, 2024-06-10:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D435i                         | 
| Firmware Version                   | 5.13.0.50                          | 
| Operating System & Version |   Linux (Ubuntu 20.04) | 
| Kernel Version (Linux Only)    |  5.15.0-94-generic                                                                  | 
| Platform                                 | PC/NVIDIA Jetson Nano 4GB                          |
| Librealsense SDK Version       |  2.54.2                                                   | 
| Language                               |  python       | 
| Segment			         |  Robot                             | 
| ROS Distro			         |  foxy                                          | 
| RealSense ROS Wrapper Version   |  4.54.1                                                 | 

### Issue Description
I'm try to prefix the namepace of the tf frames so I can use multiple cameras with separate namespaces. I'm running rs_launch.py with:
`ros2 launch realsense2_camera rs_launch.py unite_imu_method:=2 enable_accel:=true enable_gyro:=true enable_sync:=true gyro_fps:=400 accel_fps:=250 enable_infra1:=true enable_infra2:=true infra_fps:=30.0 depth_module.profile:='640x480x30'`
Then I get the frames with:
`ros2 run tf2_tools view_frames.py`
and I get something like:
[frames.pdf](https://github.com/user-attachments/files/15778637/frames.pdf)
The thing is there seems to be no option for namespacing in the configurable parameters. The only option is to manually change `camera_name` for each one but that would be tedious and I don't think I can use the `/` character like a namespace would. 

As far as I'm aware, this launch file starts up `realsense2_camera_node`, where it eventually uses `robot_state_publisher` and uses the urdf files to publish static transforms to `/tf_static`, but I can't find where this is used, because I think I could set the `tf_prefix` argument with `robot_state_publisher`. Otherwise I was thinking of manually changing the name parameter from "camera" to what I want in the line that defines the macro for the camera [here](https://github.com/IntelRealSense/realsense-ros/blob/ros2-development/realsense2_description/urdf/_d435.urdf.xacro#L29) or adding a namespace parameter (or adding a namespace or ns parameter and prefixing link-names) but again, this would be tedious and done for all robots. 

I was wondering if there was any easy way to namespace each camera so they can all publish to `/tf_static` under different frame names.

Thanks

TakShimoda, 2024-06-11:
Hi @MartyG-RealSense I tried with 4.54.1 that I had installed from packages instead of source code, and it does work like you showed in the link (nodes and topics are namespaced), but it didn't namespace the frames of tf. I was wondering if that was the intention of `camera_namespace` before I tried to install from source.

MartyG-RealSense, 2024-06-11:
I do not know if camera_namespace affects the naming of TF frames (my knowledge of TF is admittedly limited).  I do apologize.

TakShimoda, 2024-06-11:
Ok, do you think there are any alternatives to prefix the tf frames? I could still try if building from source could do that.

MartyG-RealSense, 2024-06-11:
In the ROS1 wrapper it is possible to define a **tf_prefix** parameter for cameras as a launch file arg - as described at https://github.com/IntelRealSense/realsense-ros/issues/1326 - but I do not see a similar parameter for the ROS2 wrapper.

TakShimoda, 2024-06-11:
Thanks for the suggestion. I actually did work with the group_ns tag in ROS1, but I couldn't see a way to replicate that with python launch files in ROS2. For now, I'll try manually changing the xacro files and see if that works.

TakShimoda, 2024-06-11:
I tried modifying the xacro files in /opt/ros with sudo permissions, such as randomly naming the links different names, and then sourcing setup.bash, but I realized they had no effect on the tf frames. I was wondering does the `realsense2_camera_node` even use the xacro files under `realsense2_description` or does it somehow have another way of publishing tf frames to `/tf_static`?

I also tried installing the latest version from source (v4.55.1) on my jetson nano and use it as an overlay (by sourcing `ros2_ws/install/local_setup.bash`) but when I run rs_launch.py I get the following error:
`ERROR [547339889056] (synthetic-stream.cpp:58) Exception was thrown during callback: aMalloc failed status:`
The entire output is:
```
[INFO] [launch]: All log files can be found below /home/jetson/.ros/log/2024-06-11-22-42-04-564030-nano-24047
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [24049]
[realsense2_camera_node-1] [INFO] [1718138526.464266228] [camera.camera]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1718138526.464526753] [camera.camera]: Built with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1718138526.464584827] [camera.camera]: Running with LibRealSense v2.55.1
[realsense2_camera_node-1] [WARN] [1718138526.464627900] [camera.camera]: ***************************************************
[realsense2_camera_node-1] [WARN] [1718138526.464662588] [camera.camera]: ** running with a different librealsense version **
[realsense2_camera_node-1] [WARN] [1718138526.464696287] [camera.camera]: ** than the one the wrapper was compiled with!   **
[realsense2_camera_node-1] [WARN] [1718138526.464732173] [camera.camera]: ***************************************************
[realsense2_camera_node-1] [INFO] [1718138526.527748620] [camera.camera]: Device with serial number 036522070779 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1718138526.527895758] [camera.camera]: Device with physical ID /sys/devices/70090000.xusb/usb2/2-1/2-1.4/2-1.4:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1718138526.527943051] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1718138526.528577228] [camera.camera]: Device with port number 2-1.4 was found.
[realsense2_camera_node-1] [INFO] [1718138526.528643270] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1718138526.528829732] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1718138526.529494534] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1718138526.529566098] [camera.camera]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1718138526.529606463] [camera.camera]: Device Serial No: 036522070779
[realsense2_camera_node-1] [INFO] [1718138526.529640318] [camera.camera]: Device physical port: /sys/devices/70090000.xusb/usb2/2-1/2-1.4/2-1.4:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1718138526.529674121] [camera.camera]: Device FW version: 5.13.0.50
[realsense2_camera_node-1] [INFO] [1718138526.529708080] [camera.camera]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1718138526.529739851] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1718138526.648301582] [camera.camera]: Set ROS param depth_module.depth_profile to default: 848x480x30
[realsense2_camera_node-1] [INFO] [1718138526.649987235] [camera.camera]: Set ROS param depth_module.infra_profile to default: 848x480x30
[realsense2_camera_node-1] [WARN] [1718138526.653317654] [camera.camera]: Could not set param: rgb_camera.power_line_frequency with 3 Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {rgb_camera.power_line_frequency} doesn't comply with integer range.
[realsense2_camera_node-1] [INFO] [1718138526.669521148] [camera.camera]: Set ROS param rgb_camera.color_profile to default: 1280x720x30
[realsense2_camera_node-1] [INFO] [1718138526.671387324] [camera.camera]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [INFO] [1718138526.671528889] [camera.camera]: Set ROS param accel_fps to default: 63
[realsense2_camera_node-1] [INFO] [1718138526.677204711] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1718138526.677495706] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1718138526.700230453] [camera.camera]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1718138526.704944280] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1718138526.707673908] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1718138526.720874124] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 30
[realsense2_camera_node-1] [INFO] [1718138526.735648376] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1]  11/06 22:42:07,299 ERROR [547339889056] (synthetic-stream.cpp:58) Exception was thrown during callback: aMalloc failed status: 
[realsense2_camera_node-1]  11/06 22:42:07,300 ERROR [547339889056] (synthetic-stream.cpp:58) Exception was thrown during callback: aMalloc failed status:
```

[... cut ...]
[realsense2_camera_node-1] [INFO] [1718139784.801293674] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 30
[realsense2_camera_node-1] [INFO] [1718139784.802741196] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1] [WARN] [1718139784.823911576] [camera.camera]:
```
I installed with librealsense with debian and it seems it's version 2.51.1 but with the latest ROS wrapper from source, it seems to run with 2.55.1 and I don't know if that's causing the issue.

MartyG-RealSense, 2024-06-12:
The xacro files are usually used for ROS robot simulations such as Gazebo.  

Regarding the lines below, they indicate that when 4.55.1 was built from source, the installed librealsense version at the time was 2.51.1.  Then librealsense 2.55.1 was installed but the ROS wrapper was not built again afterwards.   Each time that the librealsense version is changed, the ROS wrapper has to be reinstalled or you get different librealsense versions listed for 'built with' and 'running with'.

```
RealSense ROS v4.55.1
Built with LibRealSense v2.51.1
Running with LibRealSense v2.55.1
```

The librealsense version that should be used with wrapper 4.55.1 is librealsense 2.55.1.

TakShimoda, 2024-06-12:
> These log messages indicate that when 4.55.1 was built from source, the installed librealsense version at the time was 2.51.1. Then librealsense 2.55.1 was installed but the ROS wrapper was not built again afterwards. Each time that the librealsense version is changed, the ROS wrapper has to be reinstalled or you get different librealsense versions listed for 'built with' and 'running with'.

I don't recall installing librealsense again. I installed it once from [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md) when I first set up my jetson. I'm assuming when I installed ROS wrapper 4.55.1 from source, it automatically installed librealsense 2.55.1 but for some reason didn't build with it. 

It's strange because when I look at my default directories for librealsense, I see version 2.55, such as `/usr/lib/aarch64-linux-gnu/librealsense2.so.2.55` but I don't see 2.51.1 which I initially installed. I also rebuilt ROS wrapper 4.55.1 with only librealsense 2.55.1 in my system and it still gives the same errors. 

I was wondering if this was related to the error messages I was getting and how to fix those.

Also, since urdf is only for simulation, do you know where in the code it publishes tf data for the physical camera?

MartyG-RealSense, 2024-06-12:
If the ROS wrapper is built from source then it does not also install librealsense.  With a source build of the wrapper, you first install librealsense from source or packages and then build the ROS wrapper from source afterwards.

If you have done multiple installations on the same computer then it can sometimes be easiest just to wipe the entire storage and install everything again from the beginning, including Ubuntu.  Otherwise, you can end up with hidden glitches and tangles that defy all debugging attempts.

The wrapper file **tfs.cpp** looks like it has content relevant to TFs.

https://github.com/IntelRealSense/realsense-ros/blob/d1684b4e53111f5eb3237f76299e83a864ce929d/realsense2_camera/src/tfs.cpp#L231-L236

TakShimoda, 2024-06-12:
Thanks Marty, I have ROS wrapper 4.55.1 working on my desktop and modified some lines in tfs.cpp and it does seem to be able to rename some TF frames and even use the "/" character, so I guess I could work with it. 

I'll also try your suggestion of doing a clean install to try to get the source code version working on my jetson.

closed 2024-06-12 by TakShimoda

cjstahoviak, 2025-05-28:
For anyone looking for a solution to this that does not want to edit the source code, there is another approach.

I added a tf_prefix parameter that uses ROS2's built-in topic remapping to redirect /tf and /tf_static to namespaced topics:
```python
{'name': 'tf_prefix', 'default': "''", 'description': 'prefix for the tf frame IDs'},
```

Then in the node setup, right before the return statement:
```python
remappings = []
if _tf_prefix and _tf_prefix != "''":
    remappings = [
        ('/tf', f'/{_tf_prefix}/tf'),
        ('/tf_static', f'/{_tf_prefix}/tf_static')
    ]
```

And finally added the remapping to the node:
```python
launch_ros.actions.Node(
    package='realsense2_camera',
    namespace=LaunchConfiguration('camera_namespace' + param_name_suffix),
    name=LaunchConfiguration('camera_name' + param_name_suffix),
    executable='realsense2_camera_node',
    parameters=[params, params_from_file],
    output=_output,
    arguments=['--ros-args', '--log-level', LaunchConfiguration('log_level' + param_name_suffix)],
    emulate_tty=True,
     remappings=remappings,
     )
```

This approach is backward compatible, requires no source changes, and allows the publication to tf trees in a namespace. @MartyG-RealSense