realsenseai/realsense-ros #3120: ROS2 Lifecycle node

quantumxt, 2024-06-11:
Hi there, 

We are interested in utilizing the `realsense-ros` nodes as ROS2 Lifecycle Nodes to accommodate the dynamic addition and removal of Realsense cameras within our system at runtime. Previously, a question was raised regarding this feature (#1838), with a projected completion date of Q4 2023 (https://github.com/IntelRealSense/realsense-ros/issues/1838#issuecomment-1646757580). However, there have been no updates since July 2023.

Given this context, could you please provide an update on the status of this feature?

Thank you.

MartyG-RealSense, 2024-06-11:
Hi @quantumxt  https://github.com/IntelRealSense/realsense-ros/issues/1838 is the best place to ask this question.  Please re-post your message there as a new comment so that Intel RealSense team member SamerKhshiboun will be alerted to it.

MartyG-RealSense, 2024-06-18:
Case closed due to being handled at https://github.com/IntelRealSense/realsense-ros/issues/1838

closed 2024-06-18 by MartyG-RealSense