realsenseai/realsense-ros #3121: D455 camera cannot get imu msg from rostopic

Hanning-Liu, 2024-06-11:
| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | { D455}                          | 
| Firmware Version                   | (5.16.0.1)                          | 
| Operating System & Version |   { Ubuntu 20.04.6 LTS } | 
| Kernel Version (Linux Only)    |  (5.15.0-107-generic)                                                                       | 
| Platform                                 | NUC                      |
| Librealsense SDK Version       |  { 5.16.0.1 }                                                    | 
| Language                               |  {bash}         | 
| Segment			         |  {others }                             | 
| ROS Distro			         |  {ROS 1 noetic}                                           | 
| RealSense ROS Wrapper Version   |  {2.3.2}                                                  | 

### Issue Description
I launch my D455 camera using this launch file:
```

<launch>
  <arg name="serial_no"           default="242422302661"/>
  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default="d455"/>
  <arg name="json_file_path"      default=""/>
  <arg name="camera"              default="d455_242422302661"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>
  <arg name="output"              default="screen"/>
  <arg name="respawn"              default="true"/>

  <arg name="fisheye_width"       default="-1"/>
  <arg name="fisheye_height"      default="-1"/>
  <arg name="enable_fisheye"      default="false"/>

  <arg name="depth_width"         default="848"/>
  <arg name="depth_height"        default="480"/>
  <arg name="enable_depth"        default="true"/>

  <arg name="confidence_width"    default="-1"/>
  <arg name="confidence_height"   default="-1"/>
  <arg name="enable_confidence"   default="true"/>
  <arg name="confidence_fps"      default="-1"/>

  <arg name="infra_width"         default="848"/>
  <arg name="infra_height"        default="480"/>
  <arg name="enable_infra"        default="false"/>
  <arg name="enable_infra1"       default="false"/>
  <arg name="enable_infra2"       default="false"/>
  <arg name="infra_rgb"           default="false"/>

  <arg name="color_width"         default="848"/>
  <arg name="color_height"        default="480"/>
  <arg name="enable_color"        default="true"/>

  <arg name="fisheye_fps"         default="-1"/>
  <arg name="depth_fps"           default="30"/>
  <arg name="infra_fps"           default="30"/>
  <arg name="color_fps"           default="30"/>
  <arg name="gyro_fps"            default="400"/>
  <arg name="accel_fps"           default="200"/>
  <arg name="enable_gyro"         default="true"/>
  <arg name="enable_accel"        default="true"/>

  <arg name="enable_pointcloud"         default="false"/>
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_COLOR"/>
  <arg name="pointcloud_texture_index"  default="0"/>
  <arg name="allow_no_texture_points"   default="false"/>
  <arg name="ordered_pc"                default="false"/>

  <arg name="enable_sync"               default="false"/>
  <arg name="align_depth"               default="true"/>
[... 348 lines cut ...]
/tf_static
```

But I can get any message from the topic:

```

henry@henry-NUC8v5PNK:~/kalibr_workspace/test_VIO$ rostopic info /d455_242422302661/imu
Type: sensor_msgs/Imu

Publishers: 
 * /d455_242422302661/realsense2_camera_manager (http://henry-NUC8v5PNK:33605/)

Subscribers: None

henry@henry-NUC8v5PNK:~/kalibr_workspace/test_VIO$ rostopic hz /d455_242422302661/imu
subscribed to [/d455_242422302661/imu]
no new messages
no new messages
no new messages
no new messages
no new messages
```

Why could this happen, how to solve it? Thanks for your help in advance!

MartyG-RealSense, 2024-06-11:
Hi @Hanning-Liu  If you are using ROS1 Noetic with the 2.3.2 ROS1 wrapper then your librealsense SDK version should be 2.50.0 or 2.51.1 and your firmware driver version should be 5.13.0.50.  This is because the ROS1 wrapper is no longer updated (only the ROS2 wrapper is actively updated), so the ROS1 wrapper has not been updated for recent librealsense and firmware versions.

Hanning-Liu, 2024-06-12:
@MartyG-RealSense Thank you for your help!

I can get the IMU data from `rostopic` after reinstalling the librealsense SDK. But my librealsense SDK version is not the same as what you mentioned.
Here is my version infomation after I input the `roslaunch` command:
[ INFO] [1718157091.132644463]: RealSense ROS v2.3.2
[ INFO] [1718157091.132736139]: Built with LibRealSense v2.55.1
[ INFO] [1718157091.132788226]: Running with LibRealSense v2.55.1

```
henry@henry-NUC8v5PNK:~/kalibr_workspace/test_VIO$ roslaunch custonD455.launch 
... logging to /home/henry/.ros/log/7287cd86-285c-11ef-b304-537834ed4b16/roslaunch-henry-NUC8v5PNK-8623.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://henry-NUC8v5PNK:46369/

SUMMARY
========

PARAMETERS
 * /d455_242422302566/realsense2_camera/accel_fps: 100
 * /d455_242422302566/realsense2_camera/accel_frame_id: d455_242422302566...
 * /d455_242422302566/realsense2_camera/accel_optical_frame_id: d455_242422302566...
 * /d455_242422302566/realsense2_camera/align_depth: True
 * /d455_242422302566/realsense2_camera/aligned_depth_to_color_frame_id: d455_242422302566...
 * /d455_242422302566/realsense2_camera/aligned_depth_to_fisheye1_frame_id: d455_242422302566...
 * /d455_242422302566/realsense2_camera/aligned_depth_to_fisheye2_frame_id: d455_242422302566...
 * /d455_242422302566/realsense2_camera/aligned_depth_to_fisheye_frame_id: d455_242422302566...
 * /d455_242422302566/realsense2_camera/aligned_depth_to_infra1_frame_id: d455_242422302566...
 * /d455_242422302566/realsense2_camera/aligned_depth_to_infra2_frame_id: d455_242422302566...
 * /d455_242422302566/realsense2_camera/allow_no_texture_points: False
 * /d455_242422302566/realsense2_camera/base_frame_id: d455_242422302566...
 * /d455_242422302566/realsense2_camera/calib_odom_file: 
 * /d455_242422302566/realsense2_camera/clip_distance: -2.0
 * /d455_242422302566/realsense2_camera/color_fps: 30
 * /d455_242422302566/realsense2_camera/color_frame_id: d455_242422302566...
 * /d455_242422302566/realsense2_camera/color_height: 480
 * /d455_242422302566/realsense2_camera/color_optical_frame_id: d455_242422302566...
 * /d455_242422302566/realsense2_camera/color_width: 848
 * /d455_242422302566/realsense2_camera/confidence_fps: -1
 * /d455_242422302566/realsense2_camera/confidence_height: -1
 * /d455_242422302566/realsense2_camera/confidence_width: -1
 * /d455_242422302566/realsense2_camera/depth_fps: 30
 * /d455_242422302566/realsense2_camera/depth_frame_id: d455_242422302566...
 * /d455_242422302566/realsense2_camera/depth_height: 480
 * /d455_242422302566/realsense2_camera/depth_optical_frame_id: d455_242422302566...
 * /d455_242422302566/realsense2_camera/depth_width: 848
 * /d455_242422302566/realsense2_camera/device_type: d455
 * /d455_242422302566/realsense2_camera/enable_accel: True
 * /d455_242422302566/realsense2_camera/enable_color: True
 * /d455_242422302566/realsense2_camera/enable_confidence: True
 * /d455_242422302566/realsense2_camera/enable_depth: True
 * /d455_242422302566/realsense2_camera/enable_fisheye1: False
 * /d455_242422302566/realsense2_camera/enable_fisheye2: False
 * /d455_242422302566/realsense2_camera/enable_fisheye: False
 * /d455_242422302566/realsense2_camera/enable_gyro: True
 * /d455_242422302566/realsense2_camera/enable_infra1: False
 * /d455_242422302566/realsense2_camera/enable_infra2: False
 * /d455_242422302566/realsense2_camera/enable_infra: False
 * /d455_242422302566/realsense2_camera/enable_pointcloud: False
 * /d455_242422302566/realsense2_camera/enable_pose: False
 * /d455_242422302566/realsense2_camera/enable_sync: False
 * /d455_242422302566/realsense2_camera/filters: pointcloud
 * /d455_242422302566/realsense2_camera/fisheye1_frame_id: d455_242422302566...
 * /d455_242422302566/realsense2_camera/fisheye1_optical_frame_id: d455_242422302566...
 * /d455_242422302566/realsense2_camera/fisheye2_frame_id: d455_242422302566...
 * /d455_242422302566/realsense2_camera/fisheye2_optical_frame_id: d455_242422302566...
 * /d455_242422302566/realsense2_camera/fisheye_fps: -1
 * /d455_242422302566/realsense2_camera/fisheye_frame_id: d455_242422302566...
[... 72 lines cut ...]
[ INFO] [1718157091.253667667]: Sync Mode: On
[ INFO] [1718157091.253724294]: Device Sensors: 
[ INFO] [1718157091.259303179]: Stereo Module was found.
[ INFO] [1718157091.263152596]: RGB Camera was found.
[ INFO] [1718157091.263286937]: Motion Module was found.
[ INFO] [1718157091.263310482]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1718157091.263349799]: Add Filter: pointcloud
[ INFO] [1718157091.263387847]: num_filters: 2
[ INFO] [1718157091.263404505]: Setting Dynamic reconfig parameters.
[ INFO] [1718157091.306693737]: Done Setting Dynamic reconfig parameters.
[ INFO] [1718157091.307152084]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1718157091.307552033]: color stream is enabled - width: 848, height: 480, fps: 30, Format: RGB8
[ INFO] [1718157091.308561047]: gyro stream is enabled - fps: 200
[ INFO] [1718157091.308594060]: accel stream is enabled - fps: 100
[ INFO] [1718157091.308615792]: setupPublishers...
[ INFO] [1718157091.310697602]: Expected frequency for depth = 30.00000
[ INFO] [1718157091.339652957]: Expected frequency for color = 30.00000
[ INFO] [1718157091.358401111]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1718157091.377659709]: Start publisher IMU
[ INFO] [1718157091.378436299]: setupStreams...
 12/06 09:51:31,379 WARNING [140199745025792] (ds-calib-parsers.cpp:35) IMU Calibration is not available, default intrinsic and extrinsic will be used.
[ INFO] [1718157091.400911424]: SELECTED BASE:Depth, 0
[ INFO] [1718157091.411113764]: RealSense Node Is Up!

```

MartyG-RealSense, 2024-06-12:
The most recent librealsense version meant to be used with wrapper 2.3.2 is 2.51.1.  However, if the wrapper works well with 2.3.2 on your particular computer then the mis-match between versions does not matter.  I'm pleased to hear that you were successful  :)

MartyG-RealSense, 2024-06-21:
Hi @Hanning-Liu  Do you require further assistance with this case, please?  Thanks!

Hanning-Liu, 2024-06-21:
@MartyG-RealSense Thank you for your help! Please close this issue!

MartyG-RealSense, 2024-06-21:
You are very welcome.  Thanks very much  :)

closed 2024-06-21 by MartyG-RealSense

kevin-chau, 2024-07-20:
@Hanning-Liu @MartyG-RealSense I'm running into a similar issue. How did you update the librealsense to 2.55.1? I can't figure out how to build realsense-ros 2.3.2 with any other version than librealsense 2.50.0

MartyG-RealSense, 2024-07-20:
Hi @kevin-chau  A simple way to build librealsense 2.55.1 would be to use the **libuvc backend** build script at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

You would then have to build the ROS wrapper 2.3.2 from source code afterwards using the **Method 2** instructions.

I should emphasize though that 2.55.1 is not designed for use with wrapper 2.3.2 as it should be used with 2.50.0 or 2.51.1.  2.55.1 and firmware 5.16.0.1 do sometimes work with 2.3.2 though, even though they are not meant to.

kevin-chau, 2024-07-22:
OK thanks! I'll try to build 2.51.1 with wrapper 2.3.2.