realsenseai/realsense-ros #3123: Error with "roslaunch realsense2_camera opensource_tracking.launch"

Shrioe, 2024-06-13:
Hi,

I am new to VSLAM. I am following the instructions of "SLAM with D435i tutorial"
roslaunch realsense2_camera opensource_tracking.launch,

I encountered some errors as shown below:

ERROR] [1718263335.611787184]: Critical Error, NaNs were detected in the output state of the filter. This was likely due to poorly conditioned process, noise, or sensor covariances.
[ WARN] [1718263335.612158811]: Failed to meet update rate! Took 0.0086891700000000012677
Error:   TF_NAN_INPUT: Ignoring transform for child_frame_id "camera_link" from authority "unknown_publisher" because of a nan value in the transform (nan nan nan) (-nan nan -nan nan)
         at line 240 in /tmp/binarydeb/ros-noetic-tf2-0.7.7/src/buffer_core.cpp
Error:   TF_DENORMALIZED_QUATERNION: Ignoring transform for child_frame_id "camera_link" from authority "unknown_publisher" because of an invalid quaternion in the transform (-nan nan -nan nan)
         at line 255 in /tmp/binarydeb/ros-noetic-tf2-0.7.7/src/buffer_core.cpp

And I also faced issue when trying to replay the rosbag file after running the line: "roslaunch realsense2_camera opensource_tracking.launch offline:=true" , whereby it will only open a new rviz file without showing anything. May I ask what am I supposed to get at this step?
[image]

Could you give me some suggestion？

Thx!

MartyG-RealSense, 2024-06-13:
Hi @Shrioe  After the D435i SLAM guide was published, the gyro and accel IMU topics were later set to disabled by default in the RealSense ROS wrapper.  Please try enabling the topics using the launch instruction below to see whether it makes a difference to the errors that you are receiving.

`roslaunch realsense2_camera opensource_tracking.launch enable_accel:=true enable_gyro:=true unite_imu_method:=linear_interpolation`

Shrioe, 2024-06-13:
Hello,

Thank you for your reply! Just to let you know, I'm using a Raspberry Pi 4B with Ubuntu 20.04.

Recently, I tried reinstalling everything, but I encountered new error messages, as shown in the figure below.

Now, when RViz appears, the display panel only shows "All enabled" under "frame". The options for "map" and "camera_link" are missing.

Additionally, the image is not displayed even though I typed "/camera/color/image_raw".

Could you please provide some suggestions?

Thank you so much!
[image]

Shrioe, 2024-06-13:
Hello,

@MartyG-RealSense Thank you for your reply! Just to let you know, I'm using a Raspberry Pi 4B with Ubuntu 20.04.

Recently, I tried reinstalling everything, but I encountered new error messages, as shown in the figure below.

Now, when RViz appears, the display panel only shows "All enabled" under "frame". The options for "map" and "camera_link" are missing.

Additionally, the image is not displayed even though I typed "/camera/color/image_raw".

Could you please provide some suggestions?

Thank you so much!

[image]

Shrioe, 2024-06-13:
Hello,

Thank you for your reply! Just to let you know, I'm using a Raspberry Pi 4B with Ubuntu 20.04.

For ypur information after implementing the code you mentioned above the errors are still there, and the rviz will close by itself automatically after launching for few seconds.

Could you please provide some suggestions?

Thank you so much!

MartyG-RealSense, 2024-06-14:
At https://github.com/IntelRealSense/realsense-ros/issues/2494 another RealSense ROS user with Ubuntu 20.04 experienced `Critical Error, NaNs were detected in the output state of the filter` on a PC, so the cause may not be that you are using a Pi 4 board.  SLAM in general is a processing-intensive activity, so this error could occur even on a full-size computer.

Looking at the full list of errors, most of them look like what I woud call 'cascade errors' caused by the first NaN error.  In other words, if the initial 'Critical Error: NaNs were detected' error had not occurred then the subsequent errors that refer to NaN likely would not occur.

The 'Failed to meet update rate!' warning might also be occurring because of the Critical Error.

Does stability improve if you use `unite_imu_method:=copy` in your launch instruction instead of setting it to linear_interpolation.

As you are using a Pi 4b, that could also be a cause of the errors because RealSense cameras usually have problems with accessing IMU data when used with Raspberry Pi boards.

jingweiteo, 2024-06-15:
Hello @MartyG-RealSense , by referring to #2494, i tried ti edit the rs_camera.launch file as shown below: 
------------------------------------------------------------------------
  <arg name="depth_width"         default="640"/>
  <arg name="depth_height"        default="480"/>
  <arg name="enable_depth"        default="true"/>

  <arg name="confidence_width"    default="-1"/>
  <arg name="confidence_height"   default="-1"/>
  <arg name="enable_confidence"   default="true"/>
  <arg name="confidence_fps"      default="-1"/>

  <arg name="infra_width"         default="640"/>
  <arg name="infra_height"        default="480"/>
  <arg name="enable_infra"        default="false"/>
  <arg name="enable_infra1"       default="false"/>
  <arg name="enable_infra2"       default="false"/>
  <arg name="infra_rgb"           default="false"/>

  <arg name="color_width"         default="640"/>
  <arg name="color_height"        default="480"/>
  <arg name="enable_color"        default="true"/>

  <arg name="fisheye_fps"         default="-1"/>
  <arg name="depth_fps"           value="15"/>
  <arg name="infra_fps"           value="15"/>
  <arg name="color_fps"           value="15"/>
  <arg name="gyro_fps"            value="200"/>
  <arg name="accel_fps"           value="200"/>
  <arg name="enable_gyro"         value="true"/>
  <arg name="enable_accel"        value="true"/>

  <arg name="enable_pointcloud"         default="false"/>
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_COLOR"/>
  <arg name="pointcloud_texture_index"  default="0"/>
  <arg name="allow_no_texture_points"   default="false"/>
  <arg name="ordered_pc"                default="false"/>

  <arg name="enable_sync"               default="false"/>
  <arg name="align_depth"               default="false"/>

  <arg name="publish_tf"                default="true"/>
  <arg name="tf_publish_rate"           default="0"/>

  <arg name="filters"                   default=""/>
  <arg name="clip_distance"             default="-2"/>
  <arg name="linear_accel_cov"          default="0.01"/>
  <arg name="initial_reset"             default="false"/>
------------------------------------------------------------------------

But when I run "roslaunch realsense2_camera opensource_tracking.launch enable_accel:=true enable_gyro:=true unite_imu_method:=linear_interpolation", there is no more NaN values errors but still: 
"
15/06 14:26:33,788 WARNING [281472450236784] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
 [ WARN] [1718432786.188037476]: Failed to meet update rate! Took 0.0067153130000000005317
 15/06 14:26:33,890 WARNING [281472450236784] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b

In rviz, 
1. There is no map shown. 
2. I cannot tick the "map" or "camera_link" because under "frame", there is only "All enabled". 
3. Then, under "image", there is no image shown even though i changed to "cameracolor/image_raw"
4.

Shrioe, 2024-06-15:
Hello @MartyG-RealSense , by referring to #2494, i tried to edit the rs_camera.launch file as shown below:
------------------------------------------------------------------------
But when I run "roslaunch realsense2_camera opensource_tracking.launch enable_accel:=true enable_gyro:=true unite_imu_method:=linear_interpolation", there is no more NaN values errors but still:
"
15/06 14:26:33,788 WARNING [281472450236784] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
[ WARN] [1718432786.188037476]: Failed to meet update rate! Took 0.0067153130000000005317
15/06 14:26:33,890 WARNING [281472450236784] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b

In rviz,

There is no map shown.
I cannot tick the "map" or "camera_link" because under "frame", there is only "All enabled".
Then, under "image", there is no image shown even though i changed to "cameracolor/image_raw"

[image]

MartyG-RealSense, 2024-06-15:
There is a past case of the **opensource_tracking.launch** file at the link below that experienced the same warnings that you had.  In their particular case the problem was apparently caused by the USB port that they were using. 

https://support.intelrealsense.com/hc/en-us/community/posts/8338902895379-D435I-IMU-ISSUE

Shrioe, 2024-06-15:
I using the cable given while I bought the d435i, and I have check the USB device details:
[image]

Shrioe, 2024-06-15:
I have tried solutions from #2494, #2386, and https://support.intelrealsense.com/hc/en-us/community/posts/8338902895379-D435I-IMU-ISSUE, already, since the image cannot appear if I modify enable_gyro from false to true, I only adjusted the fps value only, and when I was trying to run "roslaunch realsense2_camera", it looks like everythingrevert back tk my initial problems which is the "NaN error", and now the rviz will close automatically by itself after it pop out for few seconds only

MartyG-RealSense, 2024-06-15:
The **unite_imu_method** command is unite_imu_method:=linear_interpolation

The D435i SLAM guide is very old and its component installation instructions are for ROS1 Kinetic from around the year 2016.  For newer ROS1 versions such as Noetic I recommend adjusting the installation instructions to use the newer Noetic versions of the components if you have not done so already.

mu_filter_madgwick: sudo apt-get install ros-noetic-imu-filter-madgwick
rtabmap_ros: sudo apt-get install ros-noetic-rtabmap-ros
robot_localization: sudo apt-get install ros-noetic-robot-localization

***********

An alternative to the D435i SLAM guide that does not need an IMU may be **ORB_SLAM2_ROS**, which is compatible with ROS1 and the IMU-less D435 camera model.

https://github.com/appliedAI-Initiative/orb_slam_2_ros

There is also a newer version called ORB_SLAM3 that has been tested with Noetic.

https://github.com/thien94/orb_slam3_ros

Shrioe, 2024-06-15:
I'm currently using ros noetic and I installed 
mu_filter_madgwick: sudo apt-get install ros-noetic-imu-filter-madgwick
rtabmap_ros: sudo apt-get install ros-noetic-rtabmap-ros
robot_localization: sudo apt-get install ros-noetic-robot-localization
Too from the beginning. 

Actually I was previously working on orbslam3 too for few months, but I always failed to build the orbslam3 folder after many trials,  that's why I switch to the website SLAM with d435i.

MartyG-RealSense, 2024-06-15:
I would not recommend going back to Ubuntu 16.04 and Kinetic as the Kinetic branch of ROS1 is End of Life (retired).

Because of the problems with accessing the IMU of RealSense cameras on Raspberry Pi boards, any solution that uses gyro and accel data likely would not be suitable for Pi 4 or Pi 5.

Instead of using ROS, perhaps you could record a bag file of 2D data in the **RealSense Viewer** tool using its 'Record' button.   You could then play back the bag in the Viewer and switch into the Viewer's 3D pointcloud mode to render the 2D bag data as a 3D pointcloud in real-time.  The 3D pointcloud mode enables the pointcloud to be exported to a **.ply** pointcloud file that can then be loaded into 3D software for further editing, such as converting the pointcloud into a solid object.

Shrioe, 2024-06-15:
I see thank you so much for your clarification! May i ask is  there any suitable orbuseful website I can refer to on how to use realsensen viewer to record and process 2d data into a 3d point cloud?

MartyG-RealSense, 2024-06-15:
The bag recording process in the Viewer is simple.

1.  Enable the stream types that you want to record to bag file and then click on the **Record** button at the top of the Viewer's options side-panel to begin recording those streams.

[image]

2.  To stop recording and create the bag file, click the Record button again.

3.  To play back the bag file in the Viewer, you can drag and drop the bag file into the center panel of the Viewer to automatically start playback.

4.  Once playback has begun, click on the **3D** option in the top corner of the Viewer window to switch to 3D pointcloud mode and automatically render the 2D data as a pointcloud.

[image]

5.  Click the **Export** button at the top center of the Viewer window during playback to export to a pointcloud .ply file.

[image]

*********

To find the location on your computer that the bag file will be saved to, click on the gear-wheel icon at the top corner of the Viewer and select **Settings** from its drop-down menu to open the settings configuration interface.

[image]

When the settings interface pops up, click on the **Playback and Record** tab and look at the text beside 'Default recording folder'.

[image]

*****

You might also find the link below's guide to exporting a 3D pointcloud from the Viewer's 3D mode to .ply file helpful.

https://www.andreasjakl.com/capturing-3d-point-cloud-intel-realsense-converting-mesh-meshlab/

jingweiteo, 2024-06-15:
Thank you so much for the guidance! I think I might still need to form a final map from the .ply file

Shrioe, 2024-06-15:
Thank you so much for the guidance! I think I might still need to form a final map from the .ply file

MartyG-RealSense, 2024-06-16:
1.  MeshLab is not a SLAM tool.  If you need to do VSLAM and cannot use the IMU then **stella-vslam** would be worth looking at as it does not require an IMU.

https://github.com/stella-cv/stella_vslam

2.  The procedure from the D435i SLAM guide that you describe is correct.  However, bag files recorded with RealSense SDK programs such as the RealSense Viewer have some differences to ROS rosbags and so may not be fully compatible with playback in ROS.

3.  When the `opensource_tracking.launch` file is launched, it imports settings from the rs_camera.launch file and configures some additional settings that enable the camera to be used with RTABMAP.

The D435i SLAM guide has not been updated since January 2019, so it cannot be guaranteed that it will work correctly in the present day.   

When It is functioning correctly, it is meant to look like the rotating map image under the text **and the resulting point cloud** in the guide.

[image]

4.  Please paste your image into the comment box again as it did not display in your last comment.

Shrioe, 2024-06-16:
Thank you fr your response! 

2. Is that mean I cannot proceed with the ros code in slam with d435i guide using the .bag file recorded in relsensenviewer and I must go through the recording using "roslaunch realsense2_camera opensource_tracking.launch"?

3. I think the one i dont understand is about the code "roslaunch realsense2_camera opensource_tracking.launch offline:=true", which is run after replay the rosbag, normally even if I have done recorded a bag file using rviz, then after I reply the recorded rosbag and run this line of code: "roslaunch realsense2_camera opensource_tracking.launch offline:=true", I can't gt any outpyr that allows me to form the 2d map or pointcloud because I think in the slamwith d435i it says: when the system is up, then only I cannot run the code to save the 2d map, therefore may I get some advices from you at this step? Regarding how can I convert my recorded .bag file (not from realsense-viewer) to the 2d map?

MartyG-RealSense, 2024-06-16:
2.  I think whether a bag file recorded in the Viewer will play in ROS will depend on the contents of the messages stored in the bag.  There is no harm in trying playback of the file in ROS to see what happens.
 
3.  I believe that you would need to include the IMU topic enabling commands in the offline launch in order for the bag's IMU data to be published.

`roslaunch realsense2_camera opensource_tracking.launch enable_gyro:=true enable_accel:=true unite_imu_method:=linear_interpolation offline:=true`

Shrioe, 2024-06-16:
2. Yes I have tried to run the .bag file I recorded through realsense viewer,  and followed ny the code:"roslaunch realsense2_camera opensource_tracking.launch offiline:=true." It's  just that we go back to the same question as I mentioned above in number 3, even though i can replay the bag, but I don't know how to use this code to get the final map

3. Alright will try

Shrioe, 2024-06-17:
Hello @MartyG-RealSense , any update regarding number 3?

MartyG-RealSense, 2024-06-17:
If the D435i SLAM guide and ORB_SLAM do not work for you then the last available option with a chance of success may be the SLAM tool **Kimera-VIO-ROS**, which is compatible with Noetic.

https://github.com/MIT-SPARK/Kimera-VIO-ROS

Shrioe, 2024-06-17:
Alright thank you for the suggestions,  I'll try for that!

Anyway based on my previous comment, can you tell me whether I'm doing the replay of rosbag and forming the map in a correct way?

MartyG-RealSense, 2024-06-17:
I did not see a problem in your replay procedure.  However, because a RealSense SDK bag was being used instead of a ROS rosbag it may not succeed because of the differences between SDK bags and rosbags.

Shrioe, 2024-06-17:
Actually the same thing happens even if I use the ROS rosbag too.

MartyG-RealSense, 2024-06-17:
I think that having ros-noetic-librealsense2 and ros-noetic-realsense2-camera on your computer means that the librealsense SDK and the ROS wrapper were both installed at the same time with the command below.

`sudo apt-get install ros-$ROS_DISTRO-realsense2-camera`

This installation method is called **Method 1**.  Have you used this method, please?

Method 1 does not install realsense-viewer though.  So if you used Method 1 and also have realsense-viewer on your computer, this would suggest that librealsense had been previously installed by a different method.  In that situation there could be two different librealsense installations on the computer that are conflicting with each other.

Shrioe, 2024-06-17:
I followed the ros 1 legacy as suggested in the link: https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy

And yes I have tried sudo apt-get install ros-$ROS_DISTRO-realsense2-camera.

And yes I can also launch the realsense viewer on my ubuntu, is that mean I already have two redundant librealsense installation? 

I gt the below messages in the terminal:

pi@raspberrypi:-$ dpkg -1 | grep realsense

ii ros-noetic-librealsense2

2.50.0-1focal.20211115.135230

arm64 Library for capturing data from the Intel(R) RealSense(TM) S R300, D400, L500 Depth cameras and T2xx Tracking devices.

2.3.2-1focal.20240111.191354

ii ros-noetic-realsense2-camera arm64 RealSense Camera package allowing access to Intel T265 Track ing module and SR300 and D400 3D cameras

MartyG-RealSense, 2024-06-17:
If you have realsense-viewer then yes, it is likely that you have two librealsense installations on your computer that were installed by different methods.

A table of librealsense versions at https://github.com/IntelRealSense/realsense-ros/issues/3026#issuecomment-1972696034 that was produced by the same grep command that you used and has the same version numbers as you indicates to me that you have not got more than one librealsense version installed.  

When you run realsense-viewer, which version number is it?  The version number will be at the top of the viewer window.

Shrioe, 2024-06-17:
For your information, I have these info:
" Realsense Ros v2.3.2
Built with librealsense v2.50.0
Running with librealsense v2.50.0
Device usb type: 3.2"

Shrioe, 2024-06-17:
But how about the roslaunch realsense2_camera opensource_tracking.launch?

MartyG-RealSense, 2024-06-18:
opensource_tracking.launch contains settings that enable the camera to be used with RTABMAP, so I guess that if `rtabmap.launch` is used with rs_camera.launch (which lacks those RTABMAP settings) then rtabmap.launch takes care of configuring those settings instead.

closed 2024-07-04 by MartyG-RealSense