realsenseai/realsense-ros #3127: Connect to device via physical ID

tonynajjar, 2024-06-14:
Currently we can connect to a device via serial number, device type or usb port. I have 2 D457 GMSL/FAKRA that need to be recognized so I can only use the serial number option. I'd like to ask for the feature to lookup via the physical id. Thanks

> [realsense2_camera_node-4] [INFO] [1718371350.148631351] [perception.front_left]: Device with physical ID /dev/video-rs-depth-1 was found.
[realsense2_camera_node-4] [INFO] [1718371350.148690520] [perception.front_left]: Device with name Intel RealSense D457 was found.

MartyG-RealSense, 2024-06-14:
Hi @tonynajjar  What do you believe the advantages would be of finding a D457 by its physical ID compared to using the serial number, please?

tonynajjar, 2024-06-14:
Same advantage as with specifying device type - when you have a fleet of robot and the cameras are always hooked to the same ports (or the udev rules remaps symlinks to knowns ports), then we don't have to configure the serial numbers for each robot if we use the physical ID

MartyG-RealSense, 2024-06-14:
Thanks very much.  I will highlight your request to the RealSense ROS team.

dmipx, 2024-06-17:
Hi @tonynajjar.
We want to understand better what specific feature can help you.
For now, it is looking like that:
```
rs-enumerate-devices
    Name                          :     Intel RealSense D455
    Physical Port                 :     [/sys/devices/platform/bus@0/3610000.usb/usb2/2-3/2-3.2/2-3.2.1/2-3.2.1:1.0/video4linux/video6](mailto:/sys/devices/platform/bus@0/3610000.usb/usb2/2-3/2-3.2/2-3.2.1/2-3.2.1:1.0/video4linux/video6)

    Name                          :     Intel RealSense D457
    Physical Port                 :     /dev/video-rs-depth-0

```
The actual device path is:
```
ls -l /sys/class/video4linux/
(GMSL) video0 -> /sys/devices/platform/tegra-capture-vi/video4linux/video0
(USB) video6 -> /sys/devices/platform/bus@0/3610000.usb/usb2/2-3/2-3.2/2-3.2.1/2-3.2.1:1.0/video4linux/video6
```
Would you like to have pointers for each camera, USB and GMSL

```
/dev/video-rs-depth-0
/dev/video-rs-depth-1
/dev/video-rs-depth-2
/dev/video-rs-depth-3
etc..
```
Or you ask to have long links like for USB:
`/sys/devices/platform/bus@0/3610000.usb/usb2/2-3/2-3.2/2-3.2.1/2-3.2.1:1.0/video4linux/video6`
And for GMSL:
`/sys/devices/platform/tegra-capture-vi/video4linux/video0`
In GMSL case, you cannot tell which video nodes corresponds to what cameras.
To collect sensors to their cameras we scan media link and extrapolate parentship.
This is different for USB devices - you can know from device path.

tonynajjar, 2024-06-17:
My which would be to specify `/dev/video-rs-depth-0` but I'm happy with whatever string I don't need to configure for each machine.

MartyG-RealSense, 2026-10-31:
Case closed due to inactivity for over 2 years.

closed 2026-10-31 by MartyG-RealSense