realsenseai/realsense-ros #3128: The ROS wrapper for RealSense D435 does not starts, it gives ERRORS      -   D435+ROS Noetic+Ubuntu 20.04

TDVL, 2024-06-15:
Dear Support Team,
I have a RealSense D435 camera, I have installed Ubuntu 20.04 and ROS Noetiuc, since I need to start the camera in ROS environment to later do the mapping with RTABMap.
I have installed the camera and its drivers using this link and the camera works well, the RealSense Viewer starts.
https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages

Later, I have tried to install the ROS package for RealSense cameras using this link with these steps in Terminal in catkin_ws folder:

https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy
mkdir -p ~/catkin_ws/src
cd ~/catkin_ws/src/
git clone https://github.com/IntelRealSense/realsense-ros.git
cd realsense-ros/
git checkout `git tag | sort -V | grep -P "^2.\d+\.\d+" | tail -1`
cd ..
catkin_init_workspace
cd ..
catkin_make clean
catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release
catkin_make install
echo "source ~/catkin_ws/devel/setup.bash" >> ~/.bashrc
source ~/.bashrc

After I have launched the ROS package using "roslaunch realsense2_camera rs_camera.launch" in Terminal, I have got the next error:

laci@laci-Inspiron-3881:~$ cd catkin_ws
laci@laci-Inspiron-3881:~/catkin_ws$ source devel/setup.bash
laci@laci-Inspiron-3881:~/catkin_ws$ roslaunch realsense2_camera rs_camera.launch
... logging to /home/laci/.ros/log/a94fb65c-2aec-11ef-a4a5-d1260e01d327/roslaunch-laci-Inspiron-3881-3589.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://laci-Inspiron-3881:46189/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
[... 64 lines cut ...]

process[camera/realsense2_camera_manager-1]: started with pid [3603]
process[camera/realsense2_camera-2]: started with pid [3604]
[ INFO] [1718438170.016560413]: Initializing nodelet with 16 worker threads.
[ERROR] [1718438170.033963650]: Failed to load nodelet [/camera/realsense2_camera] of type [realsense2_camera/RealSenseNodeFactory] even after refreshing the cache: Could not find library corresponding to plugin realsense2_camera/RealSenseNodeFactory. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1718438170.033979148]: The error before refreshing the cache was: Could not find library corresponding to plugin realsense2_camera/RealSenseNodeFactory. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[FATAL] [1718438170.034104860]: Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
[camera/realsense2_camera-2] process has died [pid 3604, exit code 255, cmd /opt/ros/noetic/lib/nodelet/nodelet load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=/home/laci/.ros/log/a94fb65c-2aec-11ef-a4a5-d1260e01d327/camera-realsense2_camera-2.log].
log file: /home/laci/.ros/log/a94fb65c-2aec-11ef-a4a5-d1260e01d327/camera-realsense2_camera-2*.log

Something is wrong, but I cannot find the solution on forums.
Please, what is the problem? Please, help me to start the D435 in ROS Noetic for pointcloud publication. 
Please, can you provide me the exact steps in Terminal for ROS Noetic RealSense package installation in catkin_ws and use?
Also, in which launch file can I change the depth resolution, FPS, min and max depth range for visualization and the color stream resolution?

Also, previously I have tried the installation using the "sudo apt-get install ros-$ROS_DISTRO-realsense2-camera" and later the "sudo apt-get install ros-$ROS_DISTRO-realsense2-description" and I have got the warnings and errors provided in the image fsKVoedp0Y7kpHvJ.png . The point cloud was not visible, and I need the pointcloud for the RTABMap. Later, the Ubuntu crashed, and I tried the previously described installation.
Also, using the "sudo" command is it possible to install the RealSense package in catkin_ws, since after using this option, the realsense folder was not in catkin_ws.

I am not an Ubuntu and ROS user, I only need to start the D435 in ROS Noetic with the RTABMap ROS Noetic package.
I hope you can help me with this.
Previously, I have used the D435 in Windows to record depth images (not pointclouds) without problems, however now I have a problem.
I am trying more than a week, and I cannot start the camera in ROS Noetic.
Thank you advance.
Best regads
[image]

MartyG-RealSense, 2024-06-15:
Hi @TDVL  When you used the `sudo apt-get install ros-$ROS_DISTRO-realsense2-camera` command to install the ROs1 wrapper for Noetic, had you already installed the librealsense SDK?  The **sudo apt-get** instruction installs both librealsense and the wrapper at the same time.  So if you had already installed the SDK, you could end up with two conflicting installations of librealsense on the same computer.

I cannot see any errors in the procedure that you used to build the wrapper in the catkin workspace with Method 2.  There could again be conflicts though if you had already installed the wrapper with apt-get before doing a source code build.

If possible to do so then I would recommend wiping your storage and installing everything again from the beginning including Ubuntu 20.04.  This will help to ensure that there are not hidden tangles that are causing conflicts and preventing the wrapper from working correctly.

You can configure the stream resolution and FPS from the launch instruction instead of editing the launch file by specifying three factors for each stream type - width, height and FPS.  For example, to set both depth and color streams to 640x480 resolution and 30 FPS:

`roslaunch realsense2_camera rs_camera.launch depth_width:=640 depth_height:=480 depth_fps:=30 color_width:=640 color_height:=480 color_fps:=30`

TDVL, 2024-06-15:
Hello Marty,
I had reinstalled Ubuntu, now only the build version in catkin_ws has been done. The error you saw is based on that.

How can I remove the realsense package from Ubuntu? I think there should be a way?

Also, when I use the sudo apt-get install ros-$ROS_DISTRO-realsense2-camera, how can I put it in the catkin_ws?

MartyG-RealSense, 2024-06-15:
The instruction below from the distribution_linux.md instruction page should remove all packages with 'realsense' in their name, whether they are librealsense or ROS wrapper packages.

`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

Source code installations of the ROS wrapper can be removed simply by deleting the entire **catkin_ws** catkin workspace folder.

Removing all librealsense and ROS wrapper installations and then reinstalling them together at the same time with `sudo apt-get install ros-$ROS_DISTRO-realsense2-camera` is recommendable.  This sudo command is the only one that you need to use to install both librealsense and wrapper simultaneously.  With this method, you do not need to install librealsense with the distribution_linux instructions.

This method does not install the RealSense Viewer tool though, so if you need the Viewer then you will have to separately install librealsense from packages with the distribution_linux instructions and the wrapper from source code with the Method 2 instructions.

TDVL, 2024-06-15:
Thank you.
I have removed all packages and installed with your recommended method.
This is the error:

 aci@laci-Inspiron-3881:~$ sudo apt-get install ros-noetic-realsense2-camera
[sudo] password for laci: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  at dctrl-tools dkms libatk-bridge2.0-dev libatspi2.0-dev libdbus-1-dev
  libegl1-mesa-dev libepoxy-dev libfl2 libglfw3 libgtk-3-dev libv4l2rds0
  libwayland-bin libwayland-dev libxkbcommon-dev libxtst-dev v4l-utils
  wayland-protocols x11proto-record-dev
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  ros-noetic-librealsense2
The following NEW packages will be installed:
  ros-noetic-librealsense2 ros-noetic-realsense2-camera
0 upgraded, 2 newly installed, 0 to remove and 5 not upgraded.
Need to get 13.2 MB of archives.
After this operation, 38.8 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://packages.ros.org/ros/ubuntu focal/main amd64 ros-noetic-librealsense2 amd64 2.50.0-1focal.20211115.134455 [12.8 MB]
Get:2 http://packages.ros.org/ros/ubuntu focal/main amd64 ros-noetic-realsense2-camera amd64 2.3.2-1focal.20240111.183800 [333 kB]
Fetched 13.2 MB in 15s (858 kB/s)                                              
Selecting previously unselected package ros-noetic-librealsense2.
(Reading database ... 272745 files and directories currently installed.)
Preparing to unpack .../ros-noetic-librealsense2_2.50.0-1focal.20211115.134455_amd64.deb ...
Unpacking ros-noetic-librealsense2 (2.50.0-1focal.20211115.134455) ...
Selecting previously unselected package ros-noetic-realsense2-camera.
Preparing to unpack .../ros-noetic-realsense2-camera_2.3.2-1focal.20240111.183800_amd64.deb ...
Unpacking ros-noetic-realsense2-camera (2.3.2-1focal.20240111.183800) ...
Setting up ros-noetic-librealsense2 (2.50.0-1focal.20211115.134455) ...
Setting up ros-noetic-realsense2-camera (2.3.2-1focal.20240111.183800) ...
Processing triggers for libc-bin (2.31-0ubuntu9.16) ...
laci@laci-Inspiron-3881:~$ cd catkin_ws
laci@laci-Inspiron-3881:~/catkin_ws$ roslaunch realsense2_camera rs_camera.launch depth_width:=1280 depth_height:=720 depth_fps:=30 color_width:=1280 color_height:=720 color_fps:=30 filters:=pointcloud
... logging to /home/laci/.ros/log/18071b76-2b1a-11ef-859f-410701829d94/roslaunch-laci-Inspiron-3881-3269.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://laci-Inspiron-3881:43477/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 720
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 1280
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
[... 59 lines cut ...]
 * /camera/realsense2_camera/stereo_module/gain/1: 16
 * /camera/realsense2_camera/stereo_module/gain/2: 16
 * /camera/realsense2_camera/tf_publish_rate: 0.0
 * /camera/realsense2_camera/topic_odom_in: odom_in
 * /camera/realsense2_camera/unite_imu_method: 
 * /camera/realsense2_camera/usb_port_id: 
 * /camera/realsense2_camera/wait_for_device_timeout: -1.0
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

process[camera/realsense2_camera_manager-1]: started with pid [3283]
process[camera/realsense2_camera-2]: started with pid [3284]
[ INFO] [1718457698.586174461]: Initializing nodelet with 16 worker threads.
[ERROR] [1718457698.605816453]: Failed to load nodelet [/camera/realsense2_camera] of type [realsense2_camera/RealSenseNodeFactory] even after refreshing the cache: Failed to load library /opt/ros/noetic/lib//librealsense2_camera.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = librealsense2.so.2.50: cannot open shared object file: No such file or directory)
[... cut ...]
[FATAL] [1718457698.605973831]: Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
[camera/realsense2_camera-2] process has died [pid 3284, exit code 255, cmd /opt/ros/noetic/lib/nodelet/nodelet load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=/home/laci/.ros/log/18071b76-2b1a-11ef-859f-410701829d94/camera-realsense2_camera-2.log].
log file: /home/laci/.ros/log/18071b76-2b1a-11ef-859f-410701829d94/camera-realsense2_camera-2*.log

TDVL, 2024-06-15:
Maybe this command should be executed too?
sudo apt-get install ros-$ROS_DISTRO-realsense2-description
However, you 
mentioned this is not mandatory

MartyG-RealSense, 2024-06-15:
The error `Failed to load nodelet [/camera/realsense2_camera] of type [realsense2_camera/RealSenseNodeFactory]` is very difficult to diagnose because it does nor provide much information.  It is the equivalent of a 'Something went wrong' message in terms of the difficulty of working out the cause of a problem, unfortunately.

Some RealSense users have solved such problems by going to the catkin workspace folder **/catkin_ws** and inputting the command below.

`source devel/setup.bash`

TDVL, 2024-06-15:
OK, thank you. I have done that and the package started.
This is now the warning and error:
aci@laci-Inspiron-3881:~/catkin_ws$ roslaunch realsense2_camera rs_camera.launch depth_width:=1280 depth_height:=720 depth_fps:=30 color_width:=1280 color_height:=720 color_fps:=30 filters:=pointcloud
... logging to /home/laci/.ros/log/73c82cbe-2b1c-11ef-859f-410701829d94/roslaunch-laci-Inspiron-3881-5991.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://laci-Inspiron-3881:34441/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 720
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 1280
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 720
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 1280
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
[... 73 lines cut ...]
[ INFO] [1718460679.649231223]: Done Setting Dynamic reconfig parameters.
[ INFO] [1718460679.649943676]: depth stream is enabled - width: 1280, height: 720, fps: 30, Format: Z16
[ INFO] [1718460679.650790741]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1718460679.650811721]: setupPublishers...
[ INFO] [1718460679.652548058]: Expected frequency for depth = 30.00000
[ INFO] [1718460679.676144387]: Expected frequency for color = 30.00000
[ INFO] [1718460679.690206360]: setupStreams...
 15/06 16:11:19,948 WARNING [140038255920896] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
[ INFO] [1718460679.998693125]: SELECTED BASE:Depth, 0
[ INFO] [1718460680.005145737]: RealSense Node Is Up!
[ WARN] [1718460680.079239340]: 
 15/06 16:11:20,080 WARNING [140038255920896] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 15/06 16:11:20,131 WARNING [140038255920896] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 15/06 16:11:20,182 WARNING [140038255920896] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 15/06 16:11:20,403 WARNING [140038255920896] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 15/06 16:11:20,453 WARNING [140038255920896] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
[... cut ...]
 15/06 16:11:20,958 WARNING [140038255920896] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 15/06 16:11:21,109 WARNING [140038255920896] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 15/06 16:11:21,160 WARNING [140038255920896] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0

Also, I cannot see the pointcloud in the rviz, the image is attached.
[image]

TDVL, 2024-06-15:
Where is the point cloud, since I need it for RTABMap mapping procedure? I am trying to see it in rviz, however I can only see the color and depth images.
The Global status in rviz has an error, it is noted with red color

MartyG-RealSense, 2024-06-15:
When launching with `rs_camera.launch` and enabling the pointcloud filter, Fixed Frame should be set to **camera_link** instead of map.

https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy?tab=readme-ov-file#point-cloud

TDVL, 2024-06-15:
Thank, I have tried it, the pointcloud is not visible, the error and warning is still on.
What could be the error?

laci@laci-Inspiron-3881:~/catkin_ws$ source devel/setup.bash
laci@laci-Inspiron-3881:~/catkin_ws$ roslaunch realsense2_camera rs_camera.launch filters:=pointcloud
... logging to /home/laci/.ros/log/73c82cbe-2b1c-11ef-859f-410701829d94/roslaunch-laci-Inspiron-3881-8095.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://laci-Inspiron-3881:38633/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
[... 75 lines cut ...]
[ INFO] [1718466080.608961974]: Done Setting Dynamic reconfig parameters.
[ INFO] [1718466080.609339529]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1718466080.609701649]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1718466080.609721865]: setupPublishers...
[ INFO] [1718466080.611291371]: Expected frequency for depth = 30.00000
[ INFO] [1718466080.634689528]: Expected frequency for color = 30.00000
[ INFO] [1718466080.651845878]: setupStreams...
 15/06 17:41:20,985 WARNING [140278312720128] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
[ INFO] [1718466081.035707019]: SELECTED BASE:Depth, 0
[ INFO] [1718466081.040567364]: RealSense Node Is Up!
[ WARN] [1718466081.106484504]: 
 15/06 17:41:21,107 WARNING [140278312720128] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
[ INFO] [1718466081.130442002]: publishing unordered pointcloud.
 15/06 17:41:21,158 WARNING [140278312720128] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 15/06 17:41:21,209 WARNING [140278312720128] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 15/06 17:41:21,334 WARNING [140278312720128] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 15/06 17:41:21,385 WARNING [140278312720128] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 15/06 17:41:21,537 WARNING [140278312720128] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
[... cut ...]
 15/06 17:41:21,840 WARNING [140278312720128] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 15/06 17:41:21,890 WARNING [140278312720128] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 15/06 17:41:22,042 WARNING [140278312720128] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 15/06 17:41:22,092 WARNING [140278312720128] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0

[image]

TDVL, 2024-06-15:
I have got the pointcloud now, however I have some hardware errors.

laci@laci-Inspiron-3881:~/catkin_ws$ roslaunch realsense2_camera rs_camera.launch depth_width:=1280 depth_height:=720 depth_fps:=30 color_width:=1280 color_height:=720 color_fps:=30 clip_distance:=2 filters:=pointcloud
... logging to /home/laci/.ros/log/9dd58ec2-2b2e-11ef-859f-410701829d94/roslaunch-laci-Inspiron-3881-9857.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://laci-Inspiron-3881:46421/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: 2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 720
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 1280
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 720
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 1280
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
[... 90 lines cut ...]
 15/06 17:50:30,535 WARNING [139673997399808] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 15/06 17:50:30,586 WARNING [139673997399808] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 15/06 17:50:30,636 WARNING [139673997399808] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 15/06 17:50:30,687 WARNING [139673997399808] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 15/06 17:50:30,738 WARNING [139673997399808] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 15/06 17:50:30,788 WARNING [139673997399808] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 15/06 17:50:30,812 ERROR [139673456834304] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 130
 15/06 17:50:30,839 WARNING [139673997399808] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 15/06 17:50:30,890 WARNING [139673997399808] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 15/06 17:50:30,940 WARNING [139673997399808] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 15/06 17:50:30,991 WARNING [139673997399808] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
[... cut ...]
 15/06 17:50:31,553 WARNING [139673997399808] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
[ WARN] [1718466631.755189858]: Hardware Notification:Depth stream start failure,1.71847e+12,Error,Hardware Error
 15/06 17:50:31,805 WARNING [139673997399808] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 15/06 17:50:32,056 WARNING [139673997399808] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0

MartyG-RealSense, 2024-06-15:
**control_transfer returned** warnings, when continuously generating in the log, indicate that there may be a communication problem between the camera and computer / computing device.  This could be an issue with the USB port or the USB cable.

If the pointcloud is generating correctly in RViz despite the continuous log warnings though then it is safe to ignore them.

TDVL, 2024-06-15:
Thank you very much. You helped me.
The camera is connected to USB 3.2 port, it should be fine.
Also, do you have a recommendation how can I use the launched D435 in this way with RTABMap in ROS?

TDVL, 2024-06-15:
Thank you.
The RTABMap has started, however there some warnings. I cannot find the topic that published the mapping results. Maybe there is a problem due to the errors.

[ WARN] [1718477730.035089462]: /rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=10).
/rtabmap/rtabmap subscribed to (approx sync):
   /rtabmap/odom \
   /camera/rgb/image_rect_color \
   /camera/depth_registered/image_raw \
   /camera/rgb/camera_info \
   /rtabmap/odom_info
[ WARN] [1718477732.715280893]: /rtabmap/rgbd_odometry: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. 
/rtabmap/rgbd_odometry subscribed to (approx sync):
   /camera/rgb/image_rect_color \
   /camera/depth_registered/image_raw \
   /camera/rgb/camera_info
[ WARN] [1718477733.736815560]: /rtabmap/rtabmap_viz: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=10).
/rtabmap/rtabmap_viz subscribed to (approx sync):
   /rtabmap/odom \
   /camera/rgb/image_rect_color \
   /camera/depth_registered/image_raw \
   /camera/rgb/camera_info \
   /rtabmap/odom_info
^C[rtabmap/rtabmap_viz-3] killing on exit
[ INFO] [1718477734.455974141]: rtabmapviz: ctrl-c catched! Exiting Qt app...
[rtabmap/rtabmap-2] killing on exit
[rtabmap/rgbd_odometry-1] killing on exit
rtabmap: Saving database/long-term memory... (located at /home/laci/.ros/rtabmap.db)
rtabmap: Saving database/long-term memory...done! (located at /home/laci/.ros/rtabmap.db, 0 MB)
shutting down processing monitor...
... shutting down processing monitor complete
done
laci@laci-Inspiron-3881:~/catkin_ws$

Shrioe, 2024-06-16:
Hello @TDVL, I am having the problem with you, whereby, once the rviz is launch, it will only show a small part of the pointcloud only, and I couldn't get the whole pointcloud of the surrounding just like what being shown in the SLAM with D435i guide website, 
[image]

may I ask did you able to solve this issue, by any chance?

MartyG-RealSense, 2024-06-16:
@TDVL  Apparently the 3D point cloud of the stored map is published to the topic `/rtabmap/cloud_map` and the 2D flat map of the surroundings is published to `/rtabmap/grid_map`

TDVL, 2024-06-16:
Hello Marty,
Thank you, the RTABMap package starts, however cannot see the mapping procedure, there is an error. There is a rviz plugin error. Looks like the rrabmap does not see the D435? However, when I launch it, the rtabmap starts.
 

laci@laci-Inspiron-3881:~/catkin_ws$ roslaunch rtabmap_ros rtabmap.launch \ localization:=true \ rtabmapviz:=true \rviz:=true \DetectionRate:=30
... logging to /home/laci/.ros/log/ee78b0a2-2bd8-11ef-81f6-87be2e758d6d/roslaunch-laci-Inspiron-3881-7805.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://laci-Inspiron-3881:39325/

SUMMARY
========

CLEAR PARAMETERS
 * /points_xyzrgb/
 * /rtabmap/rgbd_odometry/
 * /rtabmap/rtabmap/
 * /rtabmap/rtabmap_viz/

PARAMETERS
 * /points_xyzrgb/approx_sync: True
 * /points_xyzrgb/approx_sync_max_interval: 0.0
 * /points_xyzrgb/decimation: 4.0
 * /points_xyzrgb/voxel_size: 0.0
 * /rosdistro: noetic
 * /rosversion: 1.16.0
 * /rtabmap/rgbd_odometry/approx_sync: True
 * /rtabmap/rgbd_odometry/approx_sync_max_interval: 0.0
 * /rtabmap/rgbd_odometry/config_path: 
 * /rtabmap/rgbd_odometry/expected_update_rate: 0.0
 * /rtabmap/rgbd_odometry/frame_id: camera_link
 * /rtabmap/rgbd_odometry/ground_truth_base_frame_id: 
 * /rtabmap/rgbd_odometry/ground_truth_frame_id: 
 * /rtabmap/rgbd_odometry/guess_frame_id: 
 * /rtabmap/rgbd_odometry/guess_min_rotation: 0.0
 * /rtabmap/rgbd_odometry/guess_min_translation: 0.0
 * /rtabmap/rgbd_odometry/keep_color: False
 * /rtabmap/rgbd_odometry/max_update_rate: 0.0
 * /rtabmap/rgbd_odometry/odom_frame_id: odom
 * /rtabmap/rgbd_odometry/publish_tf: True
 * /rtabmap/rgbd_odometry/queue_size: 10
 * /rtabmap/rgbd_odometry/subscribe_rgbd: False
 * /rtabmap/rgbd_odometry/wait_for_transform_duration: 0.2
 * /rtabmap/rgbd_odometry/wait_imu_to_init: False
 * /rtabmap/rtabmap/Mem/IncrementalMemory: true
 * /rtabmap/rtabmap/Mem/InitWMWithAllNodes: false
 * /rtabmap/rtabmap/approx_sync: True
 * /rtabmap/rtabmap/config_path: 
 * /rtabmap/rtabmap/database_path: ~/.ros/rtabmap.db
 * /rtabmap/rtabmap/frame_id: camera_link
 * /rtabmap/rtabmap/gen_depth: False
 * /rtabmap/rtabmap/gen_depth_decimation: 1
 * /rtabmap/rtabmap/gen_depth_fill_holes_error: 0.1
 * /rtabmap/rtabmap/gen_depth_fill_holes_size: 0
 * /rtabmap/rtabmap/gen_depth_fill_iterations: 1
 * /rtabmap/rtabmap/gen_scan: False
 * /rtabmap/rtabmap/ground_truth_base_frame_id: 
 * /rtabmap/rtabmap/ground_truth_frame_id: 
 * /rtabmap/rtabmap/initial_pose: 
 * /rtabmap/rtabmap/landmark_angular_variance: 9999.0
 * /rtabmap/rtabmap/landmark_linear_variance: 0.0001
 * /rtabmap/rtabmap/loc_thr: 0.0
 * /rtabmap/rtabmap/map_frame_id: map
 * /rtabmap/rtabmap/odom_frame_id: 
 * /rtabmap/rtabmap/odom_frame_id_init: 
 * /rtabmap/rtabmap/odom_sensor_sync: False
 * /rtabmap/rtabmap/odom_tf_angular_variance: 0.001
 * /rtabmap/rtabmap/odom_tf_linear_variance: 0.001
[... 642 lines cut ...]
/rtabmap/rtabmap subscribed to (approx sync):
   /rtabmap/odom \
   /camera/rgb/image_rect_color \
   /camera/depth_registered/image_raw \
   /camera/rgb/camera_info \
   /rtabmap/odom_info
[ WARN] [1718541603.957108274]: /rtabmap/rtabmap_viz: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=10).
/rtabmap/rtabmap_viz subscribed to (approx sync):
   /rtabmap/odom \
   /camera/rgb/image_rect_color \
   /camera/depth_registered/image_raw \
   /camera/rgb/camera_info \
   /rtabmap/odom_info
^C[points_xyzrgb-5] killing on exit
[rviz-4] killing on exit
[rtabmap/rtabmap_viz-3] killing on exit
[rtabmap/rtabmap-2] killing on exit
[rtabmap/rgbd_odometry-1] killing on exit
[ INFO] [1718541604.273063428]: rtabmapviz: ctrl-c catched! Exiting Qt app...
rtabmap: Saving database/long-term memory... (located at /home/laci/.ros/rtabmap.db)
rtabmap: Saving database/long-term memory...done! (located at /home/laci/.ros/rtabmap.db, 0 MB)
shutting down processing monitor...
... shutting down processing monitor complete
done
laci@laci-Inspiron-3881:~/catkin_ws$

MartyG-RealSense, 2024-06-16:
The warning **/rtabmap/rgbd_odometry: Did not receive data since 5 seconds!** may be related to the availability of processing resources.  Others who have experienced it suggest checking **rqt_graph** to see what is happening during processing.  If 30 FPS is being used then a publishing frequency of 30 hz would be expected to be displayed on the rqt_graph.  Significantly less than 30 may suggest that there is a problem that is causing the errors.

http://wiki.ros.org/rqt_graph

TDVL, 2024-06-16:
Thank you. I have done the reset. everything is the same. I have reduced the FPS and the DetectionRate for RTABMap, still nothing. I have started the graph, however I cannot see the frequency on graph.
Yes, the warning is on when the camera in not moving.
[image]

TDVL, 2024-06-16:
Is camera publishing everything? This is the rostopic list:

laci@laci-Inspiron-3881:~$ rostopic list
/camera/align_to_color/parameter_descriptions
/camera/align_to_color/parameter_updates
/camera/aligned_depth_to_color/camera_info
/camera/aligned_depth_to_color/image_raw
/camera/aligned_depth_to_color/image_raw/compressed
/camera/aligned_depth_to_color/image_raw/compressed/parameter_descriptions
/camera/aligned_depth_to_color/image_raw/compressed/parameter_updates
/camera/aligned_depth_to_color/image_raw/compressedDepth
/camera/aligned_depth_to_color/image_raw/compressedDepth/parameter_descriptions
/camera/aligned_depth_to_color/image_raw/compressedDepth/parameter_updates
/camera/aligned_depth_to_color/image_raw/theora
/camera/aligned_depth_to_color/image_raw/theora/parameter_descriptions
/camera/aligned_depth_to_color/image_raw/theora/parameter_updates
/camera/color/camera_info
/camera/color/image_raw
/camera/color/image_raw/compressed
/camera/color/image_raw/compressed/parameter_descriptions
/camera/color/image_raw/compressed/parameter_updates
/camera/color/image_raw/compressedDepth
/camera/color/image_raw/compressedDepth/parameter_descriptions
/camera/color/image_raw/compressedDepth/parameter_updates
/camera/color/image_raw/theora
/camera/color/image_raw/theora/parameter_descriptions
/camera/color/image_raw/theora/parameter_updates
/camera/color/metadata
/camera/depth/camera_info
/camera/depth/image_rect_raw
/camera/depth/image_rect_raw/compressed
/camera/depth/image_rect_raw/compressed/parameter_descriptions
/camera/depth/image_rect_raw/compressed/parameter_updates
/camera/depth/image_rect_raw/compressedDepth
/camera/depth/image_rect_raw/compressedDepth/parameter_descriptions
/camera/depth/image_rect_raw/compressedDepth/parameter_updates
/camera/depth/image_rect_raw/theora
/camera/depth/image_rect_raw/theora/parameter_descriptions
/camera/depth/image_rect_raw/theora/parameter_updates
/camera/depth/metadata
/camera/depth_registered/image_raw
/camera/extrinsics/depth_to_color
/camera/realsense2_camera_manager/bond
/camera/rgb/camera_info
/camera/rgb/image_rect_color
/camera/rgb_camera/auto_exposure_roi/parameter_descriptions
/camera/rgb_camera/auto_exposure_roi/parameter_updates
/camera/rgb_camera/parameter_descriptions
/camera/rgb_camera/parameter_updates
/camera/stereo_module/auto_exposure_roi/parameter_descriptions
/camera/stereo_module/auto_exposure_roi/parameter_updates
/camera/stereo_module/parameter_descriptions
/camera/stereo_module/parameter_updates
/diagnostics
/gps/fix
/imu/data
/rosout
/rosout_agg
/rtabmap/cloud_ground
/rtabmap/cloud_map
/rtabmap/cloud_obstacles
/rtabmap/elevation_map
/rtabmap/global_path
/rtabmap/global_path_nodes
/rtabmap/global_pose
/rtabmap/goal
/rtabmap/goal_node
/rtabmap/goal_out
/rtabmap/goal_reached
/rtabmap/grid_map
/rtabmap/grid_prob_map
/rtabmap/info
/rtabmap/initialpose
/rtabmap/labels
/rtabmap/landmarks
/rtabmap/local_grid_empty
/rtabmap/local_grid_ground
/rtabmap/local_grid_obstacle
/rtabmap/local_path
/rtabmap/local_path_nodes
/rtabmap/localization_pose
/rtabmap/mapData
/rtabmap/mapGraph
/rtabmap/mapOdomCache
/rtabmap/mapPath
/rtabmap/octomap_binary
/rtabmap/octomap_empty_space
/rtabmap/octomap_full
/rtabmap/octomap_global_frontier_space
/rtabmap/octomap_grid
/rtabmap/octomap_ground
/rtabmap/octomap_obstacles
/rtabmap/octomap_occupied_space
/rtabmap/odom
/rtabmap/odom_info
/rtabmap/odom_info_lite
/rtabmap/odom_last_frame
/rtabmap/odom_local_map
/rtabmap/odom_local_scan_map
/rtabmap/odom_rgbd_image
/rtabmap/odom_sensor_data/compressed
/rtabmap/odom_sensor_data/features
/rtabmap/odom_sensor_data/raw
/rtabmap/proj_map
/rtabmap/republish_node_data
/rtabmap/scan_map
/tag_detections
/tf
/tf_static
/user_data_async

The camera is publishing the pointcloud, however the rtabmap does not start. Also, there is a prbolem with camera publishing INFO, here is the screeenshot
[image]

TDVL, 2024-06-16:
I have launched the rtabmap using this method, the problem is still the same:

roslaunch rtabmap_ros rtabmap.launch \ depth_topic:=/camera/depth/image_rect_raw/ \ rgb_topic:=/camera/color/image_raw \ camera_info_topic:=/camera/color/camera_info  \ rtabmapviz:=true \ localization:=true \DetectionRate:=30\queue_size:=20\approx_sync:=false\depth_topic:=/camera/aligned_depth_to_color/image_raw \ rgb_topic:=/camera/color/image_raw \ camera_info_topic:=/camera/color/camera_info

TDVL, 2024-06-16:
Hello again,
I have tried to find the problem, the rostopic reports that the color image raw is not published yet, however it is published, you can see it in rviz, in the left bottom part of the rviz window
[image]
[image]
. What could be the problem? I was searching on forums, but nothing solved the issue.

MartyG-RealSense, 2024-06-17:
The published list of topics looks fine, with nothing apparently missing.

Assuming that you are publishing the RealSense ROS topics with the **rs_camera.launch** launch file, have you tried **opensource_tracking.launch** instead?  This launch file uses the settings of rs_camera and configures some additional settings for using the camera with RTABMAP.

`roslaunch realsense2_camera opensource_tracking.launch`

TDVL, 2024-06-17:
Thank you Marty,
The RTABMap started, however it consumes my HDD. Also, where are stored these maps, since I noticed that the drive free space is rapidly decreazing? How can I delete the data from memory and how can I reduce the memory usage? In this way the hdd will be full in a few minutes!!!
Also, how can I strore a map in order to later process it, analyze it, etc.?

Hence, now, I have a problem with RTABMap, since it is consuming my HDD, even when I stopped it. Maybe some log files are filling the disk.
Do you know something about this issue?
How can I delete log files, and other not necessary files from Ubuntu?
Thank you in advance.

[image]
[image]

 WARN] (2024-06-17 07:22:17.440) OdometryF2M.cpp:562::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=11) between -1 and 2155"
[ INFO] [1718601737.441288268]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.052256s
[ WARN] (2024-06-17 07:22:17.499) OdometryF2M.cpp:562::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=15) between -1 and 2156"
[ INFO] [1718601737.499849708]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.053314s
[ WARN] (2024-06-17 07:22:17.559) OdometryF2M.cpp:562::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=16) between -1 and 2157"
[ INFO] [1718601737.560560431]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.058925s
[ WARN] (2024-06-17 07:22:17.617) OdometryF2M.cpp:562::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=14) between -1 and 2158"
[ INFO] [1718601737.618138764]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.053185s
[ WARN] [1718601737.648171641]: Could not get transform from odom to camera_link after 0.200000 seconds (for stamp=1718601736.926017)! Error="Lookup would require extrapolation 125.820122957s into the future. Requested time 1718601736.926017046 but the latest data is at time 1718601611.105894089, when looking up transform from frame [camera_link] to frame [odom]. canTransform returned after 0.200221 timeout was 0.2.".
[ WARN] (2024-06-17 07:22:17.669) OdometryF2M.cpp:562::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=13) between -1 and 2159"
[ INFO] [1718601737.669939704]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.050450s
[ERROR] (2024-06-17 07:22:17.672) Rtabmap.cpp:1348::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 3908 is ignored!
[ INFO] [1718601737.672956782]: rtabmap (4): Rate=1.00s, Limit=0.000s, Conversion=0.0026s, RTAB-Map=0.0001s, Maps update=0.0000s pub=0.0000s (local map=1, WM=1)
[ WARN] (2024-06-17 07:22:17.726) OdometryF2M.cpp:562::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=12) between -1 and 2160"
[ INFO] [1718601737.726448570]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.053264s
[ WARN] (2024-06-17 07:22:17.781) OdometryF2M.cpp:562::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=12) between -1 and 2161"
[ INFO] [1718601737.781381990]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.053204s
[ WARN] (2024-06-17 07:22:17.831) OdometryF2M.cpp:562::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=10) between -1 and 2162"
[ INFO] [1718601737.831914929]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.049199s
[ WARN] [1718601737.852298874]: Could not get transform from odom to camera_link after 0.200000 seconds (for stamp=1718601736.992727)! Error="Lookup would require extrapolation 125.886832475s into the future. Requested time 1718601736.992726564 but the latest data is at time 1718601611.105894089, when looking up transform from frame [camera_link] to frame [odom]. canTransform returned after 0.201891 timeout was 0.2.".
[ WARN] [1718601737.857434445]: /rtabmap/rtabmap_viz: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
/rtabmap/rtabmap_viz subscribed to (exact sync):
/rtabmap/odom
/camera/color/image_raw
/camera/aligned_depth_to_color/image_raw
/camera/color/camera_info
/rtabmap/odom_info

MartyG-RealSense, 2024-06-17:
RTABMAP saves maps to **~/.ros/rtabmap.db** by default.

https://github.com/introlab/rtabmap/issues/449 discusses reducing memory usage.

Suggestions for clearing Ubuntu log files can found in a guide here:

https://recoverit.wondershare.com/file-recovery/linux-clear-logs.html

TDVL, 2024-06-17:
Something used my whole ssd, I need to reinstall the Ubuntu, ROS Noetic and RealSense.
It works, however I cannot display the color depth frame, there is an error in in rviz.
Also, how can I change the permission, since my launch files are all read-only. The Ubuntu is on external SSD, this should not be a problem.
[image]

roslaunch realsense2_camera rs_camera.launch depth_width:=1280 depth_height:=720
... logging to /home/laci/.ros/log/2c90116c-2ccb-11ef-84a3-73e53b5e8e76/roslaunch-laci-Inspiron-3881-38218.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://laci-Inspiron-3881:44859/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 720
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 1280
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
[... 77 lines cut ...]
[ WARN] [1718643821.506497542]: Using default profile instead.
[ INFO] [1718643821.506768972]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1718643821.507103774]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1718643821.507139791]: setupPublishers...
[ INFO] [1718643821.508693797]: Expected frequency for depth = 30.00000
[ INFO] [1718643821.531288464]: Expected frequency for color = 30.00000
[ INFO] [1718643821.544987938]: setupStreams...
 17/06 19:03:41,884 WARNING [140177120945920] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1718643821.935390843]: SELECTED BASE:Depth, 0
[ INFO] [1718643821.940844296]: RealSense Node Is Up!
[ WARN] [1718643822.002875350]: 
 17/06 19:03:42,003 WARNING [140177120945920] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 17/06 19:03:42,053 WARNING [140177120945920] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 17/06 19:03:42,104 WARNING [140177120945920] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 17/06 19:03:42,216 WARNING [140177120945920] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 17/06 19:03:42,267 WARNING [140177120945920] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[... cut ...]
 17/06 19:03:42,619 WARNING [140177120945920] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 17/06 19:03:42,669 WARNING [140177120945920] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
^C[camera/realsense2_camera-2] killing on exit
[camera/realsense2_camera_manager-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

TDVL, 2024-06-19:
Hello,
Thank you, now it works.
Another question. Is it possible to run the D435 in ROS Noetic on Virtualbox VM 6.1, with extension pack for USB 3? I have tried to install the RealSense ROS Noetic package with sudo apt-get install ros-$ROS_DISTRO-realsense2-camera, there is an error and the installation has not started. Is there a hardware problem due to the VM and related to D435, or there is a compatibility problem?
Other packages work well on this Virtualbox VM (i.e. several yolo versions with webcam)

MartyG-RealSense, 2024-06-20:
Typically, as long as a VM has USB support then RealSense cameras shoud be able to work with it.  https://github.com/IntelRealSense/librealsense/issues/8851 is an example where VirtualBox worked with RealSense when USB support was enabled.

The link below is another case where RealSense worked on VirtualBox.

https://support.intelrealsense.com/hc/en-us/community/posts/17013512540179/comments/17150357670163

At https://github.com/IntelRealSense/librealsense/issues/10829 a RealSense user also tried installing the ROS1 wrapper on a VirtualBox VM and used `sudo apt-get install ros-$ROS_DISTRO-realsense2-camera` to do so.   Their installation worked with the RealSense Viewer tool but on a ROS launch they got a **Failed to set power state** camera launch error.  

In their particular case, updating their camera's firmware driver resolved the problem.

If you are using the ROS1 wrapper then SDK 2.50.0 or 2.51.1 and firmware version 5.13.0.50 should be used with it.

MartyG-RealSense, 2024-06-25:
Hi @TDVL  Do you require further assistance with this case, please?  Thanks!

TDVL, 2024-06-25:
Hello Marty,
The camera is working in ROS, however it has now a temperature warning, from starting the camera (ASIC temperature something...). The camera is cold, I don't know what could be the problem. I have turned it off,  few minutes again after the launching the temperature warning is still on.
The transfer error is still on, as I have sent you in previous comments. I will make some screenshot images if it is necessary. 

How can enable the infrared cameras in Terminal and set their resolution to 1280x720 (like I did with the color and depth streams) during the launch command?

MartyG-RealSense, 2024-06-26:
The temperature warnings are not related not to excessive temperature but instead occur because the camera was not ready to provide a temperature reading at the moment that it was requested.  As advised by the creator of the RealSense ROS1 wrapper at https://github.com/IntelRealSense/realsense-ros/issues/894#issuecomment-527315705 the temperature reading is only updated every 10 seconds.  So these warnings are usually safe to ignore and do not prevent the camera from operating normally.

The Infrared 2 (right-side) infrared will only be available on a USB 3.2 connection.  If you are using USB3 and the ROS1 wrapper then you can enable the two infrared streams and set them to 1280x720 and 30 FPS by adding the commands below to your launch instruction.

`enable_infra1:=true enable_infra2:=true infra_width:=1280 infra_height:=720 infra_fps:=30`

TDVL, 2024-07-01:
Thank you Marty.
One more question. The RTABMap tutorial suggests to turn off the IR emitter and use the infra cameras. http://wiki.ros.org/rtabmap_ros/Tutorials/StereoHandHeldMapping

How can I do this in Terminal when I am using the launch command?
roslaunch realsense2_camera rs_camera.launch depth_width:=1280 depth_height:=720 depth_fps:=5 color_width:=1280 color_height:=720 color_fps:=5 clip_distance:=6 align_depth:=true filters:=pointcloud initial_reset:=true enable_infra1:=true enable_infra2:=true

The rosrun rqt_reconfigure rqt_reconfigure, # select Camera->stereo_module, set emitter_enabled to OFF is OK, but how can I directly disable the IR emitter in Terminal command line?
Thank you in advance.

MartyG-RealSense, 2024-07-01:
Please try adding **emitter_enabled:=false** to your roslaunch instruction.

TDVL, 2024-07-01:
OK, thank you. I will try it,

TDVL, 2024-07-09:
Thank you, I think everything is OK now. Maybe later if there will be some issues.

MartyG-RealSense, 2024-07-09:
I'm pleased to hear that everything is okay.  I will close this issue for now.  Please feel free to re-open it or create a new issue if there is a problem at a future date.  Thanks again!