realsenseai/realsense-ros #3129: D455 camera cannot get imu msg from rostopic

GeLuzhou, 2024-06-16:
I want get IMU stream by using ROS1. I build ROS Wrapper 2.3.2 ,  I know Librealsense should be 2.50.0 and Firmware version 5.13 to suit ROS Wrapper 2.3.2 . but my linux kernel is 5.15 ,Librealsense 2.50.0 seems not to be suitable for  linux kernel  5.15. What can I do ?
----------------------------------------------------------------------------------------------------

| Required Info                         |  D455 camera cannot get imu msg from rostopic                                                                            |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | { D455 }                          | 
| Firmware Version                   | (5.16)                          | 
| Operating System & Version |   { ubuntu 20.04 } | 
| Kernel Version (Linux Only)    |  ( 5.15 )                                                                       | 
| Platform                                 | PC                     |
| Librealsense SDK Version       |  { 2.<55>.<1> }                                                    | 
| Language                               |  {C++ }         | 
| Segment			         |  {Robot/Smartphone/VR/AR/others }                             | 
| ROS Distro			         |  { noetic }                                           | 
| RealSense ROS Wrapper Version   |  { 2.3.2 }                                                  | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2024-06-16:
Hi @GeLuzhou  The IMU topics are disabled by default in the ROS wrapper.  You can publish them in the ROS1 wrapper by adding the commands below to your roslaunch instruction.

`enable_accel:=true enable_gyro:=true unite_imu_method:=linear_interpolation`

GeLuzhou, 2024-06-16:
> Hi @GeLuzhou The IMU topics are disabled by default in the ROS wrapper. You can publish them in the ROS1 wrapper by adding the commands below to your roslaunch instruction.
> 
> `enable_accel:=true enable_gyro:=true unite_imu_method:=linear_interpolation`

Yes, I  know I need to  do this before,I can get /camera/imu in topic list,but there is nothing when I echo it

MartyG-RealSense, 2024-06-16:
Support for kernel 5.15 was added in librealsense 2.53.1.

However, there are a couple of circumstances in which the kernel version should not matter:

1.  If librealsense was built from source code with CMake with the flag **-DFORCE_RSUSB_BACKEND=TRUE** included in the CMake build instruction.  Or;

2.  If the 2.3.2 ROS wrapper was installed with the instruction `sudo apt-get install ros-$ROS_DISTRO-realsense2-camera` as this installation method is based upon RSUSB=TRUE

GeLuzhou, 2024-06-16:
> Support for kernel 5.15 was added in librealsense 2.53.1.
> 
> 但是，在以下几种情况下，内核版本无关紧要：
> 
> 1. 如果 librealsense 是从源代码构建的，带有 CMake 标志**-DFORCE_RSUSB_BACKEND=TRUE** included in the CMake build instruction.  Or;
> 2. If the 2.3.2 ROS wrapper was installed with the instruction `sudo apt-get install ros-$ROS_DISTRO-realsense2-camera` as this installation method is based upon RSUSB=TRUE

I build librealsense 2.55.1 by apt-get install ,and build 2.3.2 ROS wrapper from source. You mean I can try to build librealsense 2.50.0 from source with -DFORCE_RSUSB_BACKEND=TRUE and don't need to care my linux kernel version?

GeLuzhou, 2024-06-16:
> I build librealsense 2.55.1 by apt-get install ,and build 2.3.2 ROS wrapper from source. You mean I can try to build librealsense 2.50.0 from source with -DFORCE_RSUSB_BACKEND=TRUE and don't need to [...]

and maybe I need to downgrade my firmware version to 5.13?

MartyG-RealSense, 2024-06-16:
2.3.2 should be used with 2.50.0 or 2.51.1.  The ROS1 wrapper has not been updated for support of librealsense versions that come after 2.51.1 as development of that wrapper has ceased.  So the problem is not just with the kernel version.

If the command `sudo apt-get install ros-$ROS_DISTRO-realsense2-camera` is used then it builds librealsense 2.50.0 and wrapper 2.3.2 together at the same time, ensuring that you have a correcrt match of versions.  

Also, as that installation method is based on RSUSB=true (which bypasses the kernel), having kernel 5.15 should not matter even though librealsense 2.50.0 does not officially support it.

Yes, the firmware version to use with this wrapper should be 5.13.0.50.  Using a newer firmware than that with 2.50.0 is likely to cause errors.

GeLuzhou, 2024-06-16:
> 2.3.2 should be used with 2.50.0 or 2.51.1. The ROS1 wrapper has not been updated for support of librealsense versions that come after 2.51.1 as development of that wrapper has ceased. So the problem is not just with the kernel version.
> 
> If the command `sudo apt-get install ros-$ROS_DISTRO-realsense2-camera` is used then it builds librealsense 2.50.0 and wrapper 2.3.2 together at the same time, ensuring that you have a correct match of versions. Because it builds the SDK and wrapper together, you do not need to install librealsense separately before using this command.
> 
> Also, as that installation method is based on RSUSB=true (which bypasses the kernel), having kernel 5.15 should not matter even though librealsense 2.50.0 does not officially support it.
> 
> Yes, the firmware version to use with this wrapper should be 5.13.0.50. Using a newer firmware than that with 2.50.0 is likely to cause errors.

[ INFO] [1718526304.356713678]: Initializing nodelet with 28 worker threads.
[ INFO] [1718526304.434945511]: RealSense ROS v2.3.2
[ INFO] [1718526304.434962550]: Built with LibRealSense v2.50.0
[ INFO] [1718526304.434968366]: Running with LibRealSense v2.50.0
[ INFO] [1718526304.450955509]:  
 16/06 16:25:04,594 WARNING [140481440855808] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0
[ INFO] [1718526304.662588616]: Device with serial number 242422301964 was found.

[ INFO] [1718526304.662604534]: Device with physical ID 2-4-2 was found.
[ INFO] [1718526304.662609017]: Device with name Intel RealSense D455 was found.
[ INFO] [1718526304.662753937]: Device with port number 2-4 was found.
[ INFO] [1718526304.662760921]: Device USB type: 3.2
[ INFO] [1718526304.663732147]: getParameters...
[ INFO] [1718526304.682307432]: setupDevice...
[ INFO] [1718526304.682319771]: JSON file is not provided
[ INFO] [1718526304.682325547]: ROS Node Namespace: camera
[ INFO] [1718526304.682332180]: Device Name: Intel RealSense D455
[ INFO] [1718526304.682336740]: Device Serial No: 242422301964
[ INFO] [1718526304.682342973]: Device physical port: 2-4-2
[ INFO] [1718526304.682348350]: Device FW version: 05.16.00.01
[ INFO] [1718526304.682353348]: Device Product ID: 0x0B5C
[ INFO] [1718526304.682357766]: Enable PointCloud: Off
[ INFO] [1718526304.682363850]: Align Depth: Off
[ INFO] [1718526304.682369394]: Sync Mode: Off
[ INFO] [1718526304.682386477]: Device Sensors: 
[ INFO] [1718526304.756915660]: Stereo Module was found.
[ INFO] [1718526304.762186527]: RGB Camera was found.
[ INFO] [1718526304.762268645]: Motion Module was found.
[ INFO] [1718526304.762282565]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1718526304.762295584]: num_filters: 0
[ INFO] [1718526304.762303396]: Setting Dynamic reconfig parameters.
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
[ INFO] [1718526307.994597207]: Done Setting Dynamic reconfig parameters.
[ INFO] [1718526307.997258459]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ WARN] [1718526307.997306015]: Given stream configuration is not supported by the device!  Stream: Infrared, Stream Index: 0, Width: 640, Height: 480, FPS: 30, Format: Y8
[ INFO] [1718526307.997471077]: infra1 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ INFO] [1718526307.997626477]: infra2 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ INFO] [1718526307.998487791]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1718526307.998509960]: gyro stream is enabled - fps: 200
[ INFO] [1718526307.998522592]: accel stream is enabled - fps: 100
[ INFO] [1718526307.998536648]: setupPublishers...
[ INFO] [1718526308.000423954]: Expected frequency for depth = 30.00000
[ INFO] [1718526308.015226892]: Expected frequency for infra1 = 30.00000
[ INFO] [1718526308.020677240]: Expected frequency for infra2 = 30.00000
[ INFO] [1718526308.026836818]: Expected frequency for color = 30.00000
[ INFO] [1718526308.032368317]: Start publisher IMU
[ INFO] [1718526308.032826526]: setupStreams...
 16/06 16:25:08,386 WARNING [140481466033920] (ds5-motion.cpp:473) IMU Calibration is not available, default intrinsic and extrinsic will be used.
 16/06 16:25:09,074 WARNING [140481440855808] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 16/06 16:25:09,127 WARNING [140481440855808] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 16/06 16:25:09,179 WARNING [140481440855808] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1718526309.231461165]: SELECTED BASE:Depth, 0
[ INFO] [1718526309.234496809]: RealSense Node Is Up!
 16/06 16:25:09,235 WARNING [140481440855808] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
 16/06 16:25:09,607 WARNING [140481440855808] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
[... cut ...]
 16/06 16:27:35,236 WARNING [140481440855808] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
^C[camera/realsense2_camera_manager-2] killing on exit
[camera/realsense2_camera-3] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

Now I use dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge to remove 2.55.1, and sudo apt-get install ros-$ROS_DISTRO-realsense2-camera install 2.50.0, but when I echo /camera/imu I still get nothing from this topic

GeLuzhou, 2024-06-16:
> [ INFO] [1718526304.356713678]: Initializing nodelet with 28 worker threads. [ INFO] [1718526304.434945511]: RealSense ROS v2.3.2 [ INFO] [1718526304.434962550]: Built with LibRealSense v2.50.0 [ [...]

oh,I forget to downgrade my fw to 5.13

GeLuzhou, 2024-06-16:
> 2.3.2 should be used with 2.50.0 or 2.51.1. The ROS1 wrapper has not been updated for support of librealsense versions that come after 2.51.1 as development of that wrapper has ceased. So the [...]

when I update FW from realsense viewer , I download 5.13.0.50  .bin file ,but I get error when update process
[image]

MartyG-RealSense, 2024-06-16:
Is your D455 camera still able to be detected in the RealSense Viewer after the firmware update failure?

GeLuzhou, 2024-06-16:
> Is your D455 camera still able to be detected in the RealSense Viewer after the firmware update failure?

I use rs-fw-update to downgrade successfully,and I get imu data from the topic I want. Thank you very much!
But I couldn't figure out why realsense-viewer still with v2.55.1 on top of the viewer window. After I dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge to remove 2.55.1.
do you know why? 
Thank you very much again!

MartyG-RealSense, 2024-06-16:
A package purge typically does not remove a previous installation of realsense-viewer.  

If librealsense and the ROS wrapper are installed together with `sudo apt-get install ros-$ROS_DISTRO-realsense2-camera` then this installation method does not install realsense-viewer.  This is likely why the 2.55.1 version of the Viewer is not replaced by its 2.50.0 version.

MartyG-RealSense, 2024-06-25:
Hi @GeLuzhou  Do you require further assistance with this case, please?  Thanks!

GeLuzhou, 2024-06-25:
> Hi @GeLuzhou Do you require further assistance with this case, please? Thanks!

No,I have solved my problem.Thank you very much!!!

MartyG-RealSense, 2024-06-25:
You are very welcome.  It's excellent to hear that you achieved a solution.  Thanks very much for the update!

closed 2024-06-25 by MartyG-RealSense