realsenseai/realsense-ros #3131: Unable to build from source since PR #3125

b-adkins, 2024-06-17:
| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | N/A                          | 
| Firmware Version                   | N/A                          | 
| Operating System & Version |   Linux (Ubuntu 22) | 
| Kernel Version (Linux Only)    |  6.5.0-35-generic                                                                  | 
| Platform                                 | NUC                           |
| Librealsense SDK Version       |  2.55.1-1jammy.20240516.151706                                                    | 
| Language                               |  C++         | 
| Segment			         |  Robot                             | 
| ROS Distro			         |  Humble                                           | 
| RealSense ROS Wrapper Version   |  ec8ef1                                                  | 

### Issue Description
In summary, some types related to calibration are not found by the build system, and I believe that this commit has introduced a new dependency that needs to be either properly documented or registered for rosdep.

I experienced this error with the `HEAD` of `ros2-development`, `68473b` at time of reporting, `d30980f8` (when PR #3125 was merged), and `ec8ef1` ("support calib config read/write services"), but not the previous commit, `d1684b`.

```
$ git log --graph --decorate --pretty=oneline --abbrev-commit
*   68473bd8 (HEAD -> ros2-development, origin/ros2-development, origin/HEAD) PR #3130 from SamerKhshiboun: Install specific numpy version to avoid errors in github actions
|\  
| * 732409b6 Update main.yml
|/  
*   d30980f8 PR #3125 from SamerKhshiboun: Support calibration config read/write services
|\  
| * ec8ef109 support calib config read/write services
|/  
* d1684b4e Update README.md
```

I am working with an existing codebase that builds the realsense ROS2 node from scratch. I followed the instructions to install `realsense-ros` for Ubuntu 22.02 via apt, using the `ros-humble-librealsense2*`.

```
$ dpkg -l | grep ros-humble-librealsense2
ii  ros-humble-librealsense2                          2.55.1-1jammy.20240516.151706           amd64        Library for controlling and capturing data from the Intel(R) RealSense(TM) D400 devices.
ii  ros-humble-librealsense2-dbgsym                   2.55.1-1jammy.20240516.151706           amd64        debug symbols for ros-humble-librealsense2
```

The full error log:

```
$ colcon build --base-paths src/realsense-ros/
Starting >>> realsense2_camera_msgs
Finished <<< realsense2_camera_msgs [1.47s]                     
Starting >>> realsense2_camera
Starting >>> realsense2_description
Finished <<< realsense2_description [0.22s]                           
--- stderr: realsense2_camera                             
/home/p0/ros2_thirdparty/src/realsense-ros/realsense2_camera/src/rs_node_setup.cpp: In member function ‘void realsense2_camera::BaseRealSenseNode::CalibConfigReadService(realsense2_camera_msgs::srv::CalibConfigRead_Request_<std::allocator<void> >::SharedPtr, realsense2_camera_msgs::srv::CalibConfigRead_Response_<std::allocator<void> >::SharedPtr)’:
/home/p0/ros2_thirdparty/src/realsense-ros/realsense2_camera/src/rs_node_setup.cpp:545:9: error: ‘rs2_calibration_config’ was not declared in this scope; did you mean ‘rs2_calibration_type’?
  545 |         rs2_calibration_config calib_config = _dev.as<rs2::auto_calibrated_device>().get_calibration_config();
      |         ^~~~~~~~~~~~~~~~~~~~~~
      |         rs2_calibration_type
/home/p0/ros2_thirdparty/src/realsense-ros/realsense2_camera/src/rs_node_setup.cpp:546:68: error: ‘class rs2::auto_calibrated_device’ has no member named ‘calibration_config_to_json_string’
  546 |         res->calib_config = _dev.as<rs2::auto_calibrated_device>().calibration_config_to_json_string(calib_config);
      |                                                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/p0/ros2_thirdparty/src/realsense-ros/realsense2_camera/src/rs_node_setup.cpp:546:102: error: ‘calib_config’ was not declared in this scope
  546 | onfig = _dev.as<rs2::auto_calibrated_device>().calibration_config_to_json_string(calib_config);
      |                                                                                  ^~~~~~~~~~~~

/home/p0/ros2_thirdparty/src/realsense-ros/realsense2_camera/src/rs_node_setup.cpp: In member function ‘void realsense2_camera::BaseRealSenseNode::CalibConfigWriteService(realsense2_camera_msgs::srv::CalibConfigWrite_Request_<std::allocator<void> >::SharedPtr, realsense2_camera_msgs::srv::CalibConfigWrite_Response_<std::allocator<void> >::SharedPtr)’:
/home/p0/ros2_thirdparty/src/realsense-ros/realsense2_camera/src/rs_node_setup.cpp:560:9: error: ‘rs2_calibration_config’ was not declared in this scope; did you mean ‘rs2_calibration_type’?
  560 |         rs2_calibration_config calib_config = _dev.as<rs2::auto_calibrated_device>().json_string_to_calibration_config(req->calib_config);
      |         ^~~~~~~~~~~~~~~~~~~~~~
      |         rs2_calibration_type
/home/p0/ros2_thirdparty/src/realsense-ros/realsense2_camera/src/rs_node_setup.cpp:561:48: error: ‘class rs2::auto_calibrated_device’ has no member named ‘set_calibration_config’; did you mean ‘set_calibration_table’?
  561 |         _dev.as<rs2::auto_calibrated_device>().set_calibration_config(calib_config);
      |                                                ^~~~~~~~~~~~~~~~~~~~~~
      |                                                set_calibration_table
/home/p0/ros2_thirdparty/src/realsense-ros/realsense2_camera/src/rs_node_setup.cpp:561:71: error: ‘calib_config’ was not declared in this scope
  561 |         _dev.as<rs2::auto_calibrated_device>().set_calibration_config(calib_config);
      |                                                                       ^~~~~~~~~~~~
/home/p0/ros2_thirdparty/src/realsense-ros/realsense2_camera/src/rs_node_setup.cpp:556:121: warning: unused parameter ‘req’ [-Wunused-parameter]
  556 | figWriteService(const realsense2_camera_msgs::srv::CalibConfigWrite::Request::SharedPtr req,
      |                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~

gmake[2]: *** [CMakeFiles/realsense2_camera.dir/build.make:118: CMakeFiles/realsense2_camera.dir/src/rs_node_setup.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:159: CMakeFiles/realsense2_camera.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< realsense2_camera [7.37s, exited with code 2]

Summary: 2 packages finished [9.02s]
  1 package failed: realsense2_camera
  1 package had stderr output: realsense2_camera
```

annb3, 2024-06-18:
I have the same problem.

MartyG-RealSense, 2024-06-18:
Thanks very much for the reports.  I have highlighted it to my Intel RealSense colleagues.

MartyG-RealSense, 2024-06-18:
My colleagues responded that the ros2-development ROS wrapper now needs the **development** branch of librealsense rather than the master due to recent changes in the API.

https://github.com/IntelRealSense/librealsense/tree/development

annb3, 2024-06-18:
I am already into the development branch :( @MartyG-RealSense

annb3, 2024-06-18:
@MartyG-RealSense I try to disinstall and reinstall everything. Nothing changes. Same problem also with Jetson installation, linked only to Debian modality.

SamerKhshiboun, 2024-06-18:
can you screenshot your `git log` output ?

annb3, 2024-06-18:
[image]
Of course: here it is!

SamerKhshiboun, 2024-06-18:
please run these 2 commands and screenshot the outputs:

`dpkg -l | grep realsense`
`ldconfig -p | grep realsense`

annb3, 2024-06-18:
[image]

SamerKhshiboun, 2024-06-18:
As I see you have multiple librealsense2 version that were manually installed.
What I suggest is to run `sudo make uninstall` from each `../librealsense/build` directory you have.
Or you can remove these files manually.
Make sure to have the latest (2.56) only

annb3, 2024-06-18:
I don't have the builds anymore, as I have already run this command before.. so strange. Do you suggest to eliminate this file directly from `/usr/local/lib/libreaslsense*`?

SamerKhshiboun, 2024-06-18:
I would go for that.

annb3, 2024-06-18:
Nothing is solved. Same issue.
[image]

SamerKhshiboun, 2024-06-18:
can you please run these commands and print the output ?

```
source /opt/ros/humble/setup.bash
cd <your_ros_workspace>
rm -rf log/ install/ build/
colcon build
```

annb3, 2024-06-19:
[image]
Here the errors

SamerKhshiboun, 2024-06-19:
did you install the newest code like this ?
```
cd librealsense
mkdir build
cmake ..
make -j 4
sudo make install
```

annb3, 2024-06-19:
I have followed the guidelines: https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md switching into the **development** branch.

SamerKhshiboun, 2024-06-19:
please print the output of these this command from your ros working space:
`grep '2\.5' ./build/realsense2_camera/CMakeFiles/realsense2_camera.dir/build.make`

SamerKhshiboun, 2024-06-19:
I'm still pretty sure that the current version that the ROS finds (even if it says 2.56.1) is not the latest compiled one.
Please retry to `make install` and `sudo make install` latest code of development

annb3, 2024-06-19:
I am actually working on a docker environment and by default it doesn't take the version 2.56 of the pc but every time install again the 2.55 from the dipendencies of the realsense-ros package how can I fix it?
[image]

SamerKhshiboun, 2024-06-19:
try edit the realsense-ros/realsense2_camera/CMakeLists.exe

[image]

change the numbers to 2.56.1

`find_package(realsense2 2.56.1)`
`find_package(realsense2-gl 2.56.1)`

then try to rebuild again
with these steps:

```
source /opt/ros/humble/setup.bash
cd <your_ros_workspace>
rm -rf log/ install/ build/
colcon build
source ./install/setup.bash
```

annb3, 2024-06-19:
no way :( It continues installing the 2.55 version

SamerKhshiboun, 2024-06-19:
You need the latest librealsense2/development branch to be available inside the docker.
You can do this in two different ways I think:
- expose the installed librealsense2.56.1 to the docker  (might required advanced docker commands and understanding..)
- run the docker, and reinstall (make + make install) librealsense2/development from source inside it, then rerun colcon build.

annb3, 2024-06-19:
yes, already done way 2, this morning, without apparently success.
I'll try again in some hours.

SamerKhshiboun, 2024-06-19:
Anyway, I will submit a PR for updating the CMakeLists.txt to 2.56.0 since the realsense2_camera node needs the latest development branch [2.56] (which can be built only manually for now, because it is still not formally released to ROS apt)

jerinbenny, 2024-06-19:
Cloning these 2 repositories for sdk and wrapper and following the documentation,I got the no errors.
librealsense sdk: git clone https://github.com/IntelRealSense/librealsense -b "development"
ros2 wrapper : git clone https://github.com/IntelRealSense/realsense-ros.git -b ros2-development

[image]

annb3, 2024-06-20:
The problem is that into the docker environment I encountered several problems in installing from build. The same in the Jetson. Is it possible to do it from Debian packages specifying the development installation of the librealsense2? I think it is the real problem.

SamerKhshiboun, 2024-06-20:
There is still no formal release as Debian from our side to librealsense2 2.56 (the latest one we released into ROS apt is 2.55.1)
You can work with earlier realsense-ros/ros2-development if you want, which can work with librealsense2 2.55.
```
cd realsense-ros
git checkout d1684b4e53111f5eb3237f76299e83a864ce929d
cd <your_ros_ws>
colcon build
```

SamerKhshiboun, 2024-06-20:
or you can use the latest realsense2_camera released (4.55.1) from ROS apts
`sudo apt install sudo apt install ros-humble-realsense2-camera`

annb3, 2024-06-20:
Ok, I am backed before this PR, waiting for the stable one in Debian.
Thanks a lot for the support @SamerKhshiboun.

closed 2024-07-04 by MartyG-RealSense