realsenseai/realsense-ros #3132: cant change color profile

GiladRobee, 2024-06-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D405                         | 
| Firmware Version                   | 5.16.0.1                         | 
| Operating System & Version |  Ubuntu 22 | 
| Kernel Version (Linux Only)    |  5.10.192-tegra                                                                       | 
| Platform                                 |  NVIDIA Jetson Orin                           |
| Librealsense SDK Version       |  v2.55.1                                                   | 
| Language                               |  {C/C++/Opencv/python }         | 
| Segment			         |  Robo                             | 
| ROS Distro			         |  Humble                                       | 
| RealSense ROS Wrapper Version   |  v4.54.1                                                 | 

### Issue Description
this is my current launch file:
```
from inspect import Arguments

from click import launch
from launch.actions import IncludeLaunchDescription
from launch.launch_description_sources import PythonLaunchDescriptionSource
from launch import LaunchDescription
from launch.substitutions import LaunchConfiguration
from launch.actions import DeclareLaunchArgument
from launch_ros.actions import Node
from ament_index_python import get_package_share_directory
import os
import sys

def get_share_file(package_name, file_name):
    return os.path.join(get_package_share_directory(package_name), file_name)

# add rviz config file to config dir and call it using get_share_file method
def generate_launch_description():
    ld = LaunchDescription()

    use_rviz = False
    rviz_path = ''
    camera_name = 'Realsense'
    
    if len(sys.argv) > 1:
        for arg in sys.argv:
                if arg.startswith("use_rviz:="):
                    use_rviz = int(arg.split(":=")[1])
                    print(use_rviz)
    rviz_cfg_path = get_share_file("robee_camera",'config/robee_camera_config.rviz')
    
    realsense_run = IncludeLaunchDescription(
        PythonLaunchDescriptionSource(get_share_file("realsense2_camera","launch/rs_launch.py")),
        launch_arguments={'camera_name' : camera_name,
                            'enable_color': 'True',
                            'enable_depth': 'True',
                            'enable_sync': 'True',

                            "depth_module.enable_auto_exposure": "False",
                            "depth_module.infra_profile": "1280x720x5",
                            "depth_module.depth_profile": "1280x720x5",
                            "rgb_camera.color_profile": "1280x720x5",
                            # "depth_module.brightness": "1000",
                            # "depth_module.exposure": "100",
                            # 'initial_reset': 'true',
                            'pointcloud.enable': 'true',
                            # 'json_file_path': 'jetson_config.json', #see if param file is needed
                            'align_depth.enable': 'false'}.items()
        )
    ld.add_action(realsense_run)

    if use_rviz == True:
        rviz2 = Node(
        package='rviz2',
        executable='rviz2',
        name='rviz2',
        arguments=['-d', str(rviz_cfg_path)]
        )
        ld.add_action(rviz2)
    

    

    
    return ld
```

and I can not manage to change the RGB profile to 1280X720X(5/10/15/30)

in the read me I see that there is a namespace- rgb_camera in the params but I don't see it in the launch file or in `ros2 param list` which gives me:

```
/camera/Realsense:
  .Realsense.color.image_rect_raw.format
  .Realsense.color.image_rect_raw.jpeg_quality
  .Realsense.color.image_rect_raw.png_level
  .Realsense.color.image_rect_raw.tiff.res_unit
  .Realsense.color.image_rect_raw.tiff.xdpi
  .Realsense.color.image_rect_raw.tiff.ydpi
  .Realsense.depth.image_rect_raw.format
  .Realsense.depth.image_rect_raw.jpeg_quality
  .Realsense.depth.image_rect_raw.png_level
  .Realsense.depth.image_rect_raw.tiff.res_unit
  .Realsense.depth.image_rect_raw.tiff.xdpi
  .Realsense.depth.image_rect_raw.tiff.ydpi
  Realsense.color.image_rect_raw.enable_pub_plugins
  Realsense.depth.image_rect_raw.enable_pub_plugins
  align_depth.enable
  align_depth.frames_queue_size
  angular_velocity_cov
  base_frame_id
  camera_name
  clip_distance
  color_info_qos
  color_qos
  colorizer.color_scheme
  colorizer.enable
  colorizer.frames_queue_size
  colorizer.histogram_equalization_enabled
  colorizer.max_distance
  colorizer.min_distance
  colorizer.stream_filter
  colorizer.stream_format_filter
  colorizer.stream_index_filter
  colorizer.visual_preset
  decimation_filter.enable
  decimation_filter.filter_magnitude
  decimation_filter.frames_queue_size
  decimation_filter.stream_filter
  decimation_filter.stream_format_filter
  decimation_filter.stream_index_filter
  depth_info_qos
  depth_module.auto_exposure_limit
  depth_module.auto_exposure_limit_toggle
  depth_module.auto_exposure_roi.bottom
  depth_module.auto_exposure_roi.left
  depth_module.auto_exposure_roi.right
  depth_module.auto_exposure_roi.top
  depth_module.auto_gain_limit
  depth_module.auto_gain_limit_toggle
  depth_module.backlight_compensation
  depth_module.brightness
  depth_module.color_format
  depth_module.color_profile
  depth_module.contrast
  depth_module.depth_format
  depth_module.depth_profile
  depth_module.enable_auto_exposure
  depth_module.enable_auto_white_balance
  depth_module.error_polling_enabled
  depth_module.exposure
  depth_module.frames_queue_size
  depth_module.gain
  depth_module.gamma
  depth_module.global_time_enabled
  depth_module.hdr_enabled
  depth_module.hue
  depth_module.infra1_format
  depth_module.infra2_format
  depth_module.infra_format
  depth_module.infra_profile
  depth_module.output_trigger_enabled
  depth_module.power_line_frequency
  depth_module.saturation
  depth_module.sequence_id
  depth_module.sequence_name
  depth_module.sequence_size
  depth_module.sharpness
  depth_module.visual_preset
  depth_module.white_balance
  depth_qos
  device_type
  diagnostics_period
  disparity_filter.enable
  disparity_to_depth.enable
  enable_color
  enable_depth
  enable_infra
  enable_infra1
  enable_infra2
  enable_rgbd
  enable_sync
  filter_by_sequence_id.enable
  filter_by_sequence_id.frames_queue_size
  filter_by_sequence_id.sequence_id
  hdr_merge.enable
  hdr_merge.frames_queue_size
  hold_back_imu_for_frames
  hole_filling_filter.enable
  hole_filling_filter.frames_queue_size
  hole_filling_filter.holes_fill
  hole_filling_filter.stream_filter
  hole_filling_filter.stream_format_filter
  hole_filling_filter.stream_index_filter
  infra1_info_qos
  infra1_qos
  infra2_info_qos
  infra2_qos
  infra_info_qos
  infra_qos
  initial_reset
  json_file_path
  linear_accel_cov
  pointcloud.allow_no_texture_points
  pointcloud.enable
  pointcloud.filter_magnitude
  pointcloud.frames_queue_size
  pointcloud.ordered_pc
  pointcloud.pointcloud_qos
  pointcloud.stream_filter
  pointcloud.stream_format_filter
  pointcloud.stream_index_filter
  publish_tf
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  reconnect_timeout
  rosbag_filename
  serial_no
  spatial_filter.enable
  spatial_filter.filter_magnitude
  spatial_filter.filter_smooth_alpha
  spatial_filter.filter_smooth_delta
  spatial_filter.frames_queue_size
  spatial_filter.holes_fill
  spatial_filter.stream_filter
  spatial_filter.stream_format_filter
  spatial_filter.stream_index_filter
  temporal_filter.enable
  temporal_filter.filter_smooth_alpha
  temporal_filter.filter_smooth_delta
  temporal_filter.frames_queue_size
  temporal_filter.holes_fill
  temporal_filter.stream_filter
  temporal_filter.stream_format_filter
  temporal_filter.stream_index_filter
  tf_publish_rate
  unite_imu_method
  usb_port_id
  use_sim_time
  wait_for_device_timeout
```
I would appreciate all help

MartyG-RealSense, 2024-06-18:
Hi @GiladRobee  Not all parameters that are configurable are included in the launch file.  They can be configured in the launch instruction though.  For example:

`ros2 launch realsense2_camera rs_launch.py depth_module.depth_profile:=1280x720x30 depth_module.depth_profile:=1280x720x30 rgb_camera.color_profile:=1280x720x30`

GiladRobee, 2024-06-19:
thank you for the quick reply, I will check it as soon as possible.
Isn't it a bit counter-productive not having those parameters appear in the launch file or the ros2 param list?
 How can I find the full list of parameters that I can send?

MartyG-RealSense, 2024-06-19:
You can obtain the full list of parameters with the command **ros2 param list**

https://github.com/IntelRealSense/realsense-ros?tab=readme-ov-file#available-parameters

GiladRobee, 2024-06-19:
So rgb_camera is not in the ros2 param list for some reason. 
Also I just tried the launch command that you sent me and got:
```
[INFO] [launch]: All log files can be found below /root/.ros/log/2024-06-19-09-24-25-595599-ubuntu-96150
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [96151]
[realsense2_camera_node-1] [INFO] [1718789065.836746874] [camera.camera]: RealSense ROS v4.54.1
[realsense2_camera_node-1] [INFO] [1718789065.836933595] [camera.camera]: Built with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1718789065.836965755] [camera.camera]: Running with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1718789066.605714529] [camera.camera]: Device with serial number 123622270351 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1718789066.605901282] [camera.camera]: Device with physical ID 2-3.2.1-4 was found.
[realsense2_camera_node-1] [INFO] [1718789066.605934306] [camera.camera]: Device with name Intel RealSense D405 was found.
[realsense2_camera_node-1] [INFO] [1718789066.606245349] [camera.camera]: Device with port number 2-3.2.1 was found.
[realsense2_camera_node-1] [INFO] [1718789066.606346853] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1718789066.606502342] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1718789066.608004080] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1718789066.608079313] [camera.camera]: Device Name: Intel RealSense D405
[realsense2_camera_node-1] [INFO] [1718789066.608102673] [camera.camera]: Device Serial No: 123622270351
[realsense2_camera_node-1] [INFO] [1718789066.608119601] [camera.camera]: Device physical port: 2-3.2.1-4
[realsense2_camera_node-1] [INFO] [1718789066.608132209] [camera.camera]: Device FW version: 5.16.0.1
[realsense2_camera_node-1] [INFO] [1718789066.608144689] [camera.camera]: Device Product ID: 0x0B5B
[realsense2_camera_node-1] [INFO] [1718789066.608155377] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [WARN] [1718789072.431210345] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [WARN] [1718789072.432513938] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [INFO] [1718789072.451933621] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1718789072.853218188] [camera.camera]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1718789072.966540585] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1718789072.966682538] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 1280, Height: 720, FPS: 30
[realsense2_camera_node-1]  19/06 09:24:32,967 WARNING [281472745203936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1] [INFO] [1718789072.970476004] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1]  19/06 09:24:32,978 WARNING [281472745203936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
```
so the rgb camera didn't change its default profile

MartyG-RealSense, 2024-06-19:
The D405 can be a bit more awkward to configure sometimes in regard to the color stream because it does not have a separate RGB sensor and instead obtains its RGB from the depth sensor and passes the data through an Image Signal Processor (ISP) chip to generate the final RGB image.

If you plan to align depth and color together and are concerned about RGB having a different resolution, librealsense's **align processing block** should automatically calculate for differences between resolutions when **align_depth.enable** is true.  Or you could use 848x480 for the depth profile so that the resolutions match.

GiladRobee, 2024-06-19:
I need the RGB stream on a high res for a separate task. But thank you very much for your help

MartyG-RealSense, 2024-06-25:
Hi @GiladRobee  Do you require further assistance with this case, please?  Thanks!

GiladRobee, 2024-06-25:
hi @MartyG-RealSense if you confirm that there is no way for me to control the RGB profile with the D405 then you can close this issue.
Thank you very much for the assistance

MartyG-RealSense, 2024-06-25:
A RealSense ROS user at https://github.com/IntelRealSense/realsense-ros/issues/3090#issuecomment-2105521061 set the color profile for D405 by making an edit to the `rs_launch.py` launch file.

closed 2024-07-04 by MartyG-RealSense