realsenseai/realsense-ros #3137: Failed to start the depth stream with D415 after updating to RealSense ROS Wrapper Version 4.55.1

AmielG, 2024-06-20:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       |  D415                            | 
| Firmware Version                   | 5.16.0.1                          | 
| Operating System & Version |   Linux - Ubuntu 22
| Kernel Version (Linux Only)    |  6.5.0-35-generic                                                                       | 
| Platform                                 | PC                           |
| Librealsense SDK Version       |   2.55.1                                                     | 
| Language                               |  CPP         | 
| Segment			         |  Robot                             | 
| ROS Distro			         |  Humble                                           | 
| RealSense ROS Wrapper Version   |  4.55.1                                        

### Issue Description
After updating to the newest version of 4.55.1, I cannot get the depth stream from the D415 camera.

This is my rs_multi_camera_launch.py:
```
import copy
from launch import LaunchDescription
import launch_ros.actions
from launch.actions import IncludeLaunchDescription
from launch.substitutions import LaunchConfiguration, ThisLaunchFileDir
from launch.launch_description_sources import PythonLaunchDescriptionSource
import sys
import pathlib
sys.path.append(str(pathlib.Path(__file__).parent.absolute()))
import rs_launch

local_parameters = [
                    {'name': 'camera_name1', 'default': 'camera1', 'description': 'camera unique name'},
                    {'name': 'serial_no1', 'default': "_211222063973", 'description': 'Serial number for camera 1'},

                    {'name': 'camera_name2', 'default': 'camera2', 'description': 'camera unique name'},
                    {'name': 'serial_no2', 'default': "_109522060516", 'description': 'Serial number for camera 2'},
                   ]

def set_configurable_parameters(local_params):
    return dict([(param['original_name'], LaunchConfiguration(param['name'])) for param in local_params])

def duplicate_params(general_params, posix):
    local_params = copy.deepcopy(general_params)
    for param in local_params:
        param['original_name'] = param['name']
        param['name'] += posix
    return local_params
    

def generate_launch_description():
    params1 = duplicate_params(rs_launch.configurable_parameters, '1')
    params2 = duplicate_params(rs_launch.configurable_parameters, '2')
    return LaunchDescription(
        rs_launch.declare_configurable_parameters(local_parameters) +
        rs_launch.declare_configurable_parameters(params1) + 
        rs_launch.declare_configurable_parameters(params2) + 
        [
        IncludeLaunchDescription(
            PythonLaunchDescriptionSource([ThisLaunchFileDir(), '/rs_launch.py']),
            launch_arguments=set_configurable_parameters(params1).items(),
        ),
        IncludeLaunchDescription(
            PythonLaunchDescriptionSource([ThisLaunchFileDir(), '/rs_launch.py']),
            launch_arguments=set_configurable_parameters(params2).items(),
        ),
    ])
```

And this is my rs_launch.py:
[... 169 lines cut ...]
[realsense2_camera_node-2] [INFO] [1718611211.405074850] [camera2.camera2]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1718611211.504505686] [camera1.camera1]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1718611211.504551016] [camera1.camera1]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1718611211.506007935] [camera1.camera1]: Starting Sensor: Depth Module
[realsense2_camera_node-2] [INFO] [1718611211.506537106] [camera2.camera2]: Stopping Sensor: Depth Module
[realsense2_camera_node-2] [INFO] [1718611211.506572743] [camera2.camera2]: Stopping Sensor: RGB Camera
[realsense2_camera_node-2] [INFO] [1718611211.507735978] [camera2.camera2]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1718611211.507794769] [camera1.camera1]: Open profile: stream_type: Depth(0), Format: Z16, Width: 1280, Height: 720, FPS: 30
[realsense2_camera_node-2] [INFO] [1718611211.509431814] [camera2.camera2]: Open profile: stream_type: Depth(0), Format: Z16, Width: 1280, Height: 720, FPS: 30
[realsense2_camera_node-1] [INFO] [1718611211.539388818] [camera1.camera1]: Starting Sensor: RGB Camera
[realsense2_camera_node-2] [INFO] [1718611211.540996512] [camera2.camera2]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1718611211.543781914] [camera1.camera1]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 30
[realsense2_camera_node-2] [INFO] [1718611211.545312664] [camera2.camera2]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 30
[realsense2_camera_node-1] [INFO] [1718611211.546698802] [camera1.camera1]: RealSense Node Is Up!
[realsense2_camera_node-2] [INFO] [1718611211.548228524] [camera2.camera2]: RealSense Node Is Up!
[realsense2_camera_node-1] [WARN] [1718611213.506597574] [camera1.camera1]: XXX Hardware Notification:Depth stream start failure,1.71861e+12,Error,Hardware Error
[realsense2_camera_node-1] [WARN] [1718611213.506683414] [camera1.camera1]: Hardware Notification:Depth stream start failure,1.71861e+12,Error,Hardware Error
[realsense2_camera_node-2] [WARN] [1718611213.508271192] [camera2.camera2]: XXX Hardware Notification:Depth stream start failure,1.71861e+12,Error,Hardware Error
[realsense2_camera_node-2] [WARN] [1718611213.508315005] [camera2.camera2]: Hardware Notification:Depth stream start failure,1.71861e+12,Error,Hardware Error

```

After many attempts to reset the camera, update the firmware, replug it, reboot the PC, and experiment with rs_launch.py, I suspect that something in the parameters pointcloud.stream_filter or pointcloud.stream_index_filter is causing the problem, but I am not certain. 
Currently, I have downgraded to the previous version, and it has returned to working. 
I would appreciate your help to fix the problem so I can use the latest features and improvements of the newest version.

MartyG-RealSense, 2024-06-20:
Hi @AmielG  There were changes in the latest 2.55.1 version of librealsense that required the **development** branch of 2.55.1 to be installed instead of the master branch when using the 4.55.1 ROS wrapper.

https://github.com/IntelRealSense/librealsense/tree/development
 
https://github.com/IntelRealSense/realsense-ros/issues/3131#issuecomment-2180125238 and the comment below it describes how you can colcon build an earlier version of `ros2-development` that will work with 2.55, or alternatively you can use the latest realsense2_camera released (4.55.1) from ROS apts.

AmielG, 2024-06-22:
Hello @MartyG-RealSense ,

Thanks for your quick reply.

First, does it mean that version 4.55.1 is considered unstable, which is why it uses the development branch of the SDK?

Second, I didn't mention this in the opening of the issue, but when I tried to fix my problem, I uninstalled all the RealSense packages from the PC and reinstalled them using apt as you suggested. Unfortunately, it still didn't work for me. Could it be that the build installed with apt is compiled with the master branch?

MartyG-RealSense, 2024-06-22:
Wrapper 4.55.1 needed the development branch of the SDK because it used features that were only available in the development branch at that time.  

Using the development version of the SDK does not necessarily make the wrapper unstable, as all new code that is to be added to the SDK in the development branch undergoes thorough build checks before it is permitted to be merged into the SDK.  Usually though, the development branch of librealsense would only be used if there is a new feature in it that you need and you cannot wait for it to be included in the SDK's master release at a later date (when the development branch of librealsense is pushed to release then it becomes the next master branch).

In regard to your second question, can you confirm which apt instruction you are referring to, please?

`sudo apt install ros-humble-realsense2-*` installs the 'ros2-development' 4.55.1 ROS wrapper.  Despite being named 'development' though, ros2-development is actually the main branch of the ROs wrapper.   :)

AmielG, 2024-06-22:
Thank you for the clarification.

I used the following command as instructed in the README on the main page of the repository:

```
sudo apt install ros-humble-realsense2-*
```

MartyG-RealSense, 2024-06-22:
That command installs the latest main version of the ROS2 wrapper, currently 4.55.1.

How I interpret the advice at https://github.com/IntelRealSense/realsense-ros/issues/3131#issuecomment-2180128045 is that this command installs a version of the 4.55.1 wrapper that will work with the master version of SDK 2.55 instead of needing to install the development branch of the SDK.

AmielG, 2024-06-22:
Ok, thank you, I'll test it again and see if it resolves the issue.

MartyG-RealSense, 2024-06-28:
You are very welcome.  :)

closed 2024-07-04 by MartyG-RealSense