realsenseai/realsense-ros #3139: fails to build from sources

alonsnir, 2024-06-24:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking our ROS RealSense Wrapper documentation [README](https://github.com/IntelRealSense/realsense-ros/blob/ros2-development/README.md).
    * Have you looked in our [Discussions](https://github.com/IntelRealSense/realsense-ros/discussions)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/realsense-ros/issues) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D435                       | 
| Firmware Version                   | 5.14.0                          | 
| Operating System & Version |   Linux (Ubuntu 22) | 
| Kernel Version (Linux Only)    |  5.15.0-112-generic                                                                       | 
| Platform                                 | PC                        |
| Librealsense SDK Version       |  2.2                                                    | 
| Language                               |  python         | 
| Segment			         |  Robot                             | 
| ROS Distro			         |  Humble                                           | 
| RealSense ROS Wrapper Version   |  built from source the ros2-development branch                                                  | 

### Issue Description
failed to build
`colcon build
Starting >>> realsense2_camera_msgs
Finished <<< realsense2_camera_msgs [0.34s]                     
Starting >>> realsense2_camera
Starting >>> realsense2_description
Finished <<< realsense2_description [0.06s]
--- stderr: realsense2_camera                         
CMake Warning at CMakeLists.txt:121 (find_package):
  Could not find a configuration file for package "realsense2" that is
  compatible with requested version "2.56.0".

  The following configuration files were considered but not accepted:

    /usr/local/lib/cmake/realsense2/realsense2Config.cmake, version: 2.55.1

CMake Error at CMakeLists.txt:126 (message):
  

  

   Intel RealSense SDK 2.0 is missing, please install it from https://github.com/IntelRealSense/librealsense/releases

  

---
Failed   <<< realsense2_camera [0.76s, exited with code 1]

Summary: 2 packages finished [1.23s]
  1 package failed: realsense2_camera
  1 package had stderr output: realsense2_camera
`

I have opted to build the SDK as [here](https://github.com/IntelRealSense/realsense-ros?tab=readme-ov-file#option-3-build-from-source)

MartyG-RealSense, 2024-06-24:
Hi @alonsnir  The 4.51.1 wrapper installation is requiring the **development** branch of librealsense (2.56) rather than the latest master branch.

https://github.com/IntelRealSense/librealsense/tree/development

If you do not wish to build the develoment branch of librealsense then a workaround is to try building an earlier version of the ROS wrapper for 2.55 from source using the **git checkout** command below. 

```
git clone https://github.com/IntelRealSense/realsense-ros.git -b ros2-development
cd realsense-ros/
git checkout d1684b4e53111f5eb3237f76299e83a864ce929d
cd ~/ros2_ws
colcon build
```

*********

Alternatively, you can  use the latest realsense2_camera released (4.55.1) from ROS apts so that you don't need to colcon the source code.

`sudo apt install sudo apt install ros-humble-realsense2-camera`

********

Another approach to the issue is to edit the **CMakeLists.txt** file of the ROS wrapper in the way described at https://github.com/IntelRealSense/realsense-ros/issues/3131#issuecomment-2178280500

alonsnir, 2024-06-25:
Hi @MartyG-RealSense 

I did as suggested:
```
git clone https://github.com/IntelRealSense/realsense-ros.git -b ros2-development
cd realsense-ros/
git checkout d1684b4e53111f5eb3237f76299e83a864ce929d
cd ~/ros2_ws
colcon build
```
and managed to build.

This doesn't work:
```
sudo apt install sudo apt install ros-humble-realsense2-camera
E: unable to locate package
```

I'm facing a new issue however, it refuses to set desired resolution and setting camera and depth profiles to default:
```
ros2 launch realsense2_camera rs_launch.py depth_module.profile:=1280x720x15 rgb_module.profile:=1280x720x15
[INFO] [launch]: All log files can be found below /home/louis/.ros/log/2024-06-25-11-35-43-756185-louis-11422
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [11423]
[realsense2_camera_node-1] [INFO] [1719304543.835111612] [camera.camera]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1719304543.835174540] [camera.camera]: Built with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1719304543.835182171] [camera.camera]: Running with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1719304543.842386060] [camera.camera]: Device with serial number 829212070545 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1719304543.842407069] [camera.camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb4/4-3/4-3:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1719304543.842411800] [camera.camera]: Device with name Intel RealSense D435 was found.
[realsense2_camera_node-1] [INFO] [1719304543.842499465] [camera.camera]: Device with port number 4-3 was found.
[realsense2_camera_node-1] [INFO] [1719304543.842504402] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1719304543.842534939] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1719304543.842619386] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1719304543.842624996] [camera.camera]: Device Name: Intel RealSense D435
[realsense2_camera_node-1] [INFO] [1719304543.842629007] [camera.camera]: Device Serial No: 829212070545
[realsense2_camera_node-1] [INFO] [1719304543.842632761] [camera.camera]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb4/4-3/4-3:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1719304543.842636817] [camera.camera]: Device FW version: 5.14.0
[realsense2_camera_node-1] [INFO] [1719304543.842640517] [camera.camera]: Device Product ID: 0x0B07
[realsense2_camera_node-1] [INFO] [1719304543.842643938] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1719304543.923448776] [camera.camera]: Set ROS param depth_module.depth_profile to default: 848x480x30
[realsense2_camera_node-1] [INFO] [1719304543.923671812] [camera.camera]: Set ROS param depth_module.infra_profile to default: 848x480x30
[realsense2_camera_node-1] [INFO] [1719304543.926590019] [camera.camera]: Set ROS param rgb_camera.color_profile to default: 640x480x30
[realsense2_camera_node-1] [INFO] [1719304543.927953430] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1719304543.928070108] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1719304543.930294470] [camera.camera]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1719304543.931685316] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1719304543.945862277] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1719304543.950642758] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 640, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1719304543.951402933] [camera.camera]: RealSense Node Is Up!

```

Attempts to run with launch file and configure the profile there ended with same result. I'm attaching the launch and the log with debug level on.
[common_params.yaml.txt](https://github.com/user-attachments/files/15967789/common_params.yaml.txt)
[realsense2_camera_node_10780_1719303447430.tar.gz](https://github.com/user-attachments/files/15967790/realsense2_camera_node_10780_1719303447430.tar.gz)
[louis_prod_launch.py.txt](https://github.com/user-attachments/files/15967792/louis_prod_launch.py.txt)
[camera-settings.json](https://github.com/user-attachments/files/15967793/camera-settings.json)

alonsnir, 2024-06-25:
@MartyG-RealSense Thanks for the prompt reply.
Yes, I'm willing to run both rgb and depth at 1280x720x15

Changed as suggested and it took the settings but now fails with hardware error:
```
ros2 launch realsense2_camera rs_launch.py depth_module.depth_profile:=1280x720x30 rgb_camera.color_profile:=1280x720x30 depth_module.infra_profile:=1280x720x30
[INFO] [launch]: All log files can be found below /home/louis/.ros/log/2024-06-25-12-46-37-469353-louis-3352
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [3353]
[realsense2_camera_node-1] [INFO] [1719308797.765893022] [camera.camera]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1719308797.765980413] [camera.camera]: Built with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1719308797.765986618] [camera.camera]: Running with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1719308797.774770406] [camera.camera]: Device with serial number 829212070545 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1719308797.774790895] [camera.camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb4/4-3/4-3:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1719308797.774795487] [camera.camera]: Device with name Intel RealSense D435 was found.
[realsense2_camera_node-1] [INFO] [1719308797.774882929] [camera.camera]: Device with port number 4-3 was found.
[realsense2_camera_node-1] [INFO] [1719308797.774887701] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1719308797.775180034] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1719308797.775314872] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1719308797.775321061] [camera.camera]: Device Name: Intel RealSense D435
[realsense2_camera_node-1] [INFO] [1719308797.775325430] [camera.camera]: Device Serial No: 829212070545
[realsense2_camera_node-1] [INFO] [1719308797.775329204] [camera.camera]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb4/4-3/4-3:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1719308797.775333386] [camera.camera]: Device FW version: 5.14.0
[realsense2_camera_node-1] [INFO] [1719308797.775337209] [camera.camera]: Device Product ID: 0x0B07
[realsense2_camera_node-1] [INFO] [1719308797.775340796] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [WARN] [1719308797.877176770] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [WARN] [1719308797.877383349] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [WARN] [1719308797.880519738] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [INFO] [1719308797.881876643] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1719308797.881928627] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1719308797.888343078] [camera.camera]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1719308797.890446987] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 1280, Height: 720, FPS: 30
[realsense2_camera_node-1] [INFO] [1719308797.912175441] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1719308797.913907834] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 30
[realsense2_camera_node-1] [INFO] [1719308797.915708641] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1] [WARN] [1719308799.889457166] [camera.camera]: XXX Hardware Notification:Depth stream start failure,1.71931e+12,Error,Hardware Error
[realsense2_camera_node-1] [WARN] [1719308799.889534187] [camera.camera]: Hardware Notification:Depth stream start failure,1.71931e+12,Error,Hardware Error
```

MartyG-RealSense, 2024-06-25:
Does it make a difference if you add **initial_reset:=true** to the launch instruction to reset the camera hardware at launch?

alonsnir, 2024-06-25:
Sorry, my mistake, I was running it with x30. Restarted with x15 and it worked.
Thank you very much

alonsnir, 2024-06-25:
Hi @MartyG-RealSense 
I have to reopen

tested 
`
ros2 launch realsense2_camera rs_launch.py depth_module.depth_profile:=1280x720x15 rgb_camera.color_profile:=1280x720x15 align_depth.enable:=True
`
and it works fine, but when I run it with following settings, getting hardware error:

launch file:
```
from launch import LaunchDescription
from launch_ros.actions import Node
from ament_index_python import get_package_share_path

import os
import sys

# Add the directory containing launch_common_config.py to the Python path
sys.path.append(os.path.dirname(os.path.realpath(__file__)))
camera_settings_path='/home/louis/realsense_launch/camera-settings.json'
common_config_path='/home/louis/realsense_launch/common_params.yaml'

def generate_launch_description():
    return LaunchDescription([ 
        Node(
            name='realsense',
            package='realsense2_camera',
            executable='realsense2_camera_node',
            respawn=True,
            parameters=[
                {"json_file_path": camera_settings_path},
                common_config_path,
            ],
        ),
    ])
```
camera-settings.json:

```
cat ./camera-settings.json 
{
  "aux-param-autoexposure-setpoint": "1536",
  "aux-param-colorcorrection1": "0.298828",
  "aux-param-colorcorrection10": "-0",
  "aux-param-colorcorrection11": "-0",
  "aux-param-colorcorrection12": "-0",
  "aux-param-colorcorrection2": "0.293945",
  "aux-param-colorcorrection3": "0.293945",
  "aux-param-colorcorrection4": "0.114258",
  "aux-param-colorcorrection5": "-0",
  "aux-param-colorcorrection6": "-0",
  "aux-param-colorcorrection7": "-0",
  "aux-param-colorcorrection8": "-0",
  "aux-param-colorcorrection9": "-0",
  "aux-param-depthclampmax": "65536",
  "aux-param-depthclampmin": "0",
  "aux-param-disparityshift": "0",
  "controls-autoexposure-auto": "True",
  "controls-autoexposure-manual": "8500",
  "controls-color-autoexposure-auto": "True",
  "controls-color-autoexposure-manual": "166",
  "controls-color-backlight-compensation": "0",
  "controls-color-brightness": "0",
  "controls-color-contrast": "50",
  "controls-color-gain": "64",
  "controls-color-gamma": "300",
  "controls-color-hue": "0",
  "controls-color-power-line-frequency": "3",
  "controls-color-saturation": "64",
[... 92 lines cut ...]
[realsense2_camera_node-1] [INFO] [1719324760.391474598] [camera.realsense]: Device with port number 4-3 was found.
[realsense2_camera_node-1] [INFO] [1719324760.391479314] [camera.realsense]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1719324760.391509190] [camera.realsense]: getParameters...
[realsense2_camera_node-1] [INFO] [1719324760.391622089] [camera.realsense]: Publish diagnostics every 1 seconds.
[realsense2_camera_node-1] [INFO] [1719324760.697312156] [camera.realsense]: JSON file is loaded! (/home/louis/realsense_launch/camera-settings.json)
[realsense2_camera_node-1] [INFO] [1719324760.697364321] [camera.realsense]: Device Name: Intel RealSense D435
[realsense2_camera_node-1] [INFO] [1719324760.697368325] [camera.realsense]: Device Serial No: 830112070132
[realsense2_camera_node-1] [INFO] [1719324760.697371815] [camera.realsense]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb4/4-3/4-3:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1719324760.697374881] [camera.realsense]: Device FW version: 5.15.0.2
[realsense2_camera_node-1] [INFO] [1719324760.697378612] [camera.realsense]: Device Product ID: 0x0B07
[realsense2_camera_node-1] [INFO] [1719324760.697381286] [camera.realsense]: Sync Mode: Off
[realsense2_camera_node-1] [WARN] [1719324760.789791743] [camera.realsense]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [WARN] [1719324760.793563887] [camera.realsense]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [INFO] [1719324760.794913899] [camera.realsense]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1719324760.794956525] [camera.realsense]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1719324760.808407689] [camera.realsense]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1719324760.814531456] [camera.realsense]: Open profile: stream_type: Infra(1), Format: Y8, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1719324760.814552463] [camera.realsense]: Open profile: stream_type: Infra(2), Format: Y8, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1719324760.814557314] [camera.realsense]: Open profile: stream_type: Depth(0), Format: Z16, Width: 1280, Height: 720, FPS: 15
[realsense2_camera_node-1] [INFO] [1719324760.839412592] [camera.realsense]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1719324760.841300456] [camera.realsense]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 15
[realsense2_camera_node-1] [INFO] [1719324760.842116886] [camera.realsense]: RealSense Node Is Up!

[realsense2_camera_node-1] [WARN] [1719324765.833770602] [camera.realsense]: XXX Hardware Notification:Frames didn't arrived within 5 seconds,1.71932e+12,Warn,Frames Timeout
```

MartyG-RealSense, 2024-06-25:
I note that your listed firmware version in the log is **5.15.0.2**.  The firmware that should be used with librealsense 2.55.1 is **5.16.0.1**.  Please update your camera's firmware.

alonsnir, 2024-06-25:
upgraded
```
ros2 launch ./louis_prod_launch.py
[INFO] [launch]: All log files can be found below /home/louis/.ros/log/2024-06-25-18-09-22-399746-louis-5527
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [5528]
[realsense2_camera_node-1] [INFO] [1719328162.563005386] [camera.realsense]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1719328162.563095158] [camera.realsense]: Built with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1719328162.563101661] [camera.realsense]: Running with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1719328162.570268701] [camera.realsense]: Device with serial number 829212070545 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1719328162.570289755] [camera.realsense]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb4/4-3/4-3:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1719328162.570294400] [camera.realsense]: Device with name Intel RealSense D435 was found.
[realsense2_camera_node-1] [INFO] [1719328162.570417996] [camera.realsense]: Device with port number 4-3 was found.
[realsense2_camera_node-1] [INFO] [1719328162.570422913] [camera.realsense]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1719328162.570454388] [camera.realsense]: getParameters...
[realsense2_camera_node-1] [INFO] [1719328162.570578498] [camera.realsense]: Publish diagnostics every 1 seconds.
[realsense2_camera_node-1] [INFO] [1719328162.871230288] [camera.realsense]: JSON file is loaded! (/home/louis/realsense_launch/camera-settings.json)
[realsense2_camera_node-1] [INFO] [1719328162.871308012] [camera.realsense]: Device Name: Intel RealSense D435
[realsense2_camera_node-1] [INFO] [1719328162.871313337] [camera.realsense]: Device Serial No: 829212070545
[realsense2_camera_node-1] [INFO] [1719328162.871317263] [camera.realsense]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb4/4-3/4-3:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1719328162.871322036] [camera.realsense]: Device FW version: 5.16.0.1
[realsense2_camera_node-1] [INFO] [1719328162.871325676] [camera.realsense]: Device Product ID: 0x0B07
[realsense2_camera_node-1] [INFO] [1719328162.871329115] [camera.realsense]: Sync Mode: Off
[realsense2_camera_node-1] [WARN] [1719328162.963051350] [camera.realsense]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [WARN] [1719328162.966554886] [camera.realsense]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [INFO] [1719328162.967828731] [camera.realsense]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1719328162.967874173] [camera.realsense]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1719328162.981340640] [camera.realsense]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1719328162.987785991] [camera.realsense]: Open profile: stream_type: Infra(1), Format: Y8, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1719328162.987805391] [camera.realsense]: Open profile: stream_type: Infra(2), Format: Y8, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1719328162.987809287] [camera.realsense]: Open profile: stream_type: Depth(0), Format: Z16, Width: 1280, Height: 720, FPS: 15
[realsense2_camera_node-1] [INFO] [1719328163.040712332] [camera.realsense]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1719328163.043162439] [camera.realsense]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 15
[realsense2_camera_node-1] [INFO] [1719328163.044221771] [camera.realsense]: RealSense Node Is Up!
[realsense2_camera_node-1] [WARN] [1719328168.016242038] [camera.realsense]: XXX Hardware Notification:Frames didn't arrived within 5 seconds,1.71933e+12,Warn,Frames Timeout
[realsense2_camera_node-1] [WARN] [1719328173.020211874] [camera.realsense]: XXX Hardware Notification:Frames didn't arrived within 5 seconds,1.71933e+12,Warn,Frames Timeout
```

MartyG-RealSense, 2024-06-25:
Have you tried adding **initial_reset:=true** to reset the camera at launch, please?

Have you also tried not importing the json file?

alonsnir, 2024-06-25:
after reset and without the json

```
ros2 launch ./louis_prod_launch.py
[INFO] [launch]: All log files can be found below /home/louis/.ros/log/2024-06-25-18-38-18-287301-louis-5715
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [5716]
[realsense2_camera_node-1] [INFO] [1719329898.451056923] [camera.realsense]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1719329898.451167738] [camera.realsense]: Built with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1719329898.451174265] [camera.realsense]: Running with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1719329898.457416324] [camera.realsense]: Device with serial number 829212070545 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1719329898.457435310] [camera.realsense]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb4/4-3/4-3:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1719329898.457439519] [camera.realsense]: Device with name Intel RealSense D435 was found.
[realsense2_camera_node-1] [INFO] [1719329898.457556993] [camera.realsense]: Device with port number 4-3 was found.
[realsense2_camera_node-1] [INFO] [1719329898.457561743] [camera.realsense]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1719329898.457589599] [camera.realsense]: getParameters...
[realsense2_camera_node-1] [INFO] [1719329898.457692091] [camera.realsense]: Publish diagnostics every 1 seconds.
[realsense2_camera_node-1] [INFO] [1719329898.458323050] [camera.realsense]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1719329898.458338595] [camera.realsense]: Device Name: Intel RealSense D435
[realsense2_camera_node-1] [INFO] [1719329898.458343310] [camera.realsense]: Device Serial No: 829212070545
[realsense2_camera_node-1] [INFO] [1719329898.458347051] [camera.realsense]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb4/4-3/4-3:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1719329898.458350896] [camera.realsense]: Device FW version: 5.16.0.1
[realsense2_camera_node-1] [INFO] [1719329898.458354896] [camera.realsense]: Device Product ID: 0x0B07
[realsense2_camera_node-1] [INFO] [1719329898.458358169] [camera.realsense]: Sync Mode: Off
[realsense2_camera_node-1] [WARN] [1719329898.539651031] [camera.realsense]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [WARN] [1719329898.543110556] [camera.realsense]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [INFO] [1719329898.544374708] [camera.realsense]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1719329898.544418705] [camera.realsense]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1719329898.557843851] [camera.realsense]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1719329898.563834068] [camera.realsense]: Open profile: stream_type: Infra(1), Format: Y8, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1719329898.563865219] [camera.realsense]: Open profile: stream_type: Infra(2), Format: Y8, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1719329898.563882806] [camera.realsense]: Open profile: stream_type: Depth(0), Format: Z16, Width: 1280, Height: 720, FPS: 15
[realsense2_camera_node-1] [INFO] [1719329898.588748254] [camera.realsense]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1719329898.591442105] [camera.realsense]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 15
[realsense2_camera_node-1] [INFO] [1719329898.592763343] [camera.realsense]: RealSense Node Is Up!
[realsense2_camera_node-1] [WARN] [1719329903.579994698] [camera.realsense]: XXX Hardware Notification:Frames didn't arrived within 5 seconds,1.71933e+12,Warn,Frames Timeout
[realsense2_camera_node-1] [WARN] [1719329908.583896345] [camera.realsense]: XXX Hardware Notification:Frames didn't arrived within 5 seconds,1.71933e+12,Warn,Frames Timeout
```

MartyG-RealSense, 2024-06-25:
Have you checked whether the streams are being published correctly in RViz despite the Hardware Notification warning messages?

Does it make a difference if you unplug the micro-sized end of the USB cable from the side of the camera, turn the connector around the other way and re-insert it (USB-C cables are two-way insertion at the micro-sized end).

alonsnir, 2024-06-25:
@MartyG-RealSense 
at at time I was forced to build from sources and downgrade the kernel as color/depth alignment didn't work. Should I give a try now with 
Option 1: Install librealsense2 debian package from Intel servers?

could it be installation instructions here are outdated? It refers to v2.53.1 as latest [Intel® RealSense™ SDK 2.0](https://github.com/IntelRealSense/librealsense/releases/tag/v2.53.1) and that version doesn't support kernels 6.*, 
while the latest version supports it [v2.55.1 ](https://github.com/IntelRealSense/librealsense/releases/tag/v2.55.1)

alonsnir, 2024-06-25:
```
Have you checked whether the streams are being published correctly in RViz despite the Hardware Notification warning messages?

Does it make a difference if you unplug the micro-sized end of the USB cable from the side of the camera, turn the connector around the other way and re-insert it (USB-C cables are two-way insertion at the micro-sized end).
```
color stream been published but not the depth
the re-connect of the micro usb had no affect

```
ros2 topic echo /camera/realsense/depth/image_rect_raw
A message was lost!!!
	total count change:1
	total count: 41---
A message was lost!!!
	total count change:1
	total count: 42---
header:
  stamp:
    sec: 1719330971
    nanosec: 888674561
  frame_id: camera_depth_optical_frame
height: 720
width: 1280
encoding: 16UC1
is_bigendian: 0
step: 2560
data:
```

alonsnir, 2024-06-26:
@MartyG-RealSense 
if both infra been disabled, then the error goes away. Could it affect depth accuracy in any way?

MartyG-RealSense, 2024-06-26:
Having the infrared streams will not affect depth accuracy.  The IR **emitter** of the camera that aids depth analysis is a separate component from the two IR sensors and can function when the IR streams are disabled.

alonsnir, 2024-06-26:
@MartyG-RealSense 
Very strange behavior observed. After system restart it doesn't see the camera:

```
ros2 launch realsense2_camera rs_launch.py depth_module.depth_profile:=1280x720x15 rgb_camera.color_profile:=1280x720x15 align_depth.enable:=True
[INFO] [launch]: All log files can be found below /home/louis/.ros/log/2024-06-26-15-49-56-787839-louis-2291
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [2292]
[realsense2_camera_node-1] [INFO] [1719406196.860501551] [camera.camera]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1719406196.860556548] [camera.camera]: Built with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1719406196.860562884] [camera.camera]: Running with LibRealSense v2.55.1
[realsense2_camera_node-1]  26/06 15:49:56,862 ERROR [125903677945408] (librealsense-exception.h:52) get_dev_capabilities: Cannot open '/dev/video0 Last Error: Permission denied
[realsense2_camera_node-1]  26/06 15:49:56,862 ERROR [125903677945408] (librealsense-exception.h:52) get_dev_capabilities: Cannot open '/dev/video1 Last Error: Permission denied
[realsense2_camera_node-1]  26/06 15:49:56,862 ERROR [125903677945408] (librealsense-exception.h:52) get_dev_capabilities: Cannot open '/dev/video2 Last Error: Permission denied
[realsense2_camera_node-1]  26/06 15:49:56,862 ERROR [125903677945408] (librealsense-exception.h:52) get_dev_capabilities: Cannot open '/dev/video3 Last Error: Permission denied
[realsense2_camera_node-1]  26/06 15:49:56,862 ERROR [125903677945408] (librealsense-exception.h:52) get_dev_capabilities: Cannot open '/dev/video4 Last Error: Permission denied
[realsense2_camera_node-1]  26/06 15:49:56,862 ERROR [125903677945408] (librealsense-exception.h:52) get_dev_capabilities: Cannot open '/dev/video5 Last Error: Permission denied
[realsense2_camera_node-1] [WARN] [1719406196.862834249] [camera.camera]: No RealSense devices were found!
```

MartyG-RealSense, 2024-06-26:
On some computers the camera may not be detected after a restart of the computer.  A solution for this can be to shut the computer down instead of doing a restart and then boot the computer by pressing the power button.

alonsnir, 2024-06-26:
same error after complete shutdown and start

MartyG-RealSense, 2024-06-26:
On this other system you may need to repeat the **git clone** installation procedure that you described earlier at https://github.com/IntelRealSense/realsense-ros/issues/3139#issuecomment-2188351723 instead of using apt install, bearing in mind the problems with using librealsense 2.55.1 with the master branch of the SDK.

alonsnir, 2024-07-01:
Hi @MartyG-RealSense, I'm closing this one, as with your help I have managed to build from sources and run on Ubuntu 22.04 kernel 5.5 (with infra disabled).
Will keep the related [3142](https://github.com/IntelRealSense/realsense-ros/issues/3142) for getting it to work on kernel 6.5

closed 2024-07-01 by alonsnir