realsenseai/realsense-ros #3142: "package 'realsense2_camera' not found

alonsnir, 2024-06-26:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D435                       | 
| Firmware Version                   | 5.14.0                          | 
| Operating System & Version |   Linux (Ubuntu 22) | 
| Kernel Version (Linux Only)    |  6.5.0-41-generic                                                                     | 
| Platform                                 | PC                        |
| Librealsense SDK Version       |  2.2                                                    | 
| Language                               |  python         | 
| Segment			         |  Robot                             | 
| ROS Distro			         |  Humble                                           | 

### "package 'realsense2_camera' not found
Option 2: Install librealsense2 (without graphical tools and examples) debian package from ROS servers (Foxy EOL distro is not supported by this option):

`sudo apt install ros-humble-librealsense2*`

Then the error:
```
ros2 launch realsense2_camera rs_launch.py depth_module.depth_profile:=1280x720x15 rgb_camera.color_profile:=1280x720x15 align_depth.enable:=True
Package 'realsense2_camera' not found: "package 'realsense2_camera' not found, searching: ['/opt/ros/humble']"

```

MartyG-RealSense, 2024-06-26:
Hi @alonsnir  Is this the same problem that you described at https://github.com/IntelRealSense/realsense-ros/issues/3139#issuecomment-2191631794 please or a different issue?

alonsnir, 2024-07-01:
Hi @MartyG-RealSense it's sort of the same, as that's the same hardware spec, but I'm trying to get it to work with the default Ubuntu 22.04 kernels 6.5, so the OS setup is different

alonsnir, 2024-07-01:
I suggest to ignore camera firmware at this point, as I get this error when, intentionally, no camera been connected.
See [Step 2, Option 2](https://github.com/IntelRealSense/realsense-ros?tab=readme-ov-file#option-2-install-librealsense2-without-graphical-tools-and-examples-debian-package-from-ros-servers-foxy-eol-distro-is-not-supported-by-this-option), same is [Step 3, option 1](https://github.com/IntelRealSense/realsense-ros?tab=readme-ov-file#option-1-install-debian-package-from-ros-servers-foxy-eol-distro-is-not-supported-by-this-option)
after completing all three steps librealsense and the the wrapper should be installed, while I'm getting that error message:
```
ros2 launch realsense2_camera rs_launch.py depth_module.depth_profile:=1280x720x15 rgb_camera.color_profile:=1280x720x15 align_depth.enable:=True
Package 'realsense2_camera' not found: "package 'realsense2_camera' not found, searching: ['/opt/ros/humble']"
```

MartyG-RealSense, 2024-07-02:
When installing with the **sudo apt install** method, only the two installation instructions below should be used after the Ubuntu sources have been configured with the instructions at http://wiki.ros.org/Installation/Ubuntu/Sources

**1.  Install librealsense2**
`sudo apt install ros-humble-librealsense2*`

**2.  Install ROS2 wrapper**
`sudo apt install ros-humble-realsense2-*`

**********

Source code and apt package builds of librealsense should not be made on the same computer, as you could end up with more than one librealsense installation, with conflicts between the two.

alonsnir, 2024-07-04:
Ubuntu sources been configured before installing

---
This is a fresh setup, no build from sources here

MartyG-RealSense, 2024-07-04:
Do librealsense's text-based tools behave correctly even if the ROS wrapper has the package error?  For example, are you able to run **rs-enumerate-devices** to obtain information about the camera?

closed 2024-07-24 by MartyG-RealSense