realsenseai/realsense-ros #3146: Hardware Error: Frames didn't arrive within 5 seconds

Kartikinator, 2024-06-28:
| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       |   D435if                          | 
| Firmware Version                   | 5.15.0.2                         | 
| Operating System & Version |   Linux (Ubuntu 22) | 
| Kernel Version (Linux Only)    |  6.5.0-41-generic   | 
| Platform                                 | PC |
| Librealsense SDK Version       |  { 2.54.1 }   | 
| Language                               |  Python   | 
| Segment			         |  Robot | 
| ROS Distro			         |  Humble   | 
| RealSense ROS Wrapper Version   |  4.54.1    | 

### Issue Description
Whenever I run any command trying to launch rs_launch.py, I get the following error

```
[realsense2_camera_node-1] [WARN] [1719591283.151755997] [camera.camera]: Device 241122072357 is connected using a 2.1 port. Reduced performance is expected.
[realsense2_camera_node-1] [INFO] [1719591283.151807772] [camera.camera]: getParameters...
[realsense2_camera_node-1] [WARN] [1719591283.152029913] [camera.camera]: Could not set param: unite_imu_method with 0 Range: [0, 2]: parameter 'unite_imu_method' has invalid type: expected [integer] got [string]
[realsense2_camera_node-1] [INFO] [1719591283.152047069] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1719591283.152055372] [camera.camera]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1719591283.152062785] [camera.camera]: Device Serial No: 241122072357
[realsense2_camera_node-1] [INFO] [1719591283.152069949] [camera.camera]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1719591283.152077502] [camera.camera]: Device FW version: 5.15.0.2
[realsense2_camera_node-1] [INFO] [1719591283.152084033] [camera.camera]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1719591283.152089430] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1719591283.248892933] [camera.camera]: Set ROS param depth_module.depth_profile to default: 640x480x15
[realsense2_camera_node-1] [INFO] [1719591283.249542080] [camera.camera]: Set ROS param depth_module.infra_profile to default: 640x480x15
[realsense2_camera_node-1] [INFO] [1719591283.253623458] [camera.camera]: Set ROS param rgb_camera.color_profile to default: 640x480x15
[realsense2_camera_node-1] [INFO] [1719591283.254242971] [camera.camera]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [INFO] [1719591283.254301813] [camera.camera]: Set ROS param accel_fps to default: 100
[realsense2_camera_node-1] [INFO] [1719591283.257185788] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1719591283.257338096] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1719591283.257377281] [camera.camera]: Stopping Sensor: Motion Module
[realsense2_camera_node-1] [INFO] [1719591283.260873484] [camera.camera]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1719591283.262727454] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 640, Height: 480, FPS: 15
[realsense2_camera_node-1] [INFO] [1719591283.263213409] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1719591283.264634591] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 640, Height: 480, FPS: 15
[realsense2_camera_node-1]  28/06 11:14:43,296 WARNING [137679968687680] (ds-calib-parsers.cpp:35) IMU Calibration is not available, default intrinsic and extrinsic will be used.
[realsense2_camera_node-1] [INFO] [1719591283.298537296] [camera.camera]: Starting Sensor: Motion Module
[realsense2_camera_node-1] [INFO] [1719591283.301865334] [camera.camera]: Open profile: stream_type: Accel(0)Format: MOTION_XYZ32F, FPS: 100
[realsense2_camera_node-1] [INFO] [1719591283.301932720] [camera.camera]: Open profile: stream_type: Gyro(0)Format: MOTION_XYZ32F, FPS: 200
[realsense2_camera_node-1] [INFO] [1719591283.304651118] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1]  28/06 11:14:48,267 WARNING [137679605855808] (backend-v4l2.cpp:1649) Frames didn't arrived within 5 seconds
[realsense2_camera_node-1] [WARN] [1719591288.267479768] [camera.camera]: XXX Hardware Notification:Frames didn't arrived within 5 seconds,1.71959e+12,Warn,Frames Timeout

```

Specifically, this is the command that I am running 
```ros2 launch realsense2_camera rs_launch.py initial_reset:=true align_depth:=true unite_imu_method:="linear_interpolation" enable_gyro:=true enable_accel:=true```

I've changed my camera firmware to 5.15.0.2 as suggested in this issue https://github.com/IntelRealSense/realsense-ros/issues/3035#issuecomment-1980796849. However, I still get the same error.

Kartikinator, 2024-06-28:
Hi, thanks for the information but that did not fix the error.

MartyG-RealSense, 2024-06-28:
Setting `unite_imu_method:=2` should resolve the warning in your log about this parameter being set to an invalid value.

In regard to the **Hardware Notification:Frames didn't arrived within 5 seconds** error, please try launching without` initial_reset:=true` as the camera reset command only needs to be tried if you have a problem with your launch.  If the launch works correctly without the reset command then it is not required to be in the launch instruction.

Also please try reversing the insertion direction of the micro-sized end of the USB cable (USB-C cables are two-way insertion at the micro-sized end) to see whether the hardware error still occurs.

Kartikinator, 2024-06-28:
I only added the initial_reset parameter after I got this issue, but I removed it and tried it again. Same issue. Here is the exact command I enetered
```ros2 launch realsense2_camera rs_launch.py align_depth:=true unite_imu_method:=2 enable_gyro:=true enable_accel:=true
```
Here is the complete output
```[INFO] [launch]: All log files can be found below /home/wilder/.ros/log/2024-06-28-13-45-56-928966-wilder-Nitro-AN517-55-126925
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [126926]
[realsense2_camera_node-1] [INFO] [1719600357.006563328] [camera.camera]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1719600357.006616757] [camera.camera]: Built with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1719600357.006622274] [camera.camera]: Running with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1719600357.015357927] [camera.camera]: Device with serial number 241122072357 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1719600357.015380215] [camera.camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1719600357.015385059] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1719600357.015493084] [camera.camera]: Device with port number 3-2 was found.
[realsense2_camera_node-1] [INFO] [1719600357.015498347] [camera.camera]: Device USB type: 2.1
[realsense2_camera_node-1] [WARN] [1719600357.015502308] [camera.camera]: Device 241122072357 is connected using a 2.1 port. Reduced performance is expected.
[realsense2_camera_node-1] [INFO] [1719600357.015532023] [camera.camera]: getParameters...
[realsense2_camera_node-1] [WARN] [1719600357.015605114] [camera.camera]: For the 'unite_imu_method' param update to take effect, re-enable either gyro or accel stream.
[realsense2_camera_node-1] [INFO] [1719600357.015610635] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1719600357.015614667] [camera.camera]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1719600357.015618660] [camera.camera]: Device Serial No: 241122072357
[realsense2_camera_node-1] [INFO] [1719600357.015622103] [camera.camera]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1719600357.015625985] [camera.camera]: Device FW version: 5.15.0.2
[realsense2_camera_node-1] [INFO] [1719600357.015629256] [camera.camera]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1719600357.015632341] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1719600357.142569264] [camera.camera]: Set ROS param depth_module.depth_profile to default: 640x480x15
[realsense2_camera_node-1] [INFO] [1719600357.143700149] [camera.camera]: Set ROS param depth_module.infra_profile to default: 640x480x15
[realsense2_camera_node-1] [INFO] [1719600357.148993833] [camera.camera]: Set ROS param rgb_camera.color_profile to default: 640x480x15
[realsense2_camera_node-1] [INFO] [1719600357.149522447] [camera.camera]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [INFO] [1719600357.149561170] [camera.camera]: Set ROS param accel_fps to default: 100
[realsense2_camera_node-1] [INFO] [1719600357.152426727] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1719600357.152516065] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1719600357.152535873] [camera.camera]: Stopping Sensor: Motion Module
[realsense2_camera_node-1] [INFO] [1719600357.156302445] [camera.camera]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1719600357.170183399] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 640, Height: 480, FPS: 15
[realsense2_camera_node-1] [INFO] [1719600357.171035790] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1719600357.173120874] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 640, Height: 480, FPS: 15
[realsense2_camera_node-1]  28/06 13:45:57,204 WARNING [131553506096704] (ds-calib-parsers.cpp:35) IMU Calibration is not available, default intrinsic and extrinsic will be used.
[realsense2_camera_node-1] [INFO] [1719600357.206557101] [camera.camera]: Starting Sensor: Motion Module
[realsense2_camera_node-1] [INFO] [1719600357.210289064] [camera.camera]: Open profile: stream_type: Accel(0)Format: MOTION_XYZ32F, FPS: 100
[realsense2_camera_node-1] [INFO] [1719600357.210345949] [camera.camera]: Open profile: stream_type: Gyro(0)Format: MOTION_XYZ32F, FPS: 200
[realsense2_camera_node-1] [INFO] [1719600357.212537609] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1]  28/06 13:46:02,175 WARNING [131553271199296] (backend-v4l2.cpp:1649) Frames didn't arrived within 5 seconds
[realsense2_camera_node-1]  28/06 13:46:02,175 WARNING [131553279592000] (backend-v4l2.cpp:1649) Frames didn't arrived within 5 seconds
[realsense2_camera_node-1] [WARN] [1719600362.175961292] [camera.camera]: XXX Hardware Notification:Frames didn't arrived within 5 seconds,1.7196e+12,Warn,Frames Timeout
[realsense2_camera_node-1] [WARN] [1719600362.176055690] [camera.camera]: XXX Hardware Notification:Frames didn't arrived within 5 seconds,1.7196e+12,Warn,Frames Timeout
[... cut ...]
[realsense2_camera_node-1] [INFO] [1719600363.951044737] [camera.camera]: Close Sensor. 
[realsense2_camera_node-1] [INFO] [1719600363.952302961] [camera.camera]: Close Sensor - Done. 
[realsense2_camera_node-1] [INFO] [1719600363.952315106] [camera.camera]: Stop Sensor: Motion Module
[realsense2_camera_node-1] [INFO] [1719600363.952672646] [camera.camera]: Close Sensor. 
[realsense2_camera_node-1] [INFO] [1719600364.893210363] [camera.camera]: Close Sensor - Done. 
[INFO] [realsense2_camera_node-1]: process has finished cleanly [pid 126926]
```

MartyG-RealSense, 2024-06-28:
As you are using librealsense 2.55.1, the camera firmware driver should be updated to **5.16.0.1**.  The 5.15.0.2 firmware stated in the log is designed for the earlier librealsense version 2.54.1.

Kartikinator, 2024-06-28:
That's strange, I didn't notice that it was using 2.55.1. My ros2_ws folder has librealsense 2.54.1 and realsense-ros 4.54.1. Not sure why it is using librealsense 2.55.1.

Kartikinator, 2024-06-28:
I found that there is an underlying librealsense package already installed. I uninstalled it and now I am now trying to install 2.54.1 but when I run 
```./scripts/setup_udev_rules.sh```
I keep getting permission denied
```00:02: Failed to write 'change' to '/sys/devices/pnp0/00:02/uevent': Permission denied
00:03: Failed to write 'change' to '/sys/devices/pnp0/00:03/uevent': Permission denied
00:04: Failed to write 'change' to '/sys/devices/pnp0/00:04/uevent': Permission denied
00:05: Failed to write 'change' to '/sys/devices/pnp0/00:05/uevent': Permission denied
00:06: Failed to write 'change' to '/sys/devices/pnp0/00:06/uevent': Permission denied
power: Failed to write 'change' to '/sys/devices/power/uevent': Permission denied
software: Failed to write 'change' to '/sys/devices/software/uevent': Permission denied
broadcast: Failed to write 'change' to '/sys/devices/system/clockevents/broadcast/uevent': Permission denied
clockevent0: Failed to write 'change' to '/sys/devices/system/clockevents/clockevent0/uevent': Permission denied
clockevent1: Failed to write 'change' to '/sys/devices/system/clockevents/clockevent1/uevent': Permission denied
```

Any idea how to fix?

MartyG-RealSense, 2024-06-28:
This issue has only previously been reported a couple of times and both of them involved installing on a Raspberry Pi board - please see https://github.com/IntelRealSense/librealsense/issues/11393

Which computer or computing board are you installing librealsense on, please?

Kartikinator, 2024-06-28:
I am on an Acer Nitro AN517-55

MartyG-RealSense, 2024-06-29:
The source code installation instructions for librealsense state that `./scripts/setup_udev_rules.sh` should be run from the librealsense "root directory".  This typically means the very first directory in a folder, but in the case of the librealsense source code folder that is extracted from a zip file, the root is the second directory in (`librealsense > librealsense`, the one containing folders and files).

If you are installing librealsense from a source code folder, can you confirm please if you input the udev script's launch command whilst in this directory, please?

[image]

If running the script from the librealsense > librealsense directory does not make a difference, have you tried uninstalling udev with the command below and then attempting to run the udev script again?

`./scripts/setup_udev_rules.sh --uninstall`

Kartikinator, 2024-07-01:
I did run this command in the proper root directory. To double check, I tried to uninstall and reinstall, but when I run the uninstall command I get the permission denied messages once again.
```
LNXSYSTM:00: Failed to write 'change' to '/sys/devices/LNXSYSTM:00/uevent': Permission denied
10251229:00: Failed to write 'change' to '/sys/devices/LNXSYSTM:00/10251229:00/uevent': Permission denied
input23: Failed to write 'change' to '/sys/devices/LNXSYSTM:00/10251229:00/input/input23/uevent': Permission denied
event9: Failed to write 'change' to '/sys/devices/LNXSYSTM:00/10251229:00/input/input23/event9/uevent': Permission denied
INT3420:00: Failed to write 'change' to '/sys/devices/LNXSYSTM:00/INT3420:00/uevent': Permission denied
LNXPOWER:07: Failed to write 'change' to '/sys/devices/LNXSYSTM:00/LNXPOWER:07/uevent': Permission denied
LNXPWRBN:00: Failed to write 'change' to '/sys/devices/LNXSYSTM:00/LNXPWRBN:00/uevent': Permission denied
input3: Failed to write 'change' to '/sys/devices/LNXSYSTM:00/LNXPWRBN:00/input/input3/uevent': Permission denied
...
```

MartyG-RealSense, 2024-07-01:
What method did you use to install the librealsense SDK and the RealSense ROS wrapper, please?

For example, you would only need to use the **./scripts/setup_udev_rules** udev installation script if you were installing librealsense from source code using the instructions at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

Kartikinator, 2024-07-01:
When I was using that command, I tried the source method. I then realized it would be easier to use ```sudo apt``` and switched to that instead. I currently have librealsense 2.54.1 installed the latest ros wrapper installed. However, everytime I run a launch command it says its building with librealsense 2.55.1

micah-cobot, 2024-07-02:
@Kartikinator You should first uninstall every version of librealsense on your computer, ie try uninstalling with apt as well as deleting realsense* or librealsense* binaries from your `/usr/` directory which may have been installed if you built from source. 

Only after ensuring all versions are deleted from your system, you should re-build 2.54.1 from source, and make sure your ROS wrapper version is also the version compatible with SDK v2.54.1 (ROS wrapper version 4.54.1) and camera firmware version 5.15.0.2 Compatibility table is here: https://dev.intelrealsense.com/docs/firmware-releases-d400 and check the ROS wrapper tagged release notes for compatibility of the ROS wrapper

MartyG-RealSense, 2024-07-09:
Hi @Kartikinator  Do you require further assistance with this case, please?  Thanks!

hong-cr, 2024-07-10:
Hello~i am ros2 and github newbie. i'm trying to operate realsense on rviz2 to prepare contest,
but when i run the realsense (run code : ros2 launch realsense2_camera rs_launch.py depth_module.profile:=1280x720x30 pointcloud.enable:=true)

this error comes

"[WARN] [1720587386.930576392] [camera.camera]: Device 234322307006 is connected using a 2.1 port. Reduced performance is expected." (but i think this code in not that important)

"[realsense2_camera_node-1]  10/07 13:57:42,327 WARNING [128431772411456] (backend-v4l2.cpp:1649) Frames didn't arrived within 5 seconds
[realsense2_camera_node-1] [WARN] [1720587462.328055125] [camera.camera]: XXX Hardware Notification:Frames didn't arrived within 5 seconds,1.72059e+12,Warn,Frames Timeout">>> i can't solve this problem on my own ability.... i saw some solution like chage the usb line, but it didn't work... plz help me

>>this is information of mine

uname -a  : "Linux hong-27V70N-GR5SK 6.5.0-41-generic #41~22.04.2-Ubuntu SMP PREEMPT_DYNAMIC Mon Jun  3 11:32:55 UTC 2 x86_64 x86_64 x86_64 GNU/Linux"

rs-fw-update : Connected devices:
1) [USB] Intel RealSense D455 s/n 234322307006, update serial number: 226523135751, firmware version: 5.15.1

and this is exact codes on terminal about how i operate the realsense(after install ros2 humble)
1. sudo apt install ros-humble-librealsense2*
2. mkdir -p ~/ros2_ws/src
3. cd ~/ros2_ws/src/
4. git clone https://github.com/IntelRealSense/realsense-ros.git -b ros2-development
5. cd ~/ros2_ws
6. sudo apt-get install python3-rosdep -y
7. sudo rosdep init
8. rosdep update
9. rosdep install -i --from-path src --rosdistro humble --skip-keys=librealsense2 -y
10. sudo apt install python3-colcon-common-extensions
11. colcon build
12. ROS_DISTRO=humble
13. source /opt/ros/humble/setup.bash
14. cd ~/ros2_ws
15. . install/local_setup.bash
16. ros2 launch realsense2_camera rs_launch.py depth_module.profile:=1280x720x30 pointcloud.enable:=true
17. rviz2

this code from "https://github.com/IntelRealSense/realsense-ros" here's "Installation on Ubuntu"part

thank you

MartyG-RealSense, 2024-07-10:
Hi @hong-cr  The main branch of the ROS2 wrapper is now **ros2-master** instead of ros2-development.  The details about this change can be read at https://github.com/IntelRealSense/librealsense/issues/13113

I would recommend removing your current ROS2 wrapper installation by deleting the entire **/ros2_ws/src/** catkin workspace folder.  Then repeat the colcon build instructions but this time git clone from ros2-master.

`git clone https://github.com/IntelRealSense/realsense-ros.git -b ros2-master`

Kartikinator, 2024-07-12:
> Hi @Kartikinator Do you require further assistance with this case, please? Thanks!

I no longer have time to debug this issue but thanks for your help

closed 2024-07-12 by Kartikinator

MartyG-RealSense, 2024-07-13:
Okay, thanks very much for the update.

hong-cr, 2024-07-17:
@MartyG-RealSense

Oh, I solved it! thankyou for ur advices, my problem is so minoritic.... It just because of USB Port 2.0 / 3.0 problem haha!

MartyG-RealSense, 2024-07-17:
It's great to hear that you found a solution!

JianchouBai, 2025-03-28:
I have the same problem as the subject, but I have upgraded the firmware version to 5.16 and I still get the same error

INFO] [launch]: All log files can be found below /home/bjc/.ros/log/2025-03-28-16-15-34-744157-bjc-ros2-8940
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [8941]
[realsense2_camera_node-1] [INFO] [1743149735.003672405] [camera.camera]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1743149735.003794617] [camera.camera]: Built with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1743149735.003814828] [camera.camera]: Running with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1743149735.054730963] [camera.camera]: Device with serial number 250122074851 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1743149735.054836192] [camera.camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:15.0/0000:03:00.0/usb3/3-2/3-2:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1743149735.054854814] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1743149735.054981668] [camera.camera]: Device with port number 3-2 was found.
[realsense2_camera_node-1] [INFO] [1743149735.055001236] [camera.camera]: Device USB type: 2.1
[realsense2_camera_node-1] [WARN] [1743149735.055008348] [camera.camera]: Device 250122074851 is connected using a 2.1 port. Reduced performance is expected.
[realsense2_camera_node-1] [INFO] [1743149735.055054888] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1743149735.055660518] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1743149735.055694221] [camera.camera]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1743149735.055712779] [camera.camera]: Device Serial No: 250122074851
[realsense2_camera_node-1] [INFO] [1743149735.055718124] [camera.camera]: Device physical port: /sys/devices/pci0000:00/0000:00:15.0/0000:03:00.0/usb3/3-2/3-2:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1743149735.055725830] [camera.camera]: Device FW version: 5.16.0.1
[realsense2_camera_node-1] [INFO] [1743149735.055730518] [camera.camera]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1743149735.055736894] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1743149735.173758012] [camera.camera]: Set ROS param depth_module.depth_profile to default: 640x480x15
[realsense2_camera_node-1] [INFO] [1743149735.174076670] [camera.camera]: Set ROS param depth_module.infra_profile to default: 640x480x15
[realsense2_camera_node-1] [INFO] [1743149735.180517956] [camera.camera]: Set ROS param rgb_camera.color_profile to default: 640x480x15
[realsense2_camera_node-1] [INFO] [1743149735.181364040] [camera.camera]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [INFO] [1743149735.181482624] [camera.camera]: Set ROS param accel_fps to default: 100
[realsense2_camera_node-1] [INFO] [1743149735.187244865] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1743149735.187349489] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1743149735.187369225] [camera.camera]: Stopping Sensor: Motion Module
[realsense2_camera_node-1] [INFO] [1743149735.191243865] [camera.camera]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1743149735.196583352] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 640, Height: 480, FPS: 15
[realsense2_camera_node-1] [INFO] [1743149735.197862510] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1743149735.202979026] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 640, Height: 480, FPS: 15
[realsense2_camera_node-1]  28/03 16:15:35,221 WARNING [139716933936704] (ds-calib-parsers.cpp:35) IMU Calibration is not available, default intrinsic and extrinsic will be used.
[realsense2_camera_node-1] [INFO] [1743149735.223359044] [camera.camera]: Starting Sensor: Motion Module
[realsense2_camera_node-1] [INFO] [1743149735.227130823] [camera.camera]: Open profile: stream_type: Accel(0)Format: MOTION_XYZ32F, FPS: 100
[realsense2_camera_node-1] [INFO] [1743149735.227200422] [camera.camera]: Open profile: stream_type: Gyro(0)Format: MOTION_XYZ32F, FPS: 200
[realsense2_camera_node-1] [INFO] [1743149735.230298606] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1]  28/03 16:15:40,203 WARNING [139716376012352] (backend-v4l2.cpp:1649) Frames didn't arrived within 5 seconds
[realsense2_camera_node-1] [WARN] [1743149740.203319883] [camera.camera]: XXX Hardware Notification:Frames didn't arrived within 5 seconds,1.74315e+12,Warn,Frames Timeout
[realsense2_camera_node-1]  28/03 16:15:40,208 WARNING [139716292769344] (backend-v4l2.cpp:1649) Frames didn't arrived within 5 seconds
[realsense2_camera_node-1] [WARN] [1743149740.208829035] [camera.camera]: XXX Hardware Notification:Frames didn't arrived within 5 seconds,1.74315e+12,Warn,Frames Timeout
[realsense2_camera_node-1]  28/03 16:15:45,205 WARNING [139716376012352] (backend-v4l2.cpp:1649) Frames didn't arrived within 5 seconds
[realsense2_camera_node-1] [WARN] [1743149745.205267015] [camera.camera]: XXX Hardware Notification:Frames didn't arrived within 5 seconds,1.74315e+12,Warn,Frames Timeout
[... cut ...]
[realsense2_camera_node-1] [INFO] [1743149746.823260857] [camera.camera]: Close Sensor. 
[realsense2_camera_node-1] [INFO] [1743149746.826074399] [camera.camera]: Close Sensor - Done. 
[realsense2_camera_node-1] [INFO] [1743149746.826103081] [camera.camera]: Stop Sensor: Motion Module
[realsense2_camera_node-1] [INFO] [1743149746.826414680] [camera.camera]: Close Sensor. 
[realsense2_camera_node-1] [INFO] [1743149747.323950791] [camera.camera]: Close Sensor - Done. 
[INFO] [realsense2_camera_node-1]: process has finished cleanly [pid 8941]

MartyG-RealSense, 2025-03-28:
Hi @JianchouBai  It looks as though you have the gyro and accel streams of the IMU (Motion Module) enabled.  If they are enabled, how does your launch behave if they are disabled and only depth and color topics are published?

`ros2 launch realsense2_camera rs_launch.py enable_depth:=true enable_color:=true enable_gyro:=false enable_accel:=false`

MartyG-RealSense, 2026-03-31:
@Dede20231   Thank you for your detailed feedback.

Installing the RealSense SDK on your computer (or a program that was created with the RealSense SDK( is necessary for most of a RealSense camera's functions to be accessed, otherwise Ubuntu will only know how to access the camera's video as an ordinary webcam.

As mentioned above though, if a Raspberry Pi board is used then there can be problems with accessing IMU data.  The SDK can work on Pi but it is not one of our officially supported hardware platforms.

I'm pleased to hear that you found a workaround to use the camera as a HID device without pyrealsense2.