realsenseai/realsense-ros #3155: D435i emitter_on_off problem

davidlmljll, 2024-07-04:
### D435i set emitter_on_off still appear projection points in left and right grayscale images

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | { D435i  }                          | 
| Firmware Version                   | (librealsense-2.54.2)                          | 
| Operating System & Version |   { Linux (Ubuntu 22)} | 
| Kernel Version (Linux Only)    |  (5.15.122-tegra)                                                                       | 
| Platform                                 | NVIDIA Jetson orin nano                           |
| Librealsense SDK Version       |  { librealsense-2.54.2}                                                    | 
| Language                               |  {C++}         | 
| Segment			         |  {Robot }                             | 
| ROS Distro			         |  {Humble}                                           | 
| RealSense ROS Wrapper Version   |  {4.54.1}                                                  | 

### Issue Description
When setting emitter_on_off to true, the projector can be seen in the on-off state using the realsense SDK, but in the off state, the left and right grayscale images still have projection points caused by the projector. How to solve this problem? In addition, does setting frame_laser_power_mode=1 mean that the projector is turned off? This parameter setting seems to be different from the previous version?
Projector off status(frame_laser_power_mode=1):
[image]

Projector on status(frame_laser_power_mode=0):
[image]

MartyG-RealSense, 2024-07-04:
Hi @davidlmljll  **emitter_on_off** alternates the emitter rapidly on-off on a per-frame basis (one frame on, one frame off, one frame on, etc).  So it can be difficult to capture an image of a frame where the dot pattern is completely off unless program code is used to save each individual frame instead of trying to snapshot the screen in the Viewer.

The easiest way to capture an off-frame in the Viewer is to start the Infrared stream, enable Emitter On Off and then tap the keyboard space bar repeatedly to pause and unpause the stream until you pause on a frame that has no pattern.

https://github.com/IntelRealSense/librealsense/pull/3066 has information about the emitter on-off alternating mode.  The page has an animated image with a fast-strobing light, so please be aware if you have epileptic sensitivity.

When **laser_power** is set to '1', the visibility of the dots is at its absolute minimum but the projector is still enabled. Setting laser_power to '0' disables the projector.  '0' has always been the value that turns the projector off.

davidlmljll, 2024-07-05:
The recorded ros2bag has been converted to a single frame message, but projection points still appear
[image]

MartyG-RealSense, 2024-07-05:
If i is showing a single frame, could it be a frame where emitter_on_off is in the on state instead of the off state?

davidlmljll, 2024-07-08:
Yes, it is on, the off state image is as follows:
[image]

MartyG-RealSense, 2024-07-08:
Is the off-frame easier to capture if you use 6 FPS instead of 30?

davidlmljll, 2024-07-08:
After setting the FPS to 6, there are still shadow points and exposure issues
[image]

MartyG-RealSense, 2024-07-08:
If you are using the RealSense ROS2 wrapper then the only remaining option that is likely to work is to turn off the emitter completely instead of alternating it, and if possible increase the strength of lighting illumination in the scene.  RealSense 400 Series cameras can make use of ambient light in the scene to analyze surfaces for depth information, so increased lighting could act as an alternative depth-analysis aid to projecting a dot pattern.

davidlmljll, 2024-07-09:
Ok,thanks！

closed 2024-07-09 by davidlmljll