realsenseai/realsense-ros #3157: Checking for default intrinsic and extrinsic parameters

TakShimoda, 2024-07-09:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.13.0.50| 
| Operating System & Version |   Ubuntu 20.04  | 
| Kernel Version (Linux Only)    | 5.15.0-94-generic                                         | 
| Platform                                 | PC/NVIDIA Jetson Nano 4GB  |
| SDK Version                            |  2.54.2                         | 
| Language                            |  N/A                          | 
| Segment			|  Robot                         | 

### Issue Description
I performed stereo-IMU calibration with Kalibr and wanted to check the intrinsic/extrinsic parameters against the default values. I wanted to confirm if the topics (e.g. **/camera/camera/infra1/camera_info**) provide the default intrinsic parameters from factory settings: 
- camera intrinsic parameters for each of the stereo cameras $(f_x, f_y, c_x, c_y)$ 
- radial and tangential distortion coefficients $(k_1, k_2, k_3, t_1, t_2)$ 

I checked mine for all cameras (infra1, infra2, color) and they were different from other documents, such as [ORB-SLAM3's calibration tutorial](https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/Calibration_Tutorial.pdf) using d435i so I wanted to make sure these were the correct factory parameters.  

I also wanted to check the transformation from IMU to left/right/color cameras but couldn't find it anywhere. I  looked at the mechanical drawings from the [datashet](https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/Intel-RealSense-D400-Series-Datasheet.pdf) but I could only get the stereo baseline, and couldn't see the IMU in the drawings.

Thanks

MartyG-RealSense, 2024-07-10:
Hi @TakShimoda  You can ensure that your calibration parameters are the factory-new default values by using the instructions at https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487 to reset to defaults in the RealSense Viewer tool.

Each individual RealSense camera has unique intrinsic and extrinsic values due to the manufacturing process at the factory, so it is normal that the values of your camera will be different when compared to other cameras of the same model.

You can list all intrinsics and extrinsics for your camera including the IMU ones by launching the 'rs-enumerate-devices' tool in calibration mode with the command below.

`rs-enumerate-devices -c`

TakShimoda, 2024-07-10:
Thanks Marty, I can see the intrinsic and extrinsic parameters.

closed 2024-07-10 by TakShimoda