realsenseai/realsense-ros #3158: Can't execute "ros2 run realsense2_camera realsense2_camera_node"

drinkBr, 2024-07-10:
| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D435| 
| Firmware Version                   | 5.13.0.55  | 
| Operating System & Version | Ubuntu22.04 | 
| Platform                                 | Raspberry Pi4 model B |
| Librealsense SDK Version       |  2.55.1  (realsense-viewer Version)                                                  | 
| Segment			         |  Robot | 
| ROS Distro			         |  Humble                                         | 

### Issue Description
I encountered the following error when executing `ros2 run realsense2_camera realsense2_camera_node`.

I believe that it is necessary to set the value of rgb_camera.power_line_frequency, but I would like to hear your opinion.
Also, after this error was displayed, I executed realsense-viewer, and when I turned on the Stereo Module, the following error appeared. I confirmed that the same error message appeared when executing `ros2 run realsense2_camera realsense2_camera_node`, so I believe they are related. I would like to hear your opinion.

MartyG-RealSense, 2024-07-10:
Hi @drinkBr  That **rgb_camera.power_line_frequency** message can usually be ignored.  The power_line_frequency parameter is usually configured in situations where you have fluorescent lights on the ceiling that are interfering with the image (because they flicker at frequencies that are hard to see with the human eye) and you need to change its value to reduce the interference.

Do you experience the same problems if you launch with **ros2 launch** instead of ros2 run, please?

`ros2 launch realsense2_camera rs_launch.py`

drinkBr, 2024-07-11:
I executed `ros2 launch realsense2_camera rs_launch.py`, but the error message seems to be almost the same.

MartyG-RealSense, 2024-07-11:
**xioctl(UVCIOC_CTRL_QUERY)** errors are typically related to a conflict between librealsense and the Linux kernel.  Such conflicts may be avoided if librealsense is built from source code with CMake with the flag **-DFORCE_RSUSB_BACKEND=TRUE** included in the CMake build instruction.

Which method did you use to install librealsense on your Pi, please?

drinkBr, 2024-07-12:
> Such conflicts may be avoided if librealsense is built from source code with CMake with the flag -DFORCE_RSUSB_BACKEND=TRUE included in the CMake build instruction.

I understand how to avoid the error, but I don't know exactly what actions to take. Since I am a beginner with Ubuntu, I would appreciate a simplified explanation.

> Which method did you use to install librealsense on your Pi, please?

The specific steps are as follows. I have also included the URL I referenced.
-install dependencies-
```
sudo apt-get update && sudo apt-get upgrade && sudo apt-get dist-upgrade
sudo apt-get install libssl-dev libusb-1.0-0-dev libudev-dev pkg-config libgtk-3-dev
sudo apt-get install git wget cmake build-essential
sudo apt-get install libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev at
```
-install librealsense2-
```
git clone https://github.com/IntelRealSense/librealsense.git
cd librealsense
mkdir build && cd build
cmake ../ -DBUILD_EXAMPLES=true
sudo make uninstall && make clean && make && sudo make && sudo make install
cd ..
./scripts/setup_ubev_rules.sh
```
-LibUVC-backend installation-
```
wget https://github.com/IntelRealSense/librealsense/raw/master/scripts/libuvc_installation.sh
chmod +x ./libuvc_installation.sh
./libuvc_installation.sh
```

Reference
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#building-librealsense2-sdk
https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md
https://www.youtube.com/watch?v=zbTtNsWz6Ec

Thanks.

MartyG-RealSense, 2024-07-12:
To build librealsense from source code in RSUSB = true mode, you need additional flags included in your CMake build command.  At the moment you are using this:

`cmake ../ -DBUILD_EXAMPLES=true`

This command will not build librealsense in RSUSB mode though.  To do so, please try the build command below.

`cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true`

This command comes from my RSUSB installation guide at https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692

I note that after performing the CMake build, you then used the libuvc backend script.  But the CMake build process will have already installed librealsense, so doing a libuvc backend build afterwards means that you could have two separate installations of librealsense on the same computer that could potentially conflict with each other.

drinkBr, 2024-07-13:
> To build librealsense from source code in RSUSB = true mode, you need additional flags included in your CMake build command. At the moment you are using this:
>
> cmake ../ -DBUILD_EXAMPLES=true
>
> This [...]

I am currently working on this, so I will report back once it's finished.

I have a separate question. I want to obtain RGB and depth images from Realsense based on key inputs. However, I read on another page that images are obtained from the topic, which made me wonder. Since it is not possible to perform key input and image subscription at the same time, I think this might be difficult. Is there any solution to this? I would like to hear your opinion.

MartyG-RealSense, 2024-07-13:
Some RealSense Python users write a script that uses **rospy** to subscribe to ROS topics and have the script use those as the camera data.  The link below has an example of doing so.

https://stackoverflow.com/questions/62938146/getting-realsense-depth-frame-in-ros

The RealSense ROS wrapper also provides some example ROS2 'node scripts' in Python language.

 https://github.com/IntelRealSense/realsense-ros/tree/ros2-master/realsense2_camera/scripts

drinkBr, 2024-07-14:
> To build librealsense from source code in RSUSB = true mode, you need additional flags included in your CMake build command. At the moment you are using this:
>
> cmake ../ -DBUILD_EXAMPLES=true
>
> This [...]

I tested two methods for this.
First,
```
cd librealsense/build
cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true
```
After executing this, I immediately ran realsense-viewer and was able to confirm both RGB and Depth images. Then, when I ran `ros2 run realsense2_camera realsense2_camera_node`, it seemed to be successful. However, after a while, the same error as before occurred.
```
[INFO] [1720575182.089151480] [camera.camera]: Device FW version: 5.13.0.55
[INFO] [1720575182.089356979] [camera.camera]: Device Product ID: 0x0B07
[INFO] [1720575182.089552885] [camera.camera]: Sync Mode: Off
[WARN] [1720575182.304168366] [camera.camera]: Could not set param: rgb_camera.power_line_frequency with 3 Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {rgb_camera.power_line_frequency} doesn't comply with integer range.
[INFO] [1720575182.380895995] [camera.camera]: Stopping Sensor: Depth Module
[INFO] [1720575182.382941983] [camera.camera]: Stopping Sensor: RGB Camera
[INFO] [1720575182.460460718] [camera.camera]: Starting Sensor: Depth Module
[INFO] [1720575182.503783789] [camera.camera]: Open profile: stream_type: Infra(1), Format: Y8, Width: 848, Height: 480, FPS: 30
[INFO] [1720575182.504291397] [camera.camera]: Open profile: stream_type: Infra(2), Format: Y8, Width: 848, Height: 480, FPS: 30
[INFO] [1720575182.504599488] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
[WARN] [1720575182.611571490] [camera.camera]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.
 10/07 10:33:07,728 ERROR [281473139402976] (librealsense-exception.h:52) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
 10/07 10:33:07,729 ERROR [281473139402976] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
 10/07 10:33:12,848 ERROR [281473114040544] (librealsense-exception.h:52) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such file or directory
 10/07 10:33:12,848 ERROR [281473114040544] (error-handling.cpp:99) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such file or directory
 10/07 10:33:17,972 ERROR [281473147857120] (librealsense-exception.h:52) set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
 10/07 10:33:17,973 ERROR [281473147857120] (rs.cpp:237) [rs2_get_video_stream_intrinsics( from:0xffff8805c040, intr:[ -1471343544x65535  p[-1.60068e-27 9.18341e-41]  f[-1.60092e-27 9.18341e-41]  UNKNOWN [9.18341e-41 -1.60094e-27 9.18341e-41 -4.06313e-34 9.18341e-41] ] ) Backend] set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
[WARN] [1720575197.973901736] [camera.camera]: No intrinsics available for this stream profile. Using zeroed intrinsics as default.
```
When I checked realsense-viewer, I could no longer get the Depth image, and the following error occurred.
```
Backend in rs2_open_multiple(sensor:0xffffa0016110, profiles:0xffffa0162bc0,
count:1):
set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
```
Second,
```
cd librealsense/build
cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true
sudo make uninstall && make clean && make && sudo make install
```
As before, when I ran realsense-viewer, I was able to confirm both RGB and Depth images. However, when I tried executing ros2 run realsense2_camera realsense2_camera_node, the same error appeared without success. After that, I checked realsense-viewer again, and I could no longer get the Depth image, and the same error as before was displayed.

Are the commands I entered correct? I would like to hear your opinion.
Thanks.

MartyG-RealSense, 2024-07-14:
The commands that you entered are correct.

If your librealsense version is 2.55.1 though then the firmware version installed in your camera should be the current latest **5.16.0.1** version instead of the 5.13.0.55 currently installed.

MartyG-RealSense, 2024-07-21:
Hi @drinkBr  Do you require further assistance with this case, please?  Thanks!

drinkBr, 2024-07-22:
Sorry for the late reply.

> If your librealsense version is 2.55.1 though then the firmware version installed in your camera should be the current latest 5.16.0.1 version instead of the 5.13.0.55 currently installed.

I don't have a Raspberry Pi at hand right now, so I can't verify this. I will send another message later this week.
Thanks.

MartyG-RealSense, 2024-07-22:
You're very welcome.  I look forward to your next update.  Good luck!

MartyG-RealSense, 2024-07-23:
Do you have access to any other computers other than your Pi that you could attempt a firmware update on?

drinkBr, 2024-07-24:
Yes, I have a Windows PC.

MartyG-RealSense, 2024-07-24:
The firmware driver is stored inside the camera hardware and not on the computer.  So if you are able to install the RealSense Viewer tool on your Windows computer and update the firmware in it then the updated firmware will be carried back to the Pi inside the camera.

You can download and run the Viewer on its own on Windows using the **Intel.RealSense.Viewer.exe** file in the Assets file list at the link below.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.55.1

[image]

drinkBr, 2024-07-25:
I installed RealSense Viewer on my Windows computer and verified it. As a result, I was able to confirm both RGB and Depth images. It seems that the firmware is installed correctly. There was a warning message that concerns me, so I would like to hear your opinion on this.
[image]

MartyG-RealSense, 2024-07-25:
There is no harm at all in clicking the Enable button to enable frame metadata and I would recommend that you do so.  

What this message means is that when frame metadata is disabled, some types of camera metadata information will not be available, and the metadata that is provided will be 'software metadata' that has timing based on the computer's clock instead of the camera's firmware (which provides 'hardware metadata').

RGB and depth streaming can still work normally without frame metadata support being enabled.  But when it is so easy to enable on Windows in the RealSense Viewer, there is no reason not to enable it.

drinkBr, 2024-07-25:
While using realsense-viewer on the Raspberry Pi, after an error occurred with the Stereo Module, I turned on the RGB Camera and the following error was observed.

When using realsense-viewer on the Raspberry Pi, the performance is very slow. Could this be one of the causes of the error?

MartyG-RealSense, 2024-07-25:
RealSense cameras commonly have performance issues on Raspberry Pi boards.    

The same **errno=5 Last Error: Input/output error** occurred in a Pi case at https://github.com/IntelRealSense/realsense-ros/issues/2991

**xioctl(VIDIOC_S_FMT)** errors are typically related to a conflict between librealsense and the kernel.  The **libuvc backend** SDK installation method that you used earlier works best with Pi, as it enables librealsense to bypass the kernel.

drinkBr, 2024-07-25:
I have one more question. Currently, I am using a 32GB microSD on the Raspberry Pi, but should I consider using a larger capacity microSD for Ubuntu?

MartyG-RealSense, 2024-07-25:
There is not a way to tell if there is a possible kernel conflict with librealsense other than when xioctl(VIDIOC_S_FMT) errors occur.

I would think that the access speed of the microSD card will be more important to performance than the capacity if the Ubuntu OS is being run off the card.  An **SSD** drive would give faster drive access.

https://thepihut.com/collections/raspberry-pi-ssd-storage

Another limiting factor on performance would be the real memory capacity of your Pi.  Once its real memory is used up, it will start using the virtual memory of a **swapfile**, which converts free storage space on the drive into virtual memory that is slower than the real memory on the Pi board.

https://pimylifeup.com/raspberry-pi-swap-file/

drinkBr, 2024-07-30:
Sorry for the late reply. I am currently trying to set up a ROS2 environment on a USB-connected SSD that I have. I will send another message later this week.

MartyG-RealSense, 2024-07-30:
It's no problem at all.  I look forward to your next report.  Thanks very much for the update!

drinkBr, 2024-08-05:
Hello, it's been a while. Sorry for the delay in getting back to you. I have installed Ubuntu 22.04 on an SSD and set up the ROS2 environment. Before I proceed with the installation for RealSense, I would like to confirm the installation steps. Are the following steps correct?

-install dependencies-
```
sudo apt-get update && sudo apt-get upgrade && sudo apt-get dist-upgrade
sudo apt-get install libssl-dev libusb-1.0-0-dev libudev-dev pkg-config libgtk-3-dev
sudo apt-get install git wget cmake build-essential
sudo apt-get install libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev at
```
-install librealsense2-
```
git clone https://github.com/IntelRealSense/librealsense.git
cd librealsense
mkdir build && cd build
cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true
sudo make uninstall && make clean && make && sudo make && sudo make install
cd ..
./scripts/setup_ubev_rules.sh
```
-LibUVC-backend installation-
```
wget https://github.com/IntelRealSense/librealsense/raw/master/scripts/libuvc_installation.sh
chmod +x ./libuvc_installation.sh
./libuvc_installation.sh
```

Thanks.

MartyG-RealSense, 2024-08-05:
The above instructions are correct for the most part.  However, if you install dependencies and then install librealsense2 with git clone then you do not have to use the libuvc backend installation instructions.

The libuvc backend is a different, much simpler method of installing librealsense with a build script.  You do not need to follow the prerequisites instructions with that method, as the libuvc method takes care of installing the prerequisites by using the internet connection instead.

However, you will need to edit line 46 of the **libuvc_installation.sh** script to add flags to the CMake build instruction.

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh#L46

Change this:

`cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release`

To this:

`cmake ../  -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true`

drinkBr, 2024-08-06:
> xioctl(VIDIOC_S_FMT) errors are typically related to a conflict between librealsense and the kernel. The libuvc backend SDK installation method that you used earlier works best with Pi, as it enables librealsense to bypass the kernel.

Since you previously mentioned that using the libuvc backend SDK installation method would be good, I would like to proceed with that. Could you provide the necessary commands for using the libuvc backend SDK installation method in the same format as my previous comment?

When using that method, is there any difference compared to installing dependencies and then using git clone to install librealsense2?

If I install dependencies first and then use git clone to install librealsense2, is it correct to assume that the steps I outlined in my previous comment for "-install dependencies-" and "-install librealsense2-" are sufficient?

I apologize for the many questions, and I appreciate your assistance.

MartyG-RealSense, 2024-08-06:
When using the libuvc method, make sure that you have an active internet connection and then **unplug the camera**, open a terminal and input the short list of instructions below.

```
wget https://github.com/IntelRealSense/librealsense/raw/master/scripts/libuvc_installation.sh
$ chmod +x ./libuvc_installation.sh
$ ./libuvc_installation.sh
```

When the message **Librealsense script completed** is displayed, plug the RealSense camera in and run **rs-enumerate-devices** to test the librealsense installation.

*************

When using the libuvc backend method instead of installing dependencies and using git clone, the key differences with libuvc are:

- You do not need to install the dependencies yourself as the libuvc process takes care of that.
- You do not need to apply a kernel patch script to the kernel as a libuvc build of librealsense bypasses the kernel.

************

Your instructions for installing dependencies and then git cloning are fine, though when the flag `-DFORCE_RSUSB_BACKEND=TRUE` is included in the CMake build instruction then the **./scripts/setup_ubev_rules.sh** command is not used.

drinkBr, 2024-08-08:
> ```
> wget https://github.com/IntelRealSense/librealsense/raw/master/scripts/libuvc_installation.sh
> chmod +x ./libuvc_installation.sh
> ./libuvc_installation.sh
> ```

After executing the above commands, I ran `rs-enumerate-devices`, and the information for the connected RealSense was displayed. When I started `realsense-viewer` several times, in most cases, the following message appeared, and I couldn't see the image information.

MartyG-RealSense, 2024-08-08:
Did you make the edit to line 46 of the **libuvc_installation.sh** script that I suggested at https://github.com/IntelRealSense/realsense-ros/issues/3158#issuecomment-2268817314 please?  

Whilst the SDK can use text-based programs like rs-enumerate-devices when the libuvc_installation.sh script is run in its default state, it is necessary for the **-DBUILD_GRAPHICAL_EXAMPLES=TRUE** flag to be included in order for graphics support to be included in the SDK build so that graphical tools such as realsense-viewer can be run.

drinkBr, 2024-08-08:
It has been edited.

MartyG-RealSense, 2024-08-08:
Are the **rs-hello-realsense** and **rs-capture** example programs able to run?  As rs-capture uses OpenGL graphics like realsense-viewer does, if rs-hello-realsense (a plain-text program) runs but rs-capture does not run then it could suggest that the cause of the problem is with the OpenGL graphics on your computer.

drinkBr, 2024-08-09:
I ran `rs-hello-realsense` several times with the Realsense camera connected. Sometimes it displayed "The camera is facing an object 'number' meters away," while other times, even after waiting for a while, no error or message was displayed. Here are the details of those instances.

I also tried running `rs-capture`. The results were similar to when I ran `realsense-viewer`—sometimes it executed correctly, while other times, the following message appeared. Additionally, when `rs-capture` failed to run properly, the following message was displayed in the Terminal. I believe the cause of this error might be a timeout.

MartyG-RealSense, 2024-08-09:
The problems that occur do seem to consistently be related to time-outs, yes.

As the camera works normally on a Windows computer, it appears that the problem is not with the camera hardware or the firmware driver.

It is useful to know that the text-only example **rs-hello-realsense** can function correctly most of the time.  It could suggest that graphics-based programs such as RealSense Viewer and rs-capture cause the Pi to be over-burdened with processing and slow up.

Raspberry Pi 4 Model B has 2 USB 3.0 ports and 2 USB 2.0 ports.  `Does the xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out` error occur no matter what USB port the camera is plugged into? (RealSense cameras can be used on USB 2).

drinkBr, 2024-08-10:
> Does the `xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out` error occur no matter what USB port the camera is plugged into? (RealSense cameras can be used on USB 2).

I changed the port to which the RealSense camera was connected and ran `realsense-viewer`. The error did not occur when connected to either USB 3.0 or USB 2.0. However, there was a difference in performance between the two. When connected to USB 3.0, there were no issues as long as only the Depth or RGB image was displayed, but when displaying both, it sometimes froze. When connected to USB 2.0, attempting to display the Depth image usually caused it to freeze, and RealSense Viewer would often crash. When it crashed, `Segmentation fault (core dumped)` was displayed in the Terminal.

I also tried running rs-capture and rs-hello-realsense several times after changing the USB port to which the RealSense camera was connected, but the results were the same as before.

MartyG-RealSense, 2024-08-10:
When using RealSense cameras with Raspberry Pi boards, streaming depth and color, with no other processing burdens such as pointcloud generation or post-processing filters, is the most that can be achieved before problems with streaming occur.  

There is a significant difference in maximum data transfer speed between USB2 and USB3.  USB3 can transfer 5000 Mbps, whilst USB2 can only transfer up to 480 Megabits per second (Mbps).

MiTomMi, 2024-08-10:
Hello you guys. I am using Ubuntu 22.04 LTS, ROS2 Humble on my PC and Ubuntu 22.04 Server, ROS2 Humble on my Turtlebot3 Burger (Raspberry Pi 4). I followed the LibUVC-backend installation steps above on the Raspberry Pi 4. When running `ros2 launch realsense2_camera rs_launch.py`, the camera seemed to be running but I encountered the same problem as @drinkBr mentioned on this issue. 

However, when I run `realsense-viewer` on my PC, the camera worked really fine. 

I am new to ROS2 and Ubuntu so I would be appreciate a lot to receive advice from you @MartyG-RealSense. Thank you very much!

MartyG-RealSense, 2024-08-11:
Hi @MiTomMi  Does performance in the RealSense ROS wrapper improve if you launch with 640x480 and 15 FPS for both streams?

`ros2 launch realsense2_camera rs_launch.py depth_module.depth_profile:=640x480x15 rgb_camera.color_profile:=640x480x15`

MiTomMi, 2024-08-11:
Hi @MartyG-RealSense, thank you for your reply. I have tried the command: `ros2 launch realsense2_camera rs_launch.py depth_module.depth_profile:=640x480x15 rgb_camera.color_profile:=640x480x15` on my PC and Raspberry Pi 4. The camera ran perfectly on the PC as in the picture shown below.

[image]

However, the same error occurred on the Pi 4:

[realsense2_camera_node-1] [ERROR] [1723363115.466008305] [camera.camera]: ./src/rs_node_setup.cpp:487:An exception has been thrown: std::exception
[realsense2_camera_node-1] [ERROR] [1723363115.466257829] [camera.camera]: ./src/rs_node_setup.cpp:369:An exception has been thrown: std::exception
[realsense2_camera_node-1] [ERROR] [1723363115.466445375] [camera.camera]: Error starting device: std::exception

The error has been driving me crazy for several days. At first I thought the problem came from the Raspberry Pi 3B, but when I upgraded to Pi 4 the problem still happens.

I am looking forward to your advice. I appreciate your help a lot!
Thank you!

MiTomMi, 2024-08-11:
UPDATE:
Hi @MartyG-RealSense, somehow I unplugged  the camera then plugged it in again and it worked pretty fine on Pi 4.

pi@raspberrypi4:~$ ros2 launch realsense2_camera rs_launch.py depth_module.depth_profile:=640x480x15 rgb_camera.color_profile:=640x480x15
[INFO] [launch]: All log files can be found below /home/pi/.ros/log/2024-08-11-15-13-46-710047-raspberrypi3-4061
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [4063]
[realsense2_camera_node-1] [INFO] [1723364027.702030173] [camera.camera]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1723364027.702317850] [camera.camera]: Built with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1723364027.702386959] [camera.camera]: Running with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1723364028.504624460] [camera.camera]: Device with serial number 819312072177 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1723364028.504824103] [camera.camera]: Device with physical ID /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-1/2-1:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1723364028.504881694] [camera.camera]: Device with name Intel RealSense D435 was found.
[realsense2_camera_node-1] [INFO] [1723364028.505618451] [camera.camera]: Device with port number 2-1 was found.
[realsense2_camera_node-1] [INFO] [1723364028.505752799] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1723364028.505995941] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1723364028.507143946] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1723364028.507284868] [camera.camera]: Device Name: Intel RealSense D435
[realsense2_camera_node-1] [INFO] [1723364028.507344163] [camera.camera]: Device Serial No: 819312072177
[realsense2_camera_node-1] [INFO] [1723364028.507392309] [camera.camera]: Device physical port: /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-1/2-1:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1723364028.507442215] [camera.camera]: Device FW version: 5.16.0.1
[realsense2_camera_node-1] [INFO] [1723364028.507485547] [camera.camera]: Device Product ID: 0x0B07
[realsense2_camera_node-1] [INFO] [1723364028.507526954] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [WARN] [1723364028.632808948] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [INFO] [1723364028.634437884] [camera.camera]: Set ROS param depth_module.infra_profile to default: 848x480x30
[realsense2_camera_node-1] [WARN] [1723364028.645457318] [camera.camera]: Could not set param: rgb_camera.power_line_frequency with 3 Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {rgb_camera.power_line_frequency} doesn't comply with integer range.
[realsense2_camera_node-1] [WARN] [1723364028.663526649] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [INFO] [1723364028.671633053] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1723364028.672280516] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1723364028.686621284] [camera.camera]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1723364028.693726549] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 640, Height: 480, FPS: 15
[realsense2_camera_node-1] [INFO] [1723364028.718951273] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1723364028.727177525] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 640, Height: 480, FPS: 15
[realsense2_camera_node-1] [INFO] [1723364028.734255124] [camera.camera]: RealSense Node Is Up!

 
[image]

However, the same error showed up when I started over the command again.

[realsense2_camera_node-1] [ERROR] [1723363115.466008305] [camera.camera]: ./src/rs_node_setup.cpp:487:An exception has been thrown: std::exception
[realsense2_camera_node-1] [ERROR] [1723363115.466257829] [camera.camera]: ./src/rs_node_setup.cpp:369:An exception has been thrown: std::exception
[realsense2_camera_node-1] [ERROR] [1723363115.466445375] [camera.camera]: Error starting device: std::exception

Maybe there are some Intel Realsense's firmware conflicts. Here is the firmware of the camera.

pi@raspberrypi3:~/robot_intel$ rs-fw-update -l

Connected devices:
1) [USB] Intel RealSense D435 s/n 819312072177, update serial number: 818413021385, firmware version: 5.16.0.1

Thank you!

drinkBr, 2024-08-11:
Hi, @MartyG-RealSense

> When using RealSense cameras with Raspberry Pi boards, streaming depth and color, with no other processing burdens such as pointcloud generation or post-processing filters, is the most that can be achieved before problems with streaming occur.

Is there a specific method to achieve this?

MartyG-RealSense, 2024-08-11:
Hi @drinkBr  Do you mean how to avoid having pointclouds, alignment and post-processing active so that processing burden can be reduced?  If so then all of these are disabled by default so you do not have to do anything to turn them off.

MartyG-RealSense, 2024-08-11:
@MiTomMi  If you are using firmware 5.16.0.1 then librealsense 2.55.1 and ROS2 wrapper 4.55.1 should be used with it.  Are those the versions that you are using, please?

MiTomMi, 2024-08-11:
Hi @MartyG-RealSense, I am using ROS2 Wrapper 4.55.1.

pi@raspberrypi4:~$ ros2 pkg xml realsense2_camera | grep '<version>'
  <version>4.55.1</version>

MartyG-RealSense, 2024-08-11:
@MiTomMi  Does the problem after camera restart still occur if you add **initial_reset:=true** to your launch instruction to reset the camera at launch?

drinkBr, 2024-08-11:
Hi, @MartyG-RealSense 
OK. Is there anything else I can test or verify?

MiTomMi, 2024-08-11:
@MartyG-RealSense, the problem still occured. Here is the full log:

pi@raspberrypi4:~$ ros2 launch realsense2_camera rs_launch.py depth_module.depth_profile:=640x480x15 rgb_camera.color_profile:=640x480x15 initial_reset:=true
[INFO] [launch]: All log files can be found below /home/pi/.ros/log/2024-08-11-16-23-46-025104-raspberrypi3-6688
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [6689]
[realsense2_camera_node-1] [INFO] [1723368226.678293178] [camera.camera]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1723368226.678570027] [camera.camera]: Built with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1723368226.678631508] [camera.camera]: Running with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1723368227.483921207] [camera.camera]: Device with serial number 819312072177 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1723368227.484110502] [camera.camera]: Device with physical ID /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-1/2-1:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1723368227.484166149] [camera.camera]: Device with name Intel RealSense D435 was found.
[realsense2_camera_node-1] [INFO] [1723368227.484962382] [camera.camera]: Device with port number 2-1 was found.
[realsense2_camera_node-1] [INFO] [1723368227.485091363] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1723368227.485142307] [camera.camera]: Resetting device...
[realsense2_camera_node-1] [INFO] [1723368234.296970757] [camera.camera]: Device with serial number 819312072177 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1723368234.297138367] [camera.camera]: Device with physical ID /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-1/2-1:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1723368234.297191829] [camera.camera]: Device with name Intel RealSense D435 was found.
[realsense2_camera_node-1] [INFO] [1723368234.297840841] [camera.camera]: Device with port number 2-1 was found.
[realsense2_camera_node-1] [INFO] [1723368234.297940637] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1723368234.298154098] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1723368234.299071811] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1723368234.299196050] [camera.camera]: Device Name: Intel RealSense D435
[realsense2_camera_node-1] [INFO] [1723368234.299249679] [camera.camera]: Device Serial No: 819312072177
[realsense2_camera_node-1] [INFO] [1723368234.299296883] [camera.camera]: Device physical port: /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-1/2-1:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1723368234.299341049] [camera.camera]: Device FW version: 5.16.0.1
[realsense2_camera_node-1] [INFO] [1723368234.299380937] [camera.camera]: Device Product ID: 0x0B07
[realsense2_camera_node-1] [INFO] [1723368234.299423363] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [WARN] [1723368234.425169549] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [INFO] [1723368234.426805014] [camera.camera]: Set ROS param depth_module.infra_profile to default: 848x480x30
[realsense2_camera_node-1] [WARN] [1723368234.439084875] [camera.camera]: Could not set param: rgb_camera.power_line_frequency with 3 Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {rgb_camera.power_line_frequency} doesn't comply with integer range.
[realsense2_camera_node-1] [WARN] [1723368234.457600470] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [INFO] [1723368234.465949147] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1723368234.466597159] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1723368234.482211024] [camera.camera]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1723368234.489250750] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 640, Height: 480, FPS: 15
[realsense2_camera_node-1] [WARN] [1723368234.679122767] [camera.camera]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.
[realsense2_camera_node-1]  11/08 16:23:59,553 ERROR [281472308734176] (librealsense-exception.h:52) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
[realsense2_camera_node-1]  11/08 16:23:59,553 ERROR [281472308734176] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
[realsense2_camera_node-1]  11/08 16:24:04,681 ERROR [281472342550752] (librealsense-exception.h:52) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such file or directory
[realsense2_camera_node-1]  11/08 16:24:04,681 ERROR [281472342550752] (error-handling.cpp:99) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such file or directory
[realsense2_camera_node-1]  11/08 16:24:09,801 ERROR [281472359459040] (librealsense-exception.h:52) set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
[... cut ...]
[ERROR] [realsense2_camera_node-1]: process[realsense2_camera_node-1] failed to terminate '10.0' seconds after receiving 'SIGTERM', escalating to 'SIGKILL'
[INFO] [realsense2_camera_node-1]: sending signal 'SIGKILL' to process[realsense2_camera_node-1]
[ERROR] [realsense2_camera_node-1]: process has died [pid 6689, exit code -9, cmd '/opt/ros/humble/lib/realsense2_camera/realsense2_camera_node --ros-args --log-level info --ros-args -r __node:=camera -r __ns:=/camera --params-file /tmp/launch_params_0a0t6p6s --params-file /tmp/launch_params_kb4_u3td'].

I'm sorry drinkBr for causing inconveniences with long comments.

MartyG-RealSense, 2024-08-11:
@drinkBr  @MiTomMi  As both of your issues are related to timing, let's try disabling Global Time and enabling Sync of the depth and color streams by adding the parameters below to your launch.

`global_time_enabled:=false enable_sync:=true `

drinkBr, 2024-08-11:
@MartyG-RealSense, I don't have a Raspberry Pi at hand right now, so I can't verify this. I will test it tomorrow and send you a message.

MartyG-RealSense, 2024-08-11:
Thanks very much, @drinkBr

MiTomMi, 2024-08-12:
Hi @MartyG-RealSense, I tried 2 methods:

Method 1: `ros2 launch realsense2_camera rs_launch.py depth_module.depth_profile:=640x480x15 rgb_camera.color_profile:=640x480x15 global_time_enabled:=false enable_sync:=true`

Method 2: `ros2 launch realsense2_camera rs_launch.py inital_reset:=true global_time_enabled:=false enable_sync:=true`

Both methods result in these 4 major errors as in this issue:
 
[realsense2_camera_node-1]  12/08 14:55:27,553 ERROR [281472736749792] (librealsense-exception.h:52) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such file or directory
[realsense2_camera_node-1]  12/08 14:55:27,554 ERROR [281472736749792] (error-handling.cpp:99) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such file or directory
[realsense2_camera_node-1] [ERROR] [1723449342.915113340] [camera.camera]: ./src/rs_node_setup.cpp:487:An exception has been thrown: std::exception
[realsense2_camera_node-1] [ERROR] [1723449342.915517448] [camera.camera]: ./src/rs_node_setup.cpp:369:An exception has been thrown: std::exception
[realsense2_camera_node-1] [ERROR] [1723449342.916101758] [camera.camera]: Error starting device: std::exception

Could you explain what are those errors and what cause them? I have tried to debug them with many methods from github but I still get the same error.

I'm looking forward to your reply. Thanks a lot!

MartyG-RealSense, 2024-08-12:
@MiTomMi  It looks as though the error `get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such file or directory` nearly always occurs on Raspberry Pi boards.  None of the four Pi cases linked to below that experience the errors offer a solution, unfortunately.

https://github.com/IntelRealSense/realsense-ros/issues/2991
https://github.com/IntelRealSense/realsense-ros/issues/3089
https://forums.raspberrypi.com/viewtopic.php?t=370620
https://www.reddit.com/r/ROS/comments/1cp30d4/problem_getting_rgb_frames_via_intelrealsense/

There were a couple of non-Pi cases of the error where updating the camera firmware driver resolved it.

drinkBr, 2024-08-13:
Hi, @MartyG-RealSense 
Sorry for the late reply. I forgot that I haven't installed realsense-ros on the Ubuntu system on the SSD yet, so I haven't been able to verify it. Since I'm at it, I would like to confirm the installation steps for realsense-ros.

When I previously installed realsense-ros, I followed the instructions on the following page. Is that correct? Are there any necessary changes to consider if I installed realsense-related components using the LibUVC-backend installation method?

Installation on Ubuntu Step3 Option2 : https://github.com/IntelRealSense/realsense-ros

drinkBr, 2024-08-13:
Hi, @MartyG-RealSense
> You are using a Raspberry Pi 4 though, so there might be complications. Please give a colcon build a try and see what happens.

Is it correct that I should verify the colcon build before installing realsense-ros?

drinkBr, 2024-08-13:
I misunderstood. So, I just need to execute Step 3 Option 2 in the Terminal, right?

MartyG-RealSense, 2024-08-13:
Yes, that's correct.

drinkBr, 2024-08-13:
OK! I will test it tomorrow and send you a message.

haidaumitom, 2024-08-15:
Hi @MartyG-RealSense, I have tried to re-install Ubuntu desktop 22.04 LTS and ROS2 Humble on Raspberry Pi 4. I followed libuvc_installation.md. After finished installing it, I ran the command: `realsense-viewer` and it was only successful whenever I re-plug in the cable. The results are shown below:

[image]
[image]

I don't know why the frames are fractured like this. Could you figure it out?

Thanks a lot!!!

haidaumitom, 2024-08-15:
UPDATE:
Hi @MartyG-RealSense. I ran `rs-capture` and the result are really clear with no problem.

[image]

Do you know what's the problem in realsense-viewer in my case?

Thank you!!!

MartyG-RealSense, 2024-08-15:
The rs-capture image is a 2D depth image like the one in the RealSense Viewer when you select the '2D' option in the top corner of the Viewer window.  The broken images that you provided above are from the '3D' pointcloud mode of the Viewer.

Do you get a better depth image in the Viewer if you click on '2D' to switch to 2D depth mode?

I note that in your Viewer's 3D mode with the broken images, the 'Measure' option at the top of the window is greyed out and unselectable.  This suggests that the **GLSL** graphics options in your Viewer are disabled.  You can use instructions at https://github.com/IntelRealSense/librealsense/issues/8110#issuecomment-754705023 to check the status of the GLSL options and see whether they are currently enabled or disabled.  

If the boxes beside the Processing and Rendering settings are black (off) then try clicking on them to turn the boxes blue (on) and click the Apply button, then test the depth pointcloud in 3D mode again.

MartyG-RealSense, 2024-08-23:
Hi @haidaumitom  Do you require further assistance with this case, please?  Thanks!

[5 more replies]

closed 2024-08-30 by MartyG-RealSense