realsenseai/realsense-ros #3162: Setting Dynamic reconfig parameters does not return.

axkoenig, 2024-07-11:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D455                          | 
| Firmware Version                   | 05.12.06.00                        | 
| Operating System & Version | Ubuntu 20.04 | 
| Kernel Version (Linux Only)    | 5.15.0-113-generic                                                                    | 
| Platform                                 | PC |
| Librealsense SDK Version       | LibRealSense v2.50.0                                                   | 
| Language                               |  roslaunch         | 
| Segment			         |  Robot                          | 
| ROS Distro			         |  Noetic                                           | 
| RealSense ROS Wrapper Version   | RealSense ROS v2.3.2                                                 | 

### Issue Description
Hi,

I am connecting my Realsense D455 to an USB 3.2 port. I installed the ROS wrapper via `sudo apt-get install ros-noetic-realsense2-camera`. Unfortunately `roslaunch realsense2_camera rs_camera.launch` hangs at the point where it's "Setting Dynamic reconfig parameters". I waited for at least half an hour and nothing is happening. From other posts I see the next print on my screen should be `Done Setting Dynamic reconfig parameters.`, so I assume it has something to do with those. 

Also the `(Confidence, 0) sensor isn't supported by current device! -- Skipping...` seems intriguing, could this be the issue? 

I also tried the D435 and have the same issue. 

Could you please help me out? 

```bash
roslaunch realsense2_camera rs_camera.launch
... logging to /home/koenig/.ros/log/00f82b28-3f7e-11ef-a2c4-33bec58c79ba/roslaunch-turian-6719.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://130.149.238.230:36199/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
[... 74 lines cut ...]
[ INFO] [1720699603.103697371]: Device with physical ID 2-3-2 was found.
[ INFO] [1720699603.103711063]: Device with name Intel RealSense D455 was found.
[ INFO] [1720699603.104154720]: Device with port number 2-3 was found.
[ INFO] [1720699603.104180605]: Device USB type: 3.2
[ INFO] [1720699603.107581888]: getParameters...
[ INFO] [1720699603.232748026]: setupDevice...
[ INFO] [1720699603.232771051]: JSON file is not provided
[ INFO] [1720699603.232780777]: ROS Node Namespace: camera
[ INFO] [1720699603.232793171]: Device Name: Intel RealSense D455
[ INFO] [1720699603.232803551]: Device Serial No: 105322252242
[ INFO] [1720699603.232812411]: Device physical port: 2-3-2
[ INFO] [1720699603.232821714]: Device FW version: 05.12.06.00
[ INFO] [1720699603.232831500]: Device Product ID: 0x0B5C
[ INFO] [1720699603.232842122]: Enable PointCloud: Off
[ INFO] [1720699603.232853126]: Align Depth: Off
[ INFO] [1720699603.232863385]: Sync Mode: Off
[ INFO] [1720699603.232910222]: Device Sensors: 
[ INFO] [1720699603.320586343]: Stereo Module was found.
[ INFO] [1720699603.351170797]: RGB Camera was found.
[ INFO] [1720699603.351287092]: Motion Module was found.
[ INFO] [1720699603.351358948]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1720699603.351382960]: num_filters: 0
[ INFO] [1720699603.351397518]: Setting Dynamic reconfig parameters.
# no more output after this ... 
```

MartyG-RealSense, 2024-07-11:
Hi @axkoenig  Please try installing dynamic_reconfigure for Noetic manually using the command below to see whether it makes a difference to your launch.

`sudo apt install ros-noetic-ddynamic-reconfigure`

axkoenig, 2024-07-11:
Hey, thanks for the quick response! I entered your command, and got `0 upgraded, 0 newly installed, 0 to remove and 57 not upgraded.`, so everything is up-to-date. I re-ran, same problem unfortunately.

MartyG-RealSense, 2024-07-11:
You are very welcome.  :)

I note that your camera firmware driver version is 5.12.6.0.  This firmware is designed for the very old librealsense SDK version 2.36.0.    For librealsense version 2.50,0, firmware version **5.13.0.50** should be installed in your camera.  Please try updating your firmware version.

axkoenig, 2024-07-11:
Ok thanks for the pointer. Do I need to install librealsense from source to get `rs-fw-update -l`? Currently I have these 
```
dpkg -l | grep "realsense"
ii  ros-noetic-librealsense2                      2.50.0-1focal.20211115.134455         amd64        Library for capturing data from the Intel(R) RealSense(TM) SR300, D400, L500 Depth cameras and T2xx Tracking devices.
ii  ros-noetic-realsense2-camera                  2.3.2-1focal.20240111.183800          amd64        RealSense Camera package allowing access to Intel T265 Tracking module and SR300 and D400 3D cameras
```
And I get 
```
rs-fw-update -l 
rs-fw-update: command not found
```

Sorry, I'm a realsense newbie ;)

MartyG-RealSense, 2024-07-11:
It's no trouble at all.  Everyone was new once.  :)

The command `sudo apt-get install ros-noetic-realsense2-camera` installs both librealsense and the ROS1 wrapper at the same time.  Unfortunately, this installation method does not include the RealSense Viewer tool, which would make it easy to update the firmware.  

I do not know what other tools might be omitted by this install method.  You could check in the Ubuntu folder **usr/local/bin** whether rs-fw-update has been installed there.

axkoenig, 2024-07-11:
I've done the firmware upgrade after installing librealsense from source, but the issue still persists, unfortunately. I do get pretty pictures from the camera in RealSenseViewer though. 

```
[ INFO] [1720707557.317667676]: Built with LibRealSense v2.50.0
[ INFO] [1720707557.317680916]: Running with LibRealSense v2.50.0
[ INFO] [1720707557.361942992]:  
[ INFO] [1720707557.509222345]: Device with serial number 105322252242 was found.

[ INFO] [1720707557.509248355]: Device with physical ID 2-4-5 was found.
[ INFO] [1720707557.509258295]: Device with name Intel RealSense D455 was found.
[ INFO] [1720707557.509569616]: Device with port number 2-4 was found.
[ INFO] [1720707557.509584866]: Device USB type: 3.2
[ INFO] [1720707557.514891984]: getParameters...
[ INFO] [1720707557.615157223]: setupDevice...
[ INFO] [1720707557.615183453]: JSON file is not provided
[ INFO] [1720707557.615196663]: ROS Node Namespace: camera
[ INFO] [1720707557.615211853]: Device Name: Intel RealSense D455
[ INFO] [1720707557.615225303]: Device Serial No: 105322252242
[ INFO] [1720707557.615238923]: Device physical port: 2-4-5
[ INFO] [1720707557.615250173]: Device FW version: 05.13.00.50
[ INFO] [1720707557.615260643]: Device Product ID: 0x0B5C
[ INFO] [1720707557.615276853]: Enable PointCloud: Off
[ INFO] [1720707557.615292163]: Align Depth: Off
[ INFO] [1720707557.615303123]: Sync Mode: Off
[ INFO] [1720707557.615345394]: Device Sensors: 
[ INFO] [1720707557.717908661]: Stereo Module was found.
[ INFO] [1720707557.747527218]: RGB Camera was found.
[ INFO] [1720707557.747645348]: Motion Module was found.
[ INFO] [1720707557.747713948]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1720707557.747734088]: num_filters: 0
[ INFO] [1720707557.747746758]: Setting Dynamic reconfig parameters. # stuck

```

axkoenig, 2024-07-11:
Ok now I installed the newest version of the Firmware via the RealSenseViewer, and now I am getting slightly more - does this output help?

```
[ INFO] [1720708382.061823842]: Device FW version: 05.16.00.01
[ INFO] [1720708382.061834881]: Device Product ID: 0x0B5C
[ INFO] [1720708382.061846101]: Enable PointCloud: Off
[ INFO] [1720708382.061858011]: Align Depth: Off
[ INFO] [1720708382.061867920]: Sync Mode: Off
[ INFO] [1720708382.061907439]: Device Sensors: 
[ INFO] [1720708382.178653541]: Stereo Module was found.
[ INFO] [1720708382.208167364]: RGB Camera was found.
[ INFO] [1720708382.208285489]: Motion Module was found.
[ INFO] [1720708382.208354517]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1720708382.208377336]: num_filters: 0
[ INFO] [1720708382.208391165]: Setting Dynamic reconfig parameters.
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
```

MartyG-RealSense, 2024-07-11:
5.16.0.1 should not be used with SDK 2.50.0 and wrapper 2.3.2, though there are sometimes cases where it does work even though it shouldn't.

Does it make a difference if you add **initial_reset:=true** to your roslaunch instruction to reset the camera at launch.

axkoenig, 2024-07-11:
Ok I am back on the version you recommended. No, unfortunately it makes no difference. 

```
[ INFO] [1720713299.845479547]: Initializing nodelet with 12 worker threads.
[ INFO] [1720713300.071174897]: RealSense ROS v2.3.2
[ INFO] [1720713300.071201356]: Built with LibRealSense v2.50.0
[ INFO] [1720713300.071212736]: Running with LibRealSense v2.50.0
[ INFO] [1720713300.117358562]:  
[ INFO] [1720713300.363596058]: Device with serial number 105322252242 was found.

[ INFO] [1720713300.363626488]: Device with physical ID 2-3-4 was found.
[ INFO] [1720713300.363638497]: Device with name Intel RealSense D455 was found.
[ INFO] [1720713300.363973269]: Device with port number 2-3 was found.
[ INFO] [1720713300.363993299]: Device USB type: 3.2
[ INFO] [1720713300.364011628]: Resetting device...
[ INFO] [1720713306.469897214]:  
[ INFO] [1720713306.625407806]: Device with serial number 105322252242 was found.

[ INFO] [1720713306.625440265]: Device with physical ID 2-3-5 was found.
[ INFO] [1720713306.625450185]: Device with name Intel RealSense D455 was found.
[ INFO] [1720713306.625704860]: Device with port number 2-3 was found.
[ INFO] [1720713306.625717120]: Device USB type: 3.2
[ INFO] [1720713306.629064143]: getParameters...
[ INFO] [1720713306.726669657]: setupDevice...
[ INFO] [1720713306.726689966]: JSON file is not provided
[ INFO] [1720713306.726701986]: ROS Node Namespace: camera
[ INFO] [1720713306.726717776]: Device Name: Intel RealSense D455
[ INFO] [1720713306.726732865]: Device Serial No: 105322252242
[ INFO] [1720713306.726746525]: Device physical port: 2-3-5
[ INFO] [1720713306.726759455]: Device FW version: 05.13.00.50
[ INFO] [1720713306.726775774]: Device Product ID: 0x0B5C
[ INFO] [1720713306.726790154]: Enable PointCloud: Off
[ INFO] [1720713306.726801984]: Align Depth: Off
[ INFO] [1720713306.726815124]: Sync Mode: Off
[ INFO] [1720713306.726854033]: Device Sensors: 
[ INFO] [1720713306.821201743]: Stereo Module was found.
[ INFO] [1720713306.849730086]: RGB Camera was found.
[ INFO] [1720713306.849847523]: Motion Module was found.
[ INFO] [1720713306.849920022]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1720713306.849940912]: num_filters: 0
[ INFO] [1720713306.849954201]: Setting Dynamic reconfig parameters.

:-(((
```

Could the `sensor isn't supported by current device!` be a reason?

MartyG-RealSense, 2024-07-12:
The message `(Confidence, 0) sensor isn't supported by current device!` is not important as it refers to a feature that is only available on the RealSense T265 camera model, which is why it is unsupported on your launch.  It does not prevent your D455 launch from working.

If the same problem with getting stuck on dynamic reconfigure also occurs with your D435 then it is likely to be a software issue rather than a camera hardware one.

Let's try a different launch method - an **RGBD** launch instead of rs_camera.  First, you will need to install support for an RGBD launch on Noetic using the command below.

`sudo apt-get install ros-noetic-rgbd-launch`

Then launch using the **rs_rgbd.launch** file.

`roslaunch realsense2_camera rs_rgbd.launch`

axkoenig, 2024-07-12:
Hey so I tried this on three different Cameras D455, D435 and D435i and I have the same problem with all of them. My colleagues have the same setup (same kernel, same ros versions, same librealsense version) and it works for them without a problem, so it isn't a problem of the camera. 

Another intriguing thing is that once I kill the roslaunch that seems stuck, the camera won't ever re-connect - it only shows `No RealSense devices were found!` even though the device is connected. I need to reboot my machine for it to pick up the Realsense again and even recognize it. 

One more intriguing thing is that if I just unplug the USB cable while it's stuck on `Setting Dynamic reconfig parameters` it does seem to continue executing the script! But probably this is only because of how print statements are handled low-level. Just wanted to bring that up here, maybe it is a pointer to you.

```
[ INFO] [1720771513.350792257]: Device Name: Intel RealSense D435I
[ INFO] [1720771513.350804467]: Device Serial No: 939622074251
[ INFO] [1720771513.350813467]: Device physical port: 2-3-2
[ INFO] [1720771513.350821627]: Device FW version: 05.13.00.50
[ INFO] [1720771513.350832467]: Device Product ID: 0x0B3A
[ INFO] [1720771513.350843217]: Enable PointCloud: Off
[ INFO] [1720771513.350859807]: Align Depth: Off
[ INFO] [1720771513.350869876]: Sync Mode: Off
[ INFO] [1720771513.350916506]: Device Sensors: 
[ INFO] [1720771513.476485603]: Stereo Module was found.
[ INFO] [1720771513.505556051]: RGB Camera was found.
[ INFO] [1720771513.505662050]: Motion Module was found.
[ INFO] [1720771513.505733389]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1720771513.505758299]: num_filters: 0
[ INFO] [1720771513.505775569]: Setting Dynamic reconfig parameters. 

# (stuck for at least 5 mins, then I unplug USB)

 12/07 10:05:33,141 WARNING [139726828861184] (messenger-libusb.cpp:42) control_transfer returned error, index: 703, error: No such device, number: 13
get_data_usb failed, error: RS2_USB_STATUS_NO_DEVICE
 12/07 10:05:33,155 WARNING [139726828861184] (messenger-libusb.cpp:42) control_transfer returned error, index: 703, error: No such device, number: 13
get_data_usb failed, error: RS2_USB_STATUS_NO_DEVICE
 12/07 10:05:33,161 WARNING [139726828861184] (messenger-libusb.cpp:42) control_transfer returned error, index: 703, error: No such device, number: 13
get_data_usb failed, error: RS2_USB_STATUS_NO_DEVICE
 12/07 10:05:33,168 WARNING [139726828861184] (messenger-libusb.cpp:42) control_transfer returned error, index: 703, error: No such device, number: 13
get_data_usb failed, error: RS2_USB_STATUS_NO_DEVICE
 12/07 10:05:33,175 WARNING [139726828861184] (messenger-libusb.cpp:42) control_transfer returned error, index: 603, error: No such device, number: 13
get_data_usb failed, error: RS2_USB_STATUS_NO_DEVICE
 12/07 10:05:33,182 WARNING [139726828861184] (messenger-libusb.cpp:42) control_transfer returned error, index: 703, error: No such device, number: 13
get_data_usb failed, error: RS2_USB_STATUS_NO_DEVICE
 12/07 10:05:33,191 WARNING [139726828861184] (messenger-libusb.cpp:42) control_transfer returned error, index: 703, error: No such device, number: 13
get_data_usb failed, error: RS2_USB_STATUS_NO_DEVICE
 12/07 10:05:33,198 WARNING [139726828861184] (messenger-libusb.cpp:42) control_transfer returned error, index: 603, error: No such device, number: 13
get_data_usb failed, error: RS2_USB_STATUS_NO_DEVICE
[ INFO] [1720771533.220115739]: Done Setting Dynamic reconfig parameters.
[ INFO] [1720771533.221594340]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1720771533.222807922]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1720771533.222827842]: setupPublishers...
[ INFO] [1720771533.227872629]: Expected frequency for depth = 30.00000
[ INFO] [1720771533.298806784]: Expected frequency for color = 30.00000
[ INFO] [1720771533.336116275]: setupStreams...
 12/07 10:05:33,422 WARNING [139727013402368] (messenger-libusb.cpp:92) usb_request_queue returned error, endpoint: 87 error: No such device, number: 13
 12/07 10:05:33,422 WARNING [139727013402368] (uvc-device.cpp:183) rs_uvc_device exception in listen_to_interrupts method: failed to submit interrupt request, error: RS2_USB_STATUS_OTHER
[ERROR] [1720771533.430944336]: An exception has been thrown: usb device disconnected
[ERROR] [1720771533.430999126]: Exception: usb device disconnected
 12/07 10:05:33,422 WARNING [139727013402368] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: No such device, number: 13
 12/07 10:05:34,220 ERROR [139727013402368] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_NO_DEVICE
 12/07 10:05:34,220 ERROR [139726979831552] (sensor.cpp:572) acquire_power failed: failed to set power state
 12/07 10:05:34,220 ERROR [139726979831552] (error-handling.cpp:78) Error during polling error handler: failed to set power state
[ERROR] [1720771535.045122417]: The device has been disconnected!
[ERROR] [1720771535.051227209]: Exception: stop_streaming() failed. UVC device is not streaming!
[ERROR] [1720771535.051264688]: Exception: stop_streaming() failed. UVC device is not streaming!
[ERROR] [1720771535.101752048]: Exception: failed to set power state
 12/07 10:05:35,101 ERROR [139727013402368] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_NO_DEVICE
[... cut ...]
 12/07 10:05:35,220 ERROR [139727013402368] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_NO_DEVICE
 12/07 10:05:35,220 ERROR [139726979831552] (sensor.cpp:572) acquire_power failed: failed to set power state
 12/07 10:05:35,220 ERROR [139726979831552] (error-handling.cpp:78) Error during polling error handler: failed to set power state
[ WARN] [1720771536.139171536]: No RealSense devices were found!
[ WARN] [1720771542.156073505]: No RealSense devices were found!
[ WARN] [1720771548.172118563]: No RealSense devices were found!
```

MartyG-RealSense, 2024-07-12:
The log warnings **Exception: usb device disconnected** and **failed to open usb interface** indicate that the launch is aware that the camera has been disconnected and therefore cannot be accessed.

If your colleagues used the exact same installation procedure that you did and with the same configuration, are they also using the exact same model of computer that you are?

axkoenig, 2024-07-12:
For RGBD I have exactly the same problem, unfortunately... 

```
roslaunch realsense2_camera rs_rgbd.launch

... logging to /home/koenig/.ros/log/298f4e66-403e-11ef-a2c4-33bec58c79ba/roslaunch-turian-8596.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://130.149.238.230:38709/

SUMMARY
========

PARAMETERS
 * /camera/color_rectify_color/queue_size: 5
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: 30
 * /camera/realsense2_camera/confidence_height: 480
 * /camera/realsense2_camera/confidence_width: 640
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
[... 57 lines cut ...]
[ INFO] [1720782152.910527523]: Device with physical ID 2-3-2 was found.
[ INFO] [1720782152.910537634]: Device with name Intel RealSense D455 was found.
[ INFO] [1720782152.910877699]: Device with port number 2-3 was found.
[ INFO] [1720782152.910895410]: Device USB type: 3.2
[ INFO] [1720782152.948027695]: getParameters...
[ INFO] [1720782153.062032089]: setupDevice...
[ INFO] [1720782153.062059230]: JSON file is not provided
[ INFO] [1720782153.062072400]: ROS Node Namespace: camera
[ INFO] [1720782153.062088820]: Device Name: Intel RealSense D455
[ INFO] [1720782153.062101420]: Device Serial No: 105322252242
[ INFO] [1720782153.062120271]: Device physical port: 2-3-2
[ INFO] [1720782153.062145461]: Device FW version: 05.13.00.50
[ INFO] [1720782153.062158331]: Device Product ID: 0x0B5C
[ INFO] [1720782153.062175272]: Enable PointCloud: Off
[ INFO] [1720782153.062193222]: Align Depth: On
[ INFO] [1720782153.062205392]: Sync Mode: On
[ INFO] [1720782153.062292584]: Device Sensors: 
[ INFO] [1720782153.168570516]: Stereo Module was found.
[ INFO] [1720782153.199489446]: RGB Camera was found.
[ INFO] [1720782153.199612729]: Motion Module was found.
[ INFO] [1720782153.199679930]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1720782153.200014115]: num_filters: 1
[ INFO] [1720782153.200039886]: Setting Dynamic reconfig parameters.

```

axkoenig, 2024-07-12:
My colleagues have different models and different hardware specifications

MartyG-RealSense, 2024-07-12:
What could be interpreted from this is that if you all followed the same installation procedure and have the same software and the same versions but it works on their computers but not yours, there could be a conflict with your model of computer specifically.

When you built librealsense from source, did you build it with CMake?  If you did, then did you include the flag **-DFORCE_RSUSB_BACKEND=TRUE** in the CMake build instruction?

An RSUSB = true source code build of librealsense - described at https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692 - is not dependent on Linux versions or kernel versions and does not need to have a kernel patch applied to it, so it is very good in situations where the computer hardware is exotic (not typical of a shop-bought PC) or when librealsense may be conflicting with the kernel.

MartyG-RealSense, 2024-07-18:
Hi @axkoenig  Do you require further assistance with this case, please?  Thanks!

axkoenig, 2024-07-18:
Hi, 
thanks for following up! The problem still persists, and in the meantime I switched to an old ASUS Xtion.. This is far from ideal, I would love to get the Realsense working, but I'm puzzled ... 

For the librealsense installation I initially followed [these docs](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md). I just added `cmake ../ -DFORCE_RSUSB_BACKEND=TRUE` before installing and it did not make a difference. In fact, I have another strange behaviour in the Realsenseviewer (which I also experienced before) where I start it from the command line 
```bash
realsense-viewer 
 18/07 19:16:14,431 INFO [140389761197952] (context.cpp:116) ... 5-2-3
 18/07 19:16:14,431 INFO [140389761197952] (context.cpp:128) Found 1 RealSense devices (0xff requested & 0xff from device-mask in settings)
 ```
 and it finds the realsense as shown in the command line. But then in the Realsenseviewer the device is not displayed. When I reboot the machine it works (a single time) but when I disconnect the camera and re-connect I have the same problem... Maybe this rings a bell?
[image]

MartyG-RealSense, 2024-07-19:
Thanks very much for the additional information.

How does the camera behave in the Viewer if you launch it in sudo admin permissions mode with **sudo realsense-viewer**

axkoenig, 2024-07-26:
Hey there, 
sorry for my late reply. I am using the Xtion now, but still highly motivated to solve this issue to get all the benefits of the Realsense ...
1. I reboot my machine to start off fresh
2. when I execute with sudo the application is not responding 
[image]
3. I run realsense viewer without sudo and the device is not found in the GUI but the terminal 
[image]
4. Now weirdly, when I run with sudo again the application does not freeze, but I see the same screen as in 3
[image]

MartyG-RealSense, 2024-07-26:
What happens if you start realsense-viewer without the RealSense camera plugged in and then insert the camera into the USB port after the program has finished launching?  The camera should appear in the side-panel after being plugged in.

Also, have you unplugged the Xtion camera before launching realsense-viewer?

axkoenig, 2024-08-07:
Hi @MartyG-RealSense, sorry for the late reply. If I open realsense-viewer without the realsense plugged in and connect the camera after the program started up, it is not recognized and the screen stays exactly the same as above. I made sure no other cameras are plugged into the computer...

axkoenig, 2024-08-07:
But I've managed to get the camera working now on another machine, which is also connected to my ROS network, so for now this workaround is ok. I also installed via `sudo apt-get install ros-noetic-realsense2-camera` on that machine. 

But I would prefer to get this working on my main PC, of course. But if you also don't have any more ideas how we could resolve this, we can also close the issue and revisit it some other time. 

One thing I should note is that I am getting mysterious warning messages on the other machine, where it's working. This seems to be related to #1663 and #2628 but for now I will ignore these messages. Maybe they tell you something about what could be wrong on the other machine though? 

```
XXX@YYY:~$ roslaunch realsense2_camera rs_camera.launch initial_reset:=true
... logging to /home/XXX/.ros/log/567738c6-5015-11ef-a2c4-33bec58c79ba/roslaunch-panda-ctrl9-3010702.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://130.149.238.245:40747/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
[... 86 lines cut ...]
[ INFO] [1723022756.881584040]: Setting Dynamic reconfig parameters.
 07/08 11:25:59,478 WARNING [139649007724288] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
 07/08 11:26:00,710 WARNING [139649007724288] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
[ INFO] [1723022761.232871473]: Done Setting Dynamic reconfig parameters.
[ INFO] [1723022761.233544159]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1723022761.234860702]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1723022761.234913655]: setupPublishers...
[ INFO] [1723022761.236737312]: Expected frequency for depth = 30.00000
[ INFO] [1723022761.261302300]: Expected frequency for color = 30.00000
[ INFO] [1723022761.276524848]: setupStreams...
 07/08 11:26:01,608 WARNING [139649007724288] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 07/08 11:26:01,658 WARNING [139649007724288] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[... cut ...]
 07/08 11:26:02,018 WARNING [139649007724288] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 07/08 11:26:02,068 WARNING [139649007724288] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 07/08 11:26:02,118 WARNING [139649007724288] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

```

MartyG-RealSense, 2024-08-07:
The **control_transfer returned** warning, when it is generating continuously in the log, indicates that there is a communication problem between the camera and computer.  This will typically be a USB related issue, such as a glitch on the computer's USB port or a problem with the USB cable.  It is sometimes possible though to receive these warnings and still have the streams publishing normally in RViz.

The warnings may reduce in frequency if the FPS is reduced from 30 to 15.  To do this, you would need to specify a custom stream configuration in the roslaunch instruction.

`roslaunch realsense2_camera rs_camera.launch depth_width:=848 depth_height:=480 depth_fps:=15 color_width:=1280 color_height:=720 color_fps:=15`

Have you tried installing with `sudo apt-get install ros-noetic-realsense2-camera` on the main PC with the problem too?  When using this installation method, it installs both librealsense and the 2.3.2 ROS wrapper at the same time.  It does not install the RealSense Viewer though, so if you need the Viewer and the ROS wrapper on the main PC then you would have to install librealsense first (either by package or source code installation) and then build the 2.3.2 wrapper from source code afterwards.

When using the `sudo apt-get install ros-noetic-realsense2-camera` installation method it is important that librealsense has not previously been installed, otherwise you could end up with two installations of librealsense on the same computer that could conflict with each other.

axkoenig, 2024-08-13:
Yes I originally installed the package with `sudo apt-get install ros-noetic-realsense2-camera` on the first machine too and it didn't work. But now that I have it running on the other PC, let's close this issue for now. Thanks for your help!

closed 2024-08-13 by axkoenig

MartyG-RealSense, 2024-08-13:
You are very welcome.  I'm pleased to have been able to help.  Thanks very much for the update!