realsenseai/realsense-ros #3165: Version Mismatch Between Firmware and Librealsense Causing Invalid Value Error on D455 with ROS Noetic on Jetson Orin Nano

HTLife, 2024-07-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D455                          | 
| Firmware Version                   | 2.14.0                          | 
| Operating System & Version |   Ubuntu 20 | 
| Kernel Version (Linux Only)    |  5.10.192-tegra                                                                       | 
| Platform                                 | NVIDIA Jetson Orin Nano                           |
| Librealsense SDK Version       |  2.55.1                                                    | 
| Language                               |  C opencv python         | 
| Segment			         |  Robot                             | 
| ROS Distro			         |  Noetic                                           | 
| RealSense ROS Wrapper Version   |  2.3.2                                                  | 

### Issue Description
Got the following Invalid Value error message when running ros wrapper
```
(rs.cpp:237) [rs2_set_region_of_interest( sensor:0xffff480395b0, min_x:0, min_y:0, max_x:639, max_y:479 ) Invalid Value] hwmon command 0x75( 0 1df 0 27f ) failed (response -6= Invalid parameter)
[ERROR] [1721123312.425306402]: hwmon command 0x75( 0 1df 0 27f ) failed (response -6= Invalid parameter)
```
It seems likely that there is a version mismatch between the firmware and librealsense, causing the hardware operation code to be incompatible. I tried searching for related issues but couldn't find clear information on the correct version pair.

If I need to run on Ubuntu 20 and ROS Noetic, what are the recommended firmware and component versions to ensure compatibility?

<details>
<summary>The log of ros error</summary>

```bash
... logging to /home/dev/.ros/log/8e131896-4358-11ef-9de1-b48c9d1c5705/roslaunch-ubuntu-4936.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:33357/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 0
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: True
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: 5
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/confidence_fps: 30
 * /camera/realsense2_camera/confidence_height: 480
 * /camera/realsense2_camera/confidence_width: 640
 * /camera/realsense2_camera/depth_fps: 5
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
[... 96 lines cut ...]
[ INFO] [1721123310.955238591]: Sync Mode: On
[ INFO] [1721123310.955609734]: Device Sensors: 
[ INFO] [1721123310.967237669]: Stereo Module was found.
[ INFO] [1721123310.976546551]: RGB Camera was found.
[ INFO] [1721123310.977210595]: Motion Module was found.
[ INFO] [1721123310.977592075]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1721123310.978015475]: Add Filter: pointcloud
[ INFO] [1721123310.981227568]: num_filters: 2
[ INFO] [1721123310.981597623]: Setting Dynamic reconfig parameters.
[ WARN] [1721123311.108293232]: Param '/camera/rgb_camera/power_line_frequency' has value 3 that is not in the enum { {50Hz: 1} {60Hz: 2} {Disabled: 0} }. Removing this parameter from dynamic reconfigure options.
[ INFO] [1721123311.140769533]: Done Setting Dynamic reconfig parameters.
[ INFO] [1721123311.141586157]: depth stream is enabled - width: 640, height: 480, fps: 5, Format: Z16
[ INFO] [1721123311.142106999]: color stream is enabled - width: 640, height: 480, fps: 5, Format: RGB8
[ INFO] [1721123311.143203756]: setupPublishers...
[ INFO] [1721123311.148391695]: Expected frequency for depth = 5.00000
[ INFO] [1721123311.200597174]: Expected frequency for color = 5.00000
[ INFO] [1721123311.235096138]: Expected frequency for aligned_depth_to_color = 5.00000
[ INFO] [1721123311.270664915]: setupStreams...
[ INFO] [1721123311.296859240]: SELECTED BASE:Depth, 0
[ INFO] [1721123311.311888776]: RealSense Node Is Up!
[ WARN] [1721123311.798879119]: 
 16/07 05:48:32,424 ERROR [281471775093104] (rs.cpp:237) [rs2_set_region_of_interest( sensor:0xffff480395b0, min_x:0, min_y:0, max_x:639, max_y:479 ) Invalid Value] hwmon command 0x75( 0 1df 0 27f ) failed (response -6= Invalid parameter)
[ERROR] [1721123312.425306402]: hwmon command 0x75( 0 1df 0 27f ) failed (response -6= Invalid parameter)
```
</details>

MartyG-RealSense, 2024-07-16:
Hi @HTLife  The recommended configuration for ROS1 wrapper 2.3.2 is librealsense 2.50.0 or 2.51.1, and firmware 5.13.0.50.

Your current configuration (librealsense 2.55.1 and firmware 5.16.0.1) is not designed for use with the 2.3.2 wrapper, though some RealSense ROS users do manage to achieve a working setup with this configuration even though it should not work.

Does the error still occur if you edut your launch file to set the **initial_reset** parameter to true to reset the camera at launch?

HTLife, 2024-07-17:
@MartyG-RealSense Thanks for your quick response.
I've tried to add 
```xml
<arg name="initial_reset"  default="true"/>
```
to my launch file, but the same `hwmon command 0x75` still appear.
<details>
<summary>ros output</summary>
NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [7194]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 995ab086-4422-11ef-8714-b48c9d1c5705
process[rosout-1]: started with pid [7209]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [7216]
process[camera/realsense2_camera-3]: started with pid [7217]
[ INFO] [1721210087.535350479]: Initializing nodelet with 6 worker threads.
[ INFO] [1721210087.920848640]: RealSense ROS v2.3.2
[ INFO] [1721210087.921051588]: Built with LibRealSense v2.55.1
[ INFO] [1721210087.921226567]: Running with LibRealSense v2.55.1
[ INFO] [1721210087.967294745]:  
[ INFO] [1721210088.001041023]: Device with serial number 234322304887 was found.

[ INFO] [1721210088.001361221]: Device with physical ID /sys/devices/platform/3610000.xhci/usb2/2-1/2-1.3/2-1.3:1.0/video4linux/video0 was found.
[ INFO] [1721210088.001686859]: Device with name Intel RealSense D455 was found.
[ INFO] [1721210088.002615517]: Device with port number 2-1.3 was found.
[ INFO] [1721210088.002919811]: Device USB type: 3.2
[ INFO] [1721210088.003222536]: Resetting device...
[ INFO] [1721210094.017113394]:  
[ INFO] [1721210094.057275922]: Device with serial number 234322304887 was found.

[ INFO] [1721210094.057380468]: Device with physical ID /sys/devices/platform/3610000.xhci/usb2/2-1/2-1.3/2-1.3:1.0/video4linux/video0 was found.
[ INFO] [1721210094.057404981]: Device with name Intel RealSense D455 was found.
[ INFO] [1721210094.057822973]: Device with port number 2-1.3 was found.
[ INFO] [1721210094.057858494]: Device USB type: 3.2
[ INFO] [1721210094.060754517]: getParameters...
[ INFO] [1721210094.138100221]: setupDevice...
[ INFO] [1721210094.557977728]: JSON file is loaded! (/home/dev/catkin_ws/src/fork_vision_pc/ros/fork_perception_ai/config/realsense_high_accuracy.json)
[ INFO] [1721210094.558221573]: ROS Node Namespace: camera
[ INFO] [1721210094.558724590]: Device Name: Intel RealSense D455
[ INFO] [1721210094.559049748]: Device Serial No: 234322304887
[ INFO] [1721210094.559358202]: Device physical port: /sys/devices/platform/3610000.xhci/usb2/2-1/2-1.3/2-1.3:1.0/video4linux/video0
[ INFO] [1721210094.559665952]: Device FW version: 5.16.0.1
[ INFO] [1721210094.559977958]: Device Product ID: 0x0B5C
[ INFO] [1721210094.560306316]: Enable PointCloud: On
[ INFO] [1721210094.560607346]: Align Depth: On
[ INFO] [1721210094.560907928]: Sync Mode: On
[ INFO] [1721210094.561557828]: Device Sensors: 
[ INFO] [1721210094.576281118]: Stereo Module was found.
[ INFO] [1721210094.583207043]: RGB Camera was found.
[ INFO] [1721210094.583770541]: Motion Module was found.
[ INFO] [1721210094.589429242]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1721210094.590032645]: Add Filter: pointcloud
[ INFO] [1721210094.590464622]: num_filters: 2
[ INFO] [1721210094.590588432]: Setting Dynamic reconfig parameters.
[ WARN] [1721210094.746376917]: Param '/camera/rgb_camera/power_line_frequency' has value 3 that is not in the enum { {50Hz: 1} {60Hz: 2} {Disabled: 0} }. Removing this parameter from dynamic reconfigure options.
[ INFO] [1721210094.775236029]: Done Setting Dynamic reconfig parameters.
[ INFO] [1721210094.776007756]: depth stream is enabled - width: 640, height: 480, fps: 5, Format: Z16
[ INFO] [1721210094.776574519]: color stream is enabled - width: 640, height: 480, fps: 5, Format: RGB8
[ INFO] [1721210094.777825775]: setupPublishers...
[ INFO] [1721210094.782545321]: Expected frequency for depth = 5.00000
[ INFO] [1721210094.842150718]: Expected frequency for color = 5.00000
[ INFO] [1721210094.879872016]: Expected frequency for aligned_depth_to_color = 5.00000
[ INFO] [1721210094.915896321]: setupStreams...
[ INFO] [1721210094.941782160]: SELECTED BASE:Depth, 0
[ INFO] [1721210094.956962771]: RealSense Node Is Up!
[ WARN] [1721210095.235518277]: 
 17/07 18:54:55,967 ERROR [281471678742896] (ds-options.cpp:93) Asic Temperature value is not valid!
 17/07 18:54:56,943 ERROR [281472264614256] (ds-options.cpp:93) Asic Temperature value is not valid!
 17/07 18:54:56,971 ERROR [281471678742896] (ds-options.cpp:93) Asic Temperature value is not valid!
 17/07 18:54:57,521 ERROR [281471669731696] (rs.cpp:237) [rs2_set_region_of_interest( sensor:0xffff4803c4f0, min_x:0, min_y:0, max_x:639, max_y:479 ) Invalid Value] hwmon command 0x75( 0 1df 0 27f ) failed (response -6= Invalid parameter)
[ERROR] [1721210097.521679625]: hwmon command 0x75( 0 1df 0 27f ) failed (response -6= Invalid parameter)
[ WARN] [1721210097.792545096]: Hardware Notification:Depth stream start failure,1.72121e+12,Error,Hardware Error
 17/07 18:54:57,984 ERROR [281471678742896] (ds-options.cpp:93) Asic Temperature value is not valid!
^C[camera/realsense2_camera-3] killing on exit
[camera/realsense2_camera_manager-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
</details>

Therefore, I'll try to follow your instructions to set all the versions back.

HTLife, 2024-07-17:
Still have no luck.
I got the following this time.
```
hwmon command 0x75( 0 1df 0 27f ) failed (response -6= Invalid parameter)
[ WARN] [1721214182.291414752]: Hardware Notification:Depth stream start failure,1.72121e+12,Error,Hardware Error
```

Updated version:
 - firmware 05.13.00.50
 - ROS1 wrapper 2.3.2
 - librealsense 2.51.1

<details>
<summary>Steps to downgrade and recompile</summary>
### Uninstall librealsense2
```bash
sudo apt-get remove librealsense2-dkms
sudo apt-get remove librealsense2-utils
sudo apt-get remove librealsense2-dev
sudo apt-get remove librealsense2-dbg

sudo apt-get purge librealsense2-dkms
sudo apt-get purge librealsense2-utils
sudo apt-get purge librealsense2-dev
sudo apt-get purge librealsense2-dbg
```

### Build and install librealsense 2.51.1
1. Download the [image]
2. Build and install
```bash
unzip librealsense-2.51.1.zip
cd librealsense-2.51.1
mkdir build
cd build
cmake ..
make -j4
sudo make install
```
3. Check installed librealsense2 version

### Downgrade firmware to 5.13.0.50
1. Download [image]
2. Extract the zip
3. Open `realsense-viewer` from terminal
4. The Viewer title bar should have "Intel RealSense Viewer v2.51.1" now
5. More => Update Firmware... => Open "D400_Series_FW_5_13_0_50" folder => Select "Signed_image_UVC_5_13_0_50.bin"
6. In "Info", Firmware Version: 05.13.00.50, and Min FW Version: 05.13.00.50

### Rebuild ros workspace
```bash
sudo ldconfig
cd ~/catkin_ws
rm -rf build devel
catkin build
source devel/setup.bash
```
</details>

```
/usr/local/lib$ ls
cmake                     librealsense2-gl.so.2.51.1  pkgconfig
libfw.a                   librealsense2.so            python2.7
libglfw3.a                librealsense2.so.2.51       python3.8
librealsense2-gl.so       librealsense2.so.2.51.1     python3.9
librealsense2-gl.so.2.51  librealsense-file.a
```

<details>
<summary>roslaunch error code</summary>
[... 138 lines cut ...]
[ INFO] [1721214178.975900992]: Align Depth: On
[ INFO] [1721214178.976222240]: Sync Mode: On
[ INFO] [1721214178.976772544]: Device Sensors: 
[ INFO] [1721214179.043531904]: Stereo Module was found.
[ INFO] [1721214179.103430688]: RGB Camera was found.
[ INFO] [1721214179.105936768]: Motion Module was found.
[ INFO] [1721214179.106457632]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1721214179.109849984]: Add Filter: pointcloud
[ INFO] [1721214179.119847520]: num_filters: 2
[ INFO] [1721214179.120362080]: Setting Dynamic reconfig parameters.
[ WARN] [1721214179.260549216]: Param '/camera/rgb_camera/power_line_frequency' has value 3 that is not in the enum { {50Hz: 1} {60Hz: 2} {Disabled: 0} }. Removing this parameter from dynamic reconfigure options.
[ INFO] [1721214179.287624064]: Done Setting Dynamic reconfig parameters.
[ INFO] [1721214179.289072992]: depth stream is enabled - width: 640, height: 480, fps: 5, Format: Z16
[ INFO] [1721214179.289796736]: color stream is enabled - width: 640, height: 480, fps: 5, Format: RGB8
[ INFO] [1721214179.290940256]: setupPublishers...
[ INFO] [1721214179.298824704]: Expected frequency for depth = 5.00000
[ INFO] [1721214179.355602144]: Expected frequency for color = 5.00000
[ INFO] [1721214179.397676544]: Expected frequency for aligned_depth_to_color = 5.00000
[ INFO] [1721214179.436973536]: setupStreams...
[ INFO] [1721214179.551443712]: SELECTED BASE:Depth, 0
[ INFO] [1721214179.576556064]: RealSense Node Is Up!
[ WARN] [1721214179.989020160]: 
[ERROR] [1721214182.066267968]: hwmon command 0x75( 0 1df 0 27f ) failed (response -6= Invalid parameter)
[ WARN] [1721214182.291414752]: Hardware Notification:Depth stream start failure,1.72121e+12,Error,Hardware Error
</details>

MartyG-RealSense, 2024-07-17:
Is there a Depth Stream Start Failure if you perform an **RGBD** launch instead of using rs_camera.launch, please?

To perform an RGBD launch in the 2.3.2 ROS1 wrapper, the steps are:

1.  Install support for an RGBD launch with the command below.

`sudo apt-get install ros-noetic-rgbd-launch`

 2.  Run the rs_rgbd launch file.

`roslaunch realsense2_camera rs_rgbd.launch`

HTLife, 2024-07-18:
@MartyG-RealSense Thanks for the support.
`roslaunch realsense2_camera rs_rgbd.launch`
works for me.
Let's me find out what causing the issue in my launch file compare with `rs_rgbd.launch`.

MartyG-RealSense, 2024-07-18:
I'm pleased to hear that `rs_rgbd.launch` works for you!

HTLife, 2024-07-22:
I identified the cause of the error message by comparing the rs_rgbd.launch file with my custom launch file. The rs_rgbd.launch file sets the depth_fps to 30 or "-1", while I attempted to set it to 5 fps. This should be acceptable, as the realsense_viewer also supports a 5 fps setting. It appears there might be a hardware control issue involving librealsense, the ROS wrapper, and the firmware.

As I can at least run the camera with 30 fps, I consider the issue as partially solved.  So I'll close this issue.
Thanks for your support.

closed 2024-07-22 by HTLife

MartyG-RealSense, 2024-07-22:
It's excellent news that you found a solution.

When setting a custom configuration, you need to configure three factors - stream width, stream height and FPS.  If only one factor, such as FPS, is configured and the others are left as -1 then the custom configuration is deemed invalid and the default stream settings are applied instead.

For example, you could set 5 FPS by adding the commands below to the roslaunch instruction for each stream type that you are using.

`depth_width:=640 depth_height:=480 depth_fps:=5 color_width:=640 color_height:=480 color_fps:=5`