realsenseai/realsense-ros #3168: Best way to install for L515 with ROS1 wrapper

ManasviniSrini, 2024-07-24:
I purchased a L515 camera and these are the steps I'm following to install realsense-ros. I'm currently using ROS noetic and would like to stick to ROS1. 
- I've downloaded the Intel® RealSense™ SDK 2.0 (v2.48.0).
- git cloned ROS-1 legacy.
- Placed the librealsense package in the ROS-1 git package
- Built it using this : catkin build -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true

I still run into a lot of errors when building the package and its fails everytime. I was wondering if this is the right way or if I'm missing something with the installation steps. 

Thanks.

MartyG-RealSense, 2024-07-25:
Hi @ManasviniSrini  The easiest installation procedure would be to use the command below, which will install librealsense 2.50.0 and the 2.3.2 ROS1 wrapper at the same time.

`sudo apt-get install ros-$ROS_DISTRO-realsense2-camera`

This installation method does not include the RealSense Viewer tool though, so if you need the Viewer then you should install librealsense first and then build the ROS1 wrapper from source code afterwards.

ManasviniSrini, 2024-07-25:
Thanks @MartyG-RealSense ! I got the package installed and its working great.

closed 2024-07-25 by ManasviniSrini

MartyG-RealSense, 2024-07-25:
You are very welcome.  I'm pleased to hear that you succeeded.  Thanks very much for the update!