realsenseai/realsense-ros #3169: RGB image and depth image do not match

CYJ00, 2024-07-26:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking our ROS RealSense Wrapper documentation [README](https://github.com/IntelRealSense/realsense-ros/blob/ros2-master/README.md).
    * Have you looked in our [Discussions](https://github.com/IntelRealSense/realsense-ros/discussions)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/realsense-ros/issues) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D415                                                                                | 
| Operating System & Version |   Ubuntu 20.04                                                                 |
| Platform                                 | Laptop                                                                            |
| ROS Distro			         |  Noteic                                                                            | 

ros-noetic-librealsense2: 2.50.0-1
ros-noetic-realsense2-camera: 2.3.2-1

### Issue Description
<Describe your issue / question / feature request / etc..>
RGB image and depth image do not match.
If I binarize the depth image into a mask and overlay it on the rgb image, I can see that the mask is off like below.

[image]

The left side of the image is alright, but the right side of the image do not match.
My first thought was that the time sync was out of whack, but seeing that it only happens on the right side of all the images, it doesn't seem to be a time sync issue.
Why is this happening? Are there other characteristics of the depth image that I should be aware of?

MartyG-RealSense, 2024-07-26:
Hi @CYJ00  Do you get the results that you require if you add **enable_sync:=true** to your roslaunch instruction, please? 
 enable_sync: gathers the closest frames of different sensors, such as infrared, color and depth, to be sent with the same timetag.

MartyG-RealSense, 2024-08-03:
Hi @CYJ00  Do you require further assistance with this case, please?  Thanks!

CYJ00, 2024-08-06:
Hello!
I tried `roslauch rs_camera.lanch enable_sync:=true`
The time of `/camera/color/image_raw` topic and `/camera/depth/image_rect_raw` topic are different, but the error is acceptable.
Thank you!

closed 2024-08-06 by CYJ00

MartyG-RealSense, 2024-08-06:
You are very welcome, @CYJ00 - I'm pleased that I could help.  Thanks very much for the update!