realsenseai/realsense-ros #3171: Jetson Orin Nano, No point cloud data in the published topic

hammockrobotics, 2024-07-29:
| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | { D435i  }                        
| Firmware Version                   | (Open RealSense Viewer --> Click info)                          | 
| Operating System & Version |   { Linux (Ubuntu 20.04) } | 
| Kernel Version (Linux Only)    |  (5.4)                                                                       | 
| Platform                                 | NVIDIA Jetson Orin Nano   8Gb                         |
| Librealsense SDK Version       |  { 2.3.2 }                                                    | 
| Language                               |  { }         | 
| Segment			         |  {Robot/Smartphone/VR/AR/others }                             | 
| ROS Distro			         |  { Noetic }                                           | 
| RealSense ROS Wrapper Version   |  {4.51.1, 4.54.1, etc..}                                                  | 

### Issue Description
I had followed the steps as mentioned here to install the wrapper for the Jetson orin nano 8Gb : https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

**I attempted to run the launch file :**
roslaunch realsense2_camera rs_camera.launch enable_pointcloud:=true

and it functions as expected. However, upon inspection of the topics, it seems to be there are no information on the point cloud. This issue is further confirmed on Rviz as I am not able to visualize the point cloud in it. 

In the realsense-viewer however, there seems to be no isues with visualizing the 3D image based on the colour map for depth image. 

I then attempeted to run the ros wrapper based on the original github, however there are more issues with that method and I beliee it is not compatible with the version used for the Jetson Orin Nano. 

Is there a way or a step that I am missing to actually have the point cloud topic able to be visible in Rviz? I require this topic for other functionalities later on. Appreciate the help.

MartyG-RealSense, 2024-07-29:
Hi @hammockrobotics  Are you able to visualize the pointcloud in RViz if you use the **filters:=pointcloud** parameter instead of enable_pointcloud and have Fixed Frame set to **camera_link**, please?

`roslaunch realsense2_camera rs_camera.launch filters:=pointcloud`

https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy?tab=readme-ov-file#point-cloud

*********

If you use these settings and still cannot visualize the pointcloud then there is an alternative method in the RealSense ROS1 wrapper of performing an **RGBD** launch instead of using rs_camera.

1.  Install support for an RGBD launch on Noetic with the command below.

`sudo apt-get install ros-noetic-rgbd-launch`

2.  Launch the **rs_rgbd.launch** launch file.  You do not need to enable the pointcloud in the launch command as it is already enabled by default in this launch file.

`roslaunch realsense2_camera rs_rgbd.launch `

The launch file should be published at **depth_image_proc** instead of /camera/depth/color/points

hammockrobotics, 2024-08-08:
hi @MartyG-RealSense the method with filters:=pointcloud does not have an effect of the display of information on the topic: /camera/depth/color/points for pointCloud2. Is there another topic i should be looking at?

I tried your method of roslaunch realsense2_camera rs_rgbd.launch and the pointcloud topic appears to be published on the topic: /camera/depth_registered/points. as pointCloud2

appreciate the help, however is there an explanation of why it works in the second method as opposed to the one provided in the main github?

MartyG-RealSense, 2024-08-08:
I'm pleased to hear that you were able to achieve a pointcloud with the rs_rgbd launch file.

There is not a known reason why rs_rgbd is a solution for pointcloud publishing on Jetson boards when enabling the pointcloud filter on an rs_camera launch does not work.  The main differences between the launch files aside from the different topic that the pointcloud is published to are that in rs_rgbd, (a) **align_depth** is true by default, (b) its pointcloud is an **ordered** cloud instead of the unordered cloud that rs_camera uses by default, and (c) **enable_sync** is true by default.

rs_rgbd also has a significant amount of additional instructions related to pointcloud processing that rs_camera does not have.

https://github.com/IntelRealSense/realsense-ros/blob/ros1-legacy/realsense2_camera/launch/rs_rgbd.launch#L182-L200