realsenseai/realsense-ros #3172: Wrong frame_id in /camera/camera/infra2/camera_info

vinaym815, 2024-07-30:
| Camera Model                      D415
| ROS Distro			        Humble

### Issue Description
The frame_id on topic /camera/camera/infra2/camera_info is wrongly published as 
camera_infra1_optical_frame

MartyG-RealSense, 2024-07-30:
Hi @vinaym815  https://github.com/IntelRealSense/realsense-ros/issues/2220#issuecomment-1009819775 explains why infra2 uses the same frame_id as infra1.

vinaym815, 2024-07-30:
If we launch the camera using 
```ros2 launch realsense2_camera rs_launch.py enable_infra1:=true enable_infra2:=true```
and echo the topic 
```ros2 topic echo /camera/camera/infra2/camera_info```
we get 
```
header:
  stamp:
    sec: 1722328243
    nanosec: 768045898
  frame_id: camera_infra1_optical_frame
...
```
The frame id should be camera_infra2_optical_frame

MartyG-RealSense, 2024-07-30:
I recall once seeing a line in the source code of the ROS wrapper that basically said infra2_optical_frame = infra1_optical_frame, hence why the infra1 value was displayed for infra2.

The following method was suggested to get the coordinate transform from right camera to left camera.

`ros2 run tf tf_echo camera_infra2_optical_frame camera_infra1_optical_frame`

vinaym815, 2024-07-30:
I have a generalized checkerboard detector package which uses CameraInfo topic and Image topic to publish the relevant transform. Wrong frame_id requires making unnecessary exception case.
Also when only enabling only the infra2 the frame_id is correct, i.e. 'camera_infra2_optical_frame'. So I think it is a small bug somewhere.
```ros2 launch realsense2_camera rs_launch.py enable_infra1:=true enable_infra2:=true```

MartyG-RealSense, 2024-07-30:
I will discuss your question with my Intel RealSense colleagues on the ROS wrapper team.  Thanks very much for your patience!

MartyG-RealSense, 2024-08-09:
Hi @vinaym815  My Intel RealSense colleagues on the ROS wrapper team said that they would look at this issue.  Thanks very much for your patience!

MartyG-RealSense, 2026-10-31:
Case closed due to inactivity for over 2 years.

closed 2026-10-31 by MartyG-RealSense