realsenseai/realsense-ros #3174: No realsense device were found

souravchandrasaha, 2024-07-31:
Camera Model | D435i
| Operating System & Version | Linux Ubuntu 20.04 |
| Kernel Version (Linux Only) | tegra
| Platform | NVIDIA Jetson  nano jetpack 6
| ROS 2 Galactic
Description:

I have used RSUSB method for installation of SDK 
 My realsense_viewer is working properly 
I have build from source - ros2-legacy 

Getting this error . How to solve it

ros2 run realsense2_camera realsense2_camera_node
[INFO] [1722248335.701375649] [camera]: RealSense ROS v4.51.1
[INFO] [1722248335.701861705] [camera]: Built with LibRealSense v2.51.1
[INFO] [1722248335.701995043] [camera]: Running with LibRealSense v2.51.1
[WARN] [1722248335.727763052] [camera]: No RealSense devices were found!
29/07 15:48:55,727 WARNING [546870127008] (ds5-factory.cpp:1152) DS5 group_devices is empty.
[WARN] [1722248341.914576579] [camera]: No RealSense devices were found!
29/07 15:49:01,914 WARNING [546870127008] (ds5-factory.cpp:1152) DS5 group_devices is empty.
[WARN] [1722248348.099195180] [camera]: No RealSense devices were found!
29/07 15:49:08,099 WARNING [546870127008] (ds5-factory.cpp:1152) DS5 group_devices is empty.
^C[INFO] [1722248352.065228821] [rclcpp]: signal_handler(signal_value=2)

MartyG-RealSense, 2024-07-31:
Hi @souravchandrasaha  The ros2-legacy wrapper was not designed for Galactic as it is a now-obsolete wrapper that had Foxy as its newest supported ROS2 version and was replaced by the **ros2-development** branch.

It appears though that you do not have ros2-legacy installed as the version numbers for that wrapper started with the number 3.  Instead, the log lists wrapper 4.51.1, which is the ros2-development wrapper.  4.51.1 **did** support ROS2 Galactic, so it is likely that your problem is not related to Galactic being used.  Ubuntu 20.04 is also the correct OS version to use with Galactic.

Can you link me to the instructions that you used for your source code build of the ROS wrapper, please?

souravchandrasaha, 2024-08-02:
i have followed this 👍 

Option 2: Install from source
Create a ROS2 workspace

mkdir -p ~/ros2_ws/src
cd ~/ros2_ws/src/
Clone the latest ROS Wrapper for Intel® RealSense™ cameras from [here](https://github.com/IntelRealSense/realsense-ros.git) into '~/ros2_ws/src/'

git clone https://github.com/IntelRealSense/realsense-ros.git -b ros2-master
cd ~/ros2_ws
Install dependencies

sudo apt-get install python3-rosdep -y
sudo rosdep init # "sudo rosdep init --include-eol-distros" for Foxy and earlier
rosdep update # "sudo rosdep update --include-eol-distros" for Foxy and earlier
rosdep install -i --from-path src --rosdistro $ROS_DISTRO --skip-keys=librealsense2 -y
Build
colcon build
Source environment
ROS_DISTRO=<YOUR_SYSTEM_ROS_DISTRO>  # set your ROS_DISTRO: jazzy, iron, humble, foxy
source /opt/ros/$ROS_DISTRO/setup.bash
cd ~/ros2_ws
. install/local_setup.bash

MartyG-RealSense, 2024-08-02:
Git cloning **ros2-master** will currently install the **4.55.1** wrapper, which does not support Galactic.  The support for Galactic was removed in the previous wrapper 4.54.1 and the wrapper currently only supports Humble and Iron.

The link below has a 'forked' version of wrapper 4.51.1 created by a RealSense ROS user that could be git-cloned instead of ros2-master.  4.51.1 has Galactic support.

https://github.com/HedgeHao/realsense-ros

souravchandrasaha, 2024-08-08:
i have tried running it on my rpi4 with ros2 humble : here is what we are getting error. we have realsense viewer running proper btw

 ros2 run realsense2_camera realsense2_camera_node
[INFO] [1723112768.991539503] [camera.camera]: RealSense ROS v4.55.1
[INFO] [1723112768.991983554] [camera.camera]: Built with LibRealSense v2.55.1
[INFO] [1723112768.992057053] [camera.camera]: Running with LibRealSense v2.55.1
 08/08 15:56:14,834 ERROR [281472865069280] (librealsense-exception.h:52) set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
 08/08 15:56:14,835 ERROR [281472865069280] (rs.cpp:237) [rs2_create_device( info_list:0xffff700214a0, index:0 ) Backend] set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
 08/08 15:56:14,836 ERROR [281472865069280] (rs.cpp:237) [rs2_delete_device( device:nullptr ) UNKNOWN] null pointer passed for argument "device"
 08/08 15:56:14,836 WARNING [281472865069280] (rs.cpp:373) null pointer passed for argument "device"
[WARN] [1723112774.836280070] [camera.camera]: Device 1/1 failed with exception: set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
[ERROR] [1723112774.836490438] [camera.camera]: The requested device with  is NOT found. Will Try again.
 08/08 15:56:26,610 ERROR [281472865069280] (librealsense-exception.h:52) set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
 08/08 15:56:26,612 ERROR [281472865069280] (rs.cpp:237) [rs2_create_device( info_list:0xffff700174e0, index:0 ) Backend] set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
 08/08 15:56:26,612 ERROR [281472865069280] (rs.cpp:237) [rs2_delete_device( device:nullptr ) UNKNOWN] null pointer passed for argument "device"
 08/08 15:56:26,612 WARNING [281472865069280] (rs.cpp:373) null pointer passed for argument "device"
[WARN] [1723112786.612850681] [camera.camera]: Device 1/1 failed with exception: set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
[ERROR] [1723112786.613164603] [camera.camera]: The requested device with  is NOT found. Will Try again.
^C[INFO] [1723112791.077299750] [rclcpp]: signal_handler(signum=2)
yash@yash-rpi:~$ ros2 launch realsense2_camera rs_launch.py enable_color:=true enable_depth:=true wait_for_device_timeout:=10. reconnect_timeout:=10.
[INFO] [launch]: All log files can be found below /home/yash/.ros/log/2024-08-08-15-56-49-655469-yash-rpi-174246
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [174273]
[realsense2_camera_node-1] [INFO] [1723112810.750025156] [camera.camera]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1723112810.752584148] [camera.camera]: Built with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1723112810.752756813] [camera.camera]: Running with LibRealSense v2.55.1
[realsense2_camera_node-1]  08/08 15:56:56,562 ERROR [281472802351328] (librealsense-exception.h:52) set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
[realsense2_camera_node-1]  08/08 15:56:56,565 ERROR [281472802351328] (rs.cpp:237) [rs2_create_device( info_list:0xffff680214a0, index:0 ) Backend] set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
[realsense2_camera_node-1]  08/08 15:56:56,565 ERROR [281472802351328] (rs.cpp:237) [rs2_delete_device( device:nullptr ) UNKNOWN] null pointer passed for argument "device"
[realsense2_camera_node-1]  08/08 15:56:56,565 WARNING [281472802351328] (rs.cpp:373) null pointer passed for argument "device"
[realsense2_camera_node-1] [WARN] [1723112816.565784488] [camera.camera]: Device 1/1 failed with exception: set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
[realsense2_camera_node-1] [ERROR] [1723112816.566080022] [camera.camera]: The requested device with  is NOT found. Will Try again.
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[realsense2_camera_node-1] [INFO] [1723112823.387341296] [rclcpp]: signal_handler(signum=2)
[realsense2_camera_node-1]  08/08 15:57:06,546 ERROR [281472802351328] (librealsense-exception.h:52) set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
[realsense2_camera_node-1]  08/08 15:57:06,552 ERROR [281472802351328] (rs.cpp:237) [rs2_create_device( info_list:0xffff680174e0, index:0 ) Backend] set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
[realsense2_camera_node-1]  08/08 15:57:06,552 ERROR [281472802351328] (rs.cpp:237) [rs2_delete_device( device:nullptr ) UNKNOWN] null pointer passed for argument "device"
[realsense2_camera_node-1]  08/08 15:57:06,552 WARNING [281472802351328] (rs.cpp:373) null pointer passed for argument "device"
[realsense2_camera_node-1] [WARN] [1723112826.553108713] [camera.camera]: Device 1/1 failed with exception: set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
[realsense2_camera_node-1] [ERROR] [1723112826.553385785] [camera.camera]: The requested device with  is NOT found. Will Try again.
[realsense2_camera_node-1] [ERROR] [1723112826.553801706] [camera.camera]: wait for device timeout of 10 secs expired
[ERROR] [realsense2_camera_node-1]: process[realsense2_camera_node-1] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[INFO] [realsense2_camera_node-1]: sending signal 'SIGTERM' to process[realsense2_camera_node-1]
[... cut ...]
 08/08 16:00:48,552 INFO [281473275424800] (backend-v4l2.cpp:753) Enumerating UVC video5 realpath=/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-2/2-2:1.3/video4linux/video5
 08/08 16:00:48,615 INFO [281473275424800] (context.cpp:116) ... /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-2/2-2:1.0/video4linux/video0
 08/08 16:00:48,615 INFO [281473275424800] (context.cpp:128) Found 1 RealSense devices (0xff requested & 0xff from device-mask in settings)

MartyG-RealSense, 2024-08-08:
Is the RealSense Viewer closed when performing the ROS launch?  It is not necessary for the Viewer to be streaming the camera during a ROS session.

Does the launch still have problems if **initial_reset:=true** is included in the ros2 launch instruction to reset the camera at launch?

souravchandrasaha, 2024-08-09:
no realsense-viewer was not running at backend at that time. and we have tried running with initial_reset :=true but didn't worked

MartyG-RealSense, 2024-08-09:
Which branch of the ROS2 wrapper did you install the 4.55.1 wrapper from, please?

Until recently 4.55.1 was installed from the **ros2-development** branch.  This caused some problems though, as the development-branch version of the librealsense SDK had to be used with it instead of the master branch, otherwise there were errors.

For this reason, the ROS2 wrapper is now installed from the **master** branch, which resolves the previous problems.

Did you install wrpper 4.55.1 on your Pi from the ros-development branch or master branch, please?

souravchandrasaha, 2024-08-16:
'ros2 launch realsense2_camera rs_launch.py enable_color:=false enable_depth:=true' this also didn't work. How is it even possible that my realsense -view working but ROS2 not able to find realsense device?

MartyG-RealSense, 2024-08-16:
It is common for Raspberry Pi boards to have problems with the RealSense ROS wrapper even if librealsense has been installed successfully and the realsense-viewer tool is working correctly.

I note that you switched to using ROS2 Humble but did not say whether you had upgraded from Ubuntu 20.04 to 22.04.  Humble should be used with 22.04.

souravchandrasaha, 2024-08-20:
yes we have ros2 humble up and running. We have able to run it but after sometime same error connection timeout is coming up

MartyG-RealSense, 2024-08-20:
Are you experiencing the **wait for device timeout of 10 secs expired** error mentioned earlier in this discussion?

How long on average does it run for before the timeout occurs?

If it can keep running for hours or days before a problem occurs then the problem is typically related to the computer / computing board hardware, its USB system or its operating system rather than there being a problem with the camera and its firmware driver.

closed 2024-09-01 by MartyG-RealSense