realsenseai/realsense-ros #3179: How to set up json file in l515?

Rhymakers, 2024-08-12:
| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | {L515 }                          | 
| Firmware Version                   | (01.05.08.01)                          | 
| Operating System & Version |   { Ubuntu 20.04 } | 
| Kernel Version (Linux Only)    |  (5.15.0-117-generic)                      | 
| Platform                                 | PC                           |
| Librealsense SDK Version       |  { 2.51.1 }                                                    | 
| Language                               |  {python }         | 
| Segment			         |  {Robot }                             | 
| ROS Distro			         |  {foxy }                                           | 
| RealSense ROS Wrapper Version   |  {4.51.1}                                                  | 

### Issue Description
The short range parameters provided by the viewer were saved as a json file. And I wanted to receive it in short range even in the ROS 2 environment. However, the json file is not loaded due to the rs400 advanced mode condition. **How can I get data from l515 with short range parameter in ROS 2?**

[image]

[image]

MartyG-RealSense, 2024-08-12:
Hi @Rhymakers  Whilst program scripts that load a json file from a string need to reference Advanced Mode, if **json_file_path** is set from the launch instruction or the launch file is edited to contain the file path then you should be able to import the json on your L515 without having to check for Advanced Mode.  Would it be possible for your project to use one of these two approaches to json loading, please?

For example:

`ros2 launch realsense2_camera rs_launch.py json_file_path:=/config.json`

Alternatively, if editing the launch file then the path of json_file_path can be defined in the **rs_launch.py** launch fie here:

https://github.com/IntelRealSense/realsense-ros/blob/ros2-development/realsense2_camera/launch/rs_launch.py#L30

Rhymakers, 2024-08-13:
I am using an example. I modified it in rs_launch.py ​​and used it, but I get a warning 'Device does not support advanced settings!'

> ros2 launch realsense2_camera rs_launch.py json_file_path:=/home/path/config.json

The above command also produces the same warning and fails to load.

MartyG-RealSense, 2024-08-13:
If loading a json is not working then an alternative is to individually define the main settings of a particular preset.  The short range L515 preset primarily reduces the values of the Laser Power and the Receiver Gain settings.  Please try the launch command below to set them to the values used by the Short Range preset in the RealSense Viewer.

`ros2 launch realsense2_camera rs_launch.py depth_module.laser_power:=93 depth_module.receiver_gain:=18`

Rhymakers, 2024-08-14:
ros2 launch realsense2_camera rs_launch.py depth_module.laser_power:=93 depth_module.receiver_gain:=18
[INFO] [launch]: All log files can be found below /home/desktop/.ros/log/2024-08-14-16-04-20-638497-desktop-MS-7C71-6282
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [6286]
[realsense2_camera_node-1] [INFO] [1723619060.758585251] [camera.camera]: RealSense ROS v4.51.1
[realsense2_camera_node-1] [INFO] [1723619060.758640224] [camera.camera]: Built with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1723619060.758650540] [camera.camera]: Running with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1723619061.190490476] [camera.camera]: Device with serial number f0246243 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1723619061.190543930] [camera.camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-9/2-9.2/2-9.2:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1723619061.190552248] [camera.camera]: Device with name Intel RealSense L515 was found.
[realsense2_camera_node-1] [INFO] [1723619061.190780091] [camera.camera]: Device with port number 2-9.2 was found.
[realsense2_camera_node-1] [INFO] [1723619061.190792795] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1723619061.190800265] [camera.camera]: Resetting device...
[realsense2_camera_node-1] [INFO] [1723619067.748354973] [camera.camera]: Device with serial number f0246243 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1723619067.748435780] [camera.camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-9/2-9.2/2-9.2:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1723619067.748464352] [camera.camera]: Device with name Intel RealSense L515 was found.
[realsense2_camera_node-1] [INFO] [1723619067.748897948] [camera.camera]: Device with port number 2-9.2 was found.
[realsense2_camera_node-1] [INFO] [1723619067.748930801] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1723619067.750714560] [camera.camera]: getParameters...
[realsense2_camera_node-1] [WARN] [1723619067.751285226] [camera.camera]: Device does not support advanced settings!
[realsense2_camera_node-1] [INFO] [1723619067.751331978] [camera.camera]: Device Name: Intel RealSense L515
[realsense2_camera_node-1] [INFO] [1723619067.751358925] [camera.camera]: Device Serial No: f0246243
[realsense2_camera_node-1] [INFO] [1723619067.751384292] [camera.camera]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-9/2-9.2/2-9.2:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1723619067.751409972] [camera.camera]: Device FW version: 01.05.08.01
[realsense2_camera_node-1] [INFO] [1723619067.751432983] [camera.camera]: Device Product ID: 0x0B64
[realsense2_camera_node-1] [INFO] [1723619067.751454772] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [WARN] [1723619067.862219861] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [WARN] [1723619067.867138414] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [INFO] [1723619067.867339037] [camera.camera]: Set ROS param gyro_fps to default: 100
[realsense2_camera_node-1] [INFO] [1723619067.867361590] [camera.camera]: Set ROS param accel_fps to default: 200
[realsense2_camera_node-1] [WARN] [1723619067.871315032] [camera.camera]: Could not set param: pointcloud.stream_index_filter with -1 Range: [-1, -2147483648]: parameter 'pointcloud.stream_index_filter' could not be set: Parameter {} doesn't comply with integer range.
[realsense2_camera_node-1] [INFO] [1723619067.871786940] [camera.camera]: Stopping Sensor: L500 Depth Sensor
[realsense2_camera_node-1] [INFO] [1723619067.924313394] [camera.camera]: Starting Sensor: L500 Depth Sensor
[realsense2_camera_node-1] [INFO] [1723619068.202725562] [camera.camera]: Open profile: stream_type: Confidence(0), Format: RAW8, Width: 1024, Height: 768, FPS: 30
[realsense2_camera_node-1] [INFO] [1723619068.202872441] [camera.camera]: Open profile: stream_type: Infra(0), Format: Y8, Width: 1024, Height: 768, FPS: 30
[realsense2_camera_node-1] [INFO] [1723619068.202923248] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 1024, Height: 768, FPS: 30
[realsense2_camera_node-1] [INFO] [1723619068.203253872] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1723619068.457692721] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1723619068.471970733] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1920, Height: 1080, FPS: 30
[realsense2_camera_node-1] [INFO] [1723619068.473147463] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1] [WARN] [1723619069.000383389] [camera.camera]: 
[realsense2_camera_node-1] [WARN] [1723619069.000427711] [camera.camera]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.

ros2 launch realsense2_camera rs_launch.py depth_module.laser_power:=93 depth_module.receiver_gain:=18
The above command does not modify any parameters.

Even if I apply it using ros2 param set /camera/camera dep~, it doesn't appear as short_range in the viewer. I want to try it with json, but where should I modify it?

MartyG-RealSense, 2024-08-14:
The ROS wrapper and the RealSense Viewer are two separate, independent programs.  Setting values for Laser Power and Receiver Gain in the ROS wrapper will not update the Viewer's settings.  I would recommend not to have the Viewer open when a ROS wrapper launch is active.

In the ROS1 wrapper you could set one of the built-in Visual Presets, as described at https://github.com/IntelRealSense/realsense-ros/issues/2055#issuecomment-968278545  

When setting a built-in preset in C++ / Python code it is done with a different method from setting a string for the json, so this may be a solution that will work on L515 in the ROS wrapper where the json_file_path method failed.

Setting the built-in preset in the ROS2 wrapper has not been done before as far as I am aware, but let's try constructing a command that might work.

ros2 launch realsense2_camera rs_launch.py depth_module.visual_preset:=5

'5' is the number of the short_range L515 preset.

I say 'might' work because I suspect that it might try to load the 400 Series 'Medium Density' preset instead, which is setting number 5 on that camera range.

closed 2024-08-30 by MartyG-RealSense