realsenseai/realsense-ros #3183: D435i can't find device in RK3588

ciyuan606, 2024-08-15:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D435i }                          | 
| Firmware Version                   | /                      | 
| Operating System & Version |  Linux (Ubuntu 22.04 | 
| Kernel Version (Linux Only)    | 5.10.110-rockchip-rk3588                 | 
| Platform                                 | orange pi                           |
| Librealsense SDK Version       | 2.55.1                                                | 
| Language                               |  python      | 
| Segment			         | Robot                         | 
| ROS Distro			         |  humble                                 | 
| RealSense ROS Wrapper Version   |4.55.1                                                 | 

### Issue Description
I use  
sudo apt install ros-humble-librealsense2*     
sudo apt install ros-humble-realsense2-*

[image]
 but
[image]
how can I found device?
use realsense-viewer
[image]

MartyG-RealSense, 2024-08-15:
Hi @ciyuan606  On Rockchip (RK) devices it is recommended to build the librealsense SDK from source code with CMake using the **RSUSB backend** method described at https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692

An easier method than the RSUSB procedure is to use the **libuvc backend** method at the link below to install using a build script called **libuvc_installation.sh**.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

If you need realsense-viewer to be included in the librealsense installation though then you will need to edit line 46 of the build script to add the following flags:

**-DBUILD_EXAMPLES=TRUE DBUILD_GRAPHICAL_EXAMPLES=TRUE**

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh#L46

YujieSuCWE, 2024-10-20:
I met the same problem of being not able to find my device. My system is ubuntu 24.04, my librealsense version is v2.56.2, my ros distro is jazzy, and my realsense-ros version is 4.56.1. What is different is that I can open realsense-viewer, but still cannot find the device on both rs_launch.py and the viewer. I am pretty sure I used RSUSB backend method when compiling librealsense.

MartyG-RealSense, 2024-10-21:
@YujieSuCWE  Thanks for the reminder that this issue is still open.  I will respond to your problem at the new issue that you created.

@ciyuan606  Did you achieve a solution on this issue, please?

closed 2024-11-04 by MartyG-RealSense