realsenseai/realsense-ros #3184: "disparity_filter.enable" : "true" does not work

Shunichi09, 2024-08-17:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking our ROS RealSense Wrapper documentation [README](https://github.com/IntelRealSense/realsense-ros/blob/ros2-master/README.md).
    * Have you looked in our [Discussions](https://github.com/IntelRealSense/realsense-ros/discussions)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/realsense-ros/issues) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D435 / D435i                           | 
| Firmware Version                   | 5.16.0.1                          | 
| Operating System & Version |   ubuntu 22.04} | 
| Kernel Version (Linux Only)    |  (e.g. 5.4)                                                                       | 
| Platform                                 | PC                  |
| Librealsense SDK Version       |   2.<?>.<?> }                                                    | 
| Language                               |  cpp and ros2      | 
| Segment			         |  {Robot}                             | 
| ROS Distro			         |  Humble }                                           | 
| RealSense ROS Wrapper Version   |  current ros2-dev branch            | 

### Issue Description

When I apply `"disparity_filter.enable": "true"`, I encounter the following error:

```
[realsense2_camera_node-1] Please try different format of this stream.
[realsense2_camera_node-1] [ERROR] [1723874490.956257341] [camera.rs_745412071535]: cv::Mat is empty. Ignoring this frame.
[realsense2_camera_node-1] [ERROR] [1723874490.956261868] [camera.rs_745412071535]: Could not fill ROS message. Frame was dropped.
[realsense2_camera_node-1] [ERROR] [1723874490.992304169] [camera.rs_745412071535]: Format DISPARITY32 is not supported in realsense2_camera node.
```

I suspect this error might be due to an incorrect format setting for either the depth or RGB streams.
My current configuration is as follows:

```
"depth_module.depth_profile": "1280x720x30",
"rgb_camera.color_profile": "1280x720x30",
"depth_module.infra_profile": "1280x720x30",
```

I searched through the documentation but couldn't find any information regarding this issue.

Could you please advise me on how to resolve this?

I really appreciate your help in advance.

MartyG-RealSense, 2024-08-17:
Hi @Shunichi09  A RealSense ROS user at https://github.com/IntelRealSense/realsense-ros/issues/2304 who also had problems with setting the disparity filter to true found that they also had to set `disparity_to_depth.enable` to true to resolve the problem, as the disparity image is then output in 16-bit format instead of the unsupported 32-bit format.

Shunichi09, 2024-08-18:
Thank you for the quick reply!!

Wow! I solved the problem. I really appreciate your help.

MartyG-RealSense, 2024-08-18:
You are very welcome, @Shunichi09 - I'm pleased that I could help.  Thanks very much for the update!

closed 2024-08-26 by MartyG-RealSense