realsenseai/realsense-ros #3185: jetson orin AGX issues after upgrading to jetpack 6.0,

Al-khattab, 2024-08-19:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking our ROS RealSense Wrapper documentation [README](https://github.com/IntelRealSense/realsense-ros/blob/ros2-master/README.md).
    * Have you looked in our [Discussions](https://github.com/IntelRealSense/realsense-ros/discussions)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/realsense-ros/issues) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | { D405 / D415 / D435 / D435i / D455 / D457 }                          | 
| Firmware Version                   | (Open RealSense Viewer --> Click info)                          | 
| Operating System & Version |   { Win (10/11) / Linux (Ubuntu 18/20/22) / MacOS } | 
| Kernel Version (Linux Only)    |  (e.g. 5.4)                                                                       | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..                           |
| Librealsense SDK Version       |  { 2.<?>.<?> }                                                    | 
| Language                               |  {C/C#/labview/opencv/pcl/python/unity }         | 
| Segment			         |  {Robot/Smartphone/VR/AR/others }                             | 
| ROS Distro			         |  {Iron/Humble/Rolling/etc.. }                                           | 
| RealSense ROS Wrapper Version   |  {4.51.1, 4.54.1, etc..}                                                  | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2024-08-19:
Hi @Al-khattab  Which model of RealSense camera do you have, please?

Al-khattab, 2024-08-19:
hi @MartyG-RealSense thank you for the fast reply,
sorry about the info, it was submitted before I was able to add the details
I have updated the issue with the relevant details, if there are any more required details, please let me know

MartyG-RealSense, 2024-08-19:
Thanks very much for the additional information.

When a RealSense camera equipped with an IMU (known as a HID device) is used with JetPack 6, a **MIPI driver** needs to be installed to add support for the IMU.  This is because JetPack 6 does not have a component called **hidraw**.

https://github.com/IntelRealSense/realsense_mipi_platform_driver

If JetPack is downgraded to JP 5.13 then the MIPI driver does not have to be installed.

If you require JetPack 6 then a way to keep JP6 whilst not having to install the MIPI driver is to build the librealsense SDK from source code with a build script using the **libuvc backend** procedure.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

Al-khattab, 2024-08-19:
I have seen you other posts mentioning this issue and I also followed your suggestion when I build it with **libuvc backend** the realsense viewer works fine with no lag but the IMU issue persisted and the ros2 cannot detect the camera.

regarding MIPI point, as far as I understand, I do not need a physical De-Serialize board, I just need to install the driver correct?
I tried installing the driver as well but I don't think I was successful since I couldn't even run the realsense-viewer 

also regarding jetpack 6, the only need for it to run ros2.

MartyG-RealSense, 2024-08-19:
A deserializer board is not needed when using the driver with a camera on a USB cable connection to add IMU support.

Your ROS2 launch log indicates that the camera was successfully detected by the launch.

```
[realsense2_camera_node-1] [INFO] [1724054725.169743952] [camera.camera]: Device Name: Intel RealSense D456
[realsense2_camera_node-1] [INFO] [1724054725.169790735] [camera.camera]: Device Serial No: 309122300620
[realsense2_camera_node-1] [INFO] [1724054725.169825103] [camera.camera]: Device physical port: 2-3.2-5
[realsense2_camera_node-1] [INFO] [1724054725.169855150] [camera.camera]: Device FW version: 5.16.0.1
[realsense2_camera_node-1] [INFO] [1724054725.169883566] [camera.camera]: Device Product ID: 0x0B5C
```

The problem that came after is that the warning **control_transfer returned** is generating continuously.  This indicates a communication problem between the camera and computer.  The problem is typically related to the USB system, either a glitch on the USB port or a problem with the cable.  

In the ROS wrapper, IMU data will not be published unless the IMU topics are enabled because they are disabled by default.

` ros2 launch realsense2_camera rs_launch.py enable_accel:=true enable_gyro:=true unite_imu_method:=2`

Al-khattab, 2024-08-19:
sorry for not elaborating properly, I have built sdk and lib multiple time with all methods which included  **libuvc backend** which gave me the camera not detected error.
the above log happens after I removed everything related to realsense and reinstalled it with the [this instructions](https://github.com/IntelRealSense/librealsense/issues/12831#issuecomment-2041140925).
I have also used different cables and cameras just to confirm and it is not a hardware issue.

also when I run the command 

`ros2 launch realsense2_camera rs_launch.py enable_accel:=true enable_gyro:=true unite_imu_method:=2`

I get the same log 

[INFO] [launch]: All log files can be found below /home/orin/.ros/log/2024-08-19-12-32-47-147291-ubuntu-44601
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [44602]
[realsense2_camera_node-1] [INFO] [1724059967.468494828] [camera.camera]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1724059967.468719289] [camera.camera]: Built with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1724059967.468755797] [camera.camera]: Running with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1724059968.064119495] [camera.camera]: Device with serial number 309122300620 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1724059968.064480264] [camera.camera]: Device with physical ID 2-3.2-6 was found.
[realsense2_camera_node-1] [INFO] [1724059968.064542787] [camera.camera]: Device with name Intel RealSense D456 was found.
[realsense2_camera_node-1] [INFO] [1724059968.065444726] [camera.camera]: Device with port number 2-3.2 was found.
[realsense2_camera_node-1] [INFO] [1724059968.065520207] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1724059968.065841428] [camera.camera]: getParameters...
[realsense2_camera_node-1] [WARN] [1724059968.066731175] [camera.camera]: For the 'unite_imu_method' param update to take effect, re-enable either gyro or accel stream.
[realsense2_camera_node-1] [INFO] [1724059968.066775332] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1724059968.066803201] [camera.camera]: Device Name: Intel RealSense D456
[realsense2_camera_node-1] [INFO] [1724059968.066823807] [camera.camera]: Device Serial No: 309122300620
[realsense2_camera_node-1] [INFO] [1724059968.066841950] [camera.camera]: Device physical port: 2-3.2-6
[realsense2_camera_node-1] [INFO] [1724059968.066859356] [camera.camera]: Device FW version: 5.16.0.1
[realsense2_camera_node-1] [INFO] [1724059968.066878075] [camera.camera]: Device Product ID: 0x0B5C
[realsense2_camera_node-1] [INFO] [1724059968.066892921] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1724059974.897371768] [camera.camera]: Set ROS param depth_module.depth_profile to default: 848x480x30
[realsense2_camera_node-1] [INFO] [1724059974.902093093] [camera.camera]: Set ROS param depth_module.infra_profile to default: 848x480x30
[realsense2_camera_node-1] [INFO] [1724059975.603069789] [camera.camera]: Set ROS param rgb_camera.color_profile to default: 1280x720x30
[realsense2_camera_node-1] [INFO] [1724059975.604472997] [camera.camera]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [INFO] [1724059975.604591419] [camera.camera]: Set ROS param accel_fps to default: 100
[realsense2_camera_node-1] [INFO] [1724059975.609261262] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1724059975.609431007] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1724059975.609486683] [camera.camera]: Stopping Sensor: Motion Module
[realsense2_camera_node-1] [INFO] [1724059975.704216250] [camera.camera]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1724059975.808501131] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-1]  19/08 12:32:55,810 WARNING [281472459860192] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  19/08 12:32:55,821 WARNING [281472459860192] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  19/08 12:32:55,831 WARNING [281472459860192] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  19/08 12:32:55,842 WARNING [281472459860192] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  19/08 12:32:55,853 WARNING [281472459860192] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  19/08 12:32:55,864 WARNING [281472459860192] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  19/08 12:32:55,875 WARNING [281472459860192] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[... cut ...]
[realsense2_camera_node-1]  19/08 12:32:56,464 WARNING [281472459860192] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  19/08 12:32:56,471 ERROR [281472057207008] (uvc-streamer.cpp:105) uvc streamer watchdog triggered on endpoint: 130
[realsense2_camera_node-1]  19/08 12:32:56,478 WARNING [281472459860192] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

MartyG-RealSense, 2024-08-19:
Is there any difference if you reset the camera at launch with **initial_reset:=true**

ros2 launch realsense2_camera rs_launch.py initial_reset:=true enable_accel:=true enable_gyro:=true unite_imu_method:=2

If there is no difference, what happens if you launch the camera with **ros2 run** instead of ros2 launch?

`ros2 run realsense2_camera realsense2_camera_node --ros-args -p enable_color:=false -p spatial_filter.enable:=true -p enable_accel:=true -p enable_gyro:=true -p unite_imu_method:=2`

Al-khattab, 2024-08-19:
there is no difference when I run it with **initial_reset:=true**
`ros2 launch realsense2_camera rs_launch.py initial_reset:=true enable_accel:=true enable_gyro:=true unite_imu_method:=2`

and also got the same result using ros2 run

`ros2 run realsense2_camera realsense2_camera_node --ros-args -p enable_color:=false -p spatial_filter.enable:=true -p enable_accel:=true -p enable_gyro:=true -p unite_imu_method:=2`

MartyG-RealSense, 2024-08-19:
Let's next test whether the **control_transfer returned** problem is being caused by the RGB sensor by turning it off.

` ros2 launch realsense2_camera rs_launch.py enable_color:=false`

Al-khattab, 2024-08-19:
same result after running 

`ros2 launch realsense2_camera rs_launch.py enable_color:=false`

MartyG-RealSense, 2024-08-19:
If the problems started when updating to JetPack 6 because of ROS2, it should be possible to use ROS2 Humble with JetPack 5.  JP6 was only released around May 2024 and Humble has been used by RealSense ROS users for a long time before that.

Al-khattab, 2024-08-19:
to my knowledge, ROS2 humble needs Ubuntu 22 which is only available in JP6 which is the reason I updated, 
if you do not mind sharing any helpful resource on how to run it in JetPack 5 it would really appreciated.

MartyG-RealSense, 2024-08-26:
Hi @Al-khattab  Do you require further assistance with this case, please?  Thanks!

Al-khattab, 2024-08-26:
Hi, I just wanted to update you on the satiation, I have managed to fix the issue after a while of tinkering.
here are the steps I did to make it work.

first I uninstalled all realsense SDK setup that was installed and removed the ROS-wrapper.
then followed the instructions here https://github.com/IntelRealSense/librealsense/issues/12831#issuecomment-2041140925
to setup the back end.

then afterwards I followed your instructions to install MIPI driver, it took me a while to set it up but I managed to install it,
here are the steps that I did, in case anyone is interested

1. go to the MIPI repo to the dev branch and open the [README_JP6.md](https://github.com/IntelRealSense/realsense_mipi_platform_driver/blob/dev/README_JP6.md), and follow the instructions.
2. **On a host machine** setup the environment using `sudo apt-get install -y build-essential bc wget flex bison curl libssl-dev xxd` 
3. **On a host machine** run the following commands
```
git clone https://github.com/IntelRealSense/realsense_mipi_platform_driver.git
cd realsense_mipi_platform_driver
git checkout dev
./setup_workspace.sh 6.0
./apply_patches.sh apply 6.0
./build_all.sh 6.0
```
4. after all the above is installed which took around 40 minutes you need to copy the files from the **Host machine** to the jetson by ssh or a simple usb as follows 

- Copy entire directory `images/6.0/rootfs/lib/modules/5.15.136-tegra/updates` from host to `/lib/modules/5.15.136-tegra/` on Orin target
- Copy entire directory `images/6.0/rootfs/lib/modules/5.15.136-tegra/extra` from host to `/lib/modules/5.15.136-tegra/` on Orin target
- Copy `tegra234-camera-d4xx-overlay.dtbo` from host to` /boot` on Orin target
-  Copy `tegra234-p3737-0000+p3701-0000-nv.dtb` from host to `/boot/` on Orin

- **On jetson** Run  `sudo /opt/nvidia/jetson-io/jetson-io.py`
- select the following options Configure Jetson AGX CSI Connector => Configure for compatible hardware => Jetson RealSense Camera D457
- Run `sudo sed -i 's/search updates/search extra updates/g' /etc/depmod.d/ubuntu.conf`
- Run `sudo depmod`
- Run `echo "d4xx" | sudo tee /etc/modules-load.d/d4xx.conf`
- and finally reboot

after reboot I run the realsense viewer and it worked properly, there were few small errors I needed to fix but I do not think it was related to the installation, I also installed the ros wrapper using both the apt installation and built it from source and both was working fine.

Al-khattab, 2024-08-26:
thank you @MartyG-RealSense for your help, I will be using this installation to test then I am thinking of switching to docker for later development since jp6 is giving me a hard time with other parts of my system.

MartyG-RealSense, 2024-08-26:
Thanks so much @Al-khattab for the update and for sharing your solution in detail  :)

The latest librealsense version 2.56.1 that was recently released officially supports JetPack 6.  A new ROS wrapper 4.56.1 for it has also been released.  So it may be worth trying these versions in future to see whether they improve your JP6 situation.

Al-khattab, 2024-08-27:
yeah saw the librealsense build, it is still a prelease so I will be waiting for the full release and try it, but for now I think the issue is solved so I will close the issue and if I have any updates I will update it here.
thank you for your help

closed 2024-08-27 by Al-khattab