realsenseai/realsense-ros #3187: Depth image is always 320x240 on Jetson Nano

TakShimoda, 2024-08-19:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.13.0.50 | 
| Operating System & Version |   Ubuntu 20.04  | 
| Kernel Version (Linux Only)    | 4.9.253                                         | 
| Platform                                 | NVIDIA Jetson Nano 4GB  |
| SDK Version                            |  2.51.1                         | 
| Language                            |  ROS2                          | 
| Segment			|  Robot                         | 

### Issue Description
I noticed that whenever I launch the realsense node with RGB-D, the depth image is always 320x240 resolution. This is even after setting `depth_module.depth_profile:=640x480x30` as a launch argument. I also tried  `depth_module.profile:=640x480x30` but this didn't work either. My full launch is:

`ros2 launch realsense2_camera rs_launch.py unite_imu_method:=2 enable_sync:=true enable_color:=true enable_accel:=true enable_gyro:=true  gyro_fps:=400 accel_fps:=250 depth_module.depth_profile:=640x480x30 depth_module.infra_profile:=640x480x30 rgb_module.color_profile:=640x480x30 depth_module.emitter_enabled:=0`

- I just added `depth_module.emitter_enabled` as a configurable launch parameter since it originally wasn't there, but I don't think this should affect anything

The strange thing is if I recover the parameter using `ros2 param get /camera/camera depth_module.depth_profile`, it shows the profile is 640x480x30. However, if I topic echo `/camera/camera/depth/camera_info` it says the resolution is 320x240, and I can confirm the depth image is that resolution. All other profiles seem to work (e.g. `rgb_camera.color_profile:=640x480x30 enable_infra1:=true enable_infra2:=true depth_module.infra_profile:=640x480x30` sets both color and infra parameters to 640x480). What's strange is 320x240 isn't even a profile for the depth module returned by `rs-enumerate-devices`.

When I use the same setup on my PC (ubuntu 20.04 firmware, 5.15.0-94-generic, firmware 5.13.0.50, SDK 2.54.2 ROS2 foxy), some things work but some don't. When I launch the same command above, it properly outputs the depth image with 640x480 resolution. However, if I try to change the depth profile to anything else returned by `rs-enumerate-devices`, such as 424x240x3, it will give the warning `XXX Hardware Notification:Frames didn't arrived within 5 seconds,1.7241e+12,Warn,Frames Timeout` and the depth image won't publish. I know it's not an issue with a wrong profile, as a wrong profile will just give an error output. 

I tried a few things to try to fix this:
- Changed some arguments, like `enable_sync` to false or `enable_depth` to true or `depth_module.emitter_enabled` to 0 or 1, but they didn't do anything
- On the Jetson, I thought it might be the memory/CPU cycles being taken up (`top` returned over 80% CPU when running and recording on rosbag) so I disabled topics like accel and gyro, but it didn't change anything.

I was wondering how I can fix this issue. Thanks.

MartyG-RealSense, 2024-08-19:
Hi @TakShimoda  Have you made any edits to the post-processing filter settings in the rs_launch.py file, please?  The depth resolution would usually halve if **decimation_filter** has been set to true instead of its default of false.  This is because the decimation filter 'downsamples' the depth resolution by a default division factor of 2 when enabled, so 640x480 becomes 320x240.

https://github.com/IntelRealSense/realsense-ros/blob/ros2-master/realsense2_camera/launch/rs_launch.py#L78

Also, the rgb_module definition in your launch instruction should be **rgb_camera**.

`rgb_camera.color_profile:=640x480x30`

In regard to the high CPU usage, this can occur on Jetson boards if the librealsense SDK's **CUDA** support has not been enabled.  It can be enabled by building librealsense with the Jetson version of the Debian packages (see the link below) or by compiling librealsense from source code with CMake with the flag **-DBUILD_WITH_CUDA=ON** included in the CMake build instruction.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md#4-install-with-debian-packages

MartyG-RealSense, 2024-08-21:
Intel strongly recommend that Jetson Nano users enable the **barrel jack** power connector using the instructions at the link below.  If your Nano has a barrel jack, have you enabled it please?

https://jetsonhacks.com/2019/04/10/jetson-nano-use-more-power/

 Yes, if you are not using point clouds or alignment then your CPU load percentage would not benefit from CUDA support being enabled and the processing load would be placed fully on the CPU.

Other than lowering your FPS to 15 or 6 (assuming you do not want to reduce the resolution lower than 640x480), there may not be much else you can do to reduce CPU load.

I note that you include `depth_module.infra_profile:=640x480x30` in your launch instruction.  The infra topics are disabled by default, so would not be published unless at least `enable_infra1:=true` was included in the launch instruction or you had edited the launch file to set infra1 to true.

TakShimoda, 2024-08-21:
Hello Marty,

For now, I'm powering the Jetson Nano with 2 GPIO pins at 3A 5V each for 30W as I thought this is the most convenient option. For the barrel jack, since the Jetson Nano's on a small robotics platform, I have to get the battery adapter plate and mount it on the robot which might be hard with the limited space, as well as expensive. Are there any performance downsides to using the GPIO pins over barrel jack connector? The dupont wires I use to connect to the GPIO pins have an AWG of 20 so I know there could be some resistance, although when I measure it it's usually around 5.1V, which should be good. 

I did set `enable_infra1` and `enable_infra2` to true, I think it was already set by default in my launch file.

So other than possibly changing the hardware setup for power, do you recommend the ROS2 components method? I may also have to use the realsense on Raspberry Pi 3B+ which are less powerful to capture ARTags on multiple robots so I was wondering if using the C++ pointers to record bags instead of passing messages over the network can speed things up by reducing network congestion.

Thanks

TakShimoda, 2024-08-21:
For the raspberry pi, I'll probably one need one image stream so I think I can just disable all other streams (e.g. just use color, disable infrared, IMU, etc..)

> If you do not need the infra1 and infra2 streams then having them set to false should reduce the processing burden on the Jetson's CPU. They are not required to be enabled in order for the depth stream to work.
- I am using infrared 1 and 2, along with RGB-D. I just have a question about them not being required for the depth stream to work: I thought depth images were rectified from the infrared cameras? Does this mean if I have enable_infrared1:=false, it won't publish the topic on ROS but will still use the image for depth?

I am recording rosbags, typically from a 1min-1m30sec, and they are about 1.5GB including all images but I make sure to have adequate storage so that's not really an issue.

I will try the zero-copy intra-process method as it has worked well for me with ROS in the past.

MartyG-RealSense, 2024-08-22:
The depth frames are generated from **raw** left and right frames in the camera hardware before the data is sent along the USB cable to the computer.  These raw infrared frames are not the same as the infrared1 and infrared2 streams, which is why depth frames can be produced when the publishing in ROS of the infrared1 and infrared2 topics is disabled.

TakShimoda, 2024-08-22:
Thanks Marty, I think all the questions here are answered. I'll open another issue if I encounter anything with implementing components.

closed 2024-08-22 by TakShimoda

MartyG-RealSense, 2024-08-22:
You are very welcome.  Thanks very much for the update and good luck!