realsenseai/realsense-ros #3191: failed to send response to /camera/camera/set_parameters (timeout): client will not receive response

akupferb, 2024-08-21:
| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D435 }                          | 
| Firmware Version                   | 05.13.00.50   | 
| Operating System & Version |   Ubuntu 22  | 
| Kernel Version (Linux Only)    |  5.15.136-tegra    | 
| Platform                                 |  NVIDIA Jetson Orin NX - Docker         |
| Librealsense SDK Version       |    v2.55.1   | 
| Segment			         |  Robot   | 
| ROS Distro			         |  Humble      | 
| RealSense ROS Wrapper Version   | v4.55.1   | 

### Issue Description
While in a Docker container, and running the following ROS2 commands from a command line script:
```
ros2 param set $CAMERA_NAME enable_infra1 false
ros2 param set $CAMERA_NAME enable_infra2 false

ros2 param set $CAMERA_NAME depth_module.infra_profile $RESOLUTION
ros2 param set $CAMERA_NAME enable_infra1 true
ros2 param set $CAMERA_NAME enable_infra2 true
```
I occasionally get the error response to one of the `param set` commands (not necessarily the first command, sometimes a couple work and then I get the error):
`[camera.camera.rclcpp]: failed to send response to /camera/camera/set_parameters (timeout): client will not receive response, at ./src/rmw_response.cpp:154, at ./src/rcl/service.c:314`

Why is this error sometimes occurring during runtime, especially when previous calls to the `set_parameters' service via `param set` have worked?

MartyG-RealSense, 2024-08-21:
Hi @akupferb  Did you install the librealsense SDK from source code using the libuvc backend or RSUSB backend methods?  If you did then a characteristic of this install method is that it checks for changes less often than a package installation of the SDK or a non-RSUSB source code build that has had a patch applied to its kernel (a build type known as the V4L or native backend).   Because libuvc / RSUSB versions of librealsense check for changes less often, an option change request may sometimes fail and have to be re-tried.

akupferb, 2024-08-21:
In this instance of the system the librealsense SDK has been installed via debian packages.

MartyG-RealSense, 2024-08-21:
Have you tried commenting out `depth_module.infra_profile $RESOLUTION` to test whether it is this change instruction or the infra ones that are timing-out?

akupferb, 2024-08-21:
It occurs inconsistently, sometimes on one of the 'set false' commands and sometimes on the resolution change. I just had a repeat occurrence of the error on the very first call.

MartyG-RealSense, 2024-08-21:
Does stability improve if you add a sleep command before each instruction?  For example, to wait for 1 second before initiating each command:

```
import time

time.sleep(1)
ros2 param set $CAMERA_NAME enable_infra1 false
time.sleep(1)
ros2 param set $CAMERA_NAME enable_infra2 false
time.sleep(1)
ros2 param set $CAMERA_NAME depth_module.infra_profile $RESOLUTION
time.sleep(1)
ros2 param set $CAMERA_NAME enable_infra1 true
time.sleep(1)
ros2 param set $CAMERA_NAME enable_infra2 true
```

closed 2024-09-05 by MartyG-RealSense