realsenseai/realsense-ros #3192: ros loading d405 json doesn't work

Nanmei, 2024-08-27:
| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | { D405  }                          | 
| Firmware Version                   | (Open RealSense Viewer --> Click info)                          | 
| Operating System & Version |   {  /Linux (Ubuntu 18/)  } | 
| Kernel Version (Linux Only)    |  (e.g. 5.4)                                                                       | 
| Platform                                 | PC / etc..                           |
| Librealsense SDK Version       |  { 2.<?>.<?> }                                                    | 
| Language                               |  { }         | 
| Segment			         |  {}                             | 
| ROS Distro			         |  {Melodic/etc.. }                                           | 
| RealSense ROS Wrapper Version   |  {4.51.1, 4.54.1, etc..}                                                  | 

This is the default parameter for D405 in realsense view
[image]

This is my modified parameter and saved as JSON
[image]

this is json file:
{
    "device": {
        "fw version": "5.15.1",
        "name": "Intel RealSense D405",
        "product line": "D400"
    },
    "parameters": {
        "aux-param-autoexposure-setpoint": "1000",
        "aux-param-colorcorrection1": "-0.0556641",
        "aux-param-colorcorrection10": "-0.194336",
        "aux-param-colorcorrection11": "-0.194336",
        "aux-param-colorcorrection12": "-0.589844",
        "aux-param-colorcorrection2": "0.560547",
        "aux-param-colorcorrection3": "0.560547",
        "aux-param-colorcorrection4": "0.170898",
        "aux-param-colorcorrection5": "-0.275391",
        "aux-param-colorcorrection6": "-0.238281",
        "aux-param-colorcorrection7": "-0.238281",
        "aux-param-colorcorrection8": "1.34766",
        "aux-param-colorcorrection9": "0.959961",
        "aux-param-depthclampmax": "65536",
        "aux-param-depthclampmin": "0",
        "aux-param-disparityshift": "0",
        "controls-autoexposure-auto": "True",
        "controls-autoexposure-manual": "33000",
        "controls-depth-gain": "16",
        "controls-depth-white-balance-auto": "True",
        "ignoreSAD": "0",
        "param-amplitude-factor": "0",
        "param-autoexposure-setpoint": "1000",
        "param-censusenablereg-udiameter": "9",
        "param-censusenablereg-vdiameter": "9",
        "param-censususize": "9",
        "param-censusvsize": "9",
        "param-depthclampmax": "65536",
        "param-depthclampmin": "0",
        "param-depthunits": "1000",
        "param-disableraucolor": "0",
        "param-disablesadcolor": "0",
        "param-disablesadnormalize": "0",
        "param-disablesloleftcolor": "0",
        "param-disableslorightcolor": "1",
        "param-disparitymode": "0",
        "param-disparityshift": "0",
        "param-lambdaad": "935",
        "param-lambdacensus": "26",
        "param-leftrightthreshold": "19",
        "param-maxscorethreshb": "887",
        "param-medianthreshold": "1021",
        "param-minscorethresha": "54",
[... 193 lines cut ...]
      </node>
    </group>

    <group if="$(eval depth_registered_processing and hw_registered_processing)">
      
      <node pkg="nodelet" type="nodelet" name="points_xyzrgb_hw_registered"
            args="load depth_image_proc/point_cloud_xyzrgb $(arg manager) $(arg bond)" respawn="$(arg respawn)">
        <remap from="rgb/image_rect_color"        to="$(arg rgb)/image_rect_color" />
        <remap from="rgb/camera_info"             to="$(arg rgb)/camera_info" />
        <remap from="depth_registered/image_rect" to="$(arg depth_registered)/image_raw" />
        <remap from="depth_registered/points"     to="$(arg depth_registered_pub)/points" />
      </node>
    </group>

  </group>

</launch>

I started the launth file and found in rqt that the parameters I just modified did not take effect

[image]

[image]

I need your help ,thanks!

MartyG-RealSense, 2024-08-27:
Hi @Nanmei  From your RealSense Viewer images, it looks as though you are modifying RGB brightness, contrast, gamma, saturation and sharpness.  The RealSense Viewer's json save tool does not export all of the settings to json though.

Please try adding the lines below with your customized values to your json file if they are not in it already.

```
  "controls-color-brightness": "-52",
    "controls-color-contrast": "0",
    "controls-color-gamma": "100",
    "controls-color-saturation": "100",
    "controls-color-sharpness": "100",
```

MartyG-RealSense, 2024-08-27:
When programming scripts for D405, RGB features can be controlled by accessing the depth sensor instead of the RGB sensor.  You may be able to do something similar with launch file edits to configure the RGB settings.  For example, instead of:

```
<rosparam>
 /camera/rgb_camera/brightness: 100
</rosparam>
```

Use:

```
<rosparam>
 /camera/stereo_module/brightness: 100
</rosparam>
```

Nanmei, 2024-08-28:
> 在为 D405 编写脚本时，可以通过访问深度传感器而不是 RGB 传感器来控制 RGB 功能。您可能能够对启动文件编辑执行类似操作来配置 RGB 设置。例如，而不是：
> 
> ```
> <rosparam>
>  /camera/rgb_camera/brightness: 100
> </rosparam>
> ```
> 
> 使用：
> 
> ```
> <rosparam>
>  /camera/stereo_module/brightness: 100
> </rosparam>
> ```

set stereo_module ,it can works,thanks very much.
But I found another issue. I have two d405 cameras, and when both of them start up at the same time, the second camera displays "Device 1/2 failed with exception: failed to set power state". I don't know why it displays like this.

[image]

MartyG-RealSense, 2024-08-28:
Mutiple cameras should be launched in separate terminals, or with the ROS1 launch file **rs_multiple_devices.launch** instead of rs_camera.

https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy?tab=readme-ov-file#work-with-multiple-cameras

closed 2024-09-02 by Nanmei