realsenseai/realsense-ros #3193: D435i Could not load library dlopen error: libwebp.so.6

Somethingout, 2024-08-27:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | { D435i  }                          | 
| Firmware Version                   | (Open RealSense Viewer --> Click info)                          | 
| Operating System & Version |   { Linux (Ubuntu 22)  } | 
| Kernel Version (Linux Only)    |  (e.g. 5.4)                                                                       | 
| Platform                                 | PC                           |
| Librealsense SDK Version       |  { 2.55.1 }                                                    | 
| Language                               |  {C/C#/labview/opencv/pcl/python/unity }         | 
| Segment			         |  {Robot/Smartphone/VR/AR/others }                             | 
| ROS Distro			         |  {Humble }                                           | 
| RealSense ROS Wrapper Version   |  {4.51.1, 4.54.1, etc..}                                                  | 

### Issue Description
I installed the sdk and cloned the ros2 wrapper packges into my workspace but when I try to start `ros2 run realsense2_camera realsense2_camera_node` I get this error: `terminate called after throwing an instance of 'class_loader::LibraryLoadException'
  what():  Could not load library dlopen error: libwebp.so.6: cannot open shared object file: No such file or directory, at ./src/shared_library.c:99
[ros2run]: Aborted
`
When i try to look for the debian package libwebp6 I don't find it, there is  libwebp7 but it doesnt solve the issue.

Note that the camera works fine with realsense-viewer.

Any idea on how to fix this issue?

Thanks.

MartyG-RealSense, 2024-08-27:
Hi @Somethingout  This precise error with **libwebp.so** has not been previously reported in relation to RealSense.

Which librealsense SDK version did you install, please?  

If you git cloned the ROS2 wrapper then I can imagine that the latest **4.56.1** wrapper version may have been installed, which is designed for use with the current latest 'interim beta' version **2.56.1** version of librealsense.

If your librealsense version is 2.55.1 then the ROS2 wrapper designed for that is **4.55.1**.

*************

Do you experience the same problem if you use the launch file instead of ros2 run?

`ros2 launch realsense2_camera rs_launch.py`

Somethingout, 2024-08-28:
Hi @MartyG-RealSense  I installed librealsense2 debian package from Intel servers (option 1) [from this README](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages) which resulted in a librealsense SDK 2.55.1.

The version specified in the XML file of the ROS2 packages is 4.55.1 (I followed option2). and when I open the realsense-viewer it is written V2.55.1 

Yes I have the same error with the launch file `ros2 launch realsense2_camera rs_launch.py`

Somethingout, 2024-08-28:
Problem solved by creating a symbolic link with the version installed in my computer. Thank you for your help!

MartyG-RealSense, 2024-08-28:
You are very welcome!

I was about to say that the most similar error in a past case to the one that you reported is at https://github.com/IntelRealSense/librealsense/issues/11918#issuecomment-1682878232 and the RealSense ROS user in that case solved it by using the command `colcon build --packages-select realsense2_camera` to update links.

closed 2024-08-28 by Somethingout