realsenseai/realsense-ros #3194: About how to set the resolution of /aligned_depth_to_color/image_raw

Poodlee, 2024-08-29:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking our ROS RealSense Wrapper documentation [README](https://github.com/IntelRealSense/realsense-ros/blob/ros2-master/README.md).
    * Have you looked in our [Discussions](https://github.com/IntelRealSense/realsense-ros/discussions)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/realsense-ros/issues) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | { D455 }                          | 
| Firmware Version                   | (5.13.0.50)                          | 
| Operating System & Version |   { Ubuntu 22.04} | 
| Kernel Version (Linux Only)    |  (6.8.0-40-generic)                                                                       | 
| Platform                                 | PC                           |
| Librealsense SDK Version       |  { 2.55.1}                                                    | 
| Language                               |  {C/python }         | 
| Segment			         |  {Robot }                             | 
| ROS Distro			         |  {Humble}                                           | 
| RealSense ROS Wrapper Version   |  {4.51.1}                                                  | 

### Issue Description
<Describe your issue / question / feature request / etc..>
Currently, realsense-ros is used within the isaac ros related package. I am writing this post because I encountered a problem with settings related to /aligned_depth_to_color/image_raw resolution during related operations.

In the current situation, a new launch file is created and the realsenseNodeFactory node is configured and executed as follows.
`
    def get_composable_nodes(interface_specs: Dict[str, Any]) -> Dict[str, ComposableNode]:
        realsense_config_file_path = os.path.join(
            get_package_share_directory('isaac_ros_realsense'),
            'config', 'realsense_mono_depth.yaml'
        )

        return {
            'camera_node': ComposableNode(
                package='realsense2_camera',
                plugin='realsense2_camera::RealSenseNodeFactory',
                name='realsense2_camera',
                namespace='',
                parameters=[
                    realsense_config_file_path
                ],
                remappings=[('color/image_raw', 'image_rect'),
                            ('color/camera_info', 'camera_info_rect')]
            ),
`

At this time, the realsense_mono_depth.yaml code is structured as follows.
`
enable_infra1: false
enable_infra2: false
align_depth.enable: true
depth_module.depth_profile: 640x480x15
rgb_camera:
  profile: '640x480x15'
  color_qos: "SYSTEM_DEFAULT"
`
[image]
I configured it as above and ran launch. After execution, if I check aligned_depth_to_color/image_raw through ros2 topic echo, the resolution result will be 1280x720. I thought I definitely changed it to 640x480x15 in the realsense_mono_depth.yaml file, but it doesn't apply.

I would appreciate it if you could let me know if there is anything I am wrong about or needs to be corrected.

MartyG-RealSense, 2024-08-29:
Hi @Poodlee  1280x720 is the default resolution of RGB color on the D455 camera model.  It is possible that your custom configuration was deemed invalid by the launch and the default stream profile was applied instead.

Please try changing your yaml file to use the color stream definition below.

`rgb_camera.color_profile: 640x480x15`

Poodlee, 2024-08-29:
@MartyG-RealSense  Thank you for quick reply. As you said, the yaml file was modified as follows:
[
[image]
](url)

Based on what you just said, are you saying that it is impossible to change the resolution of the aligned_depth_to_color/image_raw topic? The desired resolution for the package currently in use is 640x480, so if that is not possible, the package needs to be modified, so I am asking.

Poodlee, 2024-08-29:
[image]
After editing the .yaml file, I confirmed that the resolution did not change through ros2 topic echo. 
[image]

Since no settings have been made regarding log, only the following information is listed.

If there is anything I am lacking regarding this topic, please let me know and I will learn. thank you

MartyG-RealSense, 2024-08-29:
Ideally, ROS wrapper 4.55.1 should be used with librealsense SDK 2.55.1, as 4.51.1 is designed for use with the old 2.51.1 librealsense version.

Let's try setting the depth and color resolution in the launch instruction instead of yaml to see whether it changes the aligned resolution.

`ros2 launch realsense2_camera rs_launch.py depth_module.profile:=640x480x15 rgb_camera.color_profile:=640x480x15`

MartyG-RealSense, 2024-08-29:
The problem might be related to the use of 4.51.1 as I believe that the resolution profiles were defined a little differently in that old wrapper version.

`ros2 launch realsense2_camera rs_launch.py depth_module.profile:=640x480x15 rgb_camera.profile:=640x480x15`

Poodlee, 2024-08-29:
[image]

I confirmed that it works properly when I run rgb_camera.profile:=640x480x15 as you said. thank you. 

So, what part of the launch file I configured is causing the problem?

MartyG-RealSense, 2024-08-29:
Please try editing **depth_module.depth_profile** to be `depth_module.profile` and change **rgb_camera.color_profile** back to `rgb_camera.profile`

Poodlee, 2024-08-29:
[image]

[image]
I modified the .yaml file as you suggested and ran the original launch file again. As a result, the resolution was set to 1280x720 again. Is there an error in the node setup in launch.py?

MartyG-RealSense, 2024-08-29:
A yaml file imports settings into the launch so that they do not have to be defined in the launch file or in a launch command.  So the problem may be that a setting or settings is being imported from the yaml that is invalid and causes the launch to use the default resolution instead.

Poodlee, 2024-08-31:
Thank you. I have finally solved the problem. After investigating, I found that there was a part in the launch file that was configured to use the newly installed realsense-ros as a binary. Therefore, even after I modified the source code, it continued to run the binary version.

After removing the binary installation procedure, it is now working properly. Thank you.

closed 2024-08-31 by Poodlee

MartyG-RealSense, 2024-08-31:
You are very welcome, @Poodlee - it's great to hear that you found the solution.  Thanks very much for the update!