realsenseai/realsense-ros #3195: Angular Velocities Output goes wrong with D435i & D455 Camera

HappySamuel, 2024-08-30:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | { D435i / D455 }                          | 
| Firmware Version                   | 5.16.0.1                          | 
| Operating System & Version |  JetPack 5.1.3 (Ubuntu 20.04)  | 
| Kernel Version (Linux Only)    |  5.10.192-tegra                                                                       | 
| Platform                                 |  NVIDIA Jetson AGX Orin                           |
| Librealsense SDK Version       |  2.54.1                                                     | 
| Language                               |  ROS2        | 
| Segment			         |  Robot                             | 
| ROS Distro			         |  Humble                                           | 
| RealSense ROS Wrapper Version   |  4.54.1                                                  | 

### Issue Description
I have the realsense-ros installed following [Realsense Camera with CUDA-enabled](https://github.com/IntelRealSense/realsense-ros/issues/3188#issuecomment-2306824690) steps suggested here. However, when i try to enable both the gyro and accel, and having the `unite_imu_method=1 / 2` , which will output the imu topic `/front_cam/imu` . Although the camera is sitting still, for the accelerations portion, they are seems OK, but the angular velocities are awful, keep dramatically jumping around. For your info, the camera is sitting still. I tried both d435i and d455 cameras, both their angular velocities are unsable at all. Here i attached the bagfiles of both 435i and d455 cameras

[d435i_rs_imu.zip](https://github.com/user-attachments/files/16809424/d435i_rs_imu.zip)

[d455_rs_imu.zip](https://github.com/user-attachments/files/16809433/d455_rs_imu.zip)

Here are the config i used for both d435i and d455 cameras. I only enable gyro and accel at the moment with else disabled. I also have enabled the `enable_motion_correction`

```
rgb_camera:
  profile: '1280x720x15'
  global_time_enabled: true
  enable_auto_exposure: true
  enable_auto_white_balance: true
  power_line_frequency: 2
color_info_qos: "SENSOR_DATA"
color_qos: "SENSOR_DATA"

depth_module:
  profile: '640x360x60'
  global_time_enabled: true
  enable_auto_exposure: true
  hdr_enabled: false
  emitter_always_on: true
  emitter_enabled: 1  #0=OFF | 1=enable laser | 2=enable laser auto
  emitter_on_off: false

infra1_qos: "SENSOR_DATA"
infra1_info_qos: "SENSOR_DATA"
infra2_qos: "SENSOR_DATA"
infra2_info_qos: "SENSOR_DATA"

motion_module:
  global_time_enabled: true
  enable_motion_correction: true
  frames_queue_size: 16
gyro_fps: 200
accel_fps: 63
unite_imu_method: 2  #0=none | 1=copy | 2=linear_interpolation

enable_color: false
enable_depth: false
enable_infra1: false
enable_infra2: false
enable_sync: false
enable_accel: true
enable_gyro: true

pointcloud:
  enable: false
  filter_magnitude: 2  #1=OFF | 2=ON
  ordered_pc: false
  pointcloud_qos: "SENSOR_DATA"

decimation_filter:
  enable: false
  filter_magnitude: 4

clip_distance: 3.0
```

MartyG-RealSense, 2024-08-30:
Hi @HappySamuel  Does gyro stability improve if you set **global_time_enabled** for the motion module to false, please?  Having global time enabled is less important in ROS than it is in the RealSense Viewer because ROS has its own timing sync mechanism.

HappySamuel, 2024-08-30:
Hi @MartyG-RealSense 

I just tried to disable the `global_time_enabled` , however the gyro outcome is still dramastically jumping around for all 3 axes.

Besides, i checked my D435i camera for their available FPS for both gyro and accel via `realsense-viewer`
```
Gyro: 200 / 400
Accel: 63 / 250
```

What shall i do next?

MartyG-RealSense, 2024-08-30:
How does gyro behave if you set **enable_sync** to true?

HappySamuel, 2024-08-30:
I tried as per you suggest, having the `enable_sync:=true`, but the gyro outcome is still dramastically jumping around for all 3 axes.

MartyG-RealSense, 2024-08-30:
There is a similar sounding ROS case at https://github.com/IntelRealSense/realsense-ros/issues/2745 where the gyro was fluctuating continuously and using motion correction or setting unite_imu_method top copy made no difference.  

Eventually they described at https://github.com/IntelRealSense/realsense-ros/issues/2745#issuecomment-1571241211 how they minimized rotation with something called **/cmd_vel**.  Apparently it is a command used with robots such as turtlebots though, so may not be applicable to how you are using the camera.

http://wiki.ros.org/Robots/TIAGo/Tutorials/motions/cmd_vel

MartyG-RealSense, 2024-09-05:
Hi @HappySamuel  Do you require further assistance with this case, please?   Thanks!

HappySamuel, 2024-09-10:
Hi @MartyG-RealSense
 
Sorry for the late reply. I will look forward for the next upgrade on the realsense Firmware, perhaps the issue will resolve when it comes.

closed 2024-09-10 by HappySamuel

MartyG-RealSense, 2024-09-10:
It's no problem at all.  Thanks very much for the update!