realsenseai/realsense-ros #3197: /sbin/ldconfig.real: Can't create temporary cache file /etc/ld.so.cache~: Permission denied when catkin build.

rslim97, 2024-09-04:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking our ROS RealSense Wrapper documentation [README](https://github.com/IntelRealSense/realsense-ros/blob/ros2-master/README.md).
    * Have you looked in our [Discussions](https://github.com/IntelRealSense/realsense-ros/discussions)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/realsense-ros/issues) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D435i                        | 
| Firmware Version                   | (Open RealSense Viewer --> Click info)                          | 
| Operating System & Version |   Ubuntu 20
| Kernel Version (Linux Only)    |  5.15                                                                      | 
| Platform                                 | PC                           |
| Librealsense SDK Version       |  2.3.2                                                    | 
| Language                               |  C++        | 
| Segment			         |  Robot                            | 
| ROS Distro			         |  Noetic                                           | 
| RealSense ROS Wrapper Version   |  ROS1-Legacy                                                  | 

### Issue Description
Hi, I have just installed librealsense from source where I have followed the instruction in [https://github.com/IntelRealSense/librealsense/issues/11572#issuecomment-1473776741](url) to skip the kernel patching and finally using `make -j$(($(nproc)-1)) install` to install.

I've verified the librealsense installation by running `realsense-viewer` and indeed I can see the viewer window pop up. 

However when I go on to build my ros workspace using `catkin build realsense2_camera` during catkin build, librealsense2 was also being built as a dependency. There is the following error:

```
> $ catkin build realsense2_camera
> 
> ------------------------------------------------------------------------
> Profile:                     default
> Extending:             [env] /opt/ros/noetic
> Workspace:                   /home/rslim/proj/rslim/realsense_ws
> ------------------------------------------------------------------------
> Build Space:        [exists] /home/rslim/proj/rslim/realsense_ws/build
> Devel Space:        [exists] /home/rslim/proj/rslim/realsense_ws/devel
> Install Space:      [unused] /home/rslim/proj/rslim/realsense_ws/install
> Log Space:         [missing] /home/rslim/proj/rslim/realsense_ws/logs
> Source Space:       [exists] /home/rslim/proj/rslim/realsense_ws/src
> DESTDIR:            [unused] None
> ------------------------------------------------------------------------
> Devel Space Layout:          linked
> Install Space Layout:        None
> ------------------------------------------------------------------------
> Additional CMake Args:       None
> Additional Make Args:        None
> Additional catkin Make Args: None
> Internal Make Job Server:    True
> Cache Job Environments:      False
> ------------------------------------------------------------------------
> Buildlisted Packages:        None
> Skiplisted Packages:         None
> ------------------------------------------------------------------------
> Workspace configuration appears valid.
> 
> NOTE: Forcing CMake to run for each package.
> ------------------------------------------------------------------------
> [build] Found 4 packages in 0.0 seconds.                                                                      
> [build] Updating package table.                                                                               
> Starting  >>> catkin_tools_prebuild                                                                           
> Finished  <<< catkin_tools_prebuild                [ 1.5 seconds ]                                            
> Starting  >>> ddynamic_reconfigure                                                                            
> Starting  >>> librealsense2                                                                                   
> Finished  <<< ddynamic_reconfigure                 [ 8.6 seconds ]                                            
> ______________________________________________________________________________________________________________
> Warnings   << librealsense2:cmake /home/rslim/proj/rslim/realsense_ws/logs/librealsense2/build.cmake.000.log  
[... 58 lines cut ...]
> /sbin/ldconfig.real: Can't create temporary cache file /etc/ld.so.cache~: Permission denied
> CMake Error at tools/realsense-viewer/cmake_install.cmake:69 (file):
>   file INSTALL cannot set permissions on
>   "/home/rslim/Documents/librealsense2/presets": Operation not permitted.
> Call Stack (most recent call first):
>   tools/cmake_install.cmake:50 (include)
>   cmake_install.cmake:133 (include)
> 
> 
> make: *** [Makefile:74: install] Error 1
> cd /home/rslim/proj/rslim/realsense_ws/build/librealsense2; catkin build --get-env librealsense2 | catkin env -si  /usr/bin/make install; cd -
> 
> ..............................................................................................................
> Failed     << librealsense2:install                [ Exited with code 2 ]                                     
> Failed    <<< librealsense2                        [ 2 minutes and 31.7 seconds ]                             
> Abandoned <<< realsense2_camera                    [ Unrelated job failed ]                                   
> [build] Summary: 2 of 4 packages succeeded.                                                                   
> [build]   Ignored:   1 packages were skipped or are skiplisted.                                               
> [build]   Warnings:  1 packages succeeded with warnings.                                                      
> [build]   Abandoned: 1 packages were abandoned.                                                               
> [build]   Failed:    1 packages failed.                                                                       
> [build] Runtime: 2 minutes and 33.3 seconds total.                                                            
> [build] Note: Workspace packages have changed, please re-source setup files to use them.

```

MartyG-RealSense, 2024-09-04:
Hi @rslim97  Using **catkin build realsense2_camera** is not the recommended method for building the ROS1 wrapper from source code.  The correct procedure can be found at the link below, and this method will install only the 2.3.2 version of the ROS1 wrapper and not the librealsense SDK.

https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy?tab=readme-ov-file#step-2-install-intel-realsense-ros-from-sources

*******

The simplest method of installing both the librealsense SDK and the ROS wrapper at the same time is to do so from packages instead of source code, using the instruction below.

`sudo apt-get install ros-$ROS_DISTRO-realsense2-camera`

This installation method is also based on the RSUSB backend and so should also bypass the kernel.  However, this method does not include the realsense-viewer tool in the installation, so if you need it then you should install librealsense first and then build the ROS1 wrapper from source code afterwards using the instructions in the link above.

rslim97, 2024-09-05:
Hi @MartyG-RealSense thank you for your help, I've manage to install the librealsense SDK and the ros realsense package using catkin_make. Apparently the librealsense directory should not be inside the src folder for the ros package or should be excluded from being built during catkin_make. Would you mind to elaborate why catkin_make is preferred over catkin build? and why it did it work?

MartyG-RealSense, 2024-09-05:
The catkin workspace folder - `/catkin_ws/src` - should only contain RealSense data.

An approach I have seen taken by a RealSense user who needed to store non-RealSense data in the catkin workspace was to make a separate /catkin_ws at a different path.

https://github.com/IntelRealSense/realsense-ros/issues/1432 has a discussion about catkin_build versus catkin_make with the Intel RealSense developer who created the ROS1 wrapper.

rslim97, 2024-09-06:
Thank you very much for your help and for the link to the discussion. 

Closing this issue as resolved.

closed 2024-09-06 by rslim97

MartyG-RealSense, 2024-09-06:
You are very welcome, @rslim97 - I'm pleased that I could help.  Thanks very much for the update!