realsenseai/realsense-ros #3201: realsense d435i: given stream configuration is not unsupported by the device!

MingHe98, 2024-09-09:
| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | {D435i}                          | 
| Firmware Version                   | (5.16.0.1)                          | 
| Operating System & Version |   { Ubuntu 20.04} | 
| Kernel Version (Linux Only)    |  (5.10)                                                                       | 
| Platform                                 | NVIDIA Jetson                      |
| Librealsense SDK Version       |  { 2.55.1 }                                                    | 
| ROS Distro			         |  {foxy}                                           |

### Issue Description
I enabled infra1 and infra2, and also manually added enable_infra (tested without adding enable_infra as well), but the topics that are related to either infra1 or infra2 could not be published. The error says it's not supported by the device. I saw another thread saying setting infra-rgb to be false solved problem, but it did not work for my case.
[image]

[image]

[image]

[image]

MartyG-RealSense, 2024-09-09:
Hi @MingHe98  If enable_infra1 and enable_infra2 are true then **enable_infra** and **infra_rgb** should both be set to false in order to stream the infra1 and infra2 topics.  

However, the **legacy** ROS2 wrapper version 3.2.3 is very old and is obsolete.  It should be used with librealsense version **2.50.0** and camera firmware version **5.13.0.50**.  Are you using legacy wrapper 3.2.3 because of its support for Foxy, please?

MingHe98, 2024-09-10:
Yes, I manually set enable_infra and infra_rgb to be false. I believe I tested with librealsense version 2.51 with camera firmware version 5.13.0.50 but it did not work either. I upgraded to librealsense 2.55 because I could just download, install, and check via apt-get instead of more stuff to deal with source code folder. Yes, I am using legacy wrapper 3.2.3 because of its support for Foxy, and Jetson AGX Orin, but I'm having a hard time to have all the data published so that I could do VINS

MartyG-RealSense, 2024-09-10:
The current 4-series of ROS2 wrappers also supported Foxy up until version 4.54.1, which was designed for use with librealsense 2.54.1.  The link below is a "forked" version of the 4.54.1 wrapper that a RealSense ROS user created shortly before Foxy support was removed, so it should work for Foxy if you build it from source code.

https://github.com/MichaelKerley/realsense-ros

MingHe98, 2024-09-10:
It still did not publish the topics that are related to infra1 and infra2. Instead, I updated the wrapper and the librealsense SDK. I built the source from "Use the RSUSB backend without the kernel patching".

`xtechlab@xtechlab-desktop:~/ros2_ws$ ros2 launch realsense2_camera rs_launch.py 
[INFO] [launch]: All log files can be found below /home/xtechlab/.ros/log/2024-09-10-13-54-38-591566-xtechlab-desktop-204953
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [204960]
[realsense2_camera_node-1] [INFO] [1725990878.933919410] [camera.camera]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1725990878.934122419] [camera.camera]: Built with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1725990878.934158963] [camera.camera]: Running with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1725990879.617780702] [camera.camera]: Device with serial number 317222074967 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1725990879.617978336] [camera.camera]: Device with physical ID 2-3.4-12 was found.
[realsense2_camera_node-1] [INFO] [1725990879.618032160] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1725990879.619004550] [camera.camera]: Device with port number 2-3.4 was found.
[realsense2_camera_node-1] [INFO] [1725990879.619079207] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1725990879.619391657] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1725990879.620458096] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1725990879.620511120] [camera.camera]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1725990879.620539440] [camera.camera]: Device Serial No: 317222074967
[realsense2_camera_node-1] [INFO] [1725990879.620558192] [camera.camera]: Device physical port: 2-3.4-12
[realsense2_camera_node-1] [INFO] [1725990879.620574352] [camera.camera]: Device FW version: 5.16.0.1
[realsense2_camera_node-1] [INFO] [1725990879.620591440] [camera.camera]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1725990879.620606609] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1725990883.710355171] [camera.camera]: Set ROS param depth_module.depth_profile to default: 848x480x30
[realsense2_camera_node-1] [INFO] [1725990883.711540202] [camera.camera]: Set ROS param depth_module.infra_profile to default: 848x480x30
[realsense2_camera_node-1] [INFO] [1725990884.358863210] [camera.camera]: Set ROS param rgb_camera.color_profile to default: 1280x720x30
[realsense2_camera_node-1] [INFO] [1725990884.360058961] [camera.camera]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [INFO] [1725990884.360185106] [camera.camera]: Set ROS param accel_fps to default: 100
[realsense2_camera_node-1] [INFO] [1725990884.364119212] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1725990884.364318573] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1725990884.457641577] [camera.camera]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1725990884.531862570] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-1]  10/09 13:54:44,533 WARNING [281472544000416] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
[realsense2_camera_node-1] [INFO] [1725990884.534131192] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1]  10/09 13:54:44,544 WARNING [281472544000416] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
[realsense2_camera_node-1] [INFO] [1725990884.557233518] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 30
[realsense2_camera_node-1] [INFO] [1725990884.559562973] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1]  10/09 13:54:44,860 WARNING [281472544000416] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
[realsense2_camera_node-1]  10/09 13:54:44,871 WARNING [281472544000416] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
[realsense2_camera_node-1]  10/09 13:54:44,882 WARNING [281472544000416] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0`

If I added the configuration when launching the file, then there would be so many errors repeating to pop out.
`xtechlab@xtechlab-desktop:~/ros2_ws$ ros2 launch realsense2_camera rs_launch.py enable_gyro:=true enable_accel:=true unite_imu_method:=2
[INFO] [launch]: All log files can be found below /home/xtechlab/.ros/log/2024-09-10-14-01-44-026063-xtechlab-desktop-205729
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [205731]
[realsense2_camera_node-1] [INFO] [1725991304.707901399] [camera.camera]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1725991304.708117560] [camera.camera]: Built with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1725991304.708153464] [camera.camera]: Running with LibRealSense v2.55.1
[... cut ...]
[realsense2_camera_node-1]  10/09 14:01:51,724 WARNING [281472594323872] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  10/09 14:01:51,730 WARNING [281472594323872] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  10/09 14:01:51,735 ERROR [281472611109280] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: get_xu(ctrl=1) failed! Last Error: Success`

And since there is error, even though the topic /camera/camera/imu appears but echo nothing.