realsenseai/realsense-ros #3203: JetsonPack6 Ros Humble Opencv 4.8

djaniel, 2024-09-16:
Hello, 
I have been facing a weird issue, here is my setup:

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       |  D435i                           | 
| Firmware Version                   | 5.16.0.1                          | 
| Operating System & Version |   Ubuntu 22, JetsonPack 6.0 | 
| Kernel Version (Linux Only)    |  5.15.136-rt-tegra                                                                     | 
| Platform                                 | NVIDIA Jetson Orin Nano                          |
| Librealsense SDK Version       |  2.56.1?                                                  | 
| Language                               |  opencv         | 
| Segment			         |  Robot                            | 
| ROS Distro			         |  Humble                                          | 
| RealSense ROS Wrapper Version   |  4.55.1, 4.56.1   FORCE_LIBUVC=true                                            | 

### Issue Description

I am working with the Jetson Nano Orin, JetsonPack 6. When I try to build from source the `realsense-ros` wrapper I get a weird message error:

```bash
Starting >>> realsense2_camera
--- stderr: realsense2_camera                         
CMake Error at /usr/lib/cmake/opencv4/OpenCVModules.cmake:172 (message):
  The imported target "opencv_core" references the file

     "/usr/lib/libopencv_core.so.4.8.0"

  but this file does not exist.  Possible reasons include:

  * The file was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and contained

     "/usr/lib/cmake/opencv4/OpenCVModules.cmake"

  but not all the files it references.

Call Stack (most recent call first):
  /usr/lib/cmake/opencv4/OpenCVConfig.cmake:126 (include)
  /opt/ros/humble/share/cv_bridge/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /opt/ros/humble/share/cv_bridge/cmake/cv_bridgeConfig.cmake:41 (include)
  CMakeLists.txt:107 (find_package)

---
Failed   <<< realsense2_camera [0.57s, exited with code 1]
``` 

I get a sense that something is wrong with my installation. I did some validations:

```bash
$ pkg-config --modversion realsense2
2.55.1
```
Then, opencv shows weird numbers:

```bash
$dpkg -l | grep opencv
ii  libopencv-calib3d4.5d:arm64                        4.5.4+dfsg-9ubuntu4                               arm64        computer vision Camera Calibration library
ii  libopencv-contrib4.5d:arm64                        4.5.4+dfsg-9ubuntu4                               arm64        computer vision contrlib library
ii  libopencv-core4.5d:arm64                           4.5.4+dfsg-9ubuntu4                               arm64        computer vision core library
ii  libopencv-dev                                      4.8.0-1-g6371ee1                                  arm64        Open Computer Vision Library
ii  libopencv-dnn4.5d:arm64                            4.5.4+dfsg-9ubuntu4                               arm64        computer vision Deep neural network module
ii  libopencv-features2d4.5d:arm64                     4.5.4+dfsg-9ubuntu4                               arm64        computer vision Feature Detection and Descriptor Extraction library
ii  libopencv-flann4.5d:arm64                          4.5.4+dfsg-9ubuntu4                               arm64        computer vision Clustering and Search in Multi-Dimensional spaces library
ii  libopencv-highgui4.5d:arm64                        4.5.4+dfsg-9ubuntu4                               arm64        computer vision High-level GUI and Media I/O library
ii  libopencv-imgcodecs4.5d:arm64                      4.5.4+dfsg-9ubuntu4                               arm64        computer vision Image Codecs library
ii  libopencv-imgproc4.5d:arm64                        4.5.4+dfsg-9ubuntu4                               arm64        computer vision Image Processing library
ii  libopencv-ml4.5d:arm64                             4.5.4+dfsg-9ubuntu4                               arm64        computer vision Machine Learning library
ii  libopencv-objdetect4.5d:arm64                      4.5.4+dfsg-9ubuntu4                               arm64        computer vision Object Detection library
ii  libopencv-photo4.5d:arm64                          4.5.4+dfsg-9ubuntu4                               arm64        computer vision computational photography library
ii  libopencv-shape4.5d:arm64                          4.5.4+dfsg-9ubuntu4                               arm64        computer vision shape descriptors and matchers library
ii  libopencv-stitching4.5d:arm64                      4.5.4+dfsg-9ubuntu4                               arm64        computer vision image stitching library
ii  libopencv-video4.5d:arm64                          4.5.4+dfsg-9ubuntu4                               arm64        computer vision Video analysis library
ii  libopencv-videoio4.5d:arm64                        4.5.4+dfsg-9ubuntu4                               arm64        computer vision Video I/O library
ii  libopencv-viz4.5d:arm64                            4.5.4+dfsg-9ubuntu4                               arm64        computer vision 3D data visualization library
ii  python3-opencv:arm64                               4.5.4+dfsg-9ubuntu4                               arm64        Python 3 bindings for the computer vision library
ii  ros-humble-vision-opencv                           3.2.1-1jammy.20240731.052303                      arm64        Packages for interfacing ROS2 with OpenCV, a library of programming functions for real time computer vision.
```

`libcore-dev` shows v4.8.0, but the rest of the components are all 4.5.4. Weird. I reinstalled Ros-humble and the inconsistency  remains. On my workstation they are all V 4.5.x

Do you have any idea what is going on? Any comment would be greatly appreciated.
Thanks,

MartyG-RealSense, 2024-09-16:
Hi @djaniel  I see that you used the **libuvc backend** installation method to install the librealsense SDK.  This method typically works well with Jetson, especially with cameras like D435i that have an IMU.

A RealSense user at  https://github.com/IntelRealSense/librealsense/issues/13326 who had problems when using Humble and JetPack 6 last week (though did not have the same problem as you) commented that their problems were resolved if they compiled librealsense from source code with the **libuvc backend** installation method using the build script at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

***********

You also mention that you installed the ROS wrapper from source.  Was that with the **colcon build** method in the wrapper's source code installation instructions in **Step 3** at the link below?

https://github.com/IntelRealSense/realsense-ros?tab=readme-ov-file#installation-on-ubuntu

If you have used this method, you could try deleting the entire **/ros2_ws** catwin workspace folder to uninstall the wrapper and then attempt a source code build of the wrapper with colcon again to see if the errors continue.

MartyG-RealSense, 2024-09-24:
Hi @djaniel  Do you require further assistance with this case, please?  Thanks!

jack-obrien, 2024-10-01:
Hi, I think I have a solution. Try running `apt install libopencv-dev=4.5.4+dfsg-9ubuntu4`. Doing this will replace libopencv-dev version 4.8.0 with the version 4.5.4 to match the rest of the opencv install. This is the technique used by nvidia in their dev container docker build system.

I think the issue is that ROS2 installs libopencv-dev version 4.8.0 as a dependency. However jetpack 6 comes with the slightly older binaries for opencv. So when we install the correct version of libopencv-dev, opencv is now looking in the right place for binaries.

MartyG-RealSense, 2024-10-01:
Thanks so much @jack-obrien for sharing your knowledge with @djaniel   :)

djaniel, 2024-10-16:
Hi, no thank you. I did an installation from source, it worked.

MartyG-RealSense, 2024-10-17:
You are very welcome.  I'm pleased to hear that you succeeded.  Thanks very much for the update!

closed 2024-10-17 by MartyG-RealSense

cmburgul, 2025-01-17:
> Hi, I think I have a solution. Try running `apt install libopencv-dev=4.5.4+dfsg-9ubuntu4`. Doing this will replace libopencv-dev version 4.8.0 with the version 4.5.4 to match the rest of the opencv install. This is the technique used by nvidia in their dev container docker build system.
> 
> I think the issue is that ROS2 installs libopencv-dev version 4.8.0 as a dependency. However jetpack 6 comes with the slightly older shared libraries for opencv. So when we install the correct version of libopencv-dev, opencv is now looking in the right place for libraries.

I had similar issue and this solution worked