realsenseai/realsense-ros #3204: Documentation:

edgarcamilocamacho, 2024-09-18:
Hello.

When trying to follow the installation guides, it says "Supported RealSense SDK".

[image]

But when you go to the releases page, not all the releases have that section. 4.56.1 and 4.55.1 do not have it.

MartyG-RealSense, 2024-09-19:
Hi @edgarcamilocamacho  It is saying to go to the RealSense SDK 'Releases' page and download the source code for the SDK version that should be matched with a particular ROS2 wrapper version.  

For example, if you are using wrapper 4.55.1 then go to the information listing for the corresponding **2.55.1** SDK version, scroll down to the 'Assets' file list and click on the 'Source code' link to initiate a download of a zip file containing the SDK source code.

[image]

Sometimes the contents of the Assets list is hidden, so you have to left-click on 'Assets' to expand open the file list.

[image]

edgarcamilocamacho, 2024-09-19:
Yes, but for versions `4.54.1` and before, each of their release documentation explicitly said the version of the SDK they needed, for example:

`4.54.1`:
[image]

`4.51.1`:
[image]

`4.0.4`:
[image]

But the documentation of the last versions two versions `4.55.1` and `4.56.1` does not include that **Supported RealSense SDK** section.

What I understand is that, from the last versions, I just replace the major version `4` for `2` to get the corresponding SDK version, right?

MartyG-RealSense, 2024-09-20:
I understand what you mean now.  Thanks very much for the clarification.  Yes, the 'Supported RealSense SDK' section is no longer included in the ROS wrapper release notes.  You are correct, you just replace the major version number '4' with '2' to get the correct matching librealsense SDK version.

MartyG-RealSense, 2024-09-26:
Hi @edgarcamilocamacho  Do you require further assistance with this case, please?  Thanks!

edgarcamilocamacho, 2024-09-27:
Nope, the last response just clarified the situation. Thanks!

closed 2024-09-27 by edgarcamilocamacho