realsenseai/realsense-ros #3205: Last release documentation is not clear about the version of librealsense/firmware to use

edgarcamilocamacho, 2024-09-18:
In the release changes you see these two items:

[image]

But when you check the code changes on them, they are not coherent:

[image]

[image]

MartyG-RealSense, 2024-09-19:
@edgarcamilocamacho  The second and third images that you kindly provided are from the **ros2-development** branch of the ROS2 wrapper.

However, the naming of the wrapper has recently changed.  **ros2-development** used to be the default Master branch of the wrapper.  However, the Master branch is now **ros2-master**

After this change, the ros2-development branch is now the development branch for the next upcoming wrapper version, which only has source code available for it and not packages.

ros2-development will be in flux because it has changes being continuously added to it, whilst ros2-master is the branch that most RealSense ROS2 users will be using.  Therefore, the notes that can be relied upon for accuracy are the ones in the wrapper release notes (the first image that you posted above).

edgarcamilocamacho, 2024-09-19:
Yes, you right, the tag itself is correctly set:

https://github.com/IntelRealSense/realsense-ros/blob/4.56.1/realsense2_camera/CMakeLists.txt#L121
https://github.com/IntelRealSense/realsense-ros/blob/4.56.1/README.md?plain=1#L122

What generated the confussion were the links from the release documentation.

Thanks!

MartyG-RealSense, 2024-09-19:
You are very welcome.  I'm pleased I could help!