realsenseai/realsense-ros #3208: IMU problems for the D455

louislelay, 2024-09-23:
### Issue Description

| Required Info                         | Details                                                              |
|---------------------------------------|----------------------------------------------------------------------|
| **Camera Model**                      | D455                                                                 |
| **Firmware Version**                  | 5.15.0.2                                                             |
| **Operating System & Version**        | Ubuntu 22.04                                                         |
| **Kernel Version (Linux Only)**       | 6.8.0-40                                                             |
| **Platform**                          | PC                                                                   |
| **Librealsense SDK Version**          | 2.55.1                                                               |
| **Language**                          | N/A                                                                  |
| **Segment**                           | Robot                                                                |
| **ROS Distro**                        | Humble                                                               |
| **RealSense ROS Wrapper Version**     | 4.55.1                                                               |

Hello, I’m working with a RealSense D455 camera and encountered some issues when trying to use the IMU topic.

### Steps to Reproduce:

1. I launched the `rs_launch.py` file without any arguments. I noticed that the expected `/camera/camera/imu` topic doesn't appear. From my inspection, it seems that this topic is not enabled by default. If this is intended behavior, **_I suggest updating the README_** to reflect the change.

2. I then tried to enable the IMU by running the following command:
   ```bash
   ros2 launch realsense2_camera rs_launch.py enable_accel:=true enable_gyro:=true unite_imu_method:=2
   ```

### Issues Observed:

- **Frames Timeout Warning**: After running the above command, I started encountering the following delay warnings:
```
   [realsense2_camera_node-1] 22/09 21:52:30,461 WARNING [133606110398016] (backend-v4l2.cpp:1649) Frames didn't arrive within 5 seconds
   [realsense2_camera_node-1] [WARN] [1727056350.461886004] [camera.camera]: XXX Hardware Notification: Frames didn't arrive within 5 seconds,1.72706e+12,Warn,Frames Timeout
```

- **Partial IMU Data**: After launching with the `enable_accel`, `enable_gyro` and `unite_imu_method` options, new topics, including `/camera/camera/imu`, appeared. However, when echoing this topic, I noticed that only the `angular_velocity` and `linear_acceleration` fields were receiving updates, while the `orientation` field remained at all zeros:
```
header:
  stamp:
    sec: 1727056330
    nanosec: 371350528
  frame_id: camera_imu_optical_frame
orientation:
  x: 0.0
  y: 0.0
  z: 0.0
  w: 0.0
orientation_covariance:
- -1.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
angular_velocity:
  x: -0.005235987715423107
  y: -0.003490658476948738
  z: 0.005235987715423107
angular_velocity_covariance:
- 0.01
- 0.0
- 0.0
- 0.0
- 0.01
- 0.0
- 0.0
- 0.0
- 0.01
linear_acceleration:
  x: 0.45110589265823364
  y: -9.090764045715332
  z: 1.0296982526779175
linear_acceleration_covariance:
- 0.01
- 0.0
- 0.0
- 0.0
- 0.01
- 0.0
- 0.0
- 0.0
- 0.01
```

### Confusion:
- Why am I receiving updates for **angular velocity** and **linear acceleration** even though I'm getting "frames didn't arrive" warnings?
- Why is the **orientation** field not being updated, and why does it show all zeros?

Thanks in advance!

IRONCROWY, 2024-09-23:
If what i remembered is not wrong, D455 has no orientation output.

MartyG-RealSense, 2024-09-23:
Hi @louislelay   Thanks very much for your questions.

1.  You are correct, the accel and gyro IMU topics are disabled by default, because not everyone has a RealSense camera equipped with an IMU.  The infrared topics are also disabled by default.

2.  The recollection of @IRONCROWY above regarding orientation values not being provided is correct, at least for the ROS1 wrapper - please see https://github.com/IntelRealSense/realsense-ros/issues/1210#issuecomment-634465607

One ROS1 user at https://github.com/IntelRealSense/realsense-ros/issues/1694#issuecomment-779736012 installed an additional package to obtain the orientation.

In regard to IMU support in the ROS2 wrapper, it also only supports gyro, accel and the unite_imu_method parameter that publishes the /imu topic.

louislelay, 2024-09-23:
Hi @MartyG-RealSense, thank you so much for the detailed response!

I’ll go over everything in the next few days, but I do think the README might need a small update. It seems to suggest that the IMU topic is published without any parameters, which isn’t accurate from what you say. It’s a minor change, but it would make things clearer. Also, I now understand why the other messages in the topic aren't updating. That said, do you think it would be a good idea to add a node in the repository to publish orientation if needed?

MartyG-RealSense, 2024-09-24:
You are very welcome!

If you would like to suggest a change to the documentation or create and submit a mechanism to publish orientation then contributions to the RealSense wrapper from users via a **Pull Request** are welcomed.  Please see https://github.com/IntelRealSense/realsense-ros/pulls

louislelay, 2024-09-26:
I'll experiment with a few approaches and, if I find a promising solution, I'll be sure to propose it! Since all my questions have been answered, I'll close this issue. Thanks again @MartyG-RealSense!

closed 2024-09-26 by louislelay

MartyG-RealSense, 2024-09-26:
You are very welcome, @louislelay - thanks very much for the update.  Good luck!