realsenseai/realsense-ros #3209: No IMU/Gyro/Accel data output in related topics for D455

IRONCROWY, 2024-09-23:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | { D455 }                          | 
| Firmware Version                   | 5.16.0.1                          | 
| Operating System & Version |   { Ubuntu 20.04 } | 
| Kernel Version (Linux Only)    |  5.15.0-122-generic                                                                     | 
| Platform                                 | PC                           |
| Librealsense SDK Version       |  { v2.55.1 }                                                    | 
| Language                               |  {C/C#/labview/opencv/pcl/python/unity }         | 
| Segment			         |  {Robot}                             | 
| ROS Distro			         |  {noetic}                                           | 
| RealSense ROS Wrapper Version   |  {4.56.1 ros1-legacy}                                                  | 

Hello, i meet a strange problem on my D455.

Situation (1)
For my realsense-viewer, the gyro and accel part can be viewed normally as follows:
[image]

Situation (2)
However, in ros-noetic, i run launch the node, but when i echo the data in gyro and accel topic or imu topic(when unite_imu_method is set), there is nothing i can recieve.
The image is an example of /camera/imu, if unite_imu_method not set, the gyro and accel are also the same condition.
[image]

However for the other topics, some work normally:
[image]

More information:
1. Before i notice this problem, i ran the rs-imu-calibration.py in python3 and wrote the calibration data to D455.
2. Then i notice the problem, i tried to flash the 5.16.0.1 firmware to D455 in librealsense-2.55.1's realsense-viewer in order to cover the related settings, and obviously it didn't work.

rs_d4555.launch
```
<launch>
  <arg name="serial_no"           default=""/>
  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default=""/>
  <arg name="json_file_path"      default=""/>
  <arg name="camera"              default="camera"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>
  <arg name="output"              default="screen"/>
  <arg name="respawn"              default="false"/>

  <arg name="fisheye_width"       default="-1"/>
  <arg name="fisheye_height"      default="-1"/>
  <arg name="enable_fisheye"      default="false"/>

  <arg name="depth_width"         default="848"/>
  <arg name="depth_height"        default="480"/>
  <arg name="enable_depth"        default="true"/>

  <arg name="confidence_width"    default="-1"/>
  <arg name="confidence_height"   default="-1"/>
  <arg name="enable_confidence"   default="true"/>
  <arg name="confidence_fps"      default="-1"/>

  <arg name="infra_width"         default="848"/>
  <arg name="infra_height"        default="480"/>
  <arg name="enable_infra"        default="true"/>
  <arg name="enable_infra1"       default="false"/>
  <arg name="enable_infra2"       default="false"/>
  <arg name="infra_rgb"           default="false"/>

  <arg name="color_width"         default="848"/>
  <arg name="color_height"        default="480"/>
  <arg name="enable_color"        default="true"/>
[... 482 lines cut ...]
   92 | RTABMAP_DEPRECATED typedef SensorEvent CameraEvent;
      |                                        ^~~~~~~~~~~
/home/gds/catkin_ws/src/rtabmap_ros/rtabmap_legacy/src/CameraNode.cpp:226:52: warning: ‘rtabmap::CameraEvent’ is deprecated [-Wdeprecated-declarations]
  226 |    rtabmap::CameraEvent * e = (rtabmap::CameraEvent*)event;
      |                                                    ^
In file included from /opt/ros/noetic/include/rtabmap-0.21/rtabmap/core/CameraEvent.h:30,
                 from /home/gds/catkin_ws/src/rtabmap_ros/rtabmap_legacy/src/CameraNode.cpp:38:
/opt/ros/noetic/include/rtabmap-0.21/rtabmap/core/SensorEvent.h:92:40: note: declared here
   92 | RTABMAP_DEPRECATED typedef SensorEvent CameraEvent;
      |                                        ^~~~~~~~~~~
cd /home/gds/catkin_ws/build/rtabmap_legacy; catkin build --get-env rtabmap_legacy | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -

.....................................................................................................................................................................................
Finished  <<< rtabmap_legacy                         [ 8.5 seconds ]                                                                                                                 
Starting  >>> rtabmap_ros                                                                                                                                                            
Finished  <<< rtabmap_ros                            [ 1.1 seconds ]                                                                                                                 
[build] Summary: All 18 packages succeeded!                                                                                                                                          
[build]   Ignored:   None.                                                                                                                                                           
[build]   Warnings:  4 packages succeeded with warnings.                                                                                                                             
[build]   Abandoned: None.                                                                                                                                                           
[build]   Failed:    None.                                                                                                                                                           
[build] Runtime: 1 minute and 22.1 seconds total.                                                                                                                                    
gds@gds-System-Product-Name:~/catkin_ws$ 

```

IRONCROWY, 2024-09-23:
I doubt that the rs-imu-calibration.py has destroyed some settings in D455, because after i ran this code, the opensource_tracking.launch failed to work because of lacking TF information. Before all of these operations, the opensource_tracking.launch still worked.

IRONCROWY, 2024-09-23:
The reason was found, my realsense-ros wrapper version is 2.3.2, it only corresponds to librealsense sdk v2.50.0, the additional D455 firmware version is 5.13.0.50, after all of these compiled and flashed, the imu topic works.

MartyG-RealSense, 2024-09-23:
THanks very much @IRONCROWY for the update about your success!

IRONCROWY, 2024-09-24:
There is one more reason: **do not use catkin build on realsense-ros**, but catkin_make instead!