realsenseai/realsense-ros #3210: D435i on docker

Somethingout, 2024-09-23:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D435i }                          | 
| Firmware Version                   | (Open RealSense Viewer --> Click info)                          | 
| Operating System & Version |  Ubuntu 20 | 
| Kernel Version (Linux Only)    |  (e.g. 5.4)                                                                       | 
| Platform                                 | NVIDIA Jetson                            |
| Librealsense SDK Version       |  { 2.<?>.<?> }                                                    | 
| Language                               |  {C/C#/labview/opencv/pcl/python/unity }         | 
| Segment			         |  {Robot/Smartphone/VR/AR/others }                             | 
| ROS Distro			         |  {Humble }                                           | 
| RealSense ROS Wrapper Version   |  {4.51.1, 4.54.1, etc..}                                                  | 

### Issue Description

(ROS2 HUMBLE, JETSON XAVIER NX) 

I am trying to use a realsense camera d435i on a docker container, however I have few issues: 

- Even tough all topics are running it seems like it can only publish  rgb camera OR depth images (no pointcloud or imu).
-  The published rate is very low compared to when I run it without a container.

I know it is not advised to use the privilege mode but for simplicity this time I am using it: 

- docker run -it --network=host --ipc=host --device-cgroup-rule='c 13:* rmw' -v /dev:/dev  --device-cgroup-rule "c 81:* rmw" --device-cgroup-rule "c 189:* rmw" --privileged  mycontainer-jetson 

I tried running using "sudo" without success. 

The SDK is installed using "sudo apt install ros-humble-librealsense2*" 

I took a look at old issues but couldn't find an answer. any ideas n how to solve this please?

Thank you

MartyG-RealSense, 2024-09-24:
Hi @Somethingout  The apt install method that you used to install librealsense will not contain the SDK's support for **CUDA** on Nvidia Jetson.  When CUDA support is enabled, it offloads processing of pointclouds and depth-color alignment from the CPU onto the Jetson's Nvidia graphics GPU, resulting in a significant reduction in how much the CPU is utilized.

For CUDA support, you should install librealsense from special Jetson packages using the instructions at the link below.  You can also enable it by building librealsense from source code with CMake and including the flag **-DBUILD_WITH_CUDA=ON** in the CMake build instruction.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md#4-install-with-debian-packages

Enabling CUDA in librealsense also automatically applies its acceleration in the ROS wrapper.

********

Another way to install CUDA-enabled packages is to use the JetsonHacks website's **installLibrealsense.sh** build script.

https://github.com/jetsonhacks/installRealSenseSDK

*********

Regarding the pointcloud and IMU, can you confirm please that you have enabled the pointcloud and the IMU topics in your launch instruction?

`ros2 launch realsense2_camera rs_launch.py pointcloud.enable:=true enable_accel:=true enable_gyro:=true unite_imu_method:=2`

Somethingout, 2024-09-24:
Hi Marty, 

Thank you  for your answer, I will try to follow your instructions. 

Should I also do the same procedure in the docker container or only in the Jetson ? 

For your question, yes everything is enabled I have a custom launch/config  file that sets the params to true (tested on my own computer and it works fine but not in the container but I guess it is related to your first point).

MartyG-RealSense, 2024-09-24:
It depends whether you are using librealsense on both the host and the container.  If you are using them outside the container too then yes, enable CUDA on both installations.

If you are only going to be using librealsense in the Docker container then you would only need to enable CUDA in the Docker installation of librealsense.

Somethingout, 2024-09-24:
Unfortunately, I still have the same behavior where only depth image is published (and sometimes color image but not both at the same time) and also the pointcloud is not published even tough all the topics are created. 

I followed the step 4 instructions (Install with Debian Packages) of your link: https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md#4-install-with-debian-packages

I did it both on the jetson and docker. I checked on the jetson with realsense viewer and it works. but when running the ros2 launch file it doesn't work ! 

Do my guess would be that something is missing with docker...

MartyG-RealSense, 2024-09-24:
A RealSemnse user who had Docker container problems last week shared their docker run command at https://github.com/IntelRealSense/librealsense/issues/11368#issuecomment-2356994462 - it might give you ideas for some parameters to try.

Somethingout, 2024-09-24:
Thank you, however I couldn't fix it ! this is really a strange beahvior, almost like it doesn't have the required resources to publish the image, depth image and pointcloud at the same time...

MartyG-RealSense, 2024-09-24:
A RealSense user at https://github.com/IntelRealSense/librealsense/issues/13116#issuecomment-2232631202 reported that installing librealsense via Nvidia's **Isaac Dev Docker** worked well for them.

Somethingout, 2024-09-25:
So I tried to create a docker container on my computer to see if maybe it is related to docker command, turns out it runs perfectly in a docker container in my computer. However the problem is still on going on my jetson where only depth image OR color image is published (despite the creation of all other topics and params enabled). So maybe is it related to something else. 

Here some additional information in case it is a problem with versions and sdk.

My kernel version on my jetson xavier nx : 5.10.120-tegra
Cuda version : 
Cuda compilation tools, release 11.4, V11.4.315
Build cuda_11.4.r11.4/compiler.31964100_0

Thank you.

MartyG-RealSense, 2024-09-25:
Thanks very much for the information.  The Jetson kernel version and CUDA version are not usually related to problems with using Jetson with RealSense.

Sometimes defining specific **/video** channels as devices (dev) in the docker run command helps, like the instruction at the link below that somebody used on a Jetson Nano.  By default with RealSense cameras, depth is /video0, infrared is /video1 and RGB is /video2

https://hub.docker.com/r/drunknmunky32/jetsonrealsense

```
 xhost +

docker run --gpus all -it --rm --net=host --device=/dev/video0:/dev/video0 --device=/dev/video1:/dev/video1 --device=/dev/video2:/dev/video2 -v /tmp/.X11-unix:/tmp/.X11-unix -e DISPLAY=$DISPLAY drunknmunky32/jetsonrealsense:v0
```

Somethingout, 2024-09-25:
Thank you Marty, seems like it working (at least with rs tools) but how can I add ros2 on this docker image ?

MartyG-RealSense, 2024-09-25:
The link below has a suggestion for incorporating ROS2 into a docker run command.  Its structure is consistent with my memory of other ROS2 docker run commands used by RealSense users.

https://forums.developer.nvidia.com/t/newbie-how-to-use-intel-realsense-in-the-ros2-foxy-docker-container-on-jetson-nano/196575/4

Heres another example.

https://www.enthusiasticroboticist.com/blog/realsense-cameras-ros-2-foxy-and-docker-on-nvidia-jetson-nano/

```
docker pull ${REGISTRY}/ros2_realsense:latest
docker run \
  --name ros2_realsense \
  --rm \
  -it \
  --runtime nvidia \
  --network host \
  --gpus all \
  --privileged \
  -e DISPLAY \
  -v /dev/bus/usb/:/dev/bus/usb/ \
  ${REGISTRY}/ros2_realsense:latest \
  bash 
```

ashBabu, 2024-10-02:
I have had an experience with Raspberry Pi for a similar issue. Launching camera with `pointcloud.enabe:=true` did not produce (if I remember correctly) or at a very low frequency. I  produced pointcloud from depth images by [this](https://github.com/introlab/rtabmap_ros/issues/1061#issuecomment-1798352871) method. 

May be useful to you

MartyG-RealSense, 2024-10-02:
Thanks so much @ashBabu for sharing your experience with @Somethingout   :)

Somethingout, 2024-10-03:
Hi, 

Sorry for the late reply, thank you Marty and ashBabu. I managed to make it run on docker ! everything works well.

MartyG-RealSense, 2024-10-03:
No problem at all, @Somethingout - it's great news that you succeeded. Thanks very much for the update!

closed 2024-10-04 by Somethingout