realsenseai/realsense-ros #3211: can not run the realsense camera

liujl12345, 2024-09-24:
Hi! Here is a problem I can not solve.

camera Model : D405
ubuntu 20.04.6 LTS
kernel 5.15.0-122-generic
run command : ros2 run realsense2_camera realsense2_camera_node

when i try to run my camera after  I've installed ROS Wrapper for Intel® RealSense™ cameras, by command 'sudo apt install ros-foxy-realsense2-*'
then error occured like below,

[image]

what's wrong? or i have made mistake when installing the ROS Wrapper for Intel® RealSense™ cameras? someone helps me~(╥_╥)

MartyG-RealSense, 2024-09-24:
Hi @liujl12345  The log indicates that you have likely not made a mistake when installing the ROS wrapper.  These red errors occur when using the 5.16.0.1 firmware with an older librealsense SDK version such as the 2.51.1 one that you are using, as this firmware is designed for use with librealsense 2.55.1.

If you need to use librealsense 2.51.1 because you have to use Foxy then downgrading your firmware version to **5.13.0.50** should make the red errors disappear.

liujl12345, 2024-09-25:
> If you need to use librealsense 2.51.1 because you have to use Foxy then downgrading your firmware version to **5.13.0.50** should make the red errors disappear.

Thanks a lot for your help!

it's my shame to say that i couldn't find a way to downgrade my firmware version. Can you help me?

MartyG-RealSense, 2024-09-25:
You are very welcome!

If you have access to the realsense-viewer tool then instructions at https://github.com/IntelRealSense/librealsense/issues/12599#issuecomment-1902680303 explain how to downgrade the firmware to 5.13.0.50.

If you are not able to use realsense-viewer on your computer then you can also downgrade the firmware using the **rs-fw-update** tool once you have downloaded the **.bin** firmware file for 5.13.0.50 from the firmware releases page.

https://dev.intelrealsense.com/docs/firmware-releases-d400

https://github.com/IntelRealSense/librealsense/tree/master/tools/fw-update

[image]

liujl12345, 2024-09-25:
> If you need to use librealsense 2.51.1 because you have to use Foxy then downgrading your firmware version to **5.13.0.50** should make the red errors disappear.

Hi!@MartyG-RealSense, Thanks to your help, i downgraded my firmware successfully. And the red error disappear.

But I still can not use the camera, the pictures show blow,
[image]
[image]

is there something i didn't install? what's wrong?

liujl12345, 2024-09-26:
> ros2 launch realsense2_camera rs_launch.py pointcloud.enable:=true

I tried the command 'ros2 launch realsense2_camera rs_launch.py pointcloud.enable:=true', the output as shown below,
 
[image]

no errors, but it doesn't seem to be working.

MartyG-RealSense, 2024-09-26:
Is there any difference if you set the value of **pointcloud.stream_index_filter** yourself?  Its default value should be '0'.

`ros2 launch realsense2_camera rs_launch.py pointcloud.enable:=true pointcloud.stream_index_filter:=0`

liujl12345, 2024-09-27:
> ros2 launch realsense2_camera rs_launch.py pointcloud.enable:=true pointcloud.stream_index_filter:=0

there is nothing changed. it seems that this command can not change the value of pointcloud.stream_index_filter

liujl12345, 2024-09-27:
> there is nothing changed. it seems that this command can not change the value of pointcloud.stream_index_filter

but I can see the pointclouds after I launch the 'rviz', maybe this argument just doesn't matter. 

After all these, my task is identify aruco codes with my realsense D405. Can you recommend a teaching article or an example for me according to my situation? Thank you very much.

MartyG-RealSense, 2024-09-27:
pal-robotics have an **aruco_ros** Aruco marker detecter at https://github.com/pal-robotics/aruco_ros that is compatible with ROS2 Humble.