realsenseai/realsense-ros #3213: Auto-exposure takes time to adjust between two fast-moving points

adithyab94, 2024-09-25:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D415                         | 
| Firmware Version                   | 5.16.0.1                        | 
| Operating System & Version |   Ubuntu 20 
| Kernel Version (Linux Only)    |  5.15.0                                                                   | 
| Platform                                 | PC                          |
| Language                               |  C++/opencv//python      | 
| Segment			         |  Robot                            | 
| ROS Distro			         |  Noetic     ros1-legacy                                     | 

### Issue Description

I am facing an issue with my RealSense D415 camera when it is mounted on a robot that moves between two points in quick succession. The camera is capturing images at both points, but the automatic exposure adjustment is causing issues. Specifically, the auto-exposure takes time to adjust when the robot moves between these two points. As a result, the first image at the initial point appears correctly, but subsequent images at the other point are too dark or entirely black.

Only the first image is correctly exposed, while subsequent images are significantly underexposed due to the delay in the auto-exposure adjustment.

Additional information:

I cannot add external lighting or use IR filters to improve lighting conditions.
I am using ROS Noetic (ROS 1) with a RealSense D415 camera.

Request: 

I am looking for advice on how to approach this issue and which parameters I can adjust to resolve the exposure inconsistency while the camera is in motion. Any guidance would be appreciated.

MartyG-RealSense, 2024-09-25:
Hi @adithyab94  The D415 camera model has slow-moving **rolling** type shutters on both its depth and RGB sensors.  Whilst D415 excels at capturing static objects, capture during motion faster than a human's waving hand can cause the images to lag or blur.

If you can set the FPS speed to 60 when auto-exposure is enabled then this will help to reduce lag during fast motion.

If the camera is operating in a location with artificial lighting whose level is always consistent then you could also try disabling auto-exposure and using fixed manual exposure, as the camera will not need to take account of changing lighting conditions.  Please try adding to your roslaunch instruction the parameter `enable_auto_exposure:=false`

adithyab94, 2024-09-26:
Thank you very much. I added the `enable_auto_exposure:=false` and it is nor better. 

I also have another question before I close the issue. Can I set a minimum limit for the rgb camera's auto exposure value ?

I saw there is a similar parameter for stereo module `/camera/stereo_module/auto_exposure_limit` but I think it sets an upper limit. Can you please let me know can we set the minimum limit for RGB camera in such a way

MartyG-RealSense, 2024-09-26:
There is not an auto-exposure limit control for the RGB camera, unfortunately.

closed 2024-09-27 by adithyab94