realsenseai/realsense-ros #3217: [ros1-legacy] Higher CPU total usage when disabling the depth camera

captain-yoshi, 2024-09-26:
| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       | D415                          | 
| Firmware Version                   | 5.16.0.1                         | 
| Operating System & Version |   Ubuntu 20 | 
| Kernel Version (Linux Only)    |  5.15.0-119-generic                                                                       | 
| Platform                                 | PC                           |
| Librealsense SDK Version       |  2.55.1.0                                                    | 
| Language                               |  C++         | 
| Segment			         |  Robot                            | 
| ROS Distro			         |  Noetic                                           | 
| RealSense ROS Wrapper Version   |  2.3.2                                                  | 

### Issue Description
```sh
# Working correctly
$ roslaunch realsense2_camera rs_camera.launch

# Slow GUI, due top high CPU -> polkitd
$ roslaunch realsense2_camera rs_camera.launch enable_depth:=false
```
When starting the color and depth camera, the GUI interface is responsive, the average CPU is 91% while the realsense-ros package uses 34% .

When disabling the depth camera, the GUI interface slows down, the average CPU bumps up to 192% even though the realsense-ros process uses 15%. The `/usr/lib/policykit-1/polkitd --no-debug` process is now 50% (contrary to 2% when not disabled). This happens everytime. Tried rebooting the system and the camera.

You can inspect the ressources log with atop:
[atop_logs.zip](https://github.com/user-attachments/files/17152174/atop_logs.zip)

Depth = ON
[image]

Depth = OFF
[image]

MartyG-RealSense, 2024-09-26:
Hi @captain-yoshi  There have been a couple of past cases where this phenomenon has occurred.  As there is not a known fix, I would recommend having the depth stream enabled but setting it to a low resolution and FPS speed so that it is publishing but not having a significant effect on CPU percentage usage.

You could do this by launching with a custom stream configuration like the command below, which sets depth to 640x480 and 6 FPS, and color to 640x480 and 30 FPS.

`roslaunch realsense2_camera rs_camera.launch depth_width:=640 depth_height:=480 depth_fps:=6 color_width:=640 color_height:=480 color_fps:=30`

MartyG-RealSense, 2024-10-03:
Hi @captain-yoshi  Do you require further assistance with this case, please?  Thanks!

captain-yoshi, 2024-10-03:
Lowering the depth resolution helps reduce the CPU a bit. Thanks.

closed 2024-10-20 by MartyG-RealSense