realsenseai/realsense-ros #3218: D455 camera/imu topic not appearing

camron-BHF, 2024-09-27:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking our ROS RealSense Wrapper documentation [README](https://github.com/IntelRealSense/realsense-ros/blob/ros2-master/README.md).
    * Have you looked in our [Discussions](https://github.com/IntelRealSense/realsense-ros/discussions)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/realsense-ros/issues) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                                         |
|---------------------------------|------------------------------------------------------------ |
| Camera Model                       |  D455                           | 
| Firmware Version                   | 5.16.0.1                          | 
| Operating System & Version |  Linux Ubuntu 22.04 | 
| Kernel Version (Linux Only)    |  6.5.1                                                                   | 
| Platform                                 | PC                        |
| Librealsense SDK Version       |  { 2.55.1 }                                                    | 
| Language                               |  C++        | 
| Segment			         |  Robot                             | 
| ROS Distro			         |  Humble                      | 
| RealSense ROS Wrapper Version   |  4.55.1                                                  | 

### Issue Description
I am not able to have a single topic which publishes both the accel and gyro information in one such as camera/imu. Below I copied the topic list after I enabled accel, gyro and set unite_imu_method to 2.

/camera/camera/accel/imu_info
/camera/camera/accel/metadata
/camera/camera/accel/sample
/camera/camera/color/camera_info
/camera/camera/color/image_raw
/camera/camera/color/metadata
/camera/camera/depth/camera_info
/camera/camera/depth/image_rect_raw
/camera/camera/depth/metadata
/camera/camera/extrinsics/depth_to_accel
/camera/camera/extrinsics/depth_to_color
/camera/camera/extrinsics/depth_to_gyro
/camera/camera/gyro/imu_info
/camera/camera/gyro/metadata
/camera/camera/gyro/sample
/parameter_events
/rosout
/tf_static

I'm trying to use the camera + imu to get an accurate measurement of how much my robot moved (i.e. if we set the motors to move forward 1m, but actually it only moved forward 99cm) and I was hoping there was 1 topic if I can subscribe to or if I have to subscribe to both accel/sample and gyro/sample and sync them myself.
Thanks!

MartyG-RealSense, 2024-09-27:
Hi @camron-BHF  The `/imu` topic should be published if you have set unite_imu_method to 2.

Are you setting it in the launch instruction or within the launch file?  If you are doing so with the launch instruction, are you doing it like this:

`ros2 launch realsense2_camera rs_launch.py enable_accel:=true enable_gyro:=true unite_imu_method:=2`

If you have tried this launch method, does it make a difference if you set unite_imu_method to 1 (copy mode instead of linear interpolation mode)

`ros2 launch realsense2_camera rs_launch.py enable_accel:=true enable_gyro:=true unite_imu_method:=1`

If the /imu topic is still not published, what happens if you launch with a **ros2 run** command instead of ros2 launch?

`ros2 run realsense2_camera realsense2_camera_node --ros-args -p enable_accel:=true -p enable_gyro:=true -p unite_imu_method:=2`

camron-BHF, 2024-09-27:
Hi @MartyG-RealSense it seems like you have to enable to gyro etc before launching to publish to the imu node. It works now, thanks. 

On a separate note, I often have issues with my camera in terms of data transmission
`[WARN] [1727459016.571792083] [ camera.camera]: XXX Hardware Notification:Frames didn't arrived within 5 seconds, 1.72746e+12,Warn,Frames Timeout`

I get this error roughly 3/5 times which is weird since it isn't consistent but also isn't rare enough to not worry about. 

I was wondering what the solution to this is as I'm trying to echo the imu topic to see the format of imu data to be able to correctly use it but no information comes through.
Thanks!

MartyG-RealSense, 2024-09-28:
You could test whether having depth, color, gyro and accel published simultaneously is over-burdening your computer's CPU by disabling depth and color to see if it affects how often the warning is published.

`ros2 launch realsense2_camera rs_launch.py enable_depth:=false enable_color:=false enable_accel:=true enable_gyro:=true unite_imu_method:=2`

camron-BHF, 2024-10-01:
Hi Marty, that solved the issue in my Ubuntu 22.04 machine, but it does not solve the issue in my ubuntu 18.04 machine. I still get an error that the frames are not arriving in the specified time.
` 01/10 14:48:02,465 WARNING [140282146318080] (backend-hid.cpp:640) iio_hid_sensor: Frames didn't arrived within the predefined interval`
This was the command I used:
`roslaunch realsense2_camera rs_camera.launch enable_depth:=false enable_color:=false enable_accel:=true enable_gyro:=true unite_imu_method:='linear_interpolation' enable_infra1:=true enable_infra2:=true`

Also when I use rostopic list it seems that the infras don't turn on?

`/camera/accel/imu_info
/camera/gyro/imu_info
/camera/imu
/camera/motion_module/parameter_descriptions
/camera/motion_module/parameter_updates
/camera/realsense2_camera_manager/bond
/camera/rgb_camera/parameter_descriptions
/camera/rgb_camera/parameter_updates
/camera/stereo_module/parameter_descriptions
/camera/stereo_module/parameter_updates
/diagnostics
/rosout
/rosout_agg
/tf
/tf_static`

Thanks for the help

MartyG-RealSense, 2024-10-02:
Does it work if you only enable infra1 and not infra2?

If it works when infra2 is not enabled then it could suggest that your camera is being detected as being on a USB 2.1 connection instead of USB 3.2.  This is because Infra2 is not supported on a USB 2.1 connection, only on USB3.

You can check what USB mode your camera is being detected as by looking in the lower half of the roslaunch log for the line **Device USB type**.

camron-BHF, 2024-10-02:
I guess there was something changed in my launch file because when i rebuilt, sourced, and ran, all the topics appeared. And also the log shows USB 3.2. However I still get an issues that the frames aren't arriving? Are there other parameters I can turn off? My end goal is to use the VINS-Fusion package (or another package if you recommend something else) for Visual Inertial Odometry. 

When I run the camera through the realsense viewer all the images and imu data come through without issue but when I run the roslaunch command with the launch file it doesn't which puzzles me.

Thanks

MartyG-RealSense, 2024-10-02:
Is the RealSense Viewer closed when you perform the roslaunch?  The Viewer is not required to be active in order to use the ROS wrapper.

You could leave Infra1 and Infra2 disabled if you do not need them since the IMU topics are not dependent on them, and also try changing the unite_imu_method's setting from 'linear_interpolation' to 'copy' (Copy mode can provide more stable IMU data than linear_interpolation).

Also see if adding` initial_reset:=true` to the roslaunch instruction to reset the camera at launch makes a difference.

closed 2024-10-20 by MartyG-RealSense