realsenseai/realsense-ros #3223: Video Stream is not started as i launch realsense2_camera_node

RISHI-VEMPATI, 2024-10-09:
i am using a D457 intel real sense camera with USB mode (i toggled the switch). i am trying to launch realsense2_camera_node 
[image]
. its not showing any stream. I have checked my PC (Lenovo Legion 5) has USB3.1 Ports, Cable supports USB3.0 transfer.
[image]

MartyG-RealSense, 2024-10-10:
Hi @RISHI-VEMPATI  Is there a difference if you unplug the micro-sized end of the USB cable from the base of the D457, turn the connector around the opposite way and re-insert it into the camera, please?  Cameras will sometimes perform better when the micro-sized connector is inserted into the camera in a particular orientation of the two insertion directions available.

If it does not make a difference, what happens if you launch with **ros2 launch** instead launching the node with ros2 run?

`ros2 launch realsense2_camera rs_launch.py`

MartyG-RealSense, 2024-10-15:
What method did you use to install the RealSense ROS2 wrapper, please?

RISHI-VEMPATI, 2024-10-19:
https://github.com/IntelRealSense/realsense-ros. I followed this method to install the Ros2 wrapper.

MartyG-RealSense, 2024-10-20:
Did you install the ROS2 wrapper from source code or build it from packages with the command `sudo apt install ros-<ROS_DISTRO>-realsense2-*` please?

RISHI-VEMPATI, 2024-10-20:
no i didn't install it from source. i installed it using the command sudo apt install ros-humble-realsense2-*

MartyG-RealSense, 2024-10-20:
Thanks very much for confirming the ROS2 wrapper installation method.  

Regarding your earlier confirmation about installing librealsense from source, what method did you use for the installation?  

I would recommend performing the install using the 'libuvc_installation_sh' build script from the **libuvc backend** installation method at the link below if you have not done so already.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

As you are using a Jetson and the realsense-viewer tool, Line 46 of the script should be edited to be changed to the following command: 

`cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA=on`

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh#L46

MartyG-RealSense, 2024-10-28:
Hi @RISHI-VEMPATI  Do you require further assistance with this case, please?  Thanks!

RISHI-VEMPATI, 2024-10-28:
yes, the source based installation of RealSense viewer is done based on the instructions you mentioned.

MartyG-RealSense, 2024-10-28:
And you have the same problem that realsense-viewer displays the video stream but the ROS wrapper does not, please?

RISHI-VEMPATI, 2024-10-28:
yes, i am facing same issue. I have tried with D435i camera too and it still is the same. here is the output  "[realsense2_camera_node-1]  28/10 16:33:02,317 WARNING [281472762112224] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11"

MartyG-RealSense, 2024-10-29:
And you have tried including `initial_reset:=true` in your ROS launch instruction to reset the camera at launch and it makes no difference?

RISHI-VEMPATI, 2024-10-29:
no it didn't make any difference.

MartyG-RealSense, 2024-10-29:
**control_transfer returned** messages indicate a communication problem between the camera and computer, often related to an issue with the USB cable or USB port.  As D457 is only supplied with a GMSL/FAKRA cable, you are likely using a USB cable of your own choice rather than the official USB cable supplied with other RealSense 400 Series camera models.  So if you are using a USB cable that you purchased yourself, the issue may be being caused by the cable.

You mentioned at the beginning of this case that your cable is able supports USB3.0 transfer.  However, a cable that is compliant with **USB 3.1 Gen 1** should ideally be used with RealSense cameras.

Below is a quote from the Tech Specs section of the listing for the D455 model on the official RealSense Store (as a D457 is treated as a D455 when used in USB mode).

[image]

RISHI-VEMPATI, 2024-10-29:
But when I used the same cable with a Ubuntu 22.04-based PC, it worked perfectly on the PC. this problem arrived when i was using this on Jetson orin nano.

MartyG-RealSense, 2024-10-30:
Another possible cause of **Resource temporarily unavailable, number: 11** could be that the Jetson's CPU is struggling to keep up with the rate of incoming data from the camera over the USB cable.  

You could use an Ubuntu system monitoring tool such as **htop** to check whether the percentage usage of the Jetson Orin Nano's CPU is near or at 100%.

closed 2024-11-15 by MartyG-RealSense