realsenseai/realsense-ros #3224: Problem of D435i running with Ubuntu 20.04 and ROS1 Noetic. Help me plz.

byeongjun-YU, 2024-10-10:
---------------------------------- Question ----------------------------------
Hello.
I tried to run D435i with Ubuntu 20.04 and ROS1 Noetic.
However, I failed with some reasons as shown below.

What is the key problem ? How can I run D435i ?

Also, I know that Intel realsense does not support ROS1.
Is there any way to run D435i with Ubuntu 20.04 and ROS1 Noetic ???
(For example, using old version of ROS1 Wrapper or Realsense SKD.)

----------------------------------  My terminal  ----------------------------------  

$ roslaunch realsense2_camera rs_camera.launch filters:=pointcloud

... logging to /home/byeong/.ros/log/70b64880-86ca-11ef-a5d2-bfb6448840c3/roslaunch-jun-36560.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://jun:33191/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
[... 122 lines cut ...]
 10/10 14:42:35,446 WARNING [139810429720320] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 10/10 14:42:35,496 WARNING [139810429720320] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 10/10 14:42:35,547 WARNING [139810429720320] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 10/10 14:42:35,598 WARNING [139810429720320] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 10/10 14:42:35,649 WARNING [139810429720320] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 10/10 14:42:35,700 WARNING [139810429720320] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 10/10 14:42:35,750 WARNING [139810429720320] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 10/10 14:42:35,801 WARNING [139810429720320] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 10/10 14:42:35,846 ERROR [139810010281728] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 130
 10/10 14:42:35,852 WARNING [139810429720320] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 10/10 14:42:35,903 WARNING [139810429720320] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 10/10 14:42:35,954 WARNING [139810429720320] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 10/10 14:42:36,005 WARNING [139810429720320] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 10/10 14:42:36,056 WARNING [139810429720320] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 10/10 14:42:36,108 WARNING [139810429720320] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 10/10 14:42:36,159 WARNING [139810429720320] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
[ WARN] [1728538956.265165535]: Hardware Notification:Depth stream start failure,1.72854e+12,Error,Hardware Error
 10/10 14:42:36,919 WARNING [139810429720320] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 10/10 14:42:36,970 WARNING [139810429720320] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 10/10 14:42:37,123 WARNING [139810429720320] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 10/10 14:42:37,173 WARNING [139810429720320] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 10/10 14:42:37,325 WARNING [139810429720320] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 10/10 14:42:37,375 WARNING [139810429720320] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 10/10 14:42:37,527 WARNING [139810429720320] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 10/10 14:42:37,577 WARNING [139810429720320] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0

hellovuong, 2024-10-10:
> [ INFO] [1728538951.247616616]: Device USB type: 2.1
> [ WARN] [1728538951.247661174]: Device 215322078943 is connected using a 2.1 port. Reduced performance is expected.

You are using USB2. You should use the cable provided along with the camera and plug it into a USB3 port on your computer.

MartyG-RealSense, 2024-10-10:
Hi @byeongjun-YU  The RealSense ROS1 wrapper does support D435i.  However, the ROS1 wrapper has ceased development and so has not been updated for use with recent RealSense SDK (librealsense) versions.  The recommended librealsense version for use with the wrapper is 2.50.0 or 2.51.1, and the recommended firmware version is **5.13.0.50**.  I note that you are using firmware 5.14.0.0.

As @hellovuong mentioned, your USB connection speed is being detected as the slow '2.1' instead of the much faster USB 3.1 or 3.2.  Whilst librealsense and the ROS wrapper can work on a slow USB 2.1 connection, problems like the **control_transfer returned** one that your log is generating are more likely to occur.

I note that you have the pointcloud filter enabled in your rs_camera.launch instruction.  Please try instead publishing a pointcloud using the **rs_rgbd.launch** launch file using the instructions at https://github.com/IntelRealSense/realsense-ros/issues/3026#issuecomment-1972741737

********

@hellovuong  Thanks so much for providing advice to @byeongjun-YU - though the connection being 2.1 does not necessarily mean that the official USB3 cable supplied with the camera is not being used.  It could mean that the launch is not correctly detecting the USB connection type, or the camera is plugged into a USB 2.1 hub.

MartyG-RealSense, 2024-10-28:
Case closed due to no further comments received.

closed 2024-10-28 by MartyG-RealSense

wwwzxxzw, 2025-02-27:
wzx@wzx-911X:~/ws_camera$ roslaunch realsense2_camera rs_camera.launch 
... logging to /home/wzx/.ros/log/5a16c1cc-f4d4-11ef-8891-23e295e1ba48/roslaunch-wzx-911X-204376.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://wzx-911X:38379/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
[... 75 lines cut ...]
[ INFO] [1740637837.277264476]: Done Setting Dynamic reconfig parameters.
[ INFO] [1740637837.277669099]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1740637837.278013893]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1740637837.278026858]: setupPublishers...
[ INFO] [1740637837.279222862]: Expected frequency for depth = 30.00000
[ INFO] [1740637837.280847332]: Expected frequency for color = 30.00000
[ INFO] [1740637837.282033622]: setupStreams...
 27/02 14:30:37,405 WARNING [140104366544640] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
[ INFO] [1740637837.455523122]: SELECTED BASE:Depth, 0
[ INFO] [1740637837.459692437]: RealSense Node Is Up!
[ WARN] [1740637837.523010576]: 
 27/02 14:30:37,523 WARNING [140104366544640] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 27/02 14:30:37,573 WARNING [140104366544640] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 27/02 14:30:37,819 WARNING [140104366544640] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 27/02 14:30:37,869 WARNING [140104366544640] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 27/02 14:30:38,020 WARNING [140104366544640] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 27/02 14:30:38,171 WARNING [140104366544640] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
[... cut ...]
 27/02 14:30:38,774 WARNING [140104366544640] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0

I encountered a similar issue and haven't found a solution. I hope to get some help here. My system is Ubuntu 20.04, ROS Noetic, firmware version 5.13.0.50, and the ROS Noetic librealsense2 version is v2.50.0. Why do I see the warning control_transfer returned error, index: 768, error: Success, number: 0? Even though this warning appears, I can still receive RGB images, depth maps, and point clouds from the D435 in RViz, and I can retrieve the data.